margelo/react-native-vision-camera #2118: Barcode scanner returns different strings in Android an iOS when scanning a UPC-A format code

nthchild1, 2023-10-31:
### What's happening?

I am trying to scan `upc-a` (12 digit) codes which are not directly supported by this library. However, this library does support `ean-13` which is a superset standard with an extra digit (13 digit). You can convert any `upc-a` code to an `ean-13` code by just adding a zero to the beginning of the string.

The problem is that, when scanning an `upc-a` code with the `ean-13` `codeType` enabled, the scanner returns a 13 digit ean-13 code as expected for iOS (it is adding an extra zero to the start of the string). It seems that in Android it is not adding a zero to the start of the string and is returning the original 12 digit code, resulting string is different for each platform.

For example:

Scanning the following code:

[image]

returns `050000463916` in Android and `0050000463916` in iOS

### Reproduceable Code

```tsx
import 'react-native-reanimated';
import React, { useEffect } from 'react';
import { StyleSheet, View } from 'react-native';
import { useIsFocused } from '@react-navigation/native';
import { Camera, useCameraDevice, useCodeScanner } from 'react-native-vision-camera';

const ScannerScreen = () => {
  const isFocused = useIsFocused();
  const device = useCameraDevice('back');

  const codeScanner = useCodeScanner({
    codeTypes: ['ean-13'],
    onCodeScanned: (codes) => {
      console.log(codes);
    },
  });

  useEffect(() => {
    void Camera.getCameraPermissionStatus();
  }, []);

  return (
    <>
      <View style={{ flex: 1 }}>
        {isFocused && device && (
          <Camera
            style={StyleSheet.absoluteFillObject}
            device={device}
            isActive={!!device}
            codeScanner={codeScanner}
          />
        )}
      </View>
    </>
  );
};

export default ScannerScreen;
```

### Relevant log output

```shell
No logs
```

### Camera Device

```json
Any camera device
```

### Device

Any iOS or Android smartphone

### VisionCamera Version

3.3.1

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

gary-cohen, 2023-11-03:
Yes, it would be nice if the lib supported upc-a especially since `BarcodeScannerOptions` supports it.

Gambitboy, 2023-11-14:
Same, I also need ability to scan upc_a, coming from `vision-camera-code-scanner` which was deprecated in favour of v3

jimsideout, 2023-11-29:
Same issue. Seems clunky to need to remove the leading 0 in iOS.

Gambitboy, 2023-11-30:
I managed to get this working by modifying the android files to include upc-a and running patch-package.
iOS already has the ability to scan upc-a because of ean-13, I just do a check to include the extra digit.

marcshilling, 2023-12-19:
@Gambitboy can you share your patch file please?

Gambitboy, 2023-12-20:
@marcshilling Here is the patch file. Its quite a big. Not sure why, as it wasn't a lot of changes.
I still have a type issue for upc-a which I haven't resolved yet.

https://gist.github.com/Gambitboy/0a2afa2485fd7c285a52b093de1c32b2

arisyo13, 2024-01-04:
I have a similar issue. On android it can scan multiple barcodes an returns an array with all the barcodes scanned but on ios it returns only one item. 
[image]

mrousavy, 2024-01-15:
On iOS I use the native QR scanner from Apple, on Android I use Google's MLKit model. If there are differences, there's nothing I can do.

You can instead write a custom Frame Processor Plugin to use the same libs on both devices, or something like ZXing.

closed 2024-01-15 by mrousavy

Couzix, 2024-02-15:
Hi ! MLKit actually provides support for upc-a ! "The following formats are supported: [...] UPC-A (FORMAT_UPC_A)". Sorry for tagging you @mrousavy, but do you consider in supporting it in your lib ? Thanks a lot !!

mrousavy, 2024-02-15:
Hey @Couzix - [Apple's Vision library doesn't support UPC-A](https://developer.apple.com/documentation/vision/vnbarcodesymbology) though, so it would not be a cross-platform feature, which I like to avoid.

Gambitboy, 2024-02-15:
When I read through the apple docs it said it indirectly supported upc-a through ean. So once I added the missing code for android both devices types were able to scan a upc-a code.

mrousavy, 2024-02-15:
Oh, interesting. Well I guess I could add those types and just mark them as `@platform Android` then..

Gambitboy, 2024-02-15:
I'll quickly try to find the docs mentioning it, also since its working on my project. I don't mind doing a pull request. The only bit I never got to finish was type checking the `UPC-A` for android

PR #2563 (2024-03-18): Add UPC-A support

## What

This PR adds support for UPC-A barcode format.

## Changes

This PR adds the UPC-A type format to React Native, CodeType to Android and iOS, but points to EAN13 in iOS as Apple states it supports UPC-A through EAN13 [here](https://developer.apple.com/library/archive/technotes/tn2325/_index.html#//apple_ref/doc/uid/DTS40013824-CH1-IS_UPC_A_SUPPORTED_).

## Tested on

Samsung S22
iPhone X

## Related issues

Fixes #2118