margelo/react-native-vision-camera #2119: Passing array to JS layer with custom Frame Processor parsed as object (V3)❓

fedpre, 2023-10-31:
### Question

Hi, I am not sure if it is related to the library or not, but I am building a custom frame processor to pick up the body pose on iOS. 

Everything work until I need to pass a nested array structured like this: `frame_keypoints`: `[[<value_x>, <value_y>, <confidence>], [...], [...]]`. To pass the data to the JS layer, I stringify the data and then parse them when picked up. For some off reason, I can't see to pass an array from the native iOS layer to JS as it is always parsed as an object. 

Have you encoutered this bug before? I am using Swift 5 and the Swift native array constructor.

Thanks!

### What I tried

I tried to change to a NSArray, NSMutablArray, with no avail. Everything that I pass an array and then parse in the JS layer, it is changed to an object and it breaks the app.

### VisionCamera Version

3.6.4

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

zzz08900, 2023-11-01:
Yeah that's sort of behavior by design.
You still get `length` property and you still can access items by index so in most cases it should be fine.

So what are you trying to do with the "fake" arrays and how did that break your app?

fedpre, 2023-11-01:
I didn't really find a solution, however what I do is to parse each string again and stringify it which strangely enough fixes the bug.

mrousavy, 2024-01-15:
Hey - yes this is because react-native-worklets treats arrays as objects. It should not change the way you work with them though, they're still iterable and indexable.

closed 2024-01-15 by mrousavy