margelo/react-native-vision-camera #2120: touch to focus doesn't work properly🐛

NuraQ, 2023-11-01:
### What's happening?

after upgrading react native vision camera to  V3 and using the latest camera version, touch to focus is not working properly, specifically for ios, but android is also not very accurate. Note that it used to work on V2 just fine.

### Reproduceable Code

```tsx
const device = useCameraDevice('back');    
const format =  getCameraFormat(device, [
        { resizeTo: { width: camContainerPosition?.width ?? '', height: 
        camContainerPosition?.height ?? '' } },
        { photoResolution: 'max' },
        { videoResolution: 'max' },
        { autoFocusSystem: 'contrast-detection' },
        { videoStabilizationModes: ['on', 'on', 'cinematic-extended'] } 
    ]);
    const fps = format.maxFps >= 240 ? 240 : format.maxFps;

 <Camera
   format={format}
   isActive={isAndroid ? !paused : true}
   photo={true}
   orientation={'portrait'}
   ref={cameraRef}
   enableHighQualityPhotos={true}
   fps={fps}
   preset='photo'
   enableZoomGesture={true}
   onLayout={handleCameraLayout}
   device={device}
   zoom={device.neutralZoom}
   style={{ flex: 1 }}
   onTouchEnd={(x) => touchToFocus(x.nativeEvent)}
   />
```

### Relevant log output

```shell
No logs
```

### Camera Device

```json
{
  "minZoom": 1,
  "maxZoom": 121.875,
  "isMultiCam": false,
  "formats": "",
  "position": "back",
  "name": "Back Camera",
  "hasFlash": true,
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "supportsRawCapture": false,
  "supportsFocus": true,
  "neutralZoom": 1,
  "supportsLowLightBoost": false,
  "sensorOrientation": "landscape-right",
  "hasTorch": true,
  "hardwareLevel": "full",
  "id": "com.apple.avfoundation.avcapturedevice.built-in_video:0"
}
```

### Device

iPhone 13 ( ios version 17.0.3)

### VisionCamera Version

3.6.4

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [X] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

rsnr, 2023-11-02:
I'm experiencing the same. @mrousavy I sponsored you now in appreciation of the time you're spending in investigating this issue 🙏

NastuzziSamy, 2023-11-02:
Hello @rsnr , try to use `event.nativeEvent` instead of `event`, it worked for me

NuraQ, 2023-11-05:
@NastuzziSamy  I used nativeEvent and focus doesn't work, I am retrieving x and y values correctly on touch. When I try to focus on IOS while having two objects behind camera and I touch object on background, it tries to focus but it doesn't. Focus functionality used to work perfectly on V2 but it's not very accurate on V3... any other suggestions?

mrousavy, 2023-11-07:
Thanks for the sponsorship @rsnr !!! ❤️ 

I'll take a look at this issue in the coming days / this week! :)

kueda, 2023-11-08:
FWIW it would also help to have tap to focus demonstrated in the example app.