margelo/react-native-vision-camera #2126: Video Recording on iOS has 1-2 seconds cut off at end of video, audio out of sync🐛

jkerb8, 2023-11-03:
### What's happening?

Reopening this issue since it's still a problem. Recording a video on an iOS device, with the expectation that the final video file would include the frames from the moment "startRecording" was called until the moment "stopRecording" was called. 

The result is that the last 1-2 seconds of video are not present in the final recording file. I recorded a timer and stopped the video when the timer reached 5 seconds. The video file that was saved ended before the timer reached 4 seconds.  In addition, the audio is out of sync with the video, as it seems like the video recording starts 1-2 seconds before the button is pressed, and ends 1-2 seconds early. The audio is actually correct but since the video is not, it appears out of sync. I can confirm this happens with iOS 17 on a variety of iOS devices, as well as iOS 15.5 and iOS 16. I have noticed that this does not happen when the preset is set to "medium" or "low", however the video quality is terrible with those presets. 

This problem does not apply to Android devices. 

### Reproduceable Code

```tsx
<Camera
    ref={ref => {
	this.camera = ref;
    }}
    style={Style.cameraContainer}
    isActive={true}
    device={this.state.cameraDevice}
    audio={true}
    video={true}
    zoom={this.state.zoom}
    videoStabilizationMode={'auto'}
    preset={'hd-1920x1080'}
></Camera>
```

### Relevant log output

```shell
None
```

### Camera Device

```json
{
  "isMultiCam": false,
  "hasTorch": false,
  "id": "com.apple.avfoundation.avcapturedevice.built-in_video:0",
  "formats": [],
  "supportsFocus": true,
  "supportsRawCapture": false,
  "neutralZoom": 1,
  "supportsLowLightBoost": false,
  "hasFlash": false,
  "name": "Back Camera",
  "supportsParallelVideoProcessing": true,
  "position": "back",
  "minZoom": 1,
  "devices": [
    "wide-angle-camera"
  ],
  "maxZoom": 95.625,
  "supportsDepthCapture": false
}
```

### Device

iPad 9th Gen (17.0)

### VisionCamera Version

2.16.5

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2023-11-10:
> VisionCamera Version 2.16.5

I don't provide such issue debugging and fixing support for V2 anymore.
Please upgrade to V3 and see if that fixes the issue.

closed 2023-11-10 by mrousavy

yspreen, 2023-11-17:
I am having very similar issues on v3