margelo/react-native-vision-camera #2127: How to make device and format filters strict ❓

MarioCodetria, 2023-11-03:
### Question

Is there a way to make "useCameraDevice" and "useCameraFormat" strict? Currently, if I do this:
```
const ultraWideBackDevice = useCameraDevice('back', { physicalDevices: ['ultra-wide-angle-camera'] });
```
It returns a device without "ultra-wide-angle-camera" support. And if I do this:
```
const hdrFormat = useCameraFormat(device, [{ fps: 30, photoHDR: true }]);
```
I get a device without HDR support. I need to display HDR and UW buttons on screen if those are supported on the device, but this method doesn't work because they always return a device, even if the requested features are not supported. Do I have to filter the devices manually?

### VisionCamera Version

3.6.4

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

MarioCodetria, 2023-11-03:
I ended up doing this:
```
    const desiredUltraWideBackDevice = useCameraDevice('back', { physicalDevices: ['ultra-wide-angle-camera'] });
    const desiredUltraWideFrontDevice = useCameraDevice('front', { physicalDevices: ['ultra-wide-angle-camera'] });
    const { ultraWideBackDevice, ultraWideFrontDevice } = React.useMemo(() => {
        return {
            ultraWideBackDevice: desiredUltraWideBackDevice?.physicalDevices.includes('ultra-wide-angle-camera') ? desiredUltraWideBackDevice : null,
            ultraWideFrontDevice: desiredUltraWideFrontDevice?.physicalDevices.includes('ultra-wide-angle-camera') ? desiredUltraWideBackDevice : null,
        };
    }, [desiredUltraWideBackDevice, desiredUltraWideFrontDevice]);
    const isUltraWideSupported = ultraWideBackDevice != null || ultraWideFrontDevice != null;
```
It works, although it looks like it could be simplified in a single hook call. Please let me know if there's a better way to do this.

mrousavy, 2024-01-15:
Well yes, you need to implement your own hook. The point of the hook by VisionCamera is to be _safe_, as in; always find a device even if the criteria isn't met. Otherwise you won't get a camera.

closed 2024-01-15 by mrousavy