margelo/react-native-vision-camera #2129: 🐛 Camera not recording audio

ChristopherGabba, 2023-11-04:
### What's happening?

I am using the camera to record a video. If I take that video path and play it in an expo video, it plays fine but no audio plays. After hours of troubleshooting, I've narrowed it down to the fact that the video being captured by react-native-vision-camera is just not capturing audio. 

I previously had multiple packages requesting microphone permissions (expo-av and react-native-vision-camera) and I have removed expo-av's microphone permissions from app.json to only allow react-native-vision camera. I am also using react-native-volume manager to manipulate audio and volume sessions in an effort to troubleshoot.

### Reproduceable Code

```tsx
I copied the code from the react-native-vision-camera demo with the Capture Button and the Camera UI exactly. So the reproducable demo is the camera code in this package. 

Using an iPhone 12 physical device connected through expo dev client. 

I then take the file.path from the onMediaCaptured callback and play it in:

        <Video
          source={{
            uri: "file:///private/var/mobile/Containers/Data/Application/97BBA11E-AB1D-4E15-BE96-C2FAB7D8C8F8/tmp/ReactNative/61BF091A-B1D5-4B8A-A08D-F3576C9C7167.mp4",
          }}
          style={{ width: 300, height: 300 }}
          volume={1}
          useNativeControls
          isLooping
          shouldPlay={true}
          isMuted={false}
        />
and tried to mess with the volume like so during troubleshooting: 
  useEffect(() => {
    ;(async function manageVolume() {
      Audio.setAudioModeAsync({
        playsInSilentModeIOS: true,
      });
      // VolumeManager.enableInSilenceMode(true)
      // VolumeManager.showNativeVolumeUI({ enabled: true })
      await VolumeManager.setVolume(0.5)
      // VolumeManager.enable(true, true) // Enable async
      // VolumeManager.setActive(true, true) // Activate async
    })()
  }, [])
```

### Relevant log output

```shell
Actually do not know how to do this... Not much experience with xcode..
```

### Camera Device

```json
{
  "position": "back",
  "hasTorch": true,
  "supportsLowLightBoost": false,
  "hardwareLevel": "full",
  "isMultiCam": false,
  "minZoom": 1,
  "supportsFocus": true,
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "neutralZoom": 1,
  "supportsRawCapture": false,
  "formats": [
    {
      "minISO": 33,
      "maxFps": 60,
      "fieldOfView": 64.72028350830078,
      "supportsVideoHDR": false,
      "photoHeight": 3024,
[... 1535 lines cut ...]
  "name": "Back Camera",
  "hasFlash": true,
  "maxZoom": 123.75
}
```

### Device

iPhone 12 Physical Device

### VisionCamera Version

"react-native-vision-camera": "^3.0.0",

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [X] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

itsabk, 2023-11-06:
facing the same issue here

itsabk, 2023-11-06:
I fixed my issue by updating the `hasMicrophonePermission` state as per the correct permission. It was being passed to the `audio` prop in the `ReanimatedCamera` component

ChristopherGabba, 2023-11-06:
@itsabk I've confirmed on numerous occasions that I have the correct microphone permissions, reinstalled the app to get the permissions re-requested, passed true directly to the audio prop directly and just tried a bunch of combinations and I just get a muted video every time.

Several weeks ago it was working, and sometime in the past couple of weeks on the project I deleted my Android and iOS folders to go back to a pure expo project and I am thinking it might be related. I am re-following the expo installation criteria that Marc provides in the docs and it looks like I need to prebuild. I'm working through lots of build errors conflicting with a Share Extension package in my project but once I am able to successfully Expo prebuild, I'll check to see if this fixes my problem and let you know.

ChristopherGabba, 2023-11-07:
Update 11/7/2023: I got the project to properly expo prebuild and compile and still no audio after new development build. I then downloaded Expo Camera to see if I could get any audio to record and it worked perfectly (recorded audio). Very confident that it is this react native vision camera package. 

Here is my plugin code in app.json: 

```
      [
        "react-native-vision-camera",
        {
          "cameraPermissionText": "ReelFeel needs access to your Camera.",
          "enableMicrophonePermission": true,
          "microphonePermissionText": "ReelFeel needs access to your Microphone."
        }
      ],
```

And rebuild the camera screen in a SUPER simplified manner with no bells and whistles as an experiment: 
```
export const CameraTestScreen: FC<CameraScreenProps> = observer(function CameraTestScreen(
  _props, // @demo remove-current-line
) {
  const { navigation } = _props

  const camera = useRef<Camera>()
  const isFocused = useIsFocused()
  const isForeground = useIsForeground()
  const isActive = isFocused && isForeground

  const [hasMicrophonePermission, setHasMicrophonePermission] = useState(false)

  useEffect(() => {
    ;(async function getPermissions() {
      const cameraStatus = await Camera.requestCameraPermission()
      const microphoneStatus = await Camera.requestMicrophonePermission()
      if (cameraStatus === "granted" && microphoneStatus === "granted") {
        console.log("got both permissions")
        setHasMicrophonePermission(true)
      }
    })()
  }, [isFocused])

  const device = useCameraDevice("front")

  const onError = useCallback((error: CameraRuntimeError) => {
    console.error(error)
  }, [])

  const onInitialized = useCallback(() => {
    console.log("Camera initialized!")
  }, [])

  const onMediaCaptured = useCallback(
    async (media: PhotoFile | VideoFile, type: "photo" | "video") => {
      //   const asset = await MediaLibrary.createAssetAsync(media.path)
      console.log(`Media captured! ${JSON.stringify(media)}`)
    },
    [navigation],
  )

  const startRecording = useCallback(async () => {
    try {
      if (camera.current == null) throw new Error("Camera ref is null!")

      console.log("calling startRecording()...")
      camera.current.startRecording({
        // fileType: "mp4",
        onRecordingError: (error) => {
          console.error("Recording failed!", error)
        },
        onRecordingFinished: (video) => {
          console.log(`Recording successfully finished! ${video.path}`)
          onMediaCaptured(video, "video")
        },
      })
    } catch (e) {
      console.error("failed to start recording!", e, "camera")
    }
  }, [camera, onMediaCaptured])

  const stopRecording = useCallback(async () => {
    try {
      if (camera.current == null) throw new Error("Camera ref is null!")

      console.log("calling stopRecording()...")
      await camera.current.stopRecording()
      console.log("called stopRecording()!")
    } catch (e) {
      console.error("failed to stop recording!", e)
    }
  }, [camera])

  return (
    <Screen preset="fixed" contentContainerStyle={$container}>
      {device != null && hasMicrophonePermission && (
        <Camera
          ref={camera}
          style={$camera}
          device={device}
          isActive={isActive}
          onInitialized={onInitialized}
          onError={onError}
          orientation="portrait"
          video={true}
          audio={hasMicrophonePermission}
        />
      )}
      <Button style={$recordButton} onPress={startRecording} />
      <Button style={$stopRecordingButton} onPress={stopRecording} />
      <Button style={$back} onPress={() => navigation.goBack()} />
    </Screen>
  )
})

const $container: ViewStyle = {
  height: "100%",
  width: "100%",
}

const $recordButton: ViewStyle = {
  position: "absolute",
  left: 100,
  bottom: 50,
  width: 50,
  height: 50,
  borderRadius: 25,
  backgroundColor: "green",
}

const $stopRecordingButton: ViewStyle = {
  position: "absolute",
  right: 100,
  bottom: 50,
  width: 50,
  height: 50,
  borderRadius: 25,
  backgroundColor: "red",
}

const $back: ViewStyle = {
  position: "absolute",
  top: 100,
  left: 50,
  width: 50,
  height: 50,
  backgroundColor: "orange",
}

const $camera: ViewStyle = {
  flex: 1,
}
```

alefrometa, 2023-11-08:
Hi @ChristopherGabba ran into this same problem and rolling back to previous release solved it
`yarn remove react-native-vision-camera && yarn add react-native-vision-camera@3.6.3`

nica0012, 2023-11-08:
@alefrometa Thanks for the info, are you able to record video with flash on 3.6.3? I think this may have broke too in 3.6.4

ChristopherGabba, 2023-11-08:
@alefromet Thank you for the response. I just tried this and recorded a video and there still is no audio playing when I play the video back. I actually rolled versions all the way back to 3.0.0 and still had no luck...

itsabk, 2023-11-08:
FYI:
    "react-native": "0.70.5",
    "react-native-vision-camera": "2",

mrousavy, 2023-11-10:
fixed ;)

closed 2023-11-10 by mrousavy

ChristopherGabba, 2023-11-10:
@mrousavy You rock, thank you Marc!

BenQoder, 2023-11-12:
I currently have this issue, How do I install this fix

griffinbaker12, 2023-11-17:
@ChristopherGabba hey, I am running into the same issue. how were you able to fix things?

BenQoder, 2023-11-17:
> @ChristopherGabba hey, I am running into the same issue. how were you able to fix things?
> 
> 

There's a new update... with it my issue was resolved

ChristopherGabba, 2023-11-17:
Update from Mark 3.6.6 fixed the issue. I was able to run `npm install react-native-vision-camera` and then run `npx expo prebuild --clean` and then I deleted my ios and android folders. Then I created a new development build `npx eas build --profile development  --platform ios` after that, audio is officially working. Make sure your silent switch is on. Not sure if all these steps were necessary, but my issue is now resolved. Thanks @mrousavy and all

mrousavy, 2023-11-20:
Marc*

ChristopherGabba, 2023-11-20:
> Marc*

Apologies, autocorrect got me, I knew that (corrected above) :) thanks again...

Abdullo-0901, 2025-10-20:
@mrousavy @alefrometa @mrousavy 
On Android, it happens during the recording — meaning, half of the video has sound, and the other half doesn’t
  "react-native-video": "6.17.0",
 "react-native-vision-camera": "4.7.2",
"react-native": "0.81.4"