margelo/react-native-vision-camera #2132: 🐛 Stop recording not working

josh-thompson13, 2023-11-06:
### What's happening?

I have been attempting to stop a video recording with the following logic seen below. But everytime I stop the recording the app crashes and I get a thread error. 

### Reproduceable Code

```tsx
const takePhotoOrVideo = async (): Promise<void> => {
        if (cameraRef.current) {
            if (cameraState === "photo") {
                await cameraRef.current
                    .takePhoto({ flash: flashEnabled })
                    .then((photo: any) => {
                        console.log(photo);
                        setPathOfMedia(photo.path);

                        // Convert photo to byte array
                        setDisplayMedia(true);
                    })
                    .catch((err: any) => {
                        enqueueMsg("error", `Failed to take photo: ${err}`);
                    });
            } else if (cameraState === "video" && isCameraInRecordingState === false) {
                // Set camera state to be in recording
                setIsCameraInRecordingState(true);
                await cameraRef.current.startRecording({
                    flash: flashEnabled,
                    onRecordingFinished: (video: VideoFile) => {
                        setPathOfMedia(video.path);
                    },
                    onRecordingError(error: CameraCaptureError) {
                        enqueueMsg("error", `Failed to record video: ${error}`);
                    },
                });
            } else if (cameraState === "video" && isCameraInRecordingState === true) {
                setIsCameraInRecordingState(false);

                 await cameraRef.current.stopRecording();

                setDisplayMedia(true);
            }
        }
    };
```

### Relevant log output

```shell
Thread 27: "*** -[AVCaptureSession stopRunning] stopRunning may not be called between calls to beginConfiguration and commitConfiguration"
```

### Camera Device

```json
{
   "supportsLowLightBoost":false,
   "id":"com.apple.avfoundation.avcapturedevice.built-in_video:0",
   "position":"back",
   "maxZoom":16,
   "hardwareLevel":"full",
   "supportsFocus":true,
   "supportsRawCapture":false,
   "neutralZoom":1,
   "physicalDevices":[
      "wide-angle-camera"
   ],
   "sensorOrientation":"landscape-right",
   "name":"Back Camera",
   "hasFlash":true,
   "minZoom":1,
   "hasTorch":true,
   "isMultiCam":false
}
```

### Device

Ipad Pro (16.6)

### VisionCamera Version

3.6.4

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

josh-thompson13, 2023-11-06:
Downgrading vision camera to 3.2.2 seemed to solve the problem.

mrousavy, 2023-11-10:
As I always say on every single issue: **~~Please~~ share the full logs.**

robsonbbs, 2023-11-10:
Hey @mrousavy, please see if it helps:

```
VisionCamera.startRecording(options:onVideoRecorded:onError:): Starting Video recording...

VisionCamera.startRecording(options:onVideoRecorded:onError:): File path: /private/var/mobile/Containers/Data/Application/0C851915-C0A8-485C-9DF7-DBAA32F8EEC6/tmp/ReactNative/28C32550-D59F-49AF-A0B3-5CD6F344CDE1.mov

VisionCamera.startRecording(options:onVideoRecorded:onError:): Recommended Video Settings: ["AVVideoCodecKey": hvc1, "AVVideoWidthKey": 1080, "AVVideoCompressionPropertiesKey": {
    AllowFrameReordering = 1;
    AllowOpenGOP = 1;
    AverageBitRate = 7651584;
    ExpectedFrameRate = 30;
    MaxAllowedFrameQP = 41;
    MaxKeyFrameIntervalDuration = 1;
    MinAllowedFrameQP = 15;
    MinimizeMemoryUsage = 1;
    Priority = 80;
    ProfileLevel = "HEVC_Main_AutoLevel";
    RealTime = 1;
    RelaxAverageBitRateTarget = 1;
}, "AVVideoHeightKey": 1920]

VisionCamera.initializeVideoWriter(withSettings:pixelFormat:): Initialized Video AssetWriter.

VisionCamera.startAssetWriter(): Starting Asset Writer(s)...

<0x104911b00> Gesture: System gesture gate timed out.
VisionCamera.appendBuffer(_:type:timestamp:): Started RecordingSession at 142647.371578666 seconds.
VisionCamera.finish(): Finishing Recording with AssetWriter status "writing"...

*** Terminating app due to uncaught exception 'NSGenericException', reason: '*** -[AVCaptureSession stopRunning] stopRunning may not be called between calls to beginConfiguration and commitConfiguration'
*** First throw call stack:
(0x194e625e0 0x18d197c00 0x1b111f31c 0x100e11164 0x100e1ec80 0x1008610d0 0x104408f50 0x10440ab34 0x10441298c 0x104413728 0x1044205f8 0x10441fc2c 0x1fdf5e964 0x1fdf5ea04)
VisionCamera.startRecording(options:onVideoRecorded:onError:): RecordingSession finished with status completed.
VisionCamera.deactivateAudioSession(): Deactivating Audio Session...
{ path: 'file:///private/var/mobile/Containers/Data/Application/0C851915-C0A8-485C-9DF7-DBAA32F8EEC6/tmp/ReactNative/28C32550-D59F-49AF-A0B3-5CD6F344CDE1.mov',
  duration: 7.099456709 }
libc++abi: terminating due to uncaught exception of type NSException
```

Let me know if you prefer that I open a new issue.

josh-thompson13, 2023-11-11:
> As I always say on every single issue: **~Please~ share the full logs.**

What other logs are you referring to? I have provided you with the log output I got when the application crashed.

rodgomesc, 2023-11-11:
hey @josh-thompson13 he is talking about the "full" logs which is the trace from adb logcat you only share the line that you think is relevant

mrousavy, 2023-11-13:
Thanks @robsonbbs ! 👍 

Will take a look when I have some free time.

daltachange, 2023-11-20:
@mrousavy  Thank you very much. I  found the cause of  the  #2172 . 
the  logs:

[image]

Maybe this will help you. @josh-thompson13

josh-thompson13, 2023-11-20:
> @mrousavy Thank you very much. I found the cause of the #2172 . the logs:
> 
> [image]
> Maybe this will help you. @josh-thompson13

Thanks will check it out!

mrousavy, 2024-01-15:
Hey - I think I fixed this in the latest version of VisionCamera. :)

closed 2024-01-15 by mrousavy

flexingCode, 2024-06-20:
i have the same issue with the latest version, but i just have that problem on the first time to stop the camera after install, after the first crash the app works fine. do you have some idea? :( 

`*** Terminating app due to uncaught exception 'NSGenericException', reason: '*** -[AVCaptureSession stopRunning] stopRunning may not be called between calls to beginConfiguration and commitConfiguration'
*** First throw call stack:
(0x1985fe418 0x195965e88 0x1b6f56d5c 0x102f03460 0x102f12770 0x1027b54dc 0x1074a0ab0 0x1074a279c 0x1074aa668 0x1074ab3e0 0x1074b8640 0x1074b7c50 0x21f2e1c7c 0x21f2de488)
'Handle error native >', '*** -[AVCaptureSession stopRunning] stopRunning may not be called between calls to beginConfiguration and commitConfiguration\n(\n    "4   libc++abi.dylib                     0x000000021f20887c 20F3F597-0DBD-3E4F-8D43-3C379861E4BC + 75900",\n    "5   libc++abi.dylib                     0x000000021f208820 _ZSt9terminatev + 108",\n    "6   libdispatch.dylib                   0x00000001074a27b0 _dispatch_client_callout + 40",\n    "7   libdispatch.dylib                   0x00000001074aa668 _dispatch_lane_serial_drain + 828",\n    "8   libdispatch.dylib                   0x00000001074ab3e0 _dispatch_lane_invoke + 408"\n)'`

mrousavy, 2024-06-20:
I'd need full logs for that. That's why issue templates require full logs.

flexingCode, 2024-06-21:
Sorry for question, how i get the full logs?

RRaideRR, 2024-06-22:
@flexingCode hey. I had the same issue.

Do you have a reproducible code at hand?

My problem was that 
(1) I set `<Camera audio={true} />` even though I didn't have the mic permission yet.
(2) Asked the user when he wanted to record a video for his mic permission and then started recording.

I fixed the problem in the following way:

`<Camera audio={hasMicPermission} />`

When the user wants to record a video, I still ask him to grant his mic permission. After his decision I return from the method and the user needs to click again on "record video".