margelo/react-native-vision-camera #2133: 🐛 Crash in Android release builds

halaharr, 2023-11-06:
### What's happening?

Android app crashes after scanning few times using barcode scanner. It is fine until about 5 to 6 scans, but crashes after that. It happens only in Android. We are not using FrameProcessor.

### Reproduceable Code

```tsx
import { Camera, useCameraDevice, useCodeScanner } from 'react-native-vision-camera'

const codeScanner = useCodeScanner({
    codeTypes: [
      'ean-13',
      'code-128',
      'code-39',
      'code-93',
      'codabar',
      'ean-8',
      'itf',
      'upc-e',
    ],
    onCodeScanned: codes => {
      if (codes.length) {
        if(Platform.OS === 'android') {
          setKeepScanning(false);
        }
        setBarCodes(codes);
      }
    },
  });

{showCamera && (
          <>
            <Camera
              style={StyleSheet.absoluteFill}
              device={device}
              isActive={keepScanning}
              codeScanner={codeScanner}
              audio={false}
            />
            <RNHoleView
              holes={[
                {
                  x: width * 0.085,
                  y: height * 0.36,
                  width: width * 0.83,
                  height: height * 0.2,
                  borderRadius: 10
                }
              ]}
              style={styles.rnholeView}
            />
          </>
        )}
```

### Relevant log output

```shell
--------- beginning of crash
11-06 08:50:42.563 26683 26842 F libc    : Fatal signal 11 (SIGSEGV), code 1, fault addr 0x84c144f7 in tid 26842 (pool-4-thread-4), pid 26683 (om.whiskeyshelf)
11-06 08:50:42.569 26683 26683 I CreateCaptureSession: Camera 0: Creating Capture Session #1011... Hardware Level: 2} | Outputs: [android.hardware.camera2.params.OutputConfiguration@c9b29cc1]
11-06 08:50:42.569 26683 26683 I CreateCaptureSession: Using legacy API (<28)
11-06 08:50:42.619 26683 26791 D TransportRuntime.SQLiteEventStore: Storing event with priority=DEFAULT, name=FIREBASE_ML_SDK for destination cct
11-06 08:50:42.624 26683 26791 D TransportRuntime.JobInfoScheduler: Upload for context TransportContext(cct, DEFAULT, MSRodHRwczovL2ZpcmViYXNlbG9nZ2luZy5nb29nbGVhcGlzLmNvbS92MGNjL2xvZy9iYXRjaD9mb3JtYXQ9anNvbl9wcm90bzNc) is already scheduled. Returning...
11-06 08:50:42.642 26683 27209 E Legacy-CameraDevice-JNI: getNativeWindow: Surface had no valid native window.
11-06 08:50:42.642 26683 27209 E Legacy-CameraDevice-JNI: LegacyCameraDevice_nativeDetectSurfaceType: Could not retrieve native window from surface.
11-06 08:50:42.648 26683 27209 E CameraDeviceGLThread-0: Received exception on GL render thread: 
11-06 08:50:42.648 26683 27209 E CameraDeviceGLThread-0: java.lang.UnsupportedOperationException: Unknown error -22
11-06 08:50:42.648 26683 27209 E CameraDeviceGLThread-0: 	at android.hardware.camera2.legacy.LegacyExceptionUtils.throwOnError(LegacyExceptionUtils.java:77)
11-06 08:50:42.648 26683 27209 E CameraDeviceGLThread-0: 	at android.hardware.camera2.legacy.LegacyCameraDevice.detectSurfaceType(LegacyCameraDevice.java:680)
11-06 08:50:42.648 26683 27209 E CameraDeviceGLThread-0: 	at android.hardware.camera2.legacy.SurfaceTextureRenderer.drawIntoSurfaces(SurfaceTextureRenderer.java:799)
11-06 08:50:42.648 26683 27209 E CameraDeviceGLThread-0: 	at android.hardware.camera2.legacy.GLThreadManager$1.handleMessage(GLThreadManager.java:105)
11-06 08:50:42.648 26683 27209 E CameraDeviceGLThread-0: 	at android.os.Handler.dispatchMessage(Handler.java:102)
11-06 08:50:42.648 26683 27209 E CameraDeviceGLThread-0: 	at android.os.Looper.loop(Looper.java:164)
11-06 08:50:42.648 26683 27209 E CameraDeviceGLThread-0: 	at android.os.HandlerThread.run(HandlerThread.java:65)
11-06 08:50:42.648 26683 27209 I CameraDeviceState: Legacy camera service transitioning to state ERROR
11-06 08:50:42.652 26683 26683 D Surface : setPackageUsesOwnResolution() (native) for "n0x89632800" (): true
11-06 08:50:42.652 26683 26683 I CameraDeviceState: Legacy camera service transitioning to state CONFIGURING
11-06 08:50:42.652 26683 26683 E CameraDeviceState: Cannot call configure while in state: 0
11-06 08:50:42.653 26683 26751 I CreateCaptureSession: Camera 0: Capture Session #1011 configured!
11-06 08:50:42.653 26683 26751 E CameraManager: Camera 0: Error! 4
11-06 08:50:42.654 26683 26751 E CameraSession: Camera Device android.hardware.camera2.impl.CameraDeviceImpl@4398e has been disconnected!
11-06 08:50:42.654 26683 26751 E CameraSession: com.mrousavy.camera.core.CameraDisconnectedError: [session/camera-has-been-disconnected] The given Camera device (id: 0) has been disconnected! Error: UNKNOWN_CAMERA_DEVICE_ERROR
11-06 08:50:42.654 26683 26751 E CameraSession: 	at com.mrousavy.camera.extensions.CameraManager_openCameraKt$openCamera$2$callback$1.onError(CameraManager+openCamera.kt:49)
11-06 08:50:42.654 26683 26751 E CameraSession: 	at android.hardware.camera2.impl.CameraDeviceImpl$CameraDeviceCallbacks$2.run(CameraDeviceImpl.java:1787)
11-06 08:50:42.654 26683 26751 E CameraSession: 	at android.os.Handler.handleCallback(Handler.java:790)
11-06 08:50:42.654 26683 26751 E CameraSession: 	at android.os.Handler.dispatchMessage(Handler.java:99)
11-06 08:50:42.654 26683 26751 E CameraSession: 	at android.os.Looper.loop(Looper.java:164)
11-06 08:50:42.654 26683 26751 E CameraSession: 	at android.os.HandlerThread.run(HandlerThread.java:65)
11-06 08:50:42.654 26683 26751 E CameraView: invokeOnError(...):
11-06 08:50:42.654 26683 26751 W System.err: com.mrousavy.camera.core.CameraDisconnectedError: [session/camera-has-been-disconnected] The given Camera device (id: 0) has been disconnected! Error: UNKNOWN_CAMERA_DEVICE_ERROR
11-06 08:50:42.654 26683 26751 W System.err: 	at com.mrousavy.camera.extensions.CameraManager_openCameraKt$openCamera$2$callback$1.onError(CameraManager+openCamera.kt:49)
11-06 08:50:42.654 26683 26751 W System.err: 	at android.hardware.camera2.impl.CameraDeviceImpl$CameraDeviceCallbacks$2.run(CameraDeviceImpl.java:1787)
11-06 08:50:42.654 26683 26751 W System.err: 	at android.os.Handler.handleCallback(Handler.java:790)
11-06 08:50:42.654 26683 26751 W System.err: 	at android.os.Handler.dispatchMessage(Handler.java:99)
11-06 08:50:42.654 26683 26751 W System.err: 	at android.os.Looper.loop(Looper.java:164)
11-06 08:50:42.654 26683 26751 W System.err: 	at android.os.HandlerThread.run(HandlerThread.java:65)
11-06 08:50:42.655 26683 26683 I CameraSession: Successfully configured Session with 1 outputs for Camera #0!
11-06 08:50:42.655 26683 26683 W CameraSession: Preview Output is null, aborting...
11-06 08:50:42.655 26683 26683 I CameraSession: Successfully updated CameraSession Configuration! isActive: true
11-06 08:50:42.662 26683 26683 D Surface : setPackageUsesOwnResolution() (native) for "n0x89632000" (SurfaceView - com.whiskeyshelf/com.whiskeyshelf.MainActivity@3205fe3@0[26683]#0): false
11-06 08:50:42.663 26683 26683 D SurfaceView: BG destroy() Surface(name=Background for - SurfaceView - com.whiskeyshelf/com.whiskeyshelf.MainActivity@3205fe3@0) com.mrousavy.camera.core.PreviewView{3205fe3 V.E...... ........ -9,0-809,1091}
11-06 08:50:42.665 26683 26683 D SurfaceView: setPackageUsesOwnResolution() (Java SurfaceView) for com.whiskeyshelf: 960x720 (false)
11-06 08:50:42.666 26683 26683 I CameraView: Updating CameraSession...
```

### Camera Device

```json
{
  "sensorOrientation": "landscape-right",
  "hardwareLevel": "full",
  "maxZoom": 10,
  "minZoom": 0.6000000238418579,
  "supportsLowLightBoost": true,
  "neutralZoom": 1,
  "physicalDevices": [
    "ultra-wide-angle-camera",
    "wide-angle-camera",
    "telephoto-camera"
  ],
  "supportsFocus": true,
  "supportsRawCapture": true,
  "isMultiCam": true,
  "name": "BACK (0)",
  "hasFlash": true,
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
```

### Device

Samsung Galaxy S22 (Android 13)

### VisionCamera Version

3.6.4

### Can you reproduce this issue in the VisionCamera Example app?

No, I cannot reproduce the issue in the Example app

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

xtl-geiger, 2023-11-06:
Same issue here.
Does not always happen, but quite often.

Happens on multiple devices.

tatiana-lonestone, 2023-11-08:
Hello,
Same error for me, only on release 
Working on Android 14, fail on Android 10.

````
2023-11-08 11:40:29.168  7592-7592  CreateCaptureSession    com.wonderboxpartners                I  Using new API (>=28)
2023-11-08 11:40:29.179  7592-9744  BufferQueueProducer     com.wonderboxpartners                E  [ImageReader-1280x720f23m2-7592-1] queueBuffer: BufferQueue has been abandoned
2023-11-08 11:40:29.215  7592-9744  BufferQueueProducer     com.wonderboxpartners                E  [ImageReader-1280x720f23m2-7592-1] queueBuffer: BufferQueue has been abandoned
2023-11-08 11:40:29.251  7592-9766  BufferQueueProducer     com.wonderboxpartners                E  [ImageReader-1280x720f23m2-7592-1] queueBuffer: BufferQueue has been abandoned
2023-11-08 11:40:29.336  7592-9766  BufferQueueProducer     com.wonderboxpartners                E  [ImageReader-1280x720f23m2-7592-1] queueBuffer: BufferQueue has been abandoned
2023-11-08 11:40:29.354  7592-9755  libc                    com.wonderboxpartners                A  Fatal signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x78760abc3d in tid 9755 (pool-10-thread-), pid 7592 (nderboxpartners)
2023-11-08 11:40:29.400  7592-9669  CreateCaptureSession    com.wonderboxpartners                I  Camera 0: Capture Session #1002 configured!

```

mrousavy, 2023-11-10:
Hm does that just happen while it sits idle, or when you navigate away or close the app?

halaharr, 2023-11-10:
It happens when you scan barcode. It scans few barcodes fine but may be after 5 or 6th scan it crashes.

tatiana-lonestone, 2023-11-10:
For me, on Android 10, each code bar scan caused this fail before I downgrade, but this never happened on Android 14

lBroth, 2023-11-14:
same crash issue, occurring on Android 12 after the 5th or 6th scan

starrymedia, 2023-11-18:
Same issue here. It's fixed?

mrousavy, 2023-11-18:
This is an OpenGL error, I think it's because of the Preview View being destroyed by Android OS. I'll take a look

D4nt4lion, 2023-11-27:
@mrousavy are there any new insights on this?

mrousavy, 2024-01-15:
Hey I think this is fixed in the latest version of Visioncamera.

closed 2024-01-15 by mrousavy