margelo/react-native-vision-camera #2134: How to get depth data from TrueDepth & Lidar❓

IvanByRool, 2023-11-06:
### Question

Hey,

Don't know if it's doable or if someone can help me.

I'm trying the example lib on my iPhone 13 Pro (With Lidar & TrueDepth front cam) in order to get the depth data.
My Goal is to reconstruct in 3D an object based on depth data provided by the cameras.
I want to try TrueDepth & Lidar to evaluate accuracy.

First question : How can I get depth data from the picture ? 
Second question : From a Frame, in Frame Processor ?

### What I tried

The example App is running

1. I saved a picture taken with the right cam (Lidar or TrueDepth) saved it in my mac and open it (couldn't find depth data)
2. I tried to log data from Frame in the FrameProcessor without success.

### VisionCamera Version

3.6.4

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

IvanByRool, 2023-11-11:
Up ;)
do someone have a working example of this I can use ? 
thx

ionflow, 2024-01-04:
I have the same question. Also on an iPhone 13 Pro, although I am using Expo.

ionflow, 2024-01-04:
Aha! I see the Camera component has `enableDepthData`.

mrousavy, 2024-01-15:
Hey! You should be able to access this data on the native side in a Frame Processor Plugin. I _think_ depth data should be attached to the Frame, if not that'd be a small change on my end. If you need help with this, consider contacting us through my agency: https://margelo.io

closed 2024-01-15 by mrousavy

Jrh0203, 2024-02-22:
@ionflow were you eventually able to access depth data from the frame in a frame processor? Trying to access the depth data now myself as well!

bugra-az, 2024-04-12:
Hi, was the issue resolved? Why it is closed? I could not see the solution here.

BenjaOliva, 2024-12-24:
Any news here ? 👀

mrousavy, 2024-12-26:
😉 https://github.com/mrousavy/react-native-vision-camera/pull/3328