margelo/react-native-vision-camera #2136: 🐛V3, iOS 17.0.3 QR CodeScanner Black Screen

Skseo91, 2023-11-07:
### What's happening?

Only the black screen will be displayed when an application that has operated normally so far approaches the iOS 17 and iPhone 15 cameras.

We've confirmed that Android is fine with the iOS 17 version of other devices (iPhone XS, iPhone 13 Pro) as well.

To some extent, I can reproduce the problem with the code, but I don't think there's a problem with the other properties because I've confirmed that it doesn't work only with the iPhone 15 as I said before.

### Reproduceable Code

```tsx
const codeScanner = useCodeScanner({
    codeTypes: ['qr', 'ean-13'],
    onCodeScanned: (codes) => {
      setBarcodes(codes)
    }
  })

const isFocused = useIsFocused();
<Camera
            style={styles.camera}
            device={device}
            isActive={isFocused}
            // // frameProcessor={frameProcessor}
            codeScanner={codeScanner}
/>

const styles = StyleSheet.create({
    camera: {
      left: 0,
      right: 0,
      top: 0,
      bottom: 0,
      height: SCREEN_HEIGHT,
      width: SCREEN_WIDTH,
    },
})
```

### Relevant log output

```shell
I don't have an iPhone 15, and I can't show logs because I can't access the camera on the emulator. I'm terribly sorry I couldn't be of any help.
```

### Camera Device

```json
Sorry I Don't know this
```

### Device

iPhone 15(iOS 17.0.3)

### VisionCamera Version

3.6.4

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

kwb020312, 2023-11-07:
I had exactly the same error. I have a bug where the screen turns black on my iPhone 15 when scanning a QR code. Does anyone know the solution??

rochamarcotulio, 2023-11-07:
I had the same error. Does anyone know the solution??

kwb020312, 2023-11-10:
I noticed that it didn't work because I didn't update the library version. It was an inexperienced mistake I'm sorry

closed 2023-11-10 by Skseo91