margelo/react-native-vision-camera #2139: 🐛 startRecording() function not working.

randomstuff222, 2023-11-07:
### What's happening?

I have my code set up so when I click a button, it will start the recording function, and once I click the stop button, it stops the recording and saves the path. But whenever I start the recording, I run into an "unknown/capture" error.

### Reproduceable Code

```tsx
const startRecording = async () => {
    try{
      setRecordingStatus(true);
      await cameraRef.current?.startRecording({
        flash: 'off',
        onRecordingFinished: video => {
          setVideoPath(video.path);
          setRecordedVideo(video);
        },
        onRecordingError(error) {
          console.log('Recording Error: ', error);
        },
      })
    }catch (e){
      console.log('Error while trying to start recording: ', e);
    }

  };
  const stopRecording = async () => {
    setRecordingStatus(false);
    await cameraRef.current?.stopRecording();
  };
return(
<View style={[styles.container, { flexDirection: 'column' }]}>
      <View style={{ flex: 5 }}>
        <Camera 
          ref={cameraRef}
          style={StyleSheet.absoluteFill}
          device={device}
          isActive={true}
          video={true}
        />
</View>
<TouchableOpacity
              style={styles.startingRecordingButton}
              onPress={stopRecording}
            />
            <TouchableOpacity
              style={styles.waitingRecordButton}
              onPress={startRecording}
            />
</View>
)
```

### Relevant log output

```shell
Opening the app on Android...
info JS server already running.
info Installing the app...
> Task :gradle-plugin:compileKotlin UP-TO-DATE
> Task :gradle-plugin:compileJava NO-SOURCE
> Task :gradle-plugin:pluginDescriptors UP-TO-DATE
> Task :gradle-plugin:processResources UP-TO-DATE
> Task :gradle-plugin:classes UP-TO-DATE
> Task :gradle-plugin:jar UP-TO-DATE
> Task :gradle-plugin:inspectClassesForKotlinIC UP-TO-DATE

> Configure project :react-native-vision-camera
[VisionCamera] react-native-worklets-core not found, Frame Processors disabled!

> Task :app:buildCodegenCLI SKIPPED
[... 154 lines cut ...]
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
```

### Device

Galaxy A52 (Android)

### VisionCamera Version

3.6.4

### Can you reproduce this issue in the VisionCamera Example app?

No, I cannot reproduce the issue in the Example app

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2023-11-08:
Hm, "prepare failed -22" - I've seen that before

mrousavy, 2023-11-10:
Can you try to use a low-resolution `format`?

E.g. something like:

```ts
const format = useCameraFormat(device, [
  { videoResolution: { width: 640, height: 480 } }
])
```

Does that work for you?

mrousavy, 2023-11-10:
Can you play around with the bitrate settings? Maybe the bitrate is wrong?

randomstuff222, 2023-11-12:
I tried playing around with both (changing the video bitrate, and the format settings) settings, but I still get the same error.

mrousavy, 2023-11-13:
Can you maybe tell me what exactly you did instead of just saying you changed something? Which formats did you try?

randomstuff222, 2023-11-13:
I did this:
```js
const format = useCameraFormat(device, [
    { videoResolution: { width: 640, height: 480 }}
  ])
```
With the format, and also added the bitrate on the recorindg
```js
await cameraRef.current?.startRecording({
        flash: 'off',
        videoBitRate: 'low',
```
However, I was able to solve the issue by adding this to the format
```js
const format = useCameraFormat(device, [
    { videoResolution: { width: 640, height: 480 }, pixelFormat: 'native' }
  ])
```

So now is working

closed 2023-11-16 by randomstuff222

mrousavy, 2023-11-17:
Aha - so the `pixelFormat` 'yuv' was not properly working? weird...

Alejandro192, 2024-01-09:
> Can you try to use a low-resolution `format`?
> 
> E.g. something like:
> 
> ```ts
> const format = useCameraFormat(device, [
>   { videoResolution: { width: 640, height: 480 } }
> ])
> ```
> 
> Does that work for you?

This works for me

itzyashh, 2024-04-13:
`const screen = Dimensions.get('screen')`

`const format = useCameraFormat(device, [
  {
    videoAspectRatio: screen.height / screen.width,
  }
  ])`

This worked for me , I'm using samsung A42

itzyashh, 2024-04-13:
> > Can you try to use a low-resolution `format`?
> > E.g. something like:
> > ```ts
> > const format = useCameraFormat(device, [
> >   { videoResolution: { width: 640, height: 480 } }
> > ])
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > Does that work for you?
> 
> This works for me

This video resolution was working for me but I don't want to compromise on quality so I used different soln
`const screen = Dimensions.get('screen')`

`const format = useCameraFormat(device, [
  {
    videoAspectRatio: screen.height / screen.width,
  }
  ])`

This worked for me , I'm using samsung A42