margelo/react-native-vision-camera #2140: 🐛Flash not working when recording video

nica0012, 2023-11-08:
### What's happening?

After upgrading from v2 to v3, flash seems to not work anymore while recording a video.

Is anybody else running into this?

### Reproduceable Code

```tsx
(example app)
```

### Relevant log output

```shell
n/a
```

### Camera Device

```json
n/a
```

### Device

iPhone 13 Pro Max 

### VisionCamera Version

3.6.4

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

nica0012, 2023-11-10:
@mrousavy Downgrading to 3.3.1 resolves this along with audio for video. Maybe the issue is related.

prasan2421, 2023-11-14:
I have same problem, downgrading to 3.3.1 didn't help either. 😕

nica0012, 2023-11-15:
@mrousavy I upgraded to 3.6.6 to give this a try, now when recording the flash starts for about a second or two, then turns off. I noticed in the version notes there was some bug fixes on the flash, but this might be lingering still.

mrousavy, 2023-11-15:
Interesting - does audio work flawlessly now though? I should've fixed audio in https://github.com/mrousavy/react-native-vision-camera/pull/2144

mrousavy, 2023-11-15:
Might be that we need to delay this: https://github.com/mrousavy/react-native-vision-camera/blob/3dd56c3ba26e5f95fcea4871657b2ef0e7bcb3d8/package/ios/Core/CameraSession.swift#L164-L168

...to make torch work on initial mount, as reported by @metrix-hu in https://github.com/mrousavy/react-native-vision-camera/issues/2050.

That it's not working when recording is a bit weird though, I'll try to fix that.

As a workaround, you can just enable `torch={true}` when starting your recording, and setting it back to `false` when stopping on the JS side.

nica0012, 2023-11-15:
> Interesting - does audio work flawlessly now though? I should've fixed audio in #2144

Audio works great  👍

nica0012, 2023-11-15:
> Might be that we need to delay this:
>
> https://github.com/mrousavy/react-native-vision-camera/blob/3dd56c3ba26e5f95fcea4871657b2ef0e7bcb3d8/package/ios/Core/CameraSession.swift#L164-L168
>
> ...to make [...]

Sounds great I'll give this workaround a go for now, thank you 💯

mrousavy, 2024-01-15:
Hey I think I fixed this in a recent change! Flash in video is now handled on JS side (as torch)

closed 2024-01-15 by mrousavy