margelo/react-native-vision-camera #2141: 🐛 Frame processor + take photo causes crashes

bglgwyng, 2023-11-08:
### What's happening?

Calling `camera.takePhoto` after switching the device causes crashes when the frame processor is performing reads on the buffer. This behavior was noticed within my application, which involves performing ML tasks on frames.
If you run the null frame processor which does nothing within it, `camera.takePhoto` works well.
Also, running the example frame processor plugin doesn't cause crashes.

1. Run the example app with the following frame processor code
2. Switch the device from back to front
3. Call `takePhoto`

### Reproduceable Code

```tsx
const frameProcessor = useFrameProcessor((frame) => {
  'worklet'

  // Commenting this out makes crashes not happen
  frame.toArrayBuffer()
}, [])
```

### Relevant log output

```shell
2023-11-08 20:24:16.648 31765-31830 ReactNativeJS           com.mrousavy.camera.example          I  Camera: FRONT (1) | Format: (2400x1080 photo / 1920x864@60 video @ 60fps)
2023-11-08 20:24:16.659 31765-31765 PreviewView             com.mrousavy.camera.example          I  Resizing PreviewView to 1920 x 864...
2023-11-08 20:24:16.660 31765-31765 CameraView              com.mrousavy.camera.example          I  Updating CameraSession...
2023-11-08 20:24:16.660 31765-31860 CameraSession           com.mrousavy.camera.example          I  Updating CameraSession Configuration...
2023-11-08 20:24:16.661 31765-31860 CameraSession           com.mrousavy.camera.example          I  Configuring Camera #1...
2023-11-08 20:24:16.745 31765-32005 CameraManagerGlobal     com.mrousavy.camera.example          I  Camera 0 facing CAMERA_FACING_BACK state now CAMERA_STATE_IDLE for client com.mrousavy.camera.example API Level 2User Id 0
2023-11-08 20:24:16.860 31765-32005 CameraManagerGlobal     com.mrousavy.camera.example          I  postSingleUpdate device: camera id 0 status STATUS_PRESENT
2023-11-08 20:24:16.860 31765-31831 CameraDevices           com.mrousavy.camera.example          I  Camera #0: Available!
2023-11-08 20:24:16.860 31765-32005 CameraManagerGlobal     com.mrousavy.camera.example          I  Camera 0 facing CAMERA_FACING_BACK state now CAMERA_STATE_CLOSED for client com.mrousavy.camera.example API Level 2User Id 0
2023-11-08 20:24:16.861 31765-31778 .camera.example         com.mrousavy.camera.example          W  Long monitor contention with owner mrousavy/VisionCamera.main (31860) at void android.hardware.camera2.impl.CameraDeviceImpl.close()(CameraDeviceImpl.java:1463) waiters=0 in void android.hardware.camera2.impl.CameraDeviceImpl.onDeviceError(int, android.hardware.camera2.impl.CaptureResultExtras) for 124ms
2023-11-08 20:24:16.861 31765-31860 CameraManager           com.mrousavy.camera.example          I  Camera 1: Opening...
2023-11-08 20:24:16.875 31765-32122 CameraManagerGlobal     com.mrousavy.camera.example          I  postSingleUpdate device: camera id 1 status STATUS_NOT_AVAILABLE
2023-11-08 20:24:16.875 31765-31831 CameraDevices           com.mrousavy.camera.example          I  Camera #1: Unavailable!
2023-11-08 20:24:16.883 31765-32122 CameraManagerGlobal     com.mrousavy.camera.example          I  Camera 1 facing CAMERA_FACING_FRONT state now CAMERA_STATE_OPEN for client com.mrousavy.camera.example API Level 2User Id 0
2023-11-08 20:24:16.883 31765-31860 CameraManager           com.mrousavy.camera.example          I  Camera 1: Opened!
2023-11-08 20:24:16.887 31765-31860 PreviewView             com.mrousavy.camera.example          I  Resizing PreviewView to 1920 x 864...
2023-11-08 20:24:16.887 31765-31860 CameraSession           com.mrousavy.camera.example          I  Successfully configured Camera #1!
2023-11-08 20:24:16.891 31765-31860 CameraSession           com.mrousavy.camera.example          I  Configuring Session for Camera #1...
2023-11-08 20:24:16.892 31765-31860 SurfaceOutput           com.mrousavy.camera.example          I  Closing 2400x1080 PHOTO ImageReader..
2023-11-08 20:24:16.892 31765-31860 SurfaceOutput           com.mrousavy.camera.example          I  Closing 2400x1080 Video Pipeline..
2023-11-08 20:24:16.893 31765-31860 CameraSession           com.mrousavy.camera.example          I  Adding 2400 x 1080 Photo Output in Format #256...
2023-11-08 20:24:16.894 31765-31860 CameraSession           com.mrousavy.camera.example          I  Adding 1920 x 864 Video Output in Format #34...
2023-11-08 20:24:16.894 31765-31860 VideoPipeline           com.mrousavy.camera.example          I  Initializing 1920 x 864 Video Pipeline (format: NATIVE)
2023-11-08 20:24:16.894 31765-31860 VideoPipeline           com.mrousavy.camera.example          I  Using ImageReader round-trip (format: #34)
2023-11-08 20:24:16.894 31765-31860 VideoPipeline           com.mrousavy.camera.example          I  Using API 29 for GPU ImageReader...
2023-11-08 20:24:16.895 31765-31860 CameraSession           com.mrousavy.camera.example          I  Adding 1920 x 864 Preview Output...
2023-11-08 20:24:16.895 31765-31860 PreviewView             com.mrousavy.camera.example          I  Resizing PreviewView to 1920 x 864...
2023-11-08 20:24:16.898 31765-31860 CreateCaptureSession    com.mrousavy.camera.example          I  Camera 1: Creating Capture Session #1002... Hardware Level: 0} | Outputs: [android.hardware.camera2.params.OutputConfiguration@70575b5d, android.hardware.camera2.params.OutputConfiguration@fc2c2196, android.hardware.camera2.params.OutputConfiguration@8def04b6]
2023-11-08 20:24:16.898 31765-31860 CreateCaptureSession    com.mrousavy.camera.example          I  Using new API (>=28)
2023-11-08 20:24:16.994 31765-31860 CreateCaptureSession    com.mrousavy.camera.example          I  Camera 0: Capture Session #1001 closed!
2023-11-08 20:24:16.994 31765-31860 CreateCaptureSession    com.mrousavy.camera.example          I  Camera 1: Capture Session #1002 configured!
2023-11-08 20:24:16.994 31765-31860 CameraSession           com.mrousavy.camera.example          I  Successfully configured Session with 3 outputs for Camera #1!
2023-11-08 20:24:16.994 31765-31860 CameraSession           com.mrousavy.camera.example          I  Updating Video Outputs...
2023-11-08 20:24:16.994 31765-31860 VideoPipeline           com.mrousavy.camera.example          I  Setting 1920 x 864 FrameProcessor Output...
2023-11-08 20:24:16.994 31765-31860 VideoPipeline           com.mrousavy.camera.example          I  Removing RecordingSession Output...
2023-11-08 20:24:16.996 31765-31860 CameraSession           com.mrousavy.camera.example          I  Successfully updated CameraSession Configuration! isActive: true
2023-11-08 20:24:16.996 31765-31860 CameraView              com.mrousavy.camera.example          I  invokeOnInitialized()
2023-11-08 20:24:16.997 31765-32122 CameraManagerGlobal     com.mrousavy.camera.example          I  Camera 1 facing CAMERA_FACING_FRONT state now CAMERA_STATE_ACTIVE for client com.mrousavy.camera.example API Level 2User Id 0
2023-11-08 20:24:16.997 31765-31830 ReactNativeJS           com.mrousavy.camera.example          I  Camera initialized!
2023-11-08 20:24:17.323 31765-31861 VideoPipeline           com.mrousavy.camera.example          I  ImageReader::onImageAvailable!
2023-11-08 20:24:17.324 31765-31861 Frame                   com.mrousavy.camera.example          I  Buffer 1920 x 864 @ 1920
2023-11-08 20:24:17.361 31765-31861 VideoPipeline           com.mrousavy.camera.example          I  ImageReader::onImageAvailable!
2023-11-08 20:24:17.427 31765-31861 VideoPipeline           com.mrousavy.camera.example          I  ImageReader::onImageAvailable!
2023-11-08 20:24:17.464 31765-31861 VideoPipeline           com.mrousavy.camera.example          I  ImageReader::onImageAvailable!
2023-11-08 20:24:17.496 31765-31861 VideoPipeline           com.mrousavy.camera.example          I  ImageReader::onImageAvailable!
2023-11-08 20:24:17.535 31765-31861 VideoPipeline           com.mrousavy.camera.example          I  ImageReader::onImageAvailable!
[... 101 lines cut ...]
  "hasTorch": false,
  "position": "front",
  "id": "1"
}
```

### Device

Galaxy S21

### VisionCamera Version

3.6.4

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

xulihang, 2023-11-20:
Same crash with vision camera v2 + reanimated v3.

Vision camera 2.15 + reanimated v2 works okay.

LuongTruong, 2023-11-20:
I got the same issue when using the vision camera v2 + reanimated v3. The reason I can't use vision camera v3 is that it support Android from 26 and above, which is only covert 93.7%

[image]

I love the vision camera v3 a lot but can't use it at the moment. If anyone know how to make the vision camera v2 work with reanimated v3, please let me know. Many thanks!

xulihang, 2023-11-20:
The latest version of vision camera v2 can work with reanimated v3: https://github.com/mrousavy/react-native-vision-camera/pull/1759

But `takePhoto` + frame processor seem to cause a crash.

LuongTruong, 2023-11-22:
Hello guys, I am able to build the camera v2 + reanimated v3. Below is the details of my package.json

```
"react": "18.2.0",
"react-native": "0.72.7",
"react-native-reanimated": "3.5.4",
"react-native-vision-camera": "2.16.5",
"vision-camera-code-scanner": "0.2.0"
```

Below is my script to build aab and apk:

```
"android-aab": "cd android && ./gradlew clean && rm -rf .gradle && ./gradlew bundleRelease && cd ../",
"android-apk": "cd android && ./gradlew clean && rm -rf .gradle && ./gradlew assembleRelease && cd ../",
```

Hope these can help people who have the same issue as me

mrousavy, 2024-01-15:
Hey does this still crash? I made some changes here and there. Also error reporting should be better now.

mrousavy, 2024-01-17:
Maybe I just fixed this in https://github.com/mrousavy/react-native-vision-camera/pull/2408 - could you test and verify?

bglgwyng, 2024-01-18:
I verified this fix worked.

closed 2024-01-18 by bglgwyng