margelo/react-native-vision-camera #2142: 🐛 Camera stretched on Android

NuraQ, 2023-11-08:
### What's happening?

Camera is stretched on Android, behavior is not consistent as sometimes when closing the navbar and reopening the stretch is gone.
[image].

issue reproduced on these two devices:

-  Samsung Galaxy S9 SM G960F, Android version 8
-  OnePlus 5T, Android version 10

### Reproduceable Code

```tsx
const device = useCameraDevice('back');    
const format =  getCameraFormat(device, [
        { resizeTo: { width: camContainerPosition?.width ?? '', height: 
        camContainerPosition?.height ?? '' } },
        { photoResolution: 'max' },
        { videoResolution: 'max' },
        { autoFocusSystem: 'contrast-detection' },
        { videoStabilizationModes: ['on', 'on', 'cinematic-extended'] } 
    ]);
    const fps = format.maxFps >= 240 ? 240 : format.maxFps;
    const touchToFocus = async (event) => {
        if (!device.supportsFocus || !isCameraFocusReady.current) return;
        let point = {
            x: Math.round(event.pageX - camLocation.x),
            y: Math.round(event.pageY - camLocation.y)
        };
        try {
            isCameraFocusReady.current = false;
            await cameraRef?.current?.focus(point);
            setTimeout(() => {
                isCameraFocusReady.current = true;
            }, 200);
        } catch (error) {

        }
    };
 <Camera
   format={format}
   isActive={isAndroid ? !paused : true}
   photo={true}
   orientation={'portrait'}
   ref={cameraRef}
   enableHighQualityPhotos={true}
   fps={fps}
   preset='photo'
   enableZoomGesture={true}
   onLayout={handleCameraLayout}
   device={device}
   zoom={device.neutralZoom}
   style={{ flex: 1 }}
   onTouchEnd={(x) => touchToFocus(x.nativeEvent)}
   />
```

### Relevant log output

```shell
no logs
```

### Camera Device

```json
{"sensorOrientation":
"landscape-right",
"hardwareLevel: "full",
"maxZoom": 8,
"'minZoom":1,
"supportsLowLightBoost":false,
"neutralZoom":1,
'physicalDevices":['wide-angle-camera"],
"supportsFocus":true,
"supportsRawCapture" true,
"isMultiCam":false,
"name":"BACK(0)",
"hasFlash" true,
"has Torch" true,
"position":"back",
"id":"0"}
```

### Device

OnePlus 5T, Android version 10

### VisionCamera Version

3.6.4

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [X] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

rsnr, 2023-11-08:
@mrousavy Sponsoring this one as well :)

bglgwyng, 2023-11-09:
Could you check if `format` is `undefined`?

mrousavy, 2023-11-09:
@rsnr thank you!! ❤️ 

> behavior is not consistent as sometimes when closing the navbar and reopening the stretch is gone.

@NuraQ so if you minimize the app and open it again, it looks correct? Then it's definitely a race condition in `PreviewView`'s set size funcs.

I'll investigate this soon!

NuraQ, 2023-11-09:
@mrousavy  After testing on two devices, this was the result:

1. on One Plus, it was consistent.
 "Minimizing" by the right button, or sending to background by the middle button, did not have impact (still camera is stretched).

2. On Samsung, I think the back middle button doesn't always behave the same in terms of impact no stretching ( it some times fixes stretch issue and sometimes it doesn't).

mrousavy, 2023-11-10:
okok lemme take a look over the weekend or next week, its a race condition

j-jonathan, 2023-11-23:
Hey,

I've just started integrating react-native-vision-camera v3 into my project (i'm currently using react-native-camera) and encountered some issues related to aspect ratio on Android (similar to the problem described in this opened issue).

### Usage
I am using the resizeMode='contain' mode to display the entire preview:
```tsx
<Camera
    ref={this.cameraRef}
    style={{ position: 'absolute', top: 0, left: 0, right: 0, bottom: 0 }}
    device={device}
    format={format}
    isActive={true}
    photo={isPhoto}
    video={isVideo}
    audio={isVideo}
    resizeMode='contain'
/>
````

Tested on:
- HUAWEI P30 (ELE-L29) with Android 10
- react-native: 0.72.4
- react-native-vision-camera: 3.6.10

### Issues

Observed Issues when I switch from one format to another:
- The view is never resized (the call to requestLayout() doesn't seem to work).
- The aspect ratio is not always correct after a format change.

For the first issue, it appears to be a problem related to react-native.
I found a solution on this GitHub issue: https://github.com/facebook/react-native/issues/17968

Here is the patch to fix the issue:
```
diff --git a/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView.kt b/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView.kt
index 0aa8932..7b1c91e 100644
--- a/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView.kt
+++ b/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView.kt
@@ -242,4 +242,17 @@ class CameraView(context: Context) :
   override fun onCodeScanned(codes: List<Barcode>, scannerFrame: CodeScannerFrame) {
     invokeOnCodeScanned(codes, scannerFrame)
   }
+
+  override fun requestLayout() {
+    super.requestLayout()
+    post(measureAndLayout)
+  }
+
+  private val measureAndLayout = Runnable {
+    measure(
+        MeasureSpec.makeMeasureSpec(width, MeasureSpec.EXACTLY),
+        MeasureSpec.makeMeasureSpec(height, MeasureSpec.EXACTLY)
+    )
+    layout(left, top, right, bottom)
+  }
 }
```

For the second issue, it occurs when the surface is updated ("PreviewView Surface updated!").
The ratio is not necessarily updated, and I don't know why.

The only solution I found is to restart the entire device configuration.

Here is the patch to fix the issue:

```
diff --git a/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/CameraSession.kt b/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/CameraSession.kt
index 8ad60cc..15b85ff 100644
--- a/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/CameraSession.kt
+++ b/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/CameraSession.kt
@@ -187,6 +187,19 @@ class CameraSession(private val context: Context, private val cameraManager: Cam

         override fun surfaceChanged(holder: SurfaceHolder, format: Int, width: Int, height: Int) {
           Log.i(TAG, "PreviewView Surface updated! ${holder.surface} $width x $height")
+
+          launch {
+            mutex.withLock {
+              try {
+                configuration?.let { configuration ->
+                  configureOutputs(configuration)
+                  configureCaptureRequest(configuration)
+                }
+              } catch (error: Throwable) {
+                Log.e(TAG, "Failed to configure CameraSession! Error: ${error.message}", error)
+              }
+            }
+          }
         }

         override fun surfaceDestroyed(holder: SurfaceHolder) {
```

However, this is not perfect, and the camera does not load instantly (refer to the 3rd video where you can see the incorrect ratio, and then 1 or 2 seconds later, the correct ratio after reloading the camera configuration).

@mrousavy > Do you have any idea where this issue might be coming from? I've tried various approaches but haven't found an alternative fix. I'm available to experiment with other solutions if needed.

### Videos

To illustrate, here are videos of my application at different stages of the fixes:

Without any fix:
https://github.com/mrousavy/react-native-vision-camera/assets/78229184/98129bb6-8a15-4ff4-a79a-e0d8b30ceb7d

With the fix for issue 1:
https://github.com/mrousavy/react-native-vision-camera/assets/78229184/0d2e47d8-04ed-4c4e-b1c8-31895babd88e

With the fix for issue 2:
https://github.com/mrousavy/react-native-vision-camera/assets/78229184/6458f9e9-b6b6-4273-a4b5-58ea4ec72dc8

mrousavy, 2023-11-24:
Hey - thanks for the insights man! I need to take a look at this, probably just a small fix to avoid any race conditions here. Will have some time soon, maybe in 1-2 weeks

acesetmatch, 2023-12-14:
Has there been any progress on this? Still experiencing the issue

bpeck81, 2023-12-31:
Not sure if this helps, but when displaying an image taken with the camera, I found that I had to swap the height and width properties returned from the photo object.

`const photo = await cameraRef.current.takePhoto()`

JuMaruhn, 2024-01-04:
Here is a (dirty) quickfix that helps me:

Add the line:

`private val aspectRatio: Float get() = size.width.toFloat() / size.height.toFloat()`

to the class PreviewView in 
`react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/PreviewView.kt`.

Then change the methode onMeasure in the same file to:

```
@SuppressLint("DrawAllocation")
override fun onMeasure(widthMeasureSpec: Int, heightMeasureSpec: Int) {
  super.onMeasure(widthMeasureSpec, heightMeasureSpec)
  val viewWidth = MeasureSpec.getSize(widthMeasureSpec)
  val viewHeight = MeasureSpec.getSize(heightMeasureSpec)

  Log.i(TAG, "PreviewView onMeasure($viewWidth, $viewHeight)")

  val newWidth: Int
  val newHeight: Int

  val actualRatio = if (viewWidth > viewHeight) aspectRatio else 1f / aspectRatio
  if (viewWidth < viewHeight * actualRatio) {
    newHeight = viewHeight
    newWidth = (viewHeight * actualRatio).roundToInt()
  } else {
    newWidth = viewWidth
    newHeight = (viewWidth / actualRatio).roundToInt()
  }

  Log.d(TAG, "Measured dimensions set: $newWidth x $newHeight")
  setMeasuredDimension(newWidth, newHeight)
}
```

Alexis200007, 2024-01-04:
> Here is a (dirty) quickfix that helps me:
>
> Add the line:
>
> `private val aspectRatio: Float get() = size.width.toFloat() / size.height.toFloat()`
>
> to the class PreviewView in [...]

Thanks, it works for me

canhtvee, 2024-01-07:
It's a race condition, I fixed it on JS side by init the camera dimension at zero, settimeout waiting for the camera launch, and then resizing the camera view.

Jigar2311, 2024-01-10:
i am also have this issue but when i make orientation lock to landscape

iamtommcc, 2024-01-11:
@JuMaruhn's fix partially fixed my issue - it made the preview properly fill its container (I was getting blank bars on the sides), but for me the preview was still distorted.

I too experienced some inconsistency, sometimes it would fix itself by flipping between front/back, minimising app, etc.

I was able to workaround this issue by not feeding in `format` straight away, and instead doing this:

```jsx
function CameraPreview() {
  const [isInitialised, setIsInitialised] = useState(false);

  return (
    <View onLayout={() => setIsInitialised(true)}>
      <CameraView
        format={
          !isInitialised && Platform.OS === "android" ? undefined : format
        }
        resizeMode={"contain"}
        style={{
          aspectRatio: 3 / 4,
        }}
        {...yourOtherProps}
      />
    </View>
  );
}
```

This I think is very similar to @j-jonathan's approach except I think it's a little lighter/quicker than reinitialising the entire camera config.

Jigar2311, 2024-01-11:
> @Jigar2311 I had the same problem in landscape mode and the fix of @JuMaruhn fixed my problem.

Sure @Alexis200007, let me see if it works for me

mgvictor7, 2024-01-11:
I have the same problem in the version 3.7.0
In version 3.5.1 there was no this problem

I have tried to solve the problem by updating the size in the camera style 250 milliseconds after the component has been mounted
but I don't like this

```
...
const [adjustCameraSize, setAdjustCameraSize] = useState(false);

useEffect(() => {
    if (device && isFocussed && cameraPermission === Permissions.RESULTS.GRANTED) {
      setTimeout(() => {
        setAdjustCameraSize(true);
      }, 250);
    }
  }, [isFocussed, device, cameraPermission]);

...

<Camera
       style={[{ ...StyleSheet.absoluteFill }, { width: adjustCameraSize ? '100%' : '120%' }]}
       device={device}
        isActive
        codeScanner={codeScanner}
      />
```

mrousavy, 2024-01-11:
Hey! Does this fix your issue maybe? https://github.com/mrousavy/react-native-vision-camera/pull/2377 (still testing)

mrousavy, 2024-01-11:
Hey!

After 8 hours of debugging, I finally found the culprit! I fixed the preview stretching issue in this PR: https://github.com/mrousavy/react-native-vision-camera/pull/2377

If you appreciate my dedication and free support, please consider [💖 **sponsoring me on GitHub** 💖](https://github.com/sponsors/mrousavy/) so I can keep providing fixes and building out new features for my open-source libraries in my free time.

ben-qiu, 2024-01-14:
I am still experiencing the same with `3.7.1`, the preview is stretched.

Z-Hayk, 2024-01-19:
The same with `3.8.2`, the preview is stretched.

huntfpoly12, 2024-01-20:
> @JuMaruhn's fix partially fixed my issue - it made the preview properly fill its container (I was getting blank bars on the sides), but for me the preview was still distorted. This was on a Samsung Galaxy S22.
> 
> I too experienced some inconsistency, sometimes it would fix itself by flipping between front/back, minimising app, etc.
> 
> I was able to workaround this issue by not feeding in `format` straight away, and instead doing this:
> 
> ```js
> function CameraPreview() {
>   const [isInitialised, setIsInitialised] = useState(false);
> 
>   return (
>     <View onLayout={() => setIsInitialised(true)}>
>       <CameraView
>         format={
>           !isInitialised && Platform.OS === "android" ? undefined : format
>         }
>         resizeMode={"contain"}
>         style={{
>           aspectRatio: 3 / 4,
>         }}
>         {...yourOtherProps}
>       />
>     </View>
>   );
> }
> ```
> 
> This is very similar to @j-jonathan's approach except I think it's a little lighter/quicker than reinitialising the entire camera config.

 it works for me

piwko28, 2024-01-23:
> it works for me. Thanks

[huntfpoly12](https://github.com/huntfpoly12) Works for me as well as a temp. fix

chramos, 2024-01-30:
in my case I'm not specifying the `format` prop and I'm still getting the issue

"react-native-vision-camera": "^3.8.2"

JshGrn, 2024-01-30:
I am also not specifying the format prop and none of the fixes work for me on a Nokia G22. Whatever settings I use the height is slightly stretched, switching between camera and app doesn't fix either. My app is hardcoded to be portrait only.

grantsingleton, 2024-01-31:
Still happening on 3.8.2. None of the temp fixes work reliably for me. I'm sponsoring this issue @mrousavy 🙌

metuuu, 2024-02-01:
I'm having this issue also.

mrousavy, 2024-02-01:
Thanks @grantsingleton! Will look into this soon!

JshGrn, 2024-02-01:
Its definitely a race condition because whilst developing other areas of the app when I have 'completed' my work flow and passed back to the camera screen its fine. Weirdly, it actually flips between 3, stretched height, really stretched height (square becomes long rectangle) and normal.

boiboif, 2024-02-02:
Same issue after upgrade to 3.8.2.

krzychuuu132, 2024-02-04:
Same issue :/

Dingenis, 2024-02-05:
We are also experiencing this issue. We enormously respect and love what you @mrousavy are doing so our small company became a sponsor. If you have any time this issue would have our preference.

mrousavy, 2024-02-05:
Thanks @Dingenis, which company?

I'll take a look at this soon 🙏

Menardi, 2024-02-05:
Hey @mrousavy I'm a relatively new sponsor, so I'm not certain how asking for issue prioritisation works. If it's just asking here in the comments, then I'd really appreciate if you could focus on this issue too. Thank you!

mrousavy, 2024-02-05:
Thanks for the sponsorship @Menardi, I'll take a look soon! 🙏

JshGrn, 2024-02-05:
I really don't want to sound ungrateful or rude here, but this issue is widespread and totally broken for all android users. Its a massive issue. I am not sure how you don't have it in your demo app. I also really feel like this is the only repo I have ever seen where in the issues are target sponsorship amounts.

Why not make this require a license where people have to pay for it instead if the issue is your time and needing to be funded? I completely agree that time is not free so I don't expect it for nothing.

The thing which stops me sponsoring is seeing your reply's say you will do it soon, what is soon, what am I sponsoring? Will it be this month, this year?

Dingenis, 2024-02-05:
> Thanks @Dingenis, which company?
> 
> I'll take a look at this soon 🙏

Hey @mrousavy, that's great! Thank you! Our company is [Spaces Experiences](https://github.com/Spaces-Experiences).

james-cohen, 2024-02-06:
Unfortunately this is still occurring even on the new beta release, albeit differently. Before the preview was stretched horizontally, but it now stretches vertically, then on refreshing the camera it stretches horizontally.

Interestingly, if I remove the videoAspectRatio prop from the useCameraFormat hook, it still stretches vertically on initialisation, but on refresh it is then normal.

mrousavy, 2024-02-06:
Created a PR to fix this issue once and for all: https://github.com/mrousavy/react-native-vision-camera/pull/2519 💪🚀 
 **Thanks to everyone who sponsored me on GitHub to fix this bug!!** 💖

Try the PR and see if it works (by checking out that branch and running the example app)

Menardi, 2024-02-06:
Thanks for your hard work @mrousavy! I've tested this fix in the example app on a Pixel 7a (Android 14), and a Pixel 3a (Android 12) and the view is no longer stretched when using the `fix/preview-stretching branch`. 🙌 

I did notice one unexpected side effect on the 7a, which is that the default zoom level seems to be much more zoomed in on the fixed branch. I didn't see this behaviour on the 3a, so I presume it must be something to do with the 7a having two rear cameras. Setting the prop `zoom={device.neutralZoom}` fixed this so the zoom level was consistent between `main` and `fix/preview-stretching`.

I don't think it's a bug, but thought it was worth mentioning as it might come up once the fix is released. I've included some photos taken on each branch below if you're interested.

Either way, thanks very much for the fix!

<details>
<summary>
Screenshots showing default zoom differences (Pixel 7a)
</summary>

| Default zoom: main | Default zoom: fix/preview-stretching |
|-----|---|
| [image] | [image] |
</details>

mrousavy, 2024-02-06:
> Thanks for your hard work @mrousavy! I've tested this fix in the example app on a Pixel 7a (Android 14), and a Pixel 3a (Android 12) and the view is no longer stretched when using the `fix/preview-stretching branch`. 🙌

Great to hear!!! 💪 

> I did notice one unexpected side effect on the 7a, which is that the default zoom level seems to be much more zoomed in on the fixed branch. I didn't see this behaviour on the 3a, so I presume it [...]

Huh, interesting. Maybe this has something to do with me not resetting the zoom on device change properly, but it is weird nevertheless... 🤔

closed 2024-02-07 by mrousavy

correafederico25, 2024-02-08:
Hello! I added version **3.9.0-beta.2** and now I am experiencing a different phenomenon. Before, the problem was that the camera preview stretched vertically, now the problem is that it stretches horizontally.

**Photo from React native vision camera**

[image]

**Photo in normal conditions**

[image]

Any suggestions or recommendations you can give me so this will work on Android? Here is my component.

```
    <Camera
      ref={cameraRef}
      style={StyleSheet.absoluteFill}
      device={device}
      isActive={isActive}
      photo={true}
    />
```

nonam4, 2024-02-09:
> Hello! I added version **3.9.0-beta.2** and now I am experiencing a different phenomenon. Before, the problem was that the camera preview stretched vertically, now the problem is that it stretches horizontally.
> 
> **Photo from React native vision camera**
> 
> ![Screenshot_20240208-123014](https://private-user-images.githubusercontent.com/53522327/303391193-faa4febf-8cc0-420a-b244-eda26645f78a.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDc1MTE5MjAsIm5iZiI6MTcwNzUxMTYyMCwicGF0aCI6Ii81MzUyMjMyNy8zMDMzOTExOTMtZmFhNGZlYmYtOGNjMC00MjBhLWIyNDQtZWRhMjY2NDVmNzhhLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAyMDklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMjA5VDIwNDcwMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTE1MDg1MDViNTI1OTc0OGFmOTJiNjJmZWY0NjYwZGQ5OWE1OGQyN2FkMTA5ZDExYWE3NDY0YWNlZmMxOTRkODUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.XsWO7SinEEEl6DhawcELCzqv71m9lyGIkBs045SrkkY)
> 
> **Photo in normal conditions**
> 
> ![Screenshot_20240208-123024](https://private-user-images.githubusercontent.com/53522327/303391395-0eb21b67-642c-4790-900f-224e26868d5e.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDc1MTE5MjAsIm5iZiI6MTcwNzUxMTYyMCwicGF0aCI6Ii81MzUyMjMyNy8zMDMzOTEzOTUtMGViMjFiNjctNjQyYy00NzkwLTkwMGYtMjI0ZTI2ODY4ZDVlLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAyMDklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMjA5VDIwNDcwMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWY2YWI1NTAyYmE3MzRlM2JjOGU4MTcxYjc2YzUyNjAzMmQ0N2NhNTg5NzFkOTIzOTgzZGM4ODlmZWEwZWNkZWImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.m2fAMstSDJFqs1Vgw4us2NuUr66eBPuhg0EvzSVWgMw)
> 
> Any suggestions or recommendations you can give me so this will work on Android? Here is my component.
> 
> ```
>     <Camera
>       ref={cameraRef}
>       style={StyleSheet.absoluteFill}
>       device={device}
>       isActive={isActive}
>       photo={true}
>     />
> ```

same issue here on my old Galaxy J5

nonam4, 2024-02-12:
> Hello! I added version **3.9.0-beta.2** and now I am experiencing a different phenomenon. Before, the problem was that the camera preview stretched vertically, now the problem is that it stretches horizontally.

@mrousavy I'm currently with this exactly same issue but looking into 3470b4f I can see that you inverted some width/height props, reverting these props to correct width/height order all stretching issues are gone.

Here's a patch I made:
```
diff --git a/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/CameraSession.kt b/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/CameraSession.kt
index fef6998..c339ea7 100644
--- a/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/CameraSession.kt
+++ b/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/CameraSession.kt
@@ -313,7 +313,7 @@ class CameraSession(private val context: Context, private val cameraManager: Cam
       )
       outputs.add(output)
       // Size is usually landscape, so we flip it here
-      previewView?.setSurfaceSize(size.width, size.height)
+      previewView?.setSurfaceSize(size.height, size.width)
     }
 
     // CodeScanner Output
diff --git a/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/PreviewView.kt b/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/PreviewView.kt
index c862be7..2d26fe9 100644
--- a/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/PreviewView.kt
+++ b/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/PreviewView.kt
@@ -43,13 +43,13 @@ class PreviewView(context: Context, callback: SurfaceHolder.Callback) : SurfaceV
     holder.addCallback(callback)
   }
 
-  suspend fun setSurfaceSize(width: Int, height: Int) {
+  suspend fun setSurfaceSize(height: Int, width: Int) {
     withContext(Dispatchers.Main) {
-      size = Size(width, height)
+      size = Size(height, width)
       Log.i(TAG, "Setting PreviewView Surface Size to $size...")
       requestLayout()
       invalidate()
-      holder.resize(width, height)
+      holder.resize(height, width)
     } 
   }
 
@@ -73,7 +73,7 @@ class PreviewView(context: Context, callback: SurfaceHolder.Callback) : SurfaceV
 
   private fun getSize(contentSize: Size, containerSize: Size, resizeMode: ResizeMode): Size {
     // TODO: Take sensor orientation into account here
-    val contentAspectRatio = contentSize.height.toDouble() / contentSize.width
+    val contentAspectRatio = contentSize.width.toDouble() / contentSize.height
     val containerAspectRatio = containerSize.width.toDouble() / containerSize.height
 
     val widthOverHeight = when (resizeMode) {
diff --git a/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/extensions/CameraCharacteristics+getPreviewSize.kt b/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/extensions/CameraCharacteristics+getPreviewSize.kt
index ecd525c..e758c45 100644
--- a/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/extensions/CameraCharacteristics+getPreviewSize.kt
+++ b/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/extensions/CameraCharacteristics+getPreviewSize.kt
@@ -9,7 +9,7 @@ fun getMaximumPreviewSize(): Size {
   // See https://developer.android.com/reference/android/hardware/camera2/params/StreamConfigurationMap
   // According to the Android Developer documentation, PREVIEW streams can have a resolution
   // of up to the phone's display's resolution, with a maximum of 1920x1080.
-  val display1080p = Size(1920, 1080)
+  val display1080p = Size(1080, 1920)
   val displaySize = Size(
     Resources.getSystem().displayMetrics.widthPixels,
     Resources.getSystem().displayMetrics.heightPixels
diff --git a/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/extensions/SurfaceHolder+resize.kt b/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/extensions/SurfaceHolder+resize.kt
index cc0fe9c..6ef3880 100644
--- a/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/extensions/SurfaceHolder+resize.kt
+++ b/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/extensions/SurfaceHolder+resize.kt
@@ -7,7 +7,7 @@ import kotlin.coroutines.resume
 import kotlinx.coroutines.suspendCancellableCoroutine
 
 @UiThread
-suspend fun SurfaceHolder.resize(width: Int, height: Int) {
+suspend fun SurfaceHolder.resize(height: Int, width: Int) {
   return suspendCancellableCoroutine { continuation ->
     val currentSize = this.surfaceFrame
     if (currentSize.width() == width && currentSize.height() == height) {

```

mrousavy, 2024-02-13:
Okay yea then I'm guessing this is related to the Orientation issue (https://github.com/mrousavy/react-native-vision-camera/issues/1891), because the devices you are testing on have a different Camera Hardware-Sensor orientation than my devices. We need to take that into account.

inzqne, 2024-03-05:
> Okay yea then I'm guessing this is related to the Orientation issue (#1891), because the devices you are testing on have a different Camera Hardware-Sensor orientation than my devices. We need to take that into account.

@mrousavy How come all of the stretching issues (horizontal and vertical) are not present in 3.6.4? Were there any features or changes that altered the resizing methods after 3.6.4?

JshGrn, 2024-03-05:
I had these issues on 3.6.4 so it wasn't introduced there, I think its a mixture of things for multiple devices.

Interestingly though, this is now for me broken in a way that I can work around 100% of the time. The initial camera loads with the width and height inverted. The 'fix' is to set them to 0 and activate the camera, and then in a useEffect set them to the devices width and height again. This loads the camera correctly, although is lower quality to the native camera app.

JshGrn, 2024-03-05:
Its a hell of a lot more manageable now for those that do have the issue @mrousavy, so much appreciated. The camera is *much* quicker as well.

mrousavy, 2024-03-05:
Thanks for your feedback @JshGrn, building in open-source I normally only hear negative feedback/bug reports - good to hear some positive feedback once in a while ❤️ 

But yea, overall there's A LOT of effort and thinking and careful design construction that goes into the codebase of VisionCamera, especially on Android, to make it predictable, thread-safe, fast and stable. This is something that's really hard to achieve with Camera2, so I'm really happy about how it works right now, even though the Preview stretching issue is still bugging me.

inzqne, 2024-03-05:
> Its fixable by simply activating the camera with a 0 width and height and then setting it in a useEffect right away, then onwards its perfect.
> 
> 
> 
> I was getting crashes a lot when I kept the camera active whilst swapping between react-navigation stacks, but to fix the crash I simply deactivated the camera on blur and reactivated it on focus, that stopped all the crashing.
> 
> 
> 
> I definitely don't use it for the full features this package offers, but for the ones I do its fantastic, and now I can work around it then its fine for my use-case.
> 
> 
> 
> Much appreciated 

This is very helpful as I'm experiencing the same issues. Just for reference, what version of vision camera are you using with this workaround @JshGrn ? As @mrousavy mentioned, some of the older versions are unpredictable. Although it seems like I'm getting crashes on any version with slow androids when navigating through the stack so this workaround might be a step in the right direction. Thank you.

JshGrn, 2024-03-05:
> > Its fixable by simply activating the camera with a 0 width and height and then setting it in a useEffect right away, then onwards its perfect.
> > I was getting crashes a lot when I kept the camera active whilst swapping between react-navigation stacks, but to fix the crash I simply deactivated the camera on blur and reactivated it on focus, that stopped all the crashing.
> > I definitely don't use it for the full features this package offers, but for the ones I do its fantastic, and now I can work around it then its fine for my use-case.
> > Much appreciated
> 
> This is very helpful as I'm experiencing the same issues. Just for reference, what version of vision camera are you using with this workaround @JshGrn ? As @mrousavy mentioned, some of the older versions are unpredictable. Although it seems like I'm getting crashes on any version with slow androids when navigating through the stack so this workaround might be a step in the right direction. Thank you guys for your work. 🚀

I am using the latest at the time of writing (3.9.1). The app crashes if the camera stays active whilst navigating to another stack and then back to the camera, not the first time but if I was to scan -> go to stack -> scan -> go to stack, it will eventually crash. What I am now doing is:

On the camera component in the camera stack view:
```
isActive={isActive && scanEnabled}
```

and 
```
useEffect(() => {
    return navigation.addListener('focus', () => setScanEnabled(true));
}, [navigation]);

useEffect(() => {
    return navigation.addListener('blur', () => setScanEnabled(false));
}, [navigation]);
```

isActive is defined on the View which wraps all of the components that is set to true with onLayout:

```
<View onLayout={() => processLayout()}>
stuff here
</View>
```

that function essentially does some logic to deal with the aspect ratio by waiting 250ms and then setting the width and height properly. The fix to stop the crashing is by listening to the listeners of blur and focus, without those I can crash the app every time by quick navigation back and forth from the camera stack.

lucksp, 2024-03-10:
I am seeing this issue on `3.9.2`

lucksp, 2024-03-10:
Adding the suggestion

```
  useEffect(() => {
    if (Platform.OS === 'ios') return;

    const timeout = setTimeout(
      () => setStyle({ height: '100%', width: '100%' }),
      250
    );

    return () => clearTimeout(timeout);
  }, []);
```
does not resolve for Android

EvgenyRub, 2024-03-11:
The hack for android is not working for me, still stretched on first mount... what version does not have this issue?

lucksp, 2024-03-11:
> What I'm doing instead is the following and it works flawlessly on 3.9.2 by executing right when the camera is ready and won't rely on a fixed time to update the camera preview width:
> 
> ```
> <Camera
>       device={device}
>       style={{...StyleSheet.absoluteFill, width: Dimensions.get('screen').width - 1 + stretchFix}}
>       onInitialized={() => setStretchFix(1)} />
> ```

what is `stretchFix` @inzqne ?

JshGrn, 2024-03-12:
> what is `stretchFix` @inzqne ?

this will just be a useState(0) judging by the use.

abisalde, 2024-03-18:
`const [cameraActive, setCameraActive] = React.useState(false);

          <View style={styles.frame}>
              <Camera
                ref={cameraRef}
                style={cameraActive && styles.cameraPreview}
                device={device}
                isActive={isActive}
                photo={true}
                onInitialized={() => setCameraActive(true)}
              />

          </View>`
          
          My temporary solution on Android 12 and below

PetrVasilev, 2024-03-19:
Still have this issue in 3.9.1, stretched on first camera open

CMonjo, 2024-03-19:
Same problem here for 3.9.0

inzqne, 2024-03-19:
> Still have this issue in 3.9.1, stretched on first camera open

> Same problem here for 3.9.0

I posted a work-around earlier in the thread that works flawlessly on 3.9.2, might be worth trying it out. Note that it is on **3.9.2**

CMonjo, 2024-03-19:
> > Still have this issue in 3.9.1, stretched on first camera open
> 
> > Same problem here for 3.9.0
> 
> I posted a work-around earlier in the thread that works flawlessly on 3.9.2, might be worth trying it out. Note that I am using it on _**3.9.2**_ specifically.

Hey @inzqne, thanks for your reply! The package does not have version 3.9.2 on npm. How do I install it?

[11 more replies]