margelo/react-native-vision-camera #2143: 🐛

Relie9828, 2023-11-08:
### What's happening?

I'm trying to use the react-native-vision-camera library in my application, but the camera is not mounting because the onInitialized callback is not being called. I'm not exactly sure why this is happening as I've followed the documentation closely.

### Reproduceable Code

```tsx
import { Camera, useCameraDevices, useCameraPermission } from 'react-native-vision-camera';
import { View, StyleSheet, TouchableOpacity, Text, Image, } from 'react-native';
import React, { useEffect, useState, useRef } from 'react';

function App() {
    
    const { hasPermission, requestPermission } = useCameraPermission();

    const camera = useRef(null);
    const devices = useCameraDevices('back');

    const [isCameraReady, setIsCameraReady] = useState(false);
    const [showCamera, setShowCamera] = useState(false);
    const [permissionAuth, setPermissionsAuth] = useState('')
    const [imageSource, setImageSource] = useState('');

    useEffect(() => {
        requestPermission();
        const checkPermission = async () => {
            const status = await Camera.getCameraPermissionStatus();
            setPermissionsAuth(status)
            if (status !== 'granted') {
                return null
            } else if (status === 'granted') {
                setShowCamera(true)
            }
        };

        checkPermission();
    }, []);

    // Checking if the permission are true
    // To redirect to settings if camera is false but the status is granted
    console.log(hasPermission, permissionAuth);

    const capturePhoto = async () => {
        console.log(isCameraReady)
        if (camera.current !== null && isCameraReady) {
            const photo = await camera.current.takePhoto({});
            setImageSource(photo.path);
            setShowCamera(false);
            console.log(photo.path);
        }
    };

    if (devices == null) {
        return <Text style={{ color: 'white', }}>Camera not available</Text>;
    };

    return (
        <View style={styles.container}>
            {showCamera ? (
                <>
                <Camera
                    ref={camera}
                    style={[StyleSheet.absoluteFill, {backgroundColor: 'black',}]}
                    device={devices}
                    isActive={showCamera}
                    resizeMode='cover'
                    onInitialized={() => setIsCameraReady(true)}
                    photo
                />

[... 184 lines cut ...]
  "name": "Back Dual Camera",
  "minZoom": 1,
  "sensorOrientation": "landscape-right"
}
```

### Device

iPhone 14 pro (17.0.0) & iPhone 15 pro (17.0.0)

### VisionCamera Version

3.6.4

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [X] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

bglgwyng, 2023-11-09:
I've experienced a similar thing. Did you turn off/on your app and see the camera work well again? In my case, just turning off/on the app didn't work. I had to turn off/on my phone.

Relie9828, 2023-11-09:
> I've experienced a similar thing. Did you turn off/on your app and see the camera work well again? In my case, just turning off/on the app didn't work. I had to turn off/on my phone.

Yeah I tried that as well, but unfortunately both attempts did not work.

Relie9828, 2023-11-09:
I found others with similar  [https://github.com/mrousavy/react-native-vision-camera/discussions/1363](issues)

Relie9828, 2023-11-09:
#1363

mrousavy, 2023-11-10:
Huh? You're saying you use VisionCamera 3.6.x but you use the `useCameraDevices` hook? In V3 you need `useCameraDevice`.

Relie9828, 2023-11-10:
> Huh? You're saying you use VisionCamera 3.6.x but you use the `useCameraDevices` hook? In V3 you need `useCameraDevice`.

I corrected the typo bu the CameraView is still not mounted.

flaviodamaiajr, 2023-11-10:
I have the same issue. Any idea how to solve it?

mrousavy, 2023-11-15:
Can you please share a video recording of what's happening? I cannot reproduce this on an iPhone 15 Pro (17.0.0) with the versions you sent.

[image]

Relie9828, 2023-11-15:
> Can you please share a video recording of what's happening? I cannot reproduce this on an iPhone 15 Pro (17.0.0) with the versions you sent.
> 
> [image]

Unfortunate not since, the 15 pro I'm using is on emulator but the behavior is the same as 14 pro on my end.

mrousavy, 2023-11-15:
Then please do the screen recording on an iPhone 14.

mrousavy, 2023-11-15:
Ah no sorry - I meant of the Example app. You said you could reproduce it here in the Example app

gaabrielricci, 2023-11-30:
tenho o mesmo problema aqui, com IOS 17. no meu caso é no iphone 14. A câmera não está inicializando.

rodgomesc, 2023-11-30:
@Relie28 when he says "example app", he's talking specifically about the `example app provided in this repository with the camera lib, in `package/example`

mrousavy, 2024-01-05:
Hey! Thanks for reporting this issue.

I've been working the past days on making sure the Camera lifecycle is safely handled and released when needed to make sure this crash doesn't occur anymore.

I just created a PR for this - can you please this to see if that fixes the issue for you? https://github.com/mrousavy/react-native-vision-camera/pull/2339

If this fixes your issue, please consider [💖 **sponsoring me on GitHub** 💖](https://github.com/sponsors/mrousavy/) to support me / thank me for building VisionCamera and continuously improving it. 

If this does not fix your issue, please clone the repo, check out the branch `fix/blackscreen` (the PR above), run the Example app, reproduce your issue there and share the `adb logcat` logs with me (upload to pastebin or gist) so I can investigate this further.

Thank you! 🙏

closed 2024-01-08 by mrousavy