margelo/react-native-vision-camera #2146: Not able to scan certain barcode

halaharr, 2023-11-09:
### What's happening?

App not able to scan certain barcodes when the barcode is white on transparent background. It used to work with react-native-vision-camera (V2.16.4) and vision-camera-code-scanner(V0.2.0).
Please see the attached image where the barcode is white in transparent background.

[image]

### Reproduceable Code

```tsx
import { BarcodeFormat, scanBarcodes } from 'vision-camera-code-scanner'
import { runOnJS } from 'react-native-reanimated'
import { useCameraDevices, useFrameProcessor } from 'react-native-vision-camera'
import { Camera } from 'react-native-vision-camera'

  const frameProcessor = useFrameProcessor(frame => {
    'worklet'
    const detectedBarcodes = scanBarcodes(
      frame,
      [
        BarcodeFormat.CODE_128,
        BarcodeFormat.CODE_39,
        BarcodeFormat.CODE_93,
        BarcodeFormat.CODABAR,
        BarcodeFormat.EAN_13,
        BarcodeFormat.EAN_8,
        BarcodeFormat.ITF,
        BarcodeFormat.UPC_A,
        BarcodeFormat.UPC_E
      ],
      { checkInverted: true }
    )
    runOnJS(setBarcodes)(detectedBarcodes)
  }, [])

 <Camera
              style={StyleSheet.absoluteFill}
              device={device}
              isActive={keepScanning}
              frameProcessor={frameProcessor}
              frameProcessorFps={5}
              audio={false}
            />
```

### Relevant log output

```shell
2023-11-08 11:40:29.168  7592-7592  CreateCaptureSession    com.packagename                I  Using new API (>=28)
2023-11-08 11:40:29.179  7592-9744  BufferQueueProducer     com.packagename                E  [ImageReader-1280x720f23m2-7592-1] queueBuffer: BufferQueue has been abandoned
2023-11-08 11:40:29.215  7592-9744  BufferQueueProducer     com.packagename                E  [ImageReader-1280x720f23m2-7592-1] queueBuffer: BufferQueue has been abandoned
2023-11-08 11:40:29.251  7592-9766  BufferQueueProducer     com.packagename                E  [ImageReader-1280x720f23m2-7592-1] queueBuffer: BufferQueue has been abandoned
2023-11-08 11:40:29.336  7592-9766  BufferQueueProducer     com.packagename                E  [ImageReader-1280x720f23m2-7592-1] queueBuffer: BufferQueue has been abandoned
2023-11-08 11:40:29.354  7592-9755  libc                    com.packagename                A  Fatal signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x78760abc3d in tid 9755 (pool-10-thread-), pid 7592 (package name)
2023-11-08 11:40:29.400  7592-9669  CreateCaptureSession    com.packagename                I  Camera 0: Capture Session #1002 configured!
```

### Camera Device

```json
{
  "sensorOrientation": "landscape-right",
  "hardwareLevel": "full",
  "maxZoom": 10,
  "minZoom": 0.6000000238418579,
  "supportsLowLightBoost": true,
  "neutralZoom": 1,
  "physicalDevices": [
    "ultra-wide-angle-camera",
    "wide-angle-camera",
    "telephoto-camera"
  ],
  "supportsFocus": true,
  "supportsRawCapture": true,
  "isMultiCam": true,
  "name": "BACK (0)",
  "hasFlash": true,
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
```

### Device

Samsung Galaxy S22 (Android 13)

### VisionCamera Version

3.6.4

### Can you reproduce this issue in the VisionCamera Example app?

No, I cannot reproduce the issue in the Example app

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2023-11-10:
Hey - this is related to the code scanner FP plugin you are using.

mrousavy, 2023-11-10:
please create an issue there

closed 2023-11-10 by mrousavy