margelo/react-native-vision-camera #2147: 🐛 It crashes on Android after scanning a QRcode

lvothnrv, 2023-11-09:
### What's happening?

Hello, I am on the "react-native-vision-camera" version: "^3.6.4", and I use it to scan qrcodes. On iOS everything is good and on Android after a scan the application crashes and I don't understand why. I also have Crashlytics and I see that it tells me: Caused by java.lang.IllegalStateException
Screen fragments should never be restored. Follow instructions from https://github.com/software-mansion/react-native-screens/issues/17#issuecomment-424704067 to properly configure your main activity.

### Reproduceable Code

```tsx
const device = useCameraDevice('back');

  const codeScanner = useCodeScanner({
    codeTypes: ['qr', 'ean-13', 'code-128'],
    onCodeScanned: codes => {
      if (codes?.[0]?.value) {
        setDoScanBarcode(false);
        getCode(codes[0].value);
      }
    },
  });
 
{device != null && (
 <Camera
  device={device}
  isActive={true}
  codeScanner={codeScanner}
  style={qrCodeStyle.viewCamera}
  />
)}
```

### Relevant log output

```shell
11-09 21:46:09.433  4924  4924 I PreviewView: Creating PreviewView...
11-09 21:46:09.433  4924  4924 I PreviewView: PreviewView onMeasure(0, 0)
11-09 21:46:09.433  4924  4924 D PreviewView: coverSize :: 1920x1080 (0.5625), 0x0 (NaN)
11-09 21:46:09.434  4924  4924 D PreviewView: Fitted dimensions set: 0x0
11-09 21:46:09.439  4924  4924 I PreviewView: Resizing PreviewView to 1280 x 960...
11-09 21:46:09.439  4924  4924 I CameraView: Updating CameraSession...
11-09 21:46:09.441  4924  5100 I CameraSession: Updating CameraSession Configuration...
11-09 21:46:09.441  4924  5100 I CameraSession: Configuring Camera #0...
11-09 21:46:09.442  4924  5100 I CameraManager: Camera 0: Opening...
11-09 21:46:09.450  4924  4924 I CameraView: Updating CameraSession...
11-09 21:46:09.451 30491 30491 I CameraService: CameraService::connect call (PID -1 "com.areahunter", camera ID 0) for HAL version default and Camera API version 2
11-09 21:46:09.452 30491 30491 E CameraService: CameraService::connect evicting conflicting client for camera ID 0
11-09 21:46:09.452  4924  4924 I CameraView: Updating CameraSession...
11-09 21:46:09.463   300 31709 W EmulatedCamera_QemuCamera3: flush: Not implemented; ignored
11-09 21:46:09.463 30491 30491 I Camera3-Device: disconnectImpl: E
11-09 21:46:09.464 30491 30491 I CameraLatencyHistogram: ProcessCaptureRequest latency histogram (8) samples:
11-09 21:46:09.464 30491 30491 I CameraLatencyHistogram:        40     80    120    160    200    240    280    320    360    inf (max ms)
11-09 21:46:09.464 30491 30491 I CameraLatencyHistogram:      87.50   0.00   0.00   0.00   0.00   0.00   0.00   12.50   0.00   0.00 (%)
11-09 21:46:09.486 30491 30491 I Camera3-Device: disconnectImpl: X
11-09 21:46:09.488  4924  5018 I BpBinder: onLastStrongRef automatically unlinking death recipients: <uncached descriptor>
11-09 21:46:09.488 30491 30491 I CameraService: disconnect: Disconnected client for camera 0 for PID 4924
11-09 21:46:09.488 30491 30491 I Camera2ClientBase: Camera 0: Opened. Client: com.areahunter (PID 4924, UID 10155)
11-09 21:46:09.488 30491 30491 I CameraDeviceClient: CameraDeviceClient 0: Opened
11-09 21:46:09.491  4924  5036 I CameraDevices: Camera #0: Available!
11-09 21:46:09.493  4924  5036 I CameraDevices: Camera #0: Unavailable!
11-09 21:46:09.494   300 14656 E libc    : Access denied finding property "ro.camera.req.fmq.size"
11-09 21:46:09.495   300 14656 E FMQ     : grantorIdx must be less than 3
11-09 21:46:09.495   300 14656 E libc    : Access denied finding property "ro.camera.res.fmq.size"
11-09 21:46:09.495   300 14656 E FMQ     : grantorIdx must be less than 3
11-09 21:46:09.496 30491 30491 E FMQ     : grantorIdx must be less than 3
11-09 21:46:09.496 30491 30491 E FMQ     : grantorIdx must be less than 3
11-09 21:46:09.501  4924  5100 I CameraManager: Camera 0: Disconnected!
11-09 21:46:09.490   300   300 W HwBinder:300_5: type=1400 audit(0.0:350): avc: denied { read } for name="u:object_r:default_prop:s0" dev="tmpfs" ino=10352 scontext=u:r:hal_camera_default:s0 tcontext=u:object_r:default_prop:s0 tclass=file permissive=0
11-09 21:46:09.508  4924  5100 E CameraSession: Camera Device android.hardware.camera2.impl.CameraDeviceImpl@ba4f176 has been disconnected!
11-09 21:46:09.508  4924  5100 E CameraSession: com.mrousavy.camera.core.CameraDisconnectedError: [session/camera-has-been-disconnected] The given Camera device (id: 0) has been disconnected! Error: DISCONNECTED
11-09 21:46:09.508  4924  5100 E CameraSession: 	at com.mrousavy.camera.extensions.CameraManager_openCameraKt$openCamera$2$callback$1.onDisconnected(CameraManager+openCamera.kt:38)
11-09 21:46:09.508  4924  5100 E CameraSession: 	at android.hardware.camera2.impl.CameraDeviceImpl$7.run(CameraDeviceImpl.java:247)
[... 58 lines cut ...]
  "minZoom": 1,
  "hasFlash": false,
  "name": "BACK (0)"
}
```

### Device

Samsung and Emulator Android

### VisionCamera Version

3.6.4

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2023-11-10:
Hey - thanks for sharing the logs. Finally someone gets it.

> on Android after a scan the application crashes and I don't understand why

It's not really clear to me how I can reproduce this though, you scan a code and then navigate away from the Camera page?

lvothnrv, 2023-11-10:
> Hey - thanks for sharing the logs. Finally someone gets it.
> 
> > on Android after a scan the application crashes and I don't understand why
> 
> It's not really clear to me how I can reproduce this though, you scan a code and then navigate away from the Camera page?

Hello - No problem for the logs :) In fact after scanning the qrcode the application crashes. I'm trying to understand why but I don't understand where it could come from, I wonder if it's not perhaps a question of performance, too much stack I don't know. Since I have Crashlytics on my app I can see some information that might help.
I show you what I coded and then what Crashlytics tells me.

[image]

[image]
You see the SIGSEGV I think it's linked to that.
[image]

lvothnrv, 2023-11-13:
> why do you send me the code as a screenshot 😂🤦‍♂️

In fact it's the first time I send a code on github and when I did I use the tag it's doing anything to me so I didn't break my head I did a screenshoot 😂😭

mare95, 2023-11-19:
This is happening to me as well, unfortunately I cannot reproduce this so that i can debug it...
Im getting tons of crashes related to this, any ideas why? This started happening from V3.6.4

[image]

mrousavy, 2023-11-20:
Hey - I need symbolicated stack traces. There's a Sentry tutorial telling you how you can get those @mare95

Saif-Ur-Rehman19, 2023-11-20:
Is there any solution to this problem?  When the app crashes there is nothing in the log to look at for debugging

afcanop, 2023-11-20:
I have this same error, when reading QR codes, any solution?

santagli-bds, 2023-11-22:
I have this same error

Saif-Ur-Rehman19, 2023-11-22:
Hello folks. I have had the  same issue. I fixed it by disabling the isActive prop after the barcode/qrcode. Also do not close the camera automatically. Add a button to close and hide the component rendering the camera.

emaxedon, 2023-12-07:
I got the same issue, and as @Saif-Ur-Rehman19 stated, disabling `isActive` helps, but what I also did was I used a custom `useDebounce` hook and applied a timeout of like 300ms to the `useState` that I call in `onCodeScanned` so that the camera had time to disable before switching into the next component.

mrousavy, 2024-01-05:
Hey! Thanks for reporting this issue.

I've been working the past days on making sure the Camera lifecycle is safely handled and released when needed to make sure this crash doesn't occur anymore.

I just created a PR for this - can you please this to see if that fixes the issue for you? https://github.com/mrousavy/react-native-vision-camera/pull/2339

If this fixes your issue, please consider [💖 **sponsoring me on GitHub** 💖](https://github.com/sponsors/mrousavy/) to support me / thank me for building VisionCamera and continuously improving it. 

If this does not fix your issue, please clone the repo, check out the branch `fix/blackscreen` (the PR above), run the Example app, reproduce your issue there and share the `adb logcat` logs with me (upload to pastebin or gist) so I can investigate this further.

Thank you! 🙏

closed 2024-01-08 by mrousavy

tankhang1, 2024-02-18:
I had the same issue. However, after I controlled isActive by another variable, I changed isActive to false before closing this component. I will fix this problem