margelo/react-native-vision-camera #2149: 🐛 Camera is exclusively lock to react-native-vision-camera on Android

sapjunior, 2023-11-10:
### What's happening?

Hello, we're encountering an issue with react-native-vision-camera. It appears that when the camera is initiated within the app on Android, the library exclusively locks the camera, preventing other libraries (in same react-native-program) that also require camera access from gaining control.

As an example, we are using both react-native-vision-camera and react-native-torch (utilized for opening the flashlight, which camera permissions on Android).

To replicate the issue:

1. Open the camera using react-native-vision-camera.
2. Attempt to activate the flashlight using react-native-torch.
We got this error message "CAMERA_IN_USE (4): setTorchMode:2122: Torch for camera "0" is not available due to an existing camera user" even if the camera modal/screen is dismissed. (https://github.com/ludo/react-native-torch/blob/957db6330a0afd09455e42ce6537e0ca69518e3c/android/src/main/java/com/cubicphuse/RCTTorch/RCTTorchModule.java#L47)

We suspected that this function (https://github.com/mrousavy/react-native-vision-camera/blob/be1bb419c16c68f9982d0f67ff97513696f587f5/package/android/src/main/java/com/mrousavy/camera/core/CameraSession.kt#L98) is not call at all even the screen dismiss/close, which may make the react-native-vision-camera exclusive lock

### Reproduceable Code

```tsx
Example camera code from react-native-vision-camera repos with a single function call to react-native-torch
```

### Relevant log output

```shell
Error toggling flashlight Error: CAMERA_IN_USE (4): setTorchMode:2122: Torch for camera "0" is not available due to an existing camera user
```

### Camera Device

```json
{
  "sensorOrientation": "landscape-right",
  "hardwareLevel": "full",
  "maxZoom": 8,
  "minZoom": 1,
  "supportsLowLightBoost": false,
  "neutralZoom": 1,
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "supportsFocus": true,
  "supportsRawCapture": true,
  "isMultiCam": false,
  "name": "BACK (0)",
  "hasFlash": true,
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
```

### Device

Multiple Android Devices (SS Note9/Xiaomi Redmi Note9)

### VisionCamera Version

3.6.4

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

SocDario, 2023-11-15:
Just a question, are you using torch on some component which don't includes camera? If not try to use torch which is provided from vision-camera.

mrousavy, 2023-11-18:
Hey! I just found out that I really forgot to close and dispose the locked Camera resources, so I just fixed that in this PR: https://github.com/mrousavy/react-native-vision-camera/pull/2174 
Now the Camera fully & synchronously closes all resources (CameraSession, CameraDevice, OpenGL context, Video & Photo outputs, Photo Synchronizer) once the view gets removed from the React view hierarchy ("unmounted"), and things like Flash, Torch, NFC, and other Camera components should be working again.

There is still a small issue that causes once Camera component to turn into a blackscreen when navigating back and forth between two Camera components, that's a pretty rare edge case but I will still try to fix that soon when I have some free time.

If you appreciate my time, expertise and dedication to this project, pleas [💖 consider sponsoring me on GitHub 💖](https://github.com/sponsors/mrousavy) to support the development of this project.

PR #2174 (2023-11-18): fix: Close `CameraSession` if the View is removed

## What

Closes the `CameraSession` and performs necessary cleanup (`CaptureSession`, `CameraDevice`, OpenGL context, Video & Photo outputs, `RecordingSession`, `FrameProcessor` and Photo Synchronizer) to free the Camera Hardware once the View is removed from the React hierarchy ("unmounted").

I previously simply forgot to do that. lol.

## Changes

## Tested on

## Related issues

* Fixes #2149 
* Fixes https://github.com/mrousavy/react-native-vision-camera/issues/2152
* Fixes https://github.com/
closed 2023-11-18 by mrousavy