margelo/react-native-vision-camera #2152: 🐛 RNVC v3 stops NFC from working

pasquale95, 2023-11-10:
### What's happening?

I have an application that can scan NFC tags and has a Camera.
I recently moved to RNVC v3 and I experience the following problem:
- I open the app and I scan an NFC tag: everything works.
- I open the Camera screen and the `<Camera>` is mounted;
- I close the Camera screen (and the `<Camera>` is supposed to be unmounted);
- I try to scan the NFC tag and the NFC reader does not start anymore.

If I switch back to RNVC v2, this problem disappears.
This problem has been already mentioned once (#516), but in that issue the problem was related to RNVC v2, while I experience now this problem with RNVC v3.

Some additional information:
- the problem is not related to `react-native-worklets-core` since I don't use them;
- according to some posts on stack-overflow, on some phones you cannot use NFC together with Camera, which is fine in my case, but this brings me to think that maybe the Camera is not correctly unmounted in RNVC v3.
- setting `isActive` to `false` when unmounting does not solve the problem.
- I use `react-native-nfc-manager` as NFC library.

The only way I have to restart the NFC is by closing and reopening the app, and as soon as I mount the Camera again the problem pops up again.

### Reproduceable Code

```tsx
const device = useCameraDevice('back');
  const { hasPermission, requestPermission } = useCameraPermission();
  const isFocused = useIsFocused();

  useEffect(() => {
    (async () => {
      if (!hasPermission) {
        requestPermission();
      }
    })();
  }, [hasPermission, requestPermission]);

  const codeScanner = useCodeScanner({
    codeTypes: ['qr', 'ean-13'],
    onCodeScanned: async codes => {
      console.log('CODES: ' + JSON.stringify(codes));
    },
  });

  return device != null && hasPermission ? (
    <Camera
      style={styles.scanner}
      device={device}
      isActive={isFocused}
      codeScanner={codeScanner}
    />
  ) : null;
```

### Relevant log output

```shell
No errors, no weird logs in logcat, just NFC hanging forever.
```

### Camera Device

```json
{
   "sensorOrientation":"landscape-right",
   "hardwareLevel":"limited",
   "maxZoom":8,
   "minZoom":1,
   "supportsLowLightBoost":false,
   "neutralZoom":1,
   "physicalDevices":[
      "wide-angle-camera"
   ],
   "supportsFocus":true,
   "supportsRawCapture":false,
   "isMultiCam":false,
   "name":"BACK (0)",
   "hasFlash":true,
   "hasTorch":true,
   "position":"back",
   "id":"0"
}
```

### Device

Samsung A336B, Android 13 (API 33)

### VisionCamera Version

3.6.4

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2023-11-10:
Related to https://github.com/mrousavy/react-native-vision-camera/issues/516 - this one got some good details tho

mrousavy, 2023-11-10:
Okay so I guess maybe I am not closing something properly?

mrousavy, 2023-11-10:
> No errors, no weird logs in logcat, just NFC hanging forever.

Everybody leaves this out. I hate to explain myself everytime I think I'm gonna write a bot for this.

**I need the logs** because stuff like "Destroying Session" and "Creating Session" etc is relevant. I need to know what gets called.
Please update with full adb logcat logs

sapjunior, 2023-11-11:
@pasquale95 Could you please try to replace 
package/android/src/main/java/com/mrousavy/camera/core/CameraSession.kt with this one
https://gist.github.com/sapjunior/bf257c40ec423548c81504adb838d2b2
(We modify it a little bit to include close camera session in surfacedestroy logic)

We think that your issue may related to our issue in #2149

pasquale95, 2023-11-11:
> Please update with full adb logcat logs

Sorry @mrousavy, I've added the logs, from when I open the app, open the Camera and eventually close it.

pasquale95, 2023-11-11:
> @pasquale95 Could you please try to replace package/android/src/main/java/com/mrousavy/camera/core/CameraSession.kt with this one https://gist.github.com/sapjunior/bf257c40ec423548c81504adb838d2b2 (We modify it a little bit to include close camera session in surfacedestroy logic)
> 
> We think that your issue may related to our issue in #2149 which is related to CameraSession

It works! Now once I open the camera and I close it, the NFC reader works. Here the logs with the new code in case you want to compare with the old ones:

```shell
2023-11-11 18:28:24.731  1427-1480  WindowManager           system_server                        I  container=Task{8b29c0 #59 type=standard A=1010341:com.rntesterapp U=10 visible=true visibleRequested=true mode=fullscreen translucent=false sz=1}
                                                                                                    Target:
                                                                                                      mode=0 taskId=59 isTranslucent=false clipRect=[0,0][0,0] contentInsets=[0,80][0,135] prefixOrderIndex=19 position=[0,0] sourceContainerBounds=[0,0][1080,2400] screenSpaceBounds=[0,0][1080,2400] localBounds=[0,0][1080,2400]
                                                                                                      windowConfiguration={ mBounds=Rect(0, 0 - 1080, 2400) mAppBounds=Rect(0, 80 - 1080, 2265) mMaxBounds=Rect(0, 0 - 1080, 2400) mDisplayRotation=ROTATION_0 mWindowingMode=fullscreen mDisplayWindowingMode=fullscreen mActivityType=standard mAlwaysOnTop=undefined mRotation=ROTATION_0 mPopOver=undefined mStageConfig=undefined mFreeformTaskPinningState=unpinned mFreeformStashState=undefined}
                                                                                                      leash=Surface(name=Surface(name=Task=59)/@0x53d9c52 - animation-leash of app_transition)/@0xa0e8820
                                                                                                      taskInfo=TaskInfo{userId=10 taskId=59 displayId=0 isRunning=true baseIntent=Intent { act=android.intent.action.MAIN cat=[android.intent.category.LAUNCHER] flg=0x10200000 cmp=com.rntesterapp/.MainActivity } baseActivity=ComponentInfo{com.rntesterapp/com.rntesterapp.MainActivity} topActivity=ComponentInfo{com.rntesterapp/com.rntesterapp.MainActivity} origActivity=null realActivity=ComponentInfo{com.rntesterapp/com.rntesterapp.MainActivity} numActivities=1 lastActiveTime=206764561 supportsSplitScreenMultiWindow=true supportsMultiWindow=true resizeMode=1 isResizeable=true minWidth=-1 minHeight=-1 defaultMinSize=220 token=WCT{RemoteToken{6e1e67f Task{8b29c0 #59 type=standard A=1010341:com.rntesterapp U=10 visible=true visibleRequested=true mode=fullscreen translucent=false sz=1}}} topActivityType=1 pictureInPictureParams=null shouldDockBigOverlays=false launchIntoPipHostTaskId=0 displayCutoutSafeInsets=Rect(0, 80 - 0, 0) topActivityInfo=ActivityInfo{7a4064c com.rntesterapp.MainActivity} launchCookies=[android.os.BinderProxy@58a0695] positionInParent=Point(0, 0) parentTaskId=-1 isFocused=true isVisible=true isSleeping=false topActivityInSizeCompat=false topActivityEligibleForLetterboxEducation= false locusId=null displayAreaFeatureId=1 cameraCompatControlState=hidden originallySupportedSplitScreen=true hasWallpaper=false isTopTransparentActivity=false topActivityInFixedAspectRatio=false topActivityInDisplayCompat=false topActivityInBoundsCompat=false topActivityBounds=null}
                                                                                                      allowEnterPip=true
                                                                                                      windowType=-1  hasAnimatingParent=false  backgroundColor=0container=Task{43dac9e #1 type=home ?? U=0 visible=false visibleRequested=false mode=fullscreen translucent=true sz=1}
                                                                                                    Target:
                                                                                                      mode=1 taskId=7 isTranslucent=false clipRect=[0,0][0,0] contentInsets=[0,80][0,135] prefixOrderIndex=16 position=[0,0] sourceContainerBounds=[0,0][1080,2400] screenSpaceBounds=[0,0][1080,2400] localBounds=[0,0][1080,2400]
                                                                                                      windowConfiguration={ mBounds=Rect(0, 0 - 1080, 2400) mAppBounds=Rect(0, 80 - 1080, 2265) mMaxBounds=Rect(0, 0 - 1080, 2400) mDisplayRotation=ROTATION_0 mWindowingMode=fullscreen mDisplayWindowingMode=fullscreen mActivityType=home mAlwaysOnTop=undefined mRotation=ROTATION_0 mPopOver=undefined mStageConfig=undefined mFreeformTaskPinningState=unpinned mFreeformStashState=undefined}
                                                                                                      leash=Surface(name=Surface(name=Task=1)/@0xbdb186a - animation-leash of app_transition)/@0xf8798dd
2023-11-11 18:28:26.447 20555-20718 CameraManagerGlobal     com.rntesterapp                      I  Connecting to camera service
2023-11-11 18:28:26.452 20555-20569 CameraManagerGlobal     com.rntesterapp                      I  Camera 0 facing CAMERA_FACING_BACK state now CAMERA_STATE_CLOSED for client com.rntesterapp API Level 2User Id 10
[... cut ...]
2023-11-11 18:28:39.935 20555-20555 CameraView              com.rntesterapp                      I  Updating CameraSession...
2023-11-11 18:28:39.938 20555-20788 CameraSession           com.rntesterapp                      I  New Camera Session code...
2023-11-11 18:28:39.938 20555-20788 CameraSession           com.rntesterapp                      I  Updating CameraSession Configuration...
2023-11-11 18:28:39.938 20555-20788 CameraSession           com.rntesterapp                      W  Called configure(...) but nothing changed...
```

pasquale95, 2023-11-11:
@mrousavy @sapjunior when do you plan to release this patch?

mrousavy, 2023-11-13:
> @pasquale95 Could you please try to replace package/android/src/main/java/com/mrousavy/camera/core/CameraSession.kt with this one https://gist.github.com/sapjunior/bf257c40ec423548c81504adb838d2b2 [...]

What exactly is the diff you did here? Did you anything in `close()`?
If this fixes the issue, why do you guys never create PRs? Random patches flying around here won't help anyone, if this is really a good solution for the problem then contributing back to the project is what keeps open-source going.

pasquale95, 2023-11-13:
@sapjunior do you plan to create a PR for this patch? If you don't, I could take care it.
Let us know 😄 .

mrousavy, 2023-11-13:
I have a different idea that's a bit more stable

sapjunior, 2023-11-13:
After t

> What exactly is the diff you did here? Did you anything in `close()`? If this fixes the issue, why do you guys never create PRs? Random patches flying around here won't help anyone, if this is [...]

- After some testing, it seems to be unstable in some cases, for example, if we leave camera open, switch to recent app, then go back to app again. it will be crash
- We also want to create a PR, but we are not expert Android developers; we're just amateurs and not quite sure that is it working correctly or not very sorry about that

mrousavy, 2023-11-18:
Hey! I just found out that I really forgot to close and dispose the locked Camera resources, so I just fixed that in this PR: https://github.com/mrousavy/react-native-vision-camera/pull/2174 
Now the Camera fully & synchronously closes all resources (CameraSession, CameraDevice, OpenGL context, Video & Photo outputs, Photo Synchronizer) once the view gets removed from the React view hierarchy ("unmounted"), and things like Flash, Torch, NFC, and other Camera components should be working again.

There is still a small issue that causes once Camera component to turn into a blackscreen when navigating back and forth between two Camera components, that's a pretty rare edge case but I will still try to fix that soon when I have some free time.

If you appreciate my time, expertise and dedication to this project, pleas [💖 consider sponsoring me on GitHub 💖](https://github.com/sponsors/mrousavy) to support the development of this project.

closed 2023-11-18 by mrousavy

pasquale95, 2023-11-19:
Hi @mrousavy,
unfortunately I keep getting the same issue even after the latest update. I'm using RNVC v3.6.8 and once I open the Camera and close it, the NFC reader stops working. Here the logs I get this time:

1. Open the app and load the `<Camera>`:
```sh
2023-11-19 18:20:05.758  1426-1462  WindowManager           system_server                        I  container=Task{85f16de #219 type=standard A=1010337:com.rntesterapp U=10 visible=true visibleRequested=true mode=fullscreen translucent=false sz=1}
                                                                                                    Target:
                                                                                                      mode=0 taskId=219 isTranslucent=false clipRect=[0,0][0,0] contentInsets=[0,75][0,135] prefixOrderIndex=19 position=[0,0] sourceContainerBounds=[0,0][1080,2340] screenSpaceBounds=[0,0][1080,2340] localBounds=[0,0][1080,2340]
                                                                                                      windowConfiguration={ mBounds=Rect(0, 0 - 1080, 2340) mAppBounds=Rect(0, 75 - 1080, 2205) mMaxBounds=Rect(0, 0 - 1080, 2340) mDisplayRotation=ROTATION_0 mWindowingMode=fullscreen mDisplayWindowingMode=fullscreen mActivityType=standard mAlwaysOnTop=undefined mRotation=ROTATION_0 mPopOver=undefined mStageConfig=undefined mFreeformTaskPinningState=unpinned mFreeformStashState=undefined}
                                                                                                      leash=Surface(name=Surface(name=Task=219)/@0x1ad741c - animation-leash of app_transition)/@0x9e32825
                                                                                                      taskInfo=TaskInfo{userId=10 taskId=219 displayId=0 isRunning=true baseIntent=Intent { act=android.intent.action.MAIN cat=[android.intent.category.LAUNCHER] flg=0x10200000 cmp=com.rntesterapp/.MainActivity } baseActivity=ComponentInfo{com.rntesterapp/com.rntesterapp.MainActivity} topActivity=ComponentInfo{com.rntesterapp/com.rntesterapp.MainActivity} origActivity=null realActivity=ComponentInfo{com.rntesterapp/com.rntesterapp.MainActivity} numActivities=1 lastActiveTime=1756738108 supportsSplitScreenMultiWindow=true supportsMultiWindow=true resizeMode=1 isResizeable=true minWidth=-1 minHeight=-1 defaultMinSize=220 token=WCT{RemoteToken{2bb608 Task{85f16de #219 type=standard A=1010337:com.rntesterapp U=10 visible=true visibleRequested=true mode=fullscreen translucent=false sz=1}}} topActivityType=1 pictureInPictureParams=null shouldDockBigOverlays=false launchIntoPipHostTaskId=0 displayCutoutSafeInsets=Rect(0, 75 - 0, 0) topActivityInfo=ActivityInfo{64321a1 com.rntesterapp.MainActivity} launchCookies=[android.os.BinderProxy@25a8fc6] positionInParent=Point(0, 0) parentTaskId=-1 isFocused=true isVisible=true isSleeping=false topActivityInSizeCompat=false topActivityEligibleForLetterboxEducation= false locusId=null displayAreaFeatureId=1 cameraCompatControlState=hidden originallySupportedSplitScreen=true hasWallpaper=false isTopTransparentActivity=false topActivityInFixedAspectRatio=false topActivityInDisplayCompat=false topActivityInBoundsCompat=false topActivityBounds=null}
                                                                                                      allowEnterPip=true
                                                                                                      windowType=-1  hasAnimatingParent=false  backgroundColor=0container=Task{a450653 #1 type=home ?? U=0 visible=false visibleRequested=false mode=fullscreen translucent=true sz=1}
                                                                                                    Target:
                                                                                                      mode=1 taskId=7 isTranslucent=false clipRect=[0,0][0,0] contentInsets=[0,75][0,135] prefixOrderIndex=16 position=[0,0] sourceContainerBounds=[0,0][1080,2340] screenSpaceBounds=[0,0][1080,2340] localBounds=[0,0][1080,2340]
                                                                                                      windowConfiguration={ mBounds=Rect(0, 0 - 1080, 2340) mAppBounds=Rect(0, 75 - 1080, 2205) mMaxBounds=Rect(0, 0 - 1080, 2340) mDisplayRotation=ROTATION_0 mWindowingMode=fullscreen mDisplayWindowingMode=fullscreen mActivityType=home mAlwaysOnTop=undefined mRotation=ROTATION_0 mPopOver=undefined mStageConfig=undefined mFreeformTaskPinningState=unpinned mFreeformStashState=undefined}
                                                                                                      leash=Surface(name=Surface(name=Task=1)/@0xb3c772c - animation-leash of app_transition)/@0xfb8298f
2023-11-19 18:20:06.870  3873-3964  CameraManagerGlobal     com.rntesterapp                      I  Connecting to camera service
2023-11-19 18:20:06.876  3873-3889  CameraManagerGlobal     com.rntesterapp                      I  Camera 0 facing CAMERA_FACING_BACK state now CAMERA_STATE_CLOSED for client com.rntesterapp API Level 2User Id 10
2023-11-19 18:20:06.877  3873-3889  CameraManagerGlobal     com.rntesterapp                      I  Camera 1 facing CAMERA_FACING_FRONT state now CAMERA_STATE_CLOSED for client android.system API Level 2User Id 0
[... cut ...]
2023-11-19 18:21:38.316  7974-8927  CameraManagerGlobal     com.samsung.android.honeyboard       I  Camera 0 facing CAMERA_FACING_BACK state now CAMERA_STATE_CLOSED for client com.rntesterapp API Level 2User Id 10
2023-11-19 18:21:38.316  1003-4696  Camera2ClientBase       cameraserver                         I  Closed Camera 0. Client was: com.rntesterapp (PID 3873, UID 1010337)
```

From the JS persective nothing has changed, but now I see some error logs in logcat regarding `useCameraDevice()` (I use `const device = useCameraDevice('back');`) when I close the `<Camera>` at step 2.

@sapjunior did you try the latest version of RNVC? Did it solve the issue for you?

mrousavy, 2023-11-20:
```
2023-11-19 18:20:36.270 32753-503   CameraManagerGlobal     com...sung.android.vtcamerasettings  I  Camera 0 facing CAMERA_FACING_BACK state now CAMERA_STATE_CLOSED for client com.rntesterapp API Level 2User Id 10
2023-11-19 18:20:36.270  3039-9341  CameraManagerGlobal     com.sec.sve                          I  Camera 0 facing CAMERA_FACING_BACK state now CAMERA_STATE_CLOSED for client com.rntesterapp API Level 2User Id 10
2023-11-19 18:20:36.271 21632-21650 CameraManagerGlobal     com.android.cameraextensions         I  Camera 0 facing CAMERA_FACING_BACK state now CAMERA_STATE_CLOSED for client com.rntesterapp API Level 2User Id 10
2023-11-19 18:20:36.271  5112-11149 CameraManagerGlobal     com.sec.android.sdhms                I  Camera 0 facing CAMERA_FACING_BACK state now CAMERA_STATE_CLOSED for client com.rntesterapp API Level 2User Id 10
2023-11-19 18:20:36.271  2644-4932  CameraManagerGlobal     com.android.systemui                 I  Camera 0 facing CAMERA_FACING_BACK state now CAMERA_STATE_CLOSED for client com.rntesterapp API Level 2User Id 10
2023-11-19 18:20:36.272  1426-3628  CameraManagerGlobal     system_server                        I  Camera 0 facing CAMERA_FACING_BACK state now CAMERA_STATE_CLOSED for client com.rntesterapp API Level 2User Id 10

```

Hmm looks like it's closing.

The configure error is weird though, I think I know why! I'm doing stupid stuff here, let me rewrite that part properly to ensure it stays atomic.