margelo/react-native-vision-camera #2153: 🐛 App crashed(Not yet reproducible yet)

bglgwyng, 2023-11-11:
### What's happening?

My app crashed with the following error message. I don't know how to reproduce it yet, however, the crash logs have not been reported yet, I posted it as a draft.

### Reproduceable Code

```tsx
// don't know yet
```

### Relevant log output

```shell
FATAL EXCEPTION: mrousavy/VisionCamera.video
Process: com.gwyng.myapp, PID: 7047
com.facebook.jni.CppException: std::exception
    at com.mrousavy.camera.frameprocessor.VisionCameraScheduler.trigger(Native Method)
    at com.mrousavy.camera.frameprocessor.VisionCameraScheduler.$r8$lambda$Rh5vFX2pR6HZnaz5K-D0xg2S108(Unknown Source:0)
    at com.mrousavy.camera.frameprocessor.VisionCameraScheduler$$ExternalSyntheticLambda0.run(Unknown Source:2)
    at android.os.Handler.handleCallback(Handler.java:942)
    at android.os.Handler.dispatchMessage(Handler.java:99)
    at android.os.Looper.loopOnce(Looper.java:226)
    at android.os.Looper.loop(Looper.java:313)
    at android.os.HandlerThread.run(HandlerThread.java:67)
```

### Camera Device

```json
{"hardwareLevel": "full", "hasFlash": true, "hasTorch": true, "id": "0", "isMultiCam": false, "maxZoom": 8, "minZoom": 1, "name": "BACK (0)", "neutralZoom": 1, "physicalDevices": ["wide-angle-camera"], "position": "back", "sensorOrientation": "landscape-right", "supportsFocus": true, "supportsLowLightBoost": true, "supportsRawCapture": true}
```

### Device

Galaxy S21

### VisionCamera Version

3.6.4

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [X] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

iliapnmrv, 2023-12-08:
I also cannot reproduce this issue but my users get it. `RNVC v3.6.11` 

HONOR HWYAL, Android 10

mrousavy, 2024-01-05:
Hey @bglgwyng! Do you have any crash logs or adb logcat logs for this? Thanks! :)

bglgwyng, 2024-01-06:
@mrousavy The posted log output is all I have received from logcat. Do we have a way to obtain more helpful logs?
I believe it will take some time to reproduce the same behavior. I'll attempt to do so and share the results.

mrousavy, 2024-01-08:
Hm weird, yea `adb logcat` should normally give more output. Especially relevant are the steps happening before the crash.

bglgwyng, 2024-01-08:
@mrousavy I reproduced a similar behavior twice. Still don't know how to do it. I just used my app's camera page for minutes and it crashed suddenly.

The logs are [here](https://gist.github.com/bglgwyng/5d46bf4a00538d944972bc7a8d231018)
The crash logs are a bit different from the one reported here, but they share `com.mrousavy.camera.frameprocessor.VisionCameraScheduler.trigger` part in common.

bglgwyng, 2024-01-08:
I succeeded at reproducing the same crash logs. I just copied the entire logs here.
https://gist.github.com/bglgwyng/88627404b45da41f16400fe468759fce

mrousavy, 2024-01-09:
> Calling JS function after bridge has been destroyed

Interesting. Are you doing a fast refresh or something?

bglgwyng, 2024-01-14:
Fash refresh was disabled.

mrousavy, 2024-01-30:
Hey @bglgwyng - could you try to reproduce this bug again with react-native-worklets-core 0.3.0 (includes a fix) and VisionCamera from current `main` branch (includes a ton of fixes again)?

bglgwyng, 2024-02-01:
I'll test it this weekend!

mrousavy, 2024-02-13:
Hey - @bglgwyng did you test if this is fixed now in 3.9.x?

bglgwyng, 2024-02-13:
Not yet. The example app crashed or showed a black screen on my Galaxy S21 after upgrading to 3.9.0. I wasn't able to test this issue.

mrousavy, 2024-02-13:
> The example app crashed or showed a black screen on my Galaxy S21 after upgrading to 3.9.0

Wait what? Is there an issue for this? That's weird.

WanderingBit, 2024-02-14:
Having same consistent crash when frame.toArrayBuffer() called.

Device
Galaxy A02

VisionCamera Version
3.9.0-beta.3
WorkletsCore Version
0.3.0
PixelFormat
yuv

```
16:46:11.222  I  updateBoundsLayer: shouldReparent = false t = android.view.SurfaceControl$Transaction@6ecd175 sc = Surface(name=Bounds for - com.roadsight/com.roadsight.MainActivity@0)/@0x25050a frame = 17
16:46:11.406  A  pid: 22209, tid: 23986, name: ionCamera.video  >>> com.roadsight <<<
16:46:11.447  I  NativeAlloc concurrent copying GC freed 84017(3050KB) AllocSpace objects, 3(48KB) LOS objects, 49% free, 4789KB/9578KB, paused 879us total 219.229ms
16:46:11.454  W  A resource failed to call close. 
16:46:11.691  A        #07 pc 0003ba04  /data/app/~~3QWpXgiqE47BYtQPneq9fQ==/com.roadsight-eUYUxK3WmwtNk0dMVzk5cw==/base.apk!libVisionCamera.so (offset 0x197000) (BuildId: ec11a8c2b6cc1f734a7fb9c7eabf532cb0dde0f3)
16:46:11.691  A        #08 pc 0003dc63  /data/app/~~3QWpXgiqE47BYtQPneq9fQ==/com.roadsight-eUYUxK3WmwtNk0dMVzk5cw==/base.apk!libhermes.so (offset 0x3af9000) (BuildId: d3ef7cec979352a0481c2a8436987cd6475424af)
16:46:11.691  A        #09 pc 00044efd  /data/app/~~3QWpXgiqE47BYtQPneq9fQ==/com.roadsight-eUYUxK3WmwtNk0dMVzk5cw==/base.apk!libhermes.so (offset 0x3af9000) (BuildId: d3ef7cec979352a0481c2a8436987cd6475424af)
16:46:11.692  A        #10 pc 0005b15f  /data/app/~~3QWpXgiqE47BYtQPneq9fQ==/com.roadsight-eUYUxK3WmwtNk0dMVzk5cw==/base.apk!libhermes.so (offset 0x3af9000) (BuildId: d3ef7cec979352a0481c2a8436987cd6475424af)
16:46:11.692  A        #11 pc 0005a835  /data/app/~~3QWpXgiqE47BYtQPneq9fQ==/com.roadsight-eUYUxK3WmwtNk0dMVzk5cw==/base.apk!libhermes.so (offset 0x3af9000) (BuildId: d3ef7cec979352a0481c2a8436987cd6475424af)
16:46:11.692  A        #12 pc 0006176d  /data/app/~~3QWpXgiqE47BYtQPneq9fQ==/com.roadsight-eUYUxK3WmwtNk0dMVzk5cw==/base.apk!libhermes.so (offset 0x3af9000) (BuildId: d3ef7cec979352a0481c2a8436987cd6475424af)
16:46:11.692  A        #13 pc 00045121  /data/app/~~3QWpXgiqE47BYtQPneq9fQ==/com.roadsight-eUYUxK3WmwtNk0dMVzk5cw==/base.apk!libhermes.so (offset 0x3af9000) (BuildId: d3ef7cec979352a0481c2a8436987cd6475424af)
16:46:11.692  A        #14 pc 00038021  /data/app/~~3QWpXgiqE47BYtQPneq9fQ==/com.roadsight-eUYUxK3WmwtNk0dMVzk5cw==/base.apk!libhermes.so (offset 0x3af9000) (BuildId: d3ef7cec979352a0481c2a8436987cd6475424af)
16:46:11.692  A        #15 pc 0004fd44  /data/app/~~3QWpXgiqE47BYtQPneq9fQ==/com.roadsight-eUYUxK3WmwtNk0dMVzk5cw==/base.apk!libVisionCamera.so (offset 0x197000) (RNWorklet::JsiWorklet::call(std::__ndk1::shared_ptr<facebook::jsi::Function>, facebook::jsi::Runtime&, facebook::jsi::Value const&, facebook::jsi::Value const*, unsigned int)+576) (BuildId: ec11a8c2b6cc1f734a7fb9c7eabf532cb0dde0f3)
16:46:11.693  A        #16 pc 0004eb04  /data/app/~~3QWpXgiqE47BYtQPneq9fQ==/com.roadsight-eUYUxK3WmwtNk0dMVzk5cw==/base.apk!libVisionCamera.so (offset 0x197000) (RNWorklet::WorkletInvoker::call(facebook::jsi::Runtime&, facebook::jsi::Value const&, facebook::jsi::Value const*, unsigned int)+140) (BuildId: ec11a8c2b6cc1f734a7fb9c7eabf532cb0dde0f3)
16:46:11.693  A        #17 pc 000fa14f  /data/app/~~3QWpXgiqE47BYtQPneq9fQ==/com.roadsight-eUYUxK3WmwtNk0dMVzk5cw==/base.apk!librnworklets.so (offset 0x4942000) (RNWorklet::JsiObjectWrapper::setFunctionValue(facebook::jsi::Runtime&, facebook::jsi::Value const&)::'lambda'(facebook::jsi::Runtime&, facebook::jsi::Value const&, facebook::jsi::Value const*, unsigned int)::operator()(facebook::jsi::Runtime&, facebook::jsi::Value const&, facebook::jsi::Value const*, unsigned int) const+54) (BuildId: b7c8d35339b154369ee0b669ca7612ada736eaf3)
16:46:11.693  A        #18 pc 000fa107  /data/app/~~3QWpXgiqE47BYtQPneq9fQ==/com.roadsight-eUYUxK3WmwtNk0dMVzk5cw==/base.apk!librnworklets.so (offset 0x4942000) (BuildId: b7c8d35339b154369ee0b669ca7612ada736eaf3)
16:46:11.693  A        #19 pc 000fa091  /data/app/~~3QWpXgiqE47BYtQPneq9fQ==/com.roadsight-eUYUxK3WmwtNk0dMVzk5cw==/base.apk!librnworklets.so (offset 0x4942000) (facebook::jsi::Value std::__ndk1::__invoke_void_return_wrapper<facebook::jsi::Value>::__call<RNWorklet::JsiObjectWrapper::setFunctionValue(facebook::jsi::Runtime&, facebook::jsi::Value const&)::'lambda'(facebook::jsi::Runtime&, facebook::jsi::Value const&, facebook::jsi::Value const*, unsigned int)&, facebook::jsi::Runtime&, facebook::jsi::Value const&, facebook::jsi::Value const*, unsigned int>(RNWorklet::JsiObjectWrapper::setFunctionValue(facebook::jsi::Runtime&, facebook::jsi::Value const&)::'lambda'(facebook::jsi::Runtime&, facebook::jsi::Value const&, facebook::jsi::Value const*, unsigned int)&, facebook::jsi::Runtime&, facebook::jsi::Value const&, facebook::jsi::Value const*&&, unsigned int&&)+86) (BuildId: b7c8d35339b154369ee0b669ca7612ada736eaf3)
16:46:11.694  A        #20 pc 000fa033  /data/app/~~3QWpXgiqE47BYtQPneq9fQ==/com.roadsight-eUYUxK3WmwtNk0dMVzk5cw==/base.apk!librnworklets.so (offset 0x4942000) (BuildId: b7c8d35339b154369ee0b669ca7612ada736eaf3)
16:46:11.694  A        #21 pc 000f9645  /data/app/~~3QWpXgiqE47BYtQPneq9fQ==/com.roadsight-eUYUxK3WmwtNk0dMVzk5cw==/base.apk!librnworklets.so (offset 0x4942000) (std::__ndk1::__function::__func<RNWorklet::JsiObjectWrapper::setFunctionValue(facebook::jsi::Runtime&, facebook::jsi::Value const&)::'lambda'(facebook::jsi::Runtime&, facebook::jsi::Value const&, facebook::jsi::Value const*, unsigned int), std::__ndk1::allocator<RNWorklet::JsiObjectWrapper::setFunctionValue(facebook::jsi::Runtime&, facebook::jsi: [... cut ...]
16:46:11.698  A        #46 pc 0001937c  [anon:dalvik-classes9.dex extracted in memory from /data/app/~~3QWpXgiqE47BYtQPneq9fQ==/com.roadsight-eUYUxK3WmwtNk0dMVzk5cw==/base.apk!classes9.dex] (com.mrousavy.camera.core.VideoPipeline.$r8$lambda$EMriwS_FKhtjTFtppM84Z5V8tiI)
16:46:11.699  A        #49 pc 00019120  [anon:dalvik-classes9.dex extracted in memory from /data/app/~~3QWpXgiqE47BYtQPneq9fQ==/com.roadsight-eUYUxK3WmwtNk0dMVzk5cw==/base.apk!classes9.dex] (com.mrousavy.camera.core.VideoPipeline$$ExternalSyntheticLambda0.onImageAvailable+4)

```

bglgwyng, 2024-02-15:
@mrousavy I found that this error still occurs. Also, it seems to happen more frequently compared to 3.8.2. I'm not 100% sure.
I still don't know how to reproduce this behavior, so if you have any idea about it, please let me know.
Here's another crash log I got.

```
2024-02-15 19:59:16.937 28061-28183 libc                    com.gwyng.shootee                    A  Fatal signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x72757465723b29 in tid 28183 (ionCamera.video), pid 28061 (m.gwyng.shootee)
2024-02-15 19:59:17.059 28061-28061 VisionCameraProxy       com.gwyng.shootee                    D  Finding view 367...
2024-02-15 19:59:17.059 28061-28061 VisionCameraProxy       com.gwyng.shootee                    D  Found view 367!
---------------------------- PROCESS STARTED (28655) for package com.gwyng.shootee ----------------------------
2024-02-15 19:59:17.310 28061-28061 VisionCameraProxy       com.gwyng.shootee                    D  Finding view 367...
2024-02-15 19:59:17.310 28061-28061 VisionCameraProxy       com.gwyng.shootee                    D  Found view 367!
2024-02-15 19:59:17.977 28654-28654 DEBUG                   crash_dump64                         A  Cmdline: com.gwyng.shootee
2024-02-15 19:59:17.977 28654-28654 DEBUG                   crash_dump64                         A  pid: 28061, tid: 28183, name: ionCamera.video  >>> com.gwyng.shootee <<<
2024-02-15 19:59:17.977 28654-28654 DEBUG                   crash_dump64                         A        #00 pc 0000000000063f98  /data/app/~~dwVy3JiQwlcecdfhM_rGbg==/com.gwyng.shootee-dLlSo6_sTyQbP64gBIWi8Q==/base.apk!libVisionCamera.so (offset 0x596000) (vision::JVisionCameraScheduler::trigger()+120) (BuildId: cca35f6d563d4f8bf98eaccd0e670ba2219223cc)
2024-02-15 19:59:17.977 28654-28654 DEBUG                   crash_dump64                         A        #01 pc 0000000000065874  /data/app/~~dwVy3JiQwlcecdfhM_rGbg==/com.gwyng.shootee-dLlSo6_sTyQbP64gBIWi8Q==/base.apk!libVisionCamera.so (offset 0x596000) (facebook::jni::detail::MethodWrapper<void (vision::JVisionCameraScheduler::*)(), &(vision::JVisionCameraScheduler::trigger()), vision::JVisionCameraScheduler, void>::dispatch(facebook::jni::alias_ref<facebook::jni::detail::JTypeFor<facebook::jni::HybridClass<vision::JVisionCameraScheduler, facebook::jni::detail::BaseHybridClass>::JavaPart, facebook::jni::JObject, void>::_javaobject*>)+32) (BuildId: cca35f6d563d4f8bf98eaccd0e670ba2219223cc)
2024-02-15 19:59:17.978 28654-28654 DEBUG                   crash_dump64                         A        #02 pc 0000000000065914  /data/app/~~dwVy3JiQwlcecdfhM_rGbg==/com.gwyng.shootee-dLlSo6_sTyQbP64gBIWi8Q==/base.apk!libVisionCamera.so (offset 0x596000) (facebook::jni::detail::CallWithJniConversions<void (*)(facebook::jni::alias_ref<facebook::jni::detail::JTypeFor<facebook::jni::HybridClass<vision::JVisionCameraScheduler, facebook::jni::detail::BaseHybridClass>::JavaPart, facebook::jni::JObject, void>::_javaobject*>), void, facebook::jni::detail::JTypeFor<facebook::jni::HybridClass<vision::JVisionCameraScheduler, facebook::jni::detail::BaseHybridClass>::JavaPart, facebook::jni::JObject, void>::_javaobject*>::call(facebook::jni::detail::JTypeFor<facebook::jni::HybridClass<vision::JVisionCameraScheduler, facebook::jni::detail::BaseHybridClass>::JavaPart, facebook::jni::JObject, void>::_javaobject*, void (*)(facebook::jni::alias_ref<facebook::jni::detail::JTypeFor<facebook::jni::HybridClass<vision::JVisionCameraScheduler, facebook::jni::detail::BaseHybridClass>::JavaPart, facebook::jni::JObject, void>::_javaobject*>))+76) (BuildId: cca35f6d563d4f8bf98eaccd0e670ba2219223cc)
2024-02-15 19:59:17.978 28654-28654 DEBUG                   crash_dump64                         A        #03 pc 00000000000657ec  /data/app/~~dwVy3JiQwlcecdfhM_rGbg==/com.gwyng.shootee-dLlSo6_sTyQbP64gBIWi8Q==/base.apk!libVisionCamera.so (offset 0x596000) (facebook::jni::detail::FunctionWrapper<void (*)(facebook::jni::alias_ref<facebook::jni::detail::JTypeFor<facebook::jni::HybridClass<vision::JVisionCameraScheduler, facebook::jni::detail::BaseHybridClass>::JavaPart, facebook::jni::JObject, void>::_javaobject*>), facebook::jni::detail::JTypeFor<facebook::jni::HybridClass<vision::JVisionCameraScheduler, facebook::jni::detail::BaseHybridClass>::JavaPart, facebook::jni::JObject, void>::_javaobject*, void>::call(_JNIEnv*, _jobject*, void (*)(facebook::jni::alias_ref<facebook::jni::detail::JTypeFor<facebook::jni::HybridClass<vision::JVisionCameraScheduler, facebook::jni::detail::BaseHybridClass>::JavaPart, facebook::jni::JObject, void>::_javaobject*>))+60) (BuildId: cca35f6d563d4f8bf98eaccd0e670ba2219223cc)
2024-02-15 19:59:17.978 28654-28654 DEBUG                   crash_dump64                         A        #04 pc 00000000000642e8  /data/app/~~dwVy3JiQwlcecdfhM_rGbg==/com.gwyng.shootee-dLlSo6_sTyQbP64gBIWi8Q==/base.apk!libVisionCamera.so (offset 0x596000) (facebook::jni::detail::MethodWrapper<void (vision::JVisionCameraScheduler::*)(), &(vision::JVisionCameraScheduler::trigger()), vision::JVisionCameraScheduler, void>::call(_JNIEnv*, _jobject*)+36) (BuildId: cca35f6d563d4f8bf98eaccd0e670ba2219223cc)
2024-02-15 19:59:17.978 28654-28654 DEBUG                   crash_dump64                         A        #10 pc 00000000000ebce4  [anon:dalvik-classes2.dex extracted in memory from /data/app/~~dwVy3JiQwlcecdfhM_rGbg==/com.gwyng.shootee-dLlSo6_sTyQbP64gBIWi8Q==/base.apk!classes2.dex] (com.mrousavy.camera.frameprocessor.VisionCameraScheduler.$r8$lambda$QQXCER7V8WTCtwJ_0_iVk6pXtyc+0)
2024-02-15 19:59:17.978 28654-28654 DEBUG                   crash_dump64                         A        #15 pc 00000000000ebcc8  [anon:dalvik-classes2.dex extracted in memory from /data/app/~~dwVy3JiQwlcecdfhM_rGbg==/com.gwyng.shootee-dLlSo6_sTyQbP64gBIWi8Q==/base.apk!classes2.dex] (com.mrousavy.camera.frameprocessor.VisionCameraScheduler$$ExternalSyntheticLambda0.run+0)
```

mrousavy, 2024-02-15:
Is that with `enableGpuBuffers` set to `true` or `false`?

mrousavy, 2024-02-15:
@bglgwyng `vision::JVisionCameraScheduler::trigger()+120` is just so weird. I don't know in what case this is ever being called 🤨 

Is that immediately on camera startup or just after a while of running?

@WanderingBit your SIGSEV and the nativeRelease calls before might hint that the HardwareBuffer we are trying to access is already being deleted... I don't know how that can happen either. Is that also on app startup, or just after a while of running?

mrousavy, 2024-02-15:
or is that on some session change, like a prop to the Camera changes or you fast-refresh?

bglgwyng, 2024-02-15:
> Is that immediately on camera startup or just after a while of running?
I just left my app running. I needed to wait a few minutes until the error occurred.

bglgwyng, 2024-02-16:
@mrousavy 
Here's the reproducing step.
https://github.com/bglgwyng/react-native-vision-camera/commit/10e5b4bd3dfaf81ae4b9e03b2a5e8afaeca593fa
I made `useRunAtTargetFps` to replace the old `runAtTargetFps` with bugs, and I didn't replace it with the new one without bugs yet.

On my Galaxy S21, I could see the crash before the `count` value hits around 2000.

Now I'll try to make the code more minimal to detect the source of the bug more easily.

mrousavy, 2024-02-16:
I'm still not sure what's actually broken here?

Both runAtTargetFps and runAsync work for me in the example app (at least on 3.8.x, I'll double check if it works on 3.9.x later).

Also you could just use runAtTargetFps inside of runAsync ;)

WanderingBit, 2024-02-16:
> I'm still not sure what's actually broken here?
> 
> Both runAtTargetFps and runAsync work for me in the example app (at least on 3.8.x, I'll double check if it works on 3.9.x later).
> 
> Also you could just use runAtTargetFps inside of runAsync ;)

```
const frameProcessor = useFrameProcessor((frame) => {
    'worklet'
    if (model == null) return;
    runAsync(frame, () => {
      'worklet'
      runAtTargetFps(1, () => {
        'worklet'
        const resized = resize(frame, {
          scale: {
            width: 640,
            height: 640
          },
          pixelFormat: 'rgb',
          dataType: 'uint8'
        })
        const output = model.runSync([resized])
        const detections = output[0]

        const conf_threshold = 34
  
        
      })
    })
    
  }, [model])

```

No hot reload induced. It's weird model gives output without runAsync successfully but it freezes camera. For given code block above I leave the crash log below:

```
17:18:35.225  I  ImageReader::onImageAvailable!
17:18:35.228  E  [put()] Err:   330:, Yang OBC2 is put
17:18:35.233  I  [DBQ] (q2:0xacfb3000) Reallocate Slot(1), pool(0 -> 0) size(0 -> 2304000)  
17:18:35.239  I  [DBQ] (q2:0xacfb3000) Alloc Slot(1), handle=0xacf8cce0 c=0 w=720 h=1600 p=736 ys=736 vs=1600 f=268  
17:18:35.244  I  running async
17:18:35.244  I  Target scale: 640 x 640
17:18:35.244  I  Cropping to 1440 x 1440 at (240, 0)
17:18:35.244  I  Target Format: RGB
17:18:35.245  I  Target DataType: UINT8
17:18:35.245  I  Converting YUV 4:2:0 -> ARGB 8888...
17:18:35.247  I  Allocating _argbBuffer Buffer with size 11059200...
17:18:35.254  I  ImageReader::onImageAvailable!
17:18:35.257  I  [cust_ChangeTableByCond]Sensor_idx = 0 ,SHOOTING_MODE = 0,SceneMode = 0
17:18:35.258  I  [cust_ChangeSceneBy3rdpartyApp]VT_MODE = 0 ,MinFps = 30000,MaxFps = 30000,SceneMode = 7,AeMode = 1
17:18:35.258  I  [cust_mapAOSPAeMode] u4AeMode(1), u4AospAeMode(1)
17:18:35.258  I  [Scenario4AE] SensorMode(0), ShootMode(0), AeMode(1)
17:18:35.258  I  [setParams] SEC date.0222, disable auto flicker
17:18:35.261  E  [put()] Err:   330:, Yang OBC2 is put
17:18:35.266  I  [DBQ] (q2:0xacfb3000) Reallocate Slot(2), pool(0 -> 0) size(0 -> 2304000)  
17:18:35.271  I  [DBQ] (q2:0xacfb3000) Alloc Slot(2), handle=0xacf8b7e0 c=0 w=720 h=1600 p=736 ys=736 vs=1600 f=268  
17:18:35.274  I  Cropping [0, 0 @ 1920x1440] ARGB buffer to [240, 0 @ 1440x1440]...
17:18:35.274  I  Allocating _cropBuffer Buffer with size 8294400...
17:18:35.292  I  Scaling [0, 0 @ 1440x1440] ARGB buffer to [0, 0 @ 640x640]...
17:18:35.292  I  Allocating _scaleBuffer Buffer with size 1638400...
17:18:35.294  I  [cust_ChangeTableByCond]Sensor_idx = 0 ,SHOOTING_MODE = 0,SceneMode = 0
17:18:35.294  I  [cust_ChangeSceneBy3rdpartyApp]VT_MODE = 0 ,MinFps = 30000,MaxFps = 30000,SceneMode = 7,AeMode = 1
17:18:35.294  I  [cust_mapAOSPAeMode] u4AeMode(1), u4AospAeMode(1)
17:18:35.294  I  [Scenario4AE] SensorMode(0), ShootMode(0), AeMode(1)
17:18:35.294  I  [setParams] SEC date.0222, disable auto flicker
17:18:35.299  E  [put()] Err:   330:, Yang OBC2 is put
17:18:35.301  I  ImageReader::onImageAvailable!
17:18:35.304  I  Converting ARGB Buffer to Pixel Format 0...
17:18:35.304  I  Allocating _customFormatBuffer Buffer with size 1228800...
17:18:35.309  I  Wrapping Java ByteBuffer with size 1228800...
17:18:35.311  A  Fatal signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x39f75e42 in tid 10663 (mqt_js), pid 10107 (com.sampleapp)
```

bglgwyng, 2024-02-16:
@mrousavy  My description was a bit misleading. I explained the reason why I didn't use `runAtTargetFps`, just in case you wonder why. The example code was written a year ago, before [this commit](https://github.com/mrousavy/react-native-vision-camera/commit/af4e366312a2335535f80109e6bf0b3a977444f0) was made. Anyway, it's not an important part.

This behavior is observed only on Android. Have you tried the example on Android? I wonder if it's reproducible on your device. If not, then it's not an Android issue, but a Samsung issue instead again.
The example code is just a combination of a run-in-js call and shared value updates, so I don't see anything unusual either. I'm trying to minimize the example code to narrow down the hypotheses.

bglgwyng, 2024-02-17:
I found a much easier way to reproduce it.

```typescript
const [count, setCount] = useReducer((count) => count + 1, 0)
  const setCount$ = useMemo(() => Worklets.createRunInJsFn(setCount), [])

  const frameProcessor = useFrameProcessor(
    (frame) => {
      'worklet'

      console.info(count)
      setCount$()
    },
    [setCount$, count],
  )
  ```

You can see the example app crashes until `count` hits 2000.
  
`count` in deps triggers reinitializing of the frame processor. When you omit it, the crash doesn't occur.

mrousavy, 2024-02-19:
> `count` in deps triggers reinitializing of the frame processor.

So you think the Frame Processor being updated causes a crash?

mrousavy, 2024-02-19:
This is the code that gets called when the Frame Processor changes: https://github.com/mrousavy/react-native-vision-camera/blob/f6b7163660679599e48d1b75ab444ba8b5564809/package/android/src/main/cpp/frameprocessor/java-bindings/JVisionCameraProxy.cpp#L59-L70

bglgwyng, 2024-02-19:
> So you think the Frame Processor being updated causes a crash?

Not sure. But it crashes the app very easily.

The following code crahes,
```ts
const [count, setCount] = useReducer((count) => count + 1, 0)

useEffect(() => {
  const interval = setInterval(setCount, 10)
  return () => { clearInterval(interval) }
}, [])

const frameProcessor = useFrameProcessor(
  (frame) => {
    'worklet'
  },
  [count],
)
```

while this one doesn't crash
```ts
const frameProcessor = useFrameProcessor(
  (frame) => {
    'worklet'
  },
  [], // no `count` in `deps`
)
```
The example code in https://github.com/mrousavy/react-native-vision-camera/issues/2153#issuecomment-1950038141 includes the usage of `createRunInJsFn`, but now I found that it's not necessary to reproduce the crash.

However, it doesn't mean that the frame processor update is mandatory to cause a crash.
I observed the app crash (with the same error message) without frame processor updates several times.
But it's so hard to reproduce.

Updating frame processors is the fastest way to cause a crash that I know.

bglgwyng, 2024-02-19:
@mrousavy I tried this fix https://github.com/bglgwyng/react-native-vision-camera/commit/eff64de2d98da160ab0add65cb093f645b3b974bwithout without expectation, but it seems to work...?!
Do you have any idea about it?

mrousavy, 2024-02-19:
lol yes this should actually be there, I don't know why I didn't spot that mistake 😂 

can you send a PR? looks good

closed 2024-02-19 by mrousavy

WanderingBit, 2024-02-26:
> @bglgwyng `vision::JVisionCameraScheduler::trigger()+120` is just so weird. I don't know in what case this is ever being called 🤨
>
> Is that immediately on camera startup or just after a while of [...]

@mrousavy The fix did not work for me. Crash happens during initial start.

WanderingBit, 2024-02-26:
Here it is https://github.com/mrousavy/react-native-vision-camera/issues/2153#issuecomment-1948476845

prachipatel01, 2024-05-10:
Hi, any update on this? I went through the thread and can confirm the same is happening with me

PR #2591 (2024-02-19): fix: Fix Frame Processor SIGSEV crash in `VisionCameraScheduler::trigger` by locking mutex in `dispatchAsync`

## What

Fix for the crash on frame processor updates in `VisionCameraScheduler` by making sure the `trigger` and `schedule` functions are properly locked with a Mutex.

## Changes

Add a missing lock.

## Tested on

Galaxy S21

## Related issues

- Fixes #2153
- Fixes #2178
-