margelo/react-native-vision-camera #2154: 🔧 Build failed on IOS - VisionCamera.modulemap not found

MohamedShawkyBayoumi, 2023-11-11:
### How were you trying to build the app?

I tried ti build my app using npx react-native run-ios, and it failed

for your information also

I have installed a new app for testing like Example and I have just installed the `react-native-vision-camera` package then I run `pod install` and I encountered the same issue!!

### Full build logs

```tsx
npx react-native run-ios                                                                                                                         ✔  10490  10:51:22
info Found Xcode workspace "MyApp.xcworkspace"
info Found booted iPhone 8
info Launching iPhone 8
info Building (using "xcodebuild -workspace MyApp.xcworkspace -configuration Debug -scheme MyApp -destination id=0DD9887E-80C1-43D7-B254-CA04493FEC95")

info 💡 Tip: Make sure that you have set up your development environment correctly, by running react-native doctor. To read more about doctor command visit: https://github.com/react-native-community/cli/blob/main/packages/cli-doctor/README.md#doctor

error Failed to build iOS project. We ran "xcodebuild" command but it exited with error code 65. To debug build logs further, consider building your app with Xcode.app, by opening MyApp.xcworkspace.
Command line invocation:
    /Applications/Xcode.app/Contents/Developer/usr/bin/xcodebuild -workspace MyApp.xcworkspace -configuration Debug -scheme MyApp -destination id=0DD9887E-80C1-43D7-B254-CA04493FEC95

User defaults from command line:
    IDEPackageSupportUseBuiltinSCM = YES

note: Using new build system
note: Planning
Analyze workspace

Create build description
Build description signature: a9302422d2795a882e1449381c8ad517
Build description path: /Users/hassan/Library/Developer/Xcode/DerivedData/MyApp-crjsscxieqrcrsgtoxkvmnrbzdwc/Build/Intermediates.noindex/XCBuildData/a9302422d2795a882e1449381c8ad517-desc.xcbuild

note: Build preparation complete
note: Building targets in dependency order
ProcessProductPackaging "" /Users/hassan/Library/Developer/Xcode/DerivedData/MyApp-crjsscxieqrcrsgtoxkvmnrbzdwc/Build/Intermediates.noindex/MyApp.build/Debug-iphonesimulator/MyApp.build/MyApp.app.xcent (in target 'MyApp' from project 'MyApp')
    cd /Users/hassan/practice/MyApp/ios

    Entitlements:

    {
    "com.apple.security.get-task-allow" = 1;
}

    builtin-productPackagingUtility -entitlements -format xml -o /Users/hassan/Library/Developer/Xcode/DerivedData/MyApp-crjsscxieqrcrsgtoxkvmnrbzdwc/Build/Intermediates.noindex/MyApp.build/Debug-iphonesimulator/MyApp.build/MyApp.app.xcent

ProcessProductPackaging "" /Users/hassan/Library/Developer/Xcode/DerivedData/MyApp-crjsscxieqrcrsgtoxkvmnrbzdwc/Build/Intermediates.noindex/MyApp.build/Debug-iphonesimulator/MyApp.build/MyApp.app-Simulated.xcent (in target 'MyApp' from project 'MyApp')
    cd /Users/hassan/practice/MyApp/ios

    Entitlements:

    {
    "application-identifier" = "G8AZ53THAQ.org.reactjs.native.example.MyApp";
}

    builtin-productPackagingUtility -entitlements -format xml -o /Users/hassan/Library/Developer/Xcode/DerivedData/MyApp-crjsscxieqrcrsgtoxkvmnrbzdwc/Build/Intermediates.noindex/MyApp.build/Debug-iphonesimulator/MyApp.build/MyApp.app-Simulated.xcent

PhaseScriptExecution [CP]\ Check\ Pods\ Manifest.lock /Users/hassan/Library/Developer/Xcode/DerivedData/MyApp-crjsscxieqrcrsgtoxkvmnrbzdwc/Build/Intermediates.noindex/MyApp.build/Debug-iphonesimulator/MyApp.build/Script-F149B70960878098D4A76ED9.sh (in target 'MyApp' from project 'MyApp')
    cd /Users/hassan/practice/MyApp/ios
    /bin/sh -c /Users/hassan/Library/Developer/Xcode/DerivedData/MyApp-crjsscxieqrcrsgtoxkvmnrbzdwc/Build/Intermediates.noindex/MyApp.build/Debug-iphonesimulator/MyApp.build/Script-F149B70960878098D4A76ED9.sh

PhaseScriptExecution Start\ Packager /Users/hassan/Library/Developer/Xcode/DerivedData/MyApp-crjsscxieqrcrsgtoxkvmnrbzdwc/Build/Intermediates.noindex/MyApp.build/Debug-iphonesimulator/MyApp.build/Script-FD10A7F022414F080027D42C.sh (in target 'MyApp' from project 'MyApp')
    cd /Users/hassan/practice/MyApp/ios
    /bin/sh -c /Users/hassan/Library/Developer/Xcode/DerivedData/MyApp-crjsscxieqrcrsgtoxkvmnrbzdwc/Build/Intermediates.noindex/MyApp.build/Debug-iphonesimulator/MyApp.build/Script-FD10A7F022414F080027D42C.sh
Connection to localhost port 8081 [tcp/sunproxyadmin] succeeded!

CompileC /Users/hassan/Library/Developer/Xcode/DerivedData/MyApp-crjsscxieqrcrsgtoxkvmnrbzdwc/Build/Intermediates.noindex/MyApp.build/Debug-iphonesimulator/MyApp.build/Objects-normal/arm64/MyApp_vers.o /Users/hassan/Library/Developer/Xcode/DerivedData/MyApp-crjsscxieqrcrsgtoxkvmnrbzdwc/Build/Intermediates.noindex/MyApp.build/Debug-iphonesimulator/MyApp.build/DerivedSources/MyApp_vers.c normal arm64 c com.apple.compilers.llvm.clang.1_0.compiler (in target 'MyApp' from project 'MyApp')
    cd /Users/hassan/practice/MyApp/ios
    export LANG\=en_US.US-ASCII
[... 38 lines cut ...]
```

### VisionCamera Version

3.6.3

### Target platforms

iOS

### Operating system

MacOS

### Can you build the VisionCamera Example app?

No, I cannot build the Example app either

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

pnthach95, 2023-11-16:
> I had the same error but in a different situation when I upgraded from vision-camera 2 to 3. It was also looking for the VisionCamera.modulema, I checked and the file really did not exist and thus the build failed. It took me a day to realise that we had it in `OTHER_CFLAGS` as `-fmodule-map-file=\"${PODS_CONFIGURATION_BUILD_DIR}/VisionCamera/VisionCamera.modulemap\"` (I don't know at which point it got there...). Removing this cflag fixed the issue and I was able to build the app in iOS. Maybe you can check if you have something similar in your pods build settings?

Same for me, but I remove the workaround `config.build_settings["EXCLUDED_ARCHS[sdk=iphonesimulator*]"] = "arm64"` in podfile

mrousavy, 2024-01-15:
I think this is not related to VisionCamera as it builds fine for me in a new blank RN project

closed 2024-01-15 by mrousavy