margelo/react-native-vision-camera #2156: Incorrect package="com.mrousavy.camera" found in source AndroidManifest.xml

talaikis, 2023-11-12:
### What's happening?

Build on RN0.73 fails with `Incorrect package="com.mrousavy.camera" found in source AndroidManifest.xml: node_modules\react-native-vision-camera\android\src\main\AndroidManifest.xml`

### Reproduceable Code

```tsx
1. Upgrade RN to latest beta
2. npm run android
```

### Relevant log output

```shell
Incorrect package="com.mrousavy.camera" found in source AndroidManifest.xml: node_modules\react-native-vision-camera\android\src\main\AndroidManifest.xml.
Setting the namespace via the package attribute in the source AndroidManifest.xml is no longer supported.
Recommendation: remove package="com.mrousavy.camera" from the source AndroidManifest.xml: node_modules\react-native-vision-camera\android\src\main\AndroidManifest.xml. FAILURE: Build failed with an exception. * What went wrong: Execution failed for task ':react-native-vision-camera:processDebugManifest'. > A failure occurred while executing com.android.build.gradle.tasks.ProcessLibraryManifest$ProcessLibWorkAction > Incorrect package="com.mrousavy.camera" found in source AndroidManifest.xml: node_modules\react-native-vision-camera\android\src\main\AndroidManifest.xml. Setting the namespace via the package attribute in the source AndroidManifest.xml is no longer supported. Recommendation: remove package="com.mrousavy.camera" from the source AndroidManifest.xml: node_modules\react-native-vision-camera\android\src\main\AndroidManifest.xml. * Try: > Run with --stacktrace option to get the stack trace. > Run with --info or --debug option to get more log output. > Run with --scan to get full insights. > Get more help at https://help.gradle.org. BUILD FAILED in 9m 42s.
```

### Camera Device

```json
Not applicable
```

### Device

Android

### VisionCamera Version

3.6.4

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2023-11-18:
This is not a bug but a build error, wrong issue template.

closed 2023-11-18 by mrousavy

AlexShavkuta, 2024-01-28:
@talaikis Did you find any solution for this error?

talaikis, 2024-01-28:
> @talaikis Did you find any solution for this error?

This is old, but I think latest lib version doesn't have this build error. Anyway it's easy to fix if you need to use current version.

ankitlb, 2024-02-11:
@talaikis I'm facing the same issue can you suggest me something?