margelo/react-native-vision-camera #2163: 🔧 v3.6.6 - Not able to build on Android

Spxc, 2023-11-15:
### How were you trying to build the app?

./gradle assembleRelease

### Full build logs

```tsx
> Task :react-native-vision-camera:buildCMakeRelWithDebInfo[arm64-v8a] FAILED
C/C++: ninja: Entering directory `/Users/stianwiikinstebo/Documents/Backup/YourProjectNameBak-RN096/node_modules/react-native-vision-camera/android/.cxx/RelWithDebInfo/205o2y4k/arm64-v8a'
C/C++: In file included from ../../../../src/main/cpp/frameprocessor/java-bindings/JVisionCameraProxy.cpp:5:
C/C++: ../../../../src/main/cpp/frameprocessor/java-bindings/JVisionCameraProxy.h:7:10: fatal error: 'ReactCommon/CallInvokerHolder.h' file not found
C/C++: In file included from ../../../../src/main/cpp/VisionCamera.cpp:2:
C/C++: ../../../../src/main/cpp/frameprocessor/java-bindings/JVisionCameraProxy.h:7:10: fatal error: 'ReactCommon/CallInvokerHolder.h' file not found
C/C++: In file included from ../../../../src/main/cpp/frameprocessor/VisionCameraProxy.cpp:5:
C/C++: In file included from ../../../../src/main/cpp/frameprocessor/VisionCameraProxy.h:9:
C/C++: ../../../../src/main/cpp/frameprocessor/java-bindings/JVisionCameraProxy.h:7:10: fatal error: 'ReactCommon/CallInvokerHolder.h' file not found
w: Detected multiple Kotlin daemon sessions at build/kotlin/sessions

FAILURE: Build completed with 2 failures.

1: Task failed with an exception.
-----------
* What went wrong:
Execution failed for task ':react-native-vision-camera:buildCMakeRelWithDebInfo[arm64-v8a]'.
> com.android.ide.common.process.ProcessException: ninja: Entering directory `/Users/stianwiikinstebo/Documents/Backup/YourProjectNameBak-RN096/node_modules/react-native-vision-camera/android/.cxx/RelWithDebInfo/205o2y4k/arm64-v8a'
  [1/4] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/java-bindings/JVisionCameraProxy.cpp.o
  FAILED: CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/java-bindings/JVisionCameraProxy.cpp.o 
  /Users/stianwiikinstebo/Library/Android/sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/darwin-x86_64/bin/clang++ --target=aarch64-none-linux-android26 --gcc-toolchain=/Users/stianwiikinstebo/Library/Android/sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/darwin-x86_64 --sysroot=/Users/stianwiikinstebo/Library/Android/sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/darwin-x86_64/sysroot -DVISION_CAMERA_ENABLE_FRAME_PROCESSORS=false -DVisionCamera_EXPORTS -I../../../../../cpp -I../../../../src/main/cpp -I../../../../src/main/cpp/frameprocessor -I../../../../src/main/cpp/frameprocessor/java-bindings -I/Users/stianwiikinstebo/Documents/Backup/YourProjectNameBak-RN096/node_modules/react-native/ReactCommon -I/Users/stianwiikinstebo/Documents/Backup/YourProjectNameBak-RN096/node_modules/react-native/ReactCommon/callinvoker -I/Users/stianwiikinstebo/Documents/Backup/YourProjectNameBak-RN096/node_modules/react-native/ReactAndroid/src/main/jni/react/turbomodule -isystem /Users/stianwiikinstebo/.gradle/caches/transforms-3/44f87529cbf9f777306931dc59c5d740/transformed/jetified-react-android-0.73.0-rc.4-release/prefab/modules/jsi/include -isystem /Users/stianwiikinstebo/.gradle/caches/transforms-3/44f87529cbf9f777306931dc59c5d740/transformed/jetified-react-android-0.73.0-rc.4-release/prefab/modules/reactnativejni/include -isystem /Users/stianwiikinstebo/.gradle/caches/transforms-3/c7598e7d957f8ec4dcaab85973495b27/transformed/jetified-fbjni-0.5.1/prefab/modules/fbjni/include -g -DANDROID -fdata-sections -ffunction-sections -funwind-tables -fstack-protector-strong -no-canonical-prefixes -D_FORTIFY_SOURCE=2 -Wformat -Werror=format-security  -O2 -frtti -fexceptions -Wall -Wno-unused-variable -fstack-protector-all -O2 -g -DNDEBUG -fPIC -std=gnu++17 -MD -MT CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/java-bindings/JVisionCameraProxy.cpp.o -MF CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/java-bindings/JVisionCameraProxy.cpp.o.d -o CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/java-bindings/JVisionCameraProxy.cpp.o -c ../../../../src/main/cpp/frameprocessor/java-bindings/JVisionCameraProxy.cpp
  In file included from ../../../../src/main/cpp/frameprocessor/java-bindings/JVisionCameraProxy.cpp:5:
  ../../../../src/main/cpp/frameprocessor/java-bindings/JVisionCameraProxy.h:7:10: fatal error: 'ReactCommon/CallInvokerHolder.h' file not found
  #include <ReactCommon/CallInvokerHolder.h>
           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  1 error generated.
  [2/4] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/VisionCamera.cpp.o
  FAILED: CMakeFiles/VisionCamera.dir/src/main/cpp/VisionCamera.cpp.o 
  /Users/stianwiikinstebo/Library/Android/sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/darwin-x86_64/bin/clang++ --target=aarch64-none-linux-android26 --gcc-toolchain=/Users/stianwiikinstebo/Library/Android/sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/darwin-x86_64 --sysroot=/Users/stianwiikinstebo/Library/Android/sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/darwin-x86_64/sysroot -DVISION_CAMERA_ENABLE_FRAME_PROCESSORS=false -DVisionCamera_EXPORTS -I../../../../../cpp -I../../../../src/main/cpp -I../../../../src/main/cpp/frameprocessor -I../../../../src/main/cpp/frameprocessor/java-bindings -I/Users/stianwiikinstebo/Documents/Backup/YourProjectNameBak-RN096/node_modules/react-native/ReactCommon -I/Users/stianwiikinstebo/Documents/Backup/YourProjectNameBak-RN096/node_modules/react-native/ReactCommon/callinvoker -I/Users/stianwiikinstebo/Documents/Backup/YourProjectNameBak-RN096/node_modules/react-native/ReactAndroid/src/main/jni/react/turbomodule -isystem /Users/stianwiikinstebo/.gradle/caches/transforms-3/44f87529cbf9f777306931dc59c5d740/transformed/jetified-react-android-0.73.0-rc.4-release/prefab/modules/jsi/include -isystem /Users/stianwiikinstebo/.gradle/caches/transforms-3/44f87529cbf9f777306931dc59c5d740/transformed/jetified-react-android-0.73.0-rc.4-release/prefab/modules/reactnativejni/include -isystem /Users/stianwiikinstebo/.gradle/caches/transforms-3/c7598e7d957f8ec4dcaab85973495b27/transformed/jetified-fbjni-0.5.1/prefab/modules/fbjni/include -g -DANDROID -fdata-sections -ffunction-sections -funwind-tables -fstack-protector-strong -no-canonical-prefixes -D_FORTIFY_SOURCE=2 -Wformat -Werror=format-security  -O2 -frtti -fexceptions -Wall -Wno-unused-variable -fstack-protector-all -O2 -g -DNDEBUG -fPIC -std=gnu++17 -MD -MT CMakeFiles/VisionCamera.dir/src/main/cpp/VisionCamera.cpp.o -MF CMakeFiles/VisionCamera.dir/src/main/cpp/VisionCamera.cpp.o.d -o CMakeFiles/VisionCamera.dir/src/main/cpp/VisionCamera.cpp.o -c ../../../../src/main/cpp/VisionCamera.cpp
  In file included from ../../../../src/main/cpp/VisionCamera.cpp:2:
  ../../../../src/main/cpp/frameprocessor/java-bindings/JVisionCameraProxy.h:7:10: fatal error: 'ReactCommon/CallInvokerHolder.h' file not found
  #include <ReactCommon/CallInvokerHolder.h>
           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  1 error generated.
  [3/4] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/VisionCameraProxy.cpp.o
  FAILED: CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/VisionCameraProxy.cpp.o 
  /Users/stianwiikinstebo/Library/Android/sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/darwin-x86_64/bin/clang++ --target=aarch64-none-linux-android26 --gcc-toolchain=/Users/stianwiikinstebo/Library/Android/sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/darwin-x86_64 --sysroot=/Users/stianwiikinstebo/Library/Android/sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/darwin-x86_64/sysroot -DVISION_CAMERA_ENABLE_FRAME_PROCESSORS=false -DVisionCamera_EXPORTS -I../../../../../cpp -I../../../../src/main/cpp -I../../../../src/main/cpp/frameprocessor -I../../../../src/main/cpp/frameprocessor/java-bindings -I/Users/stianwiikinstebo/Documents/Backup/YourProjectNameBak-RN096/node_modules/react-native/ReactCommon -I/Users/stianwiikinstebo/Documents/Backup/YourProjectNameBak-RN096/node_modules/react-native/ReactCommon/callinvoker -I/Users/stianwiikinstebo/Documents/Backup/YourProjectNameBak-RN096/node_modules/react-native/ReactAndroid/src/main/jni/react/turbomodule -isystem /Users/stianwiikinstebo/.gradle/caches/transforms-3/44f87529cbf9f777306931dc59c5d740/transformed/jetified-react-android-0.73.0-rc.4-release/prefab/modules/jsi/include -isystem /Users/stianwiikinstebo/.gradle/caches/transforms-3/44f87529cbf9f777306931dc59c5d740/transformed/jetified-react-android-0.73.0-rc.4-release/prefab/modules/reactnativejni/include -isystem /Users/stianwiikinstebo/.gradle/caches/transforms-3/c7598e7d957f8ec4dc [... cut ...]
### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

zzz08900, 2023-11-16:
You need to upgrade your RN version to 0.72

Spxc, 2023-11-16:
Just because of Android? Is v2 supported on 0.69.12 then?

daltachange, 2023-11-16:
[image]
android release package, start video error

daltachange, 2023-11-16:
[image]
android  version 3.6.6

zzz08900, 2023-11-17:
> Just because of Android? Is v2 supported on 0.69.12 then?

No.
VisionCamera V2 requires React Native 0.70+, iOS 11+, Android API 21+, Xcode 15+.

instesti, 2023-11-17:
> > Just because of Android? Is v2 supported on 0.69.12 then?
> 
> No. VisionCamera V2 requires React Native 0.70+, iOS 11+, Android API 21+, Xcode 15+.

Okay then... So it runs FINE on 0.69.12 on iOS (so not sure you're correct here). 
The only issue is the build on Android.

zzz08900, 2023-11-17:
There are older versions of V2 that works with RN 0.69 + Android but AFAIK  those versions does not work with XCode 15 and might have problems with iOS17/iPhone 15 line up.

In theory you could cherry pick commits/make your own version of vision camera V2 that works well on both sides, but I assume that's more troublesome than bumping your RN version.

mrousavy, 2023-11-17:
Hey yup you need RN 0.71+

closed 2023-11-17 by mrousavy

jslok, 2023-12-13:
Getting the same error on RN 0.73 but it's weird that I was able to build earlier today. Didn't change anything and now I can't build. Tried deleting and reinstalling node_modules, yarn start --reset-cache, nothing works.