margelo/react-native-vision-camera #2166: 🐛 V 3.6.6 An android error when render Camera component.

daltachange, 2023-11-17:
### What's happening?

when open the screen and render the Camera component. an error happens. And can not open the camera.

[image]

### Reproduceable Code

```tsx
<Camera
          format={format}
          style={{ flex: 1 }}
          audio={true}
          video={true}
          ref={camera}
          device={device}
          isActive={true}
          fps={24}
          hdr={false}
          enableZoomGesture={true}
          onInitialized={() => {}}
          pixelFormat="yuv"
        />
```

### Relevant log output

```shell
1474-3879  NetworkAcc...tchService  I  getAverageWifiRssi:65
23021-23189 CameraSession            E  Failed to configure CameraSession! Error: no "Lcom/facebook/jni/HybridData;" field "mHybridData" in class "Lcom/mrousavy/camera/core/VideoPipeline;" or its superclasses, Config-Diff: Difference(deviceChanged=true, outputsChanged=true, sidePropsChanged=true)
                                        java.lang.NoSuchFieldError: no "Lcom/facebook/jni/HybridData;" field "mHybridData" in class "Lcom/mrousavy/camera/core/VideoPipeline;" or its superclasses
                                        	at com.mrousavy.camera.core.VideoPipeline.removeRecordingSessionOutputSurface(Native Method)
                                        	at com.mrousavy.camera.core.VideoPipeline.s(VideoPipeline.kt:64)
                                        	at ba.l.S(CameraSession.kt:28)
                                        	at ba.l.B(CameraSession.kt:869)
                                        	at ba.l.h(CameraSession.kt:1)
                                        	at ba.l$i.l(Unknown Source:12)
                                        	at kc.a.b(ContinuationImpl.kt:14)
                                        	at zc.u0.run(DispatchedTask.kt:117)
                                        	at android.os.Handler.handleCallback(Handler.java:942)
                                        	at android.os.Handler.dispatchMessage(Handler.java:99)
                                        	at android.os.Looper.loopOnce(Looper.java:211)
                                        	at android.os.Looper.loop(Looper.java:300)
                                        	at android.os.HandlerThread.run(HandlerThread.java:67)
23021-23189 System.err               W  	at com.mrousavy.camera.core.VideoPipeline.removeRecordingSessionOutputSurface(Native Method)
23021-23189 System.err               W  	at com.mrousavy.camera.core.VideoPipeline.s(VideoPipeline.kt:64)
23021-23189 CameraSession            E  Failed to configure CameraSession! Error: no "Lcom/facebook/jni/HybridData;" field "mHybridData" in class "Lcom/mrousavy/camera/core/VideoPipeline;" or its superclasses, Config-Diff: Difference(deviceChanged=true, outputsChanged=true, sidePropsChanged=true)
                                        java.lang.NoSuchFieldError: no "Lcom/facebook/jni/HybridData;" field "mHybridData" in class "Lcom/mrousavy/camera/core/VideoPipeline;" or its superclasses
                                        	at com.mrousavy.camera.core.VideoPipeline.removeRecordingSessionOutputSurface(Native Method)
                                        	at com.mrousavy.camera.core.VideoPipeline.s(VideoPipeline.kt:64)
                                        	at ba.l.S(CameraSession.kt:28)
                                        	at ba.l.B(CameraSession.kt:869)
                                        	at ba.l.h(CameraSession.kt:1)
                                        	at ba.l$i.l(Unknown Source:12)
                                        	at kc.a.b(ContinuationImpl.kt:14)
                                        	at zc.u0.run(DispatchedTask.kt:117)
                                        	at android.os.Handler.handleCallback(Handler.java:942)
                                        	at android.os.Handler.dispatchMessage(Handler.java:99)
                                        	at android.os.Looper.loopOnce(Looper.java:211)
                                        	at android.os.Looper.loop(Looper.java:300)
                                        	at android.os.HandlerThread.run(HandlerThread.java:67)
23021-23189 System.err               W  	at com.mrousavy.camera.core.VideoPipeline.removeRecordingSessionOutputSurface(Native Method)
23021-23189 System.err               W  	at com.mrousavy.camera.core.VideoPipeline.s(VideoPipeline.kt:64)
```

### Camera Device

```json
const device = useCameraDevice('back', { physicalDevices: ['wide-angle-camera'] });
  const format = useCameraFormat(device, [
    { videoResolution: { width: 1280, height: 720 } },
    { fps: 24 },
    { videoHDR: false },
    { photoHDR: false },
    { pixelFormat: 'yuv' },
  ]);
```

### Device

android 

### VisionCamera Version

3.6.6

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [X] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

daltachange, 2023-11-17:
@mrousavy

daltachange, 2023-11-17:
react-natvie 0.71.8

mrousavy, 2023-11-17:
Hey - thanks for the logs! I just fixed this in https://github.com/mrousavy/react-native-vision-camera/pull/2169 :) If you appreciate my work and dedication to the library please consider [💖 sponsoring me on GitHub 💖](https://github.com/sponsors/mrousavy) to keep the project running!

closed 2023-11-17 by mrousavy