margelo/react-native-vision-camera #2167: ❓ ERROR Frame Processor threw an error: Exception in HostFunction: java.lang.IllegalArgumentException: Only JPEG and YUV_420_888 are supported now

sonhmivirse, 2023-11-17:
### Question

I am using these package:

"react-native-reanimated": "^3.5.4",
"react-native-vision-camera": "^3.6.4",
"react-native-worklets-core": "^0.2.4"

I follow instruction in docs for create frame processor plugins with MLkit face detector and receive error:
```shell
ERROR Frame Processor threw an error: Exception in HostFunction: java.lang.IllegalArgumentException: Only JPEG and YUV_420_888 are supported now
```

### What I tried

I 've log the pixel format of the frame in code and receive type is 'native'
I 've try to change the pixelFormat in react-native code to 'yuv' and receive this error:
```shell
ERROR Frame Processor threw an error: Exception in HostFunction: Received unknown JNI type "class com.facebook.react.bridge.WritableNativeArray"! Cannot convert to jsi::Value.
```
I change to 'rgb', i receive white screen with no warning or error logs.

Anyone solve this problem or anyway to convert the frame to extract pixelFormat?

### VisionCamera Version

3.6.4

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

sonhmivirse, 2023-11-17:
@mrousavy

xseignard, 2023-11-17:
> I 've try to change the pixelFormat in react-native code to 'yuv' and receive this error:
Way to go 👌 

The error you have after seems to be a bug on your Android code.

Can you show it?

sonhmivirse, 2023-11-17:
@xseignard 
Thanks for answer, this is my android code for FaceDetectorFrameProcessorPlugin:

```java
package com.meditationaimobile;

import android.graphics.ImageFormat;
import android.graphics.PointF;
import android.graphics.Rect;
import android.media.Image;
import android.media.ImageReader;
import android.util.Log;

import androidx.annotation.NonNull;
import androidx.annotation.Nullable;

import com.facebook.react.bridge.Arguments;
import com.facebook.react.bridge.WritableMap;
import com.facebook.react.bridge.WritableNativeArray;
import com.facebook.react.bridge.WritableNativeMap;
import com.google.android.gms.tasks.Task;
import com.google.android.gms.tasks.Tasks;
import com.google.mlkit.vision.common.InputImage;
import com.google.mlkit.vision.face.Face;
import com.google.mlkit.vision.face.FaceContour;
import com.google.mlkit.vision.face.FaceDetection;
import com.google.mlkit.vision.face.FaceDetector;
import com.google.mlkit.vision.face.FaceDetectorOptions;
import com.mrousavy.camera.frameprocessor.Frame;
import com.mrousavy.camera.frameprocessor.FrameProcessorPlugin;

import java.nio.ByteBuffer;
import java.util.List;
import java.util.Map;

public class FaceDetectorFrameProcessorPlugin extends FrameProcessorPlugin {
    /**
     * The initializer of this Frame Processor Plugin.
     * This is called everytime this Frame Processor Plugin is loaded from the JS side (`initFrameProcessorPlugin(..)`).
     * Optionally override this method to implement custom initialization logic.
     *
     * @param options An options dictionary passed from the JS side, or null if none.
     */
    public FaceDetectorFrameProcessorPlugin(@Nullable Map<String, Object> options) {
        super(options);

    }

    FaceDetectorOptions options = new FaceDetectorOptions.Builder()
            .setPerformanceMode(FaceDetectorOptions.PERFORMANCE_MODE_ACCURATE)
            .setContourMode(FaceDetectorOptions.CONTOUR_MODE_ALL)
            .setClassificationMode(FaceDetectorOptions.CLASSIFICATION_MODE_ALL)
            .setMinFaceSize(0.15f)
            .build();
    FaceDetector faceDetector = FaceDetection.getClient(options);

    private WritableMap processBoundingBox (Rect boundingBox) {
        WritableMap bounds = Arguments.createMap();

        // Calculate offset (we need to center the overlay on the target)
        Double offsetX =  (boundingBox.exactCenterX() - Math.ceil(boundingBox.width())) / 2.0f;
        Double offsetY =  (boundingBox.exactCenterY() - Math.ceil(boundingBox.height())) / 2.0f;

        Double x = boundingBox.right + offsetX;
        Double y = boundingBox.top + offsetY;

        bounds.putDouble("x", boundingBox.centerX() + (boundingBox.centerX() - x));
        bounds.putDouble("y", boundingBox.centerY() + (y - boundingBox.centerY()));
        bounds.putDouble("width", boundingBox.width());
        bounds.putDouble("height", boundingBox.height());
[... 124 lines cut ...]
        // Calculate sizes
        int ySize = yBuffer.remaining();
        int uSize = uBuffer.remaining();
        int vSize = vBuffer.remaining();

        // Ensure the input buffer has enough data
        if (yuvBuffer.remaining() < ySize + uSize + vSize) {
            throw new IllegalArgumentException("Not enough data in the input buffer");
        }

        // Copy the YUV_420_888 data into the image planes
        yuvBuffer.get(yBuffer.array(), 0, ySize);
        yuvBuffer.get(vBuffer.array(), 0, vSize);
        yuvBuffer.get(uBuffer.array(), 0, uSize);

        // Release the ImageReader
        imageReader.close();

        return image;
    }
}

```

I follow this repository: https://github.com/mrousavy/react-native-vision-camera/pull/2108

xseignard, 2023-11-17:
Can you try to use plain:
```java
import java.util.ArrayList;
import java.util.HashMap;
```

instead of:
```java
import com.facebook.react.bridge.WritableNativeArray;
import com.facebook.react.bridge.WritableNativeMap;
```

?

sonhmivirse, 2023-11-17:
@xseignard 
There no error anymore but seem my code not work like i expect, no faces were detected.
Anyway, thanks so much

xseignard, 2023-11-17:
> no error anymore

Glad to read that 🎉 

> no faces were detected

Well this is not an issue with `react-native-vision-camera` anymore, it is more on your own code.

I think this issue can be closed.

closed 2023-11-17 by sonhmivirse

mrousavy, 2023-11-17:
Might be wrong Orientation or Model not correctly loading.