margelo/react-native-vision-camera #2168: 🐛 Video recording does not work in release mode

Romick2005, 2023-11-17:
### What's happening?

Cannot record camera video.
`java.lang.NoSuchFieldError: no "Lcom/facebook/jni/HybridData;" field "mHybridData" in class "Lcom/mrousavy/camera/core/VideoPipeline;"`

### Reproduceable Code

```tsx
Any camera example code.
```

### Relevant log output

```shell
11-17 12:05:27.501 21307   477 I CameraSession: Successfully configured Session with 1 outputs for Camera #1!
11-17 12:05:27.501 21307   477 I CameraSession: Updating Video Outputs...
11-17 12:05:27.501 21307   477 I VideoPipeline: Removing FrameProcessor Output...
11-17 12:05:27.501 21307   477 I VideoPipeline: Removing RecordingSession Output...
11-17 12:05:27.501   944  1093 I CamX    : [ INFO][HAL    ] camxsession.cpp:7923 ResetMetabufferPendingQueue() MetaBuffer done Maximum delay for ZSLSnapshotMipiToUnpacketRaw_0 is 0
11-17 12:05:27.501   944  1093 I CamX    : [CORE_CFG][CORE   ] camxsession.cpp:2431 Initialize() Session (0xb400007378b28f40) Initialized ZSLSnapshotMipiToUnpacketRaw
11-17 12:05:27.501   944  1232 I CamX    : [CORE_CFG][CORE   ] camxnode.cpp:806 InitializeSourcePortBufferFormat() Topology: Node::ZSLSnapshotYUVHALwithCDS_OPE2 has a source port id 1 using format 9 dim 2304x1728, stream 0xb40000737c4bb480
11-17 12:05:27.502 21307   477 E CameraSession: Failed to configure CameraSession! Error: no "Lcom/facebook/jni/HybridData;" field "mHybridData" in class "Lcom/mrousavy/camera/core/VideoPipeline;" or its superclasses, Config-Diff: Difference(deviceChanged=true, outputsChanged=true, sidePropsChanged=true)
11-17 12:05:27.502 21307   477 E CameraSession: java.lang.NoSuchFieldError: no "Lcom/facebook/jni/HybridData;" field "mHybridData" in class "Lcom/mrousavy/camera/core/VideoPipeline;" or its superclasses
11-17 12:05:27.502 21307   477 E CameraSession: 	at com.mrousavy.camera.core.VideoPipeline.removeRecordingSessionOutputSurface(Native Method)
11-17 12:05:27.502 21307   477 E CameraSession: 	at com.mrousavy.camera.core.VideoPipeline.W(Unknown Source:59)
11-17 12:05:27.502 21307   477 E CameraSession: 	at com.mrousavy.camera.core.c.c1(Unknown Source:27)
11-17 12:05:27.502 21307   477 E CameraSession: 	at com.mrousavy.camera.core.c.r0(Unknown Source:856)
11-17 12:05:27.502 21307   477 E CameraSession: 	at com.mrousavy.camera.core.c.f(Unknown Source:0)
11-17 12:05:27.502 21307   477 E CameraSession: 	at com.mrousavy.camera.core.c$i.invokeSuspend(Unknown Source:12)
11-17 12:05:27.502 21307   477 E CameraSession: 	at kotlin.coroutines.jvm.internal.a.resumeWith(Unknown Source:11)
11-17 12:05:27.502 21307   477 E CameraSession: 	at ze.x0.run(Unknown Source:93)
11-17 12:05:27.502 21307   477 E CameraSession: 	at android.os.Handler.handleCallback(Handler.java:942)
11-17 12:05:27.502 21307   477 E CameraSession: 	at android.os.Handler.dispatchMessage(Handler.java:99)
11-17 12:05:27.502 21307   477 E CameraSession: 	at android.os.Looper.loopOnce(Looper.java:211)
11-17 12:05:27.502 21307   477 E CameraSession: 	at android.os.Looper.loop(Looper.java:300)
11-17 12:05:27.502 21307   477 E CameraSession: 	at android.os.HandlerThread.run(HandlerThread.java:67)
11-17 12:05:27.502 21307   477 E CameraView: invokeOnError(...):
11-17 12:05:27.502 21307   477 W System.err: java.lang.NoSuchFieldError: no "Lcom/facebook/jni/HybridData;" field "mHybridData" in class "Lcom/mrousavy/camera/core/VideoPipeline;" or its superclasses
11-17 12:05:27.502   944  1232 I CamX    : [ INFO][CORE   ] camxnode.cpp:8566 FinalizeInitialization() Node::ZSLSnapshotYUVHALwithCDS_OPE2 pChiStreamWrapper for Port Index at 0 is 0xb40000739a163280
11-17 12:05:27.502 21307   477 W System.err: 	at com.mrousavy.camera.core.VideoPipeline.removeRecordingSessionOutputSurface(Native Method)
11-17 12:05:27.502 21307   477 W System.err: 	at com.mrousavy.camera.core.VideoPipeline.W(Unknown Source:59)
11-17 12:05:27.502 21307   477 W System.err: 	at com.mrousavy.camera.core.c.c1(Unknown Source:27)
11-17 12:05:27.502 21307   477 W System.err: 	at com.mrousavy.camera.core.c.r0(Unknown Source:856)
11-17 12:05:27.502 21307   477 W System.err: 	at com.mrousavy.camera.core.c.f(Unknown Source:0)
11-17 12:05:27.502 21307   477 W System.err: 	at com.mrousavy.camera.core.c$i.invokeSuspend(Unknown Source:12)
11-17 12:05:27.502 21307   477 W System.err: 	at kotlin.coroutines.jvm.internal.a.resumeWith(Unknown Source:11)
11-17 12:05:27.502   944  1093 I CHIUSECASE: [INFO   ] chxmetadata.cpp:1137 AllocateBuffers() [CMB_DEBUG] Buffer count 1 1
11-17 12:05:27.502 21307   477 W System.err: 	at ze.x0.run(Unknown Source:93)
11-17 12:05:27.502 21307   477 W System.err: 	at android.os.Handler.handleCallback(Handler.java:942)
11-17 12:05:27.502   944  1093 I CHIUSECASE: [INFO   ] chxmetadata.cpp:1145 AllocateBuffers() [CMB_DEBUG] Allocate buffers for client 0xb40000739cb5bb08 1
11-17 12:05:27.502 21307   477 W System.err: 	at android.os.Handler.dispatchMessage(Handler.java:99)
11-17 12:05:27.502 21307   477 W System.err: 	at android.os.Looper.loopOnce(Looper.java:211)
11-17 12:05:27.502   944  1093 I CHIUSECASE: [INFO   ] chxmetadata.cpp:1668 RegisterSharedClient() [CMB_DEBUG] Register success index 3 id 3 1
11-17 12:05:27.502 21307   477 W System.err: 	at android.os.Looper.loop(Looper.java:300)
11-17 12:05:27.502 21307   477 W System.err: 	at android.os.HandlerThread.run(HandlerThread.java:67)
```

### Camera Device

```json
{reason: 'Have nothing to do with camera'}
```

### Device

iPhone XS

### VisionCamera Version

3.6.6

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

Romick2005, 2023-11-17:
Android Studio logs:
```
unknown:ReactNative     com.app                        E  
unknown/unknown: [unknown/unknown] no "Lcom/facebook/jni/HybridData;" field "mHybridData" in class "Lcom/mrousavy/camera/core/VideoPipeline;" or its superclasses, js engine: hermes, stack:
    Wrapper@1:179508
    _createSuperInternal@1:14308146
    CameraError@1:14308610
    _createSuperInternal@1:14308146
    CameraRuntimeError@1:14309108
    onError@1:14306146
    invokeGuardedCallbackImpl@1:310848
    invokeGuardedCallback@1:310905
    invokeGuardedCallbackAndCatchFirstError@1:310937
    executeDispatch@1:311069
    executeDispatchesAndReleaseTopLevel@1:315326
    forEachAccumulated@1:312552
    anonymous@1:315678
    batchedUpdatesImpl@1:374817
    batchedUpdates@1:315246
    _receiveRootNodeIDEvent@1:315519
    receiveEvent@1:369324
    __callFunction@1:84370
    anonymous@1:82726
    __guard@1:83677
    callFunctionReturnFlushedQueue@1:82684
```

zzz08900, 2023-11-17:
Just curious, how did you get Android logs from an iPhone XS?

mrousavy, 2023-11-17:
Hey - thanks for the detailed logs! I just fixed this in https://github.com/mrousavy/react-native-vision-camera/pull/2169 :) If you appreciate my work and dedication to the library please consider [💖 sponsoring me on GitHub 💖](https://github.com/sponsors/mrousavy) to keep the project running!

mrousavy, 2023-11-17:
[image]

This is wrong. You created an Android-specific error with Android logs, it can't be an iPhone Xs haha - just like @zzz08900 said

PR #2169 (2023-11-17): fix: Fix `VideoPipeline` crash because ProGuard removed `mHybridData`

## What

Fixes a crash in release mode that was caused by ProGuard stripping the `mHybridData` field in `VideoPipeline.kt`. 

## Changes

## Tested on

## Related issues

* Fixes #2168
* Fixes https://github.com/mrousavy/react-native-vision-camera/issues/1769
* Fixes https://github.com/mrousavy/react-native-vision-camera/issues/2166
closed 2023-11-17 by mrousavy