margelo/react-native-vision-camera #2172: 🐛 An error happens when stop recording  on iOS?

daltachange, 2023-11-18:
### What's happening?

Step1: start recording
Step2: stop recording after 10s.

Can not get the video file.
[image]

### Reproduceable Code

```tsx
<Camera
          format={format}
          style={{ flex: 1 }}
          audio={true}
          video={true}
          ref={camera}
          device={device}
          isActive={true}
          fps={24}
          hdr={false}
          enableZoomGesture={true}
          onInitialized={() => {}}
          pixelFormat="yuv"
        />
```

### Relevant log output

```shell
2023-11-15 18:19:54.985308+0800 presale[44320:6571929] *** Terminating app due to uncaught exception 'NSGenericException', reason: '*** -[AVCaptureSession stopRunning] stopRunning may not be called between calls to beginConfiguration and commitConfiguration'
```

### Camera Device

```json
const device = useCameraDevice('back', { physicalDevices: ['wide-angle-camera'] });
  const format = useCameraFormat(device, [
    { videoResolution: { width: 1280, height: 720 } },
    { fps: 24 },
    { videoHDR: false },
    { photoHDR: false },
    { pixelFormat: 'yuv' },
  ]);
```

### Device

iOS 16.1

### VisionCamera Version

3.6.6

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

daltachange, 2023-11-18:
@mrousavy RN: 0.71.8

daltachange, 2023-11-18:
#2132

mrousavy, 2023-11-18:
Everytime I say the same thing, I need the full log output. Just the "Terminating App" error is not enough, you already posted this in the info.

mrousavy, 2023-11-18:
Closing until the full log output has been added.

closed 2023-11-18 by mrousavy

mrousavy, 2023-11-18:
Also duplicate of https://github.com/mrousavy/react-native-vision-camera/issues/2132