margelo/react-native-vision-camera #2175: Migrating plugin to V3❓

francesco-clementi-92, 2023-11-19:
### Question

I'm trying to migrate vision-camera-ocr to vision camera v3 so I decided to do a fork of the original package, because it seems unmaintained.

This is the fork: https://github.com/francesco-clementi-92/vision-camera-ocr

After struggling really a lot I managed to make it work for android, but the first time that I try `yarn run android` the camera is blank.

If I disable by commenting the frameProcessor prop, then enable it again, it will work correctly.

Any idea on what could be the cause?

To replicate, go to the example folder, run 

When the app starts on your device you will see a black screen. Comment the frameProcessor prop:
```
<Camera
        style={[StyleSheet.absoluteFill]}
        //frameProcessor={frameProcessor}
        device={device}
        isActive={true}
        pixelFormat="yuv"
        onLayout={(event: LayoutChangeEvent) => {
          setPixelRatio(
            event.nativeEvent.layout.width /
              PixelRatio.getPixelSizeForLayoutSize(
                event.nativeEvent.layout.width,
              ),
          );
        }}
        //orientation={'portrait'}
      />
```
save and then uncomment it again.

It will work

### VisionCamera Version

3.6.4

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2023-11-19:
Hey - thanks for the effort in migrating it to V3, but what you're describing is a bug report, not a question - so please  create a new issue with the bug report template filled out (and logs included). Thanks!

closed 2023-11-19 by mrousavy

francesco-clementi-92, 2023-11-19:
I actually think that the problem is with my migration of the package, but I’ll open a new issue

thongquach, 2024-03-27:
Hi @francesco-clementi-92, I saw that you managed to make your plugin work with V3, right?
Why the author haven't mentioned you yet in the documentation? `vision-camera-ocr` does not work with V3 and is not maintained anymore.
[image]

francesco-clementi-92, 2024-03-28:
You are right, I need to publish the package to the author documentation, but I do not have time to mantain it.
I’ll publish it and I’ll try to find someone to support the maintaining effort.
In the meantime feel free to test it

thongquach, 2024-03-29:
Thanks for your contribution @francesco-clementi-92 . The plugin does not work with the latest version of vision-camera. Therefore, I need to fallback to version 3.6.
Do you know any other plugin that can recognize texts on V3?