margelo/react-native-vision-camera #2181: ❓ Draw on the frame

sonhmivirse, 2023-11-20:
### Question

I want to draw the landmark point on the frame following the docs that use Skia, but i realize that author has remove that feature.

I am using:

Please suggest me the solution to draw on frame by either js or java in android studio code. Thanks!
"react-native-vision-camera": "^3.6.4",
"react-native-worklets-core": "^0.2.4"
"react-native-reanimated": "^3.5.4",

### What I tried

- I've try to use an overlay, then setState the landmark to update point position but it not working because the error:
```shell
 Regular javascript function 'setState' cannot be shared. Try decorating the function with the 'worklet' keyword to allow the javascript function to be used as a worklet.
```
- I've do following chat GPT that use the useSharedValue from react-native-reanimated like this:

```javascript
const posePoints = useSharedValue({});

  const frameProcessor = useFrameProcessor(frame => {
    'worklet';
    const res = scanPose(frame);
    posePoints.value = res; // Update the shared value
  });
```

But receive error:
```shell
[Reanimated] Reading from _value directly is only possible on the UI runtime
```

### VisionCamera Version

3.6.4

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

amirnouraein, 2023-12-05:
running into the same issue. they removed reanimated sharedValue support and also Skia integration. so right now there's no way to actually make use of frameProcessors if i'm not mistaken ?

rodgomesc, 2023-12-05:
`useSharedValues` should be imported from 'react-native-worklets-core' not from reanimated2

pvedula7, 2023-12-05:
I've read the documentation but get forced camera reinitializations or lack of re-rendering with useSharedValue from worklets. Opened up an issue here: https://github.com/mrousavy/react-native-vision-camera/issues/2244

amirnouraein, 2023-12-05:
so the trick that did the job for me was to use Worklets.createRunInJsFn callback to update a Reanimated shared value within the frame processor (since you can't update reanimated shared values directly from there). and then used that to animate stuff on the screen

```js
import Reanimated, {
    useSharedValue as useReanimatedSharedValue,
} from 'react-native-reanimated';

const sharedValue = useReanimatedSharedValue(0);
const updateSharedValue = Worklets.createRunInJsFn((v) => {
       sharedValue.value = v;
})
```

pvedula7, 2023-12-08:
> so the trick that did the job for me was to use Worklets.createRunInJsFn callback to update a Reanimated shared value within the frame processor (since you can't update reanimated shared values directly from there). and then used that to animate stuff on the screen (in this case no re-renders would happen)
> 
> ```js
> import Reanimated, {
>     useSharedValue as useReanimatedSharedValue,
> } from 'react-native-reanimated';
> 
> const sharedValue = useReanimatedSharedValue(0);
> const updateSharedValue = Worklets.createRunInJsFn((v) => {
>        sharedValue.value = v;
> })
> ```

Are you using useAnimatedProps() from Reanimated? Reanimated shared values update properly when using them through Worklets. I can render fixed values but anytime I use a changing reanimated shared value in useAnimatedProps the animation disappears after the first frame.

c-info, 2023-12-15:
> so the trick that did the job for me was to use Worklets.createRunInJsFn callback to update a Reanimated shared value within the frame processor (since you can't update reanimated shared values [...]

Seems solution not working 

const updateSharedValue = Worklets.createRunInJsFn(v => {
    sharedValueReanimated.value = {...v};
  });

  const frameProcessor = useFrameProcessor(frame => {
    'worklet';
    const result = objectDetector(frame, {});
    sharedValue.value = {height: 100, left: 160, top: 160, width: 100};
    updateSharedValue(sharedValue.value);
  }, []);

const animatedStyles = useAnimatedStyle(
    () => ({
      backgroundColor: 'blue',
      position: 'absolute',
      ...sharedValueReanimated.value,
    }),
    [sharedValueReanimated],
  );
  
   <Reanimated.View style={animatedStyles} />
   
   UI is not showing initially

mrousavy, 2024-01-15:
Skia support unfortunately didnt make it into V3. Maybe V4?