margelo/react-native-vision-camera #2182: 🔧 React Native Vision Camera build failure !!

miracezra, 2023-11-20:
### How were you trying to build the app?

I am currently using React Native Camera, but I am experiencing severe lag issues when scanning QR codes. Therefore, I have decided to use React Native Vision Camera, but I'm encountering errors during the installation process. I've tried many solutions I found online but couldn't resolve the issue. I'm sharing the error message below. I would greatly appreciate your help

### Full build logs

```tsx
Class 'kotlin.Unit' was compiled with an incompatible version of Kotlin. The binary version of its metadata is 1.8.0, expected version is 1.5.1.
```

### Project dependencies

```json
dependencies: {
  react-native: 
  react-native-vision-camera: ^3.0.0
},
```

### VisionCamera Version

3.3.6

### Target platforms

iOS, Android

### Operating system

MacOS

### Can you build the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

rodgomesc, 2023-11-21:
hey @miracezra  can you double check which version of vc are you using exactly? you reported 3.3.6 but it is mismatching with your project dependencies version

FidoelPepino, 2023-11-22:
I can't compile the project due to the library. I've tried all versions, but it won't let me. I'm doing it from Windows :(

mrousavy, 2024-01-11:
Maybe upgrade your Kotlin version

closed 2024-01-11 by mrousavy

uqmessias, 2024-02-23:
@FidoelPepino, did you manage to make it work? I'm having a similar issue