margelo/react-native-vision-camera #2184: Camera not loading on first render 🐛

francesco-clementi-92, 2023-11-20:
### What's happening?

I'm migrating the plugin vision camera ocr to use Vision Camera V3, so I'm not sure if this is really a bug of Vision Camera or the way that I changed the plugin.

The problem is that the first time that I start the android app, the Camera is black, if I comment the frameProcessor prop, save and the uncomment it, it will work.

The example app can be found here:
https://github.com/francesco-clementi-92/vision-camera-ocr/tree/main/example

### Reproduceable Code

```tsx
<Camera
        style={[StyleSheet.absoluteFill]}
        frameProcessor={frameProcessor}
        device={device}
        isActive={true}
        pixelFormat="yuv"
        onLayout={(event: LayoutChangeEvent) => {
          setPixelRatio(
            event.nativeEvent.layout.width /
              PixelRatio.getPixelSizeForLayoutSize(
                event.nativeEvent.layout.width,
              ),
          );
        }}
      />
```

### Relevant log output

```shell
023-11-20 08:58:57.165  1722-2009  ActivityManager         system_server                        I  Start proc 10426:com.google.android.webview:webview_service/u0a223 for service {com.google.android.webview/org.chromium.android_webview.services.VariationsSeedServer} caller=com.visioncameraocrexample
2023-11-20 08:58:57.170  9037-10376 cr_LibraryLoader        com.visioncameraocrexample           I  Successfully loaded native library
2023-11-20 08:58:57.178  9037-10376 cr_CachingUmaRecorder   com.visioncameraocrexample           I  Flushed 7 samples from 7 histograms, 0 samples were dropped.
2023-11-20 08:58:57.240  9037-10450 TrafficStats            com.visioncameraocrexample           D  tagSocket(160) with statsTag=0xffffffff, statsUid=-1
2023-11-20 08:58:57.283  9037-10375 ameraocrexample         com.visioncameraocrexample           W  CheckJNI: method to register "installJSIBindings" not in the given class. This is slow, consider changing your RegisterNatives calls.
2023-11-20 08:58:57.426  9037-10375 ReactNativeJS           com.visioncameraocrexample           I  Loading react-native-worklets-core...
2023-11-20 08:58:57.430  9037-10375 ReactNativeJS           com.visioncameraocrexample           I  Worklets loaded successfully
2023-11-20 08:58:57.440  9037-10375 VisionCameraProxy       com.visioncameraocrexample           I  Initializing VisionCameraProxy...
2023-11-20 08:58:57.440  9037-10375 VisionCameraProxy       com.visioncameraocrexample           I  Creating Worklet Context...
2023-11-20 08:58:57.445  9037-10375 VisionCameraProxy       com.visioncameraocrexample           I  Worklet Context created!
2023-11-20 08:58:57.451  9037-10375 FrameProce...inRegistry com.visioncameraocrexample           I  Looking up Frame Processor Plugin "scanOCR"...
2023-11-20 08:58:57.452  9037-10375 FrameProce...inRegistry com.visioncameraocrexample           I  Frame Processor Plugin "scanOCR" found! Initializing...
2023-11-20 08:58:57.455  9037-10375 ReactNativeJS           com.visioncameraocrexample           I  undefined
2023-11-20 08:58:57.476  9037-10500 TrafficStats            com.visioncameraocrexample           D  tagSocket(163) with statsTag=0xffffffff, statsUid=-1
2023-11-20 08:58:57.486  9037-10375 ReactNativeJS           com.visioncameraocrexample           I  Running "VisionCameraOcrExample" with {"rootTag":11}
2023-11-20 08:58:57.789  9037-10570 TrafficStats            com.visioncameraocrexample           D  tagSocket(164) with statsTag=0xffffffff, statsUid=-1
2023-11-20 08:58:57.949  9037-10375 CameraInjector          com.visioncameraocrexample           D  updateCloudCameraControllerInfoAsync: has aleardy start update task.
2023-11-20 08:58:57.950  9037-10375 CameraInjector          com.visioncameraocrexample           D  waitForResult: 
2023-11-20 08:58:57.950  9037-10375 libc                    com.visioncameraocrexample           W  Access denied finding property "vendor.camera.aux.packagelist"
2023-11-20 08:58:57.950  9037-10375 libc                    com.visioncameraocrexample           W  Access denied finding property "vendor.camera.aux.packagelistext"
2023-11-20 08:58:57.994  1722-2009  Boost                   system_server                        D  hostingType=service, hostingName={com.android.cameraextensions/com.android.cameraextensions.CameraExtensionsProxyService}, callerPackage=com.visioncameraocrexample, isSystem=false, isBoostNeeded=false.
2023-11-20 08:58:57.994  1722-2009  ActivityManager         system_server                        I  Start proc 10596:com.android.cameraextensions/u0a335 for service {com.android.cameraextensions/com.android.cameraextensions.CameraExtensionsProxyService} caller=com.visioncameraocrexample
2023-11-20 08:58:58.252  9037-10375 Compatibil...geReporter com.visioncameraocrexample           D  Compat change id reported: 206033068; UID 10423; state: ENABLED
2023-11-20 08:58:59.882  9037-10869 TrafficStats            com.visioncameraocrexample           D  tagSocket(146) with statsTag=0xffffffff, statsUid=-1
2023-11-20 08:59:02.002  9037-11130 TrafficStats            com.visioncameraocrexample           D  tagSocket(146) with statsTag=0xffffffff, statsUid=-1
2023-11-20 08:59:02.536  9037-9037  PreviewView             com.visioncameraocrexample           I  Creating PreviewView...
2023-11-20 08:59:02.537  9037-9037  PreviewView             com.visioncameraocrexample           I  PreviewView onMeasure(0, 0)
2023-11-20 08:59:02.538  9037-9037  PreviewView             com.visioncameraocrexample           D  coverSize :: 1920x1080 (0.5625), 0x0 (NaN)
2023-11-20 08:59:02.538  9037-9037  PreviewView             com.visioncameraocrexample           D  Fitted dimensions set: 0x0
2023-11-20 08:59:02.570  9037-9037  PreviewView             com.visioncameraocrexample           I  Resizing PreviewView to 1920 x 1080...
2023-11-20 08:59:02.575  9037-9037  CameraView              com.visioncameraocrexample           I  Updating CameraSession...
2023-11-20 08:59:02.589  9037-10493 CameraSession           com.visioncameraocrexample           I  Updating CameraSession Configuration...
2023-11-20 08:59:02.590  9037-10493 CameraSession           com.visioncameraocrexample           I  Configuring Camera #0...
2023-11-20 08:59:02.592  9037-10493 CameraManager           com.visioncameraocrexample           I  Camera 0: Opening...
2023-11-20 08:59:02.599  9037-10493 libc                    com.visioncameraocrexample           W  Access denied finding property "persist.vendor.camera.privapp.list"
2023-11-20 08:59:02.599  9037-10493 CameraExtImplXiaoMi     com.visioncameraocrexample           D  initCameraDevice: 0
2023-11-20 08:59:02.600  1722-4466  CameraBooster           system_server                        I  notifyCameraForegroundState isCameraForeground: true, caller : com.visioncameraocrexample, mCameraScence :3
2023-11-20 08:59:02.631  9037-10493 MiuiMultiWindowUtils    com.visioncameraocrexample           D  freeform resolution args raw data:{  "liuqin": {    "freeform_args": {      "vertical_portrait":{"aspect_ratio":0.5625, "original_ratio":0.3334,"original_scale":0.835,"top_margin":0.049,"left_margin":0.2775},      "horizontal_portrait":{"aspect_ratio":0.5625, "original_ratio":0.3334,"original_scale":0.835,"top_margin":-1,"left_margin":0.6525},      "vertical_landscape":{"aspect_ratio":-1, "original_ratio":1,"original_scale":0.468,"top_margin":0.049,"left_margin":-1},      "horizontal_landscape":{"aspect_ratio":-1, "original_ratio":1,"original_scale":0.468,"top_margin":-1,"left_margin":0.4976}    },    "mini_freeform_args":{      "vertical_portrait":{"aspect_ratio":0.5625, "original_ratio":0.1278},      "horizontal_portrait":{"aspect_ratio":0.5625, "original_ratio":0.1278},      "vertical_landscape":{"aspect_ratio":-1, "original_ratio":0.1778},      "horizontal_landscape":{"aspect_ratio":-1, "original_ratio":0.1778}    }  },  "pipa": {    "freeform_args": {      "vertical_portrait":{"aspect_ratio":0.5625, "original_ratio":0.3334,"original_scale":0.835,"top_margin":0.049,"left_margin":0.2775},      "horizontal_portrait":{"aspect_ratio":0.5625, "original_ratio":0.3334,"original_scale":0.835,"top_margin":-1,"left_margin":0.6525},      "vertical_landscape":{"aspect_ratio":-1, "original_ratio":1,"original_scale":0.468,"top_margin":0.049,"left_margin":-1},      "horizontal_landscape":{"aspect_ratio":-1, "original_ratio":1,"original_scale":0.468,"top_margin":-1,"left_margin":0.4976}    },    "mini_freeform_args":{      "vertical_portrait":{"aspect_ratio":0.562 [... cut ...]
### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

canibal2, 2023-11-23:
This error appeared after version 3.6.7. (3.6.7, 3.6.8, 3.6.9)
3.6.6 is clear.
[image]

c-goettert, 2023-11-24:
I have the same issue when using a frameprocessor on my Pixel 7A / Android 14.

rodgomesc, 2023-11-24:
same on mi9t pro

Arpine-javakhkuhi, 2023-11-24:
Same error.

abdelrahman-muntaser, 2023-11-27:
> @mrousavy This error appeared after version 3.6.7. (3.6.7, 3.6.8, 3.6.9) 3.6.6 is clear. ![WhatsApp Image 2023-11-23 at 15 07 15](https://private-user-images.githubusercontent.com/150250882/285200742-ee953622-a94e-4406-8362-175299350cd3.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDEwNzIzNjEsIm5iZiI6MTcwMTA3MjA2MSwicGF0aCI6Ii8xNTAyNTA4ODIvMjg1MjAwNzQyLWVlOTUzNjIyLWE5NGUtNDQwNi04MzYyLTE3NTI5OTM1MGNkMy5qcGVnP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQUlXTkpZQVg0Q1NWRUg1M0ElMkYyMDIzMTEyNyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyMzExMjdUMDgwMTAxWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9M2ZkOTBjYzE3YzM5ZTI1N2Q5ZjI4ZDc0MTZjMTZmOTc4N2IzMWVmOTQzZWNmOTZlNTJjYzg3Y2ZlNzExNmYxYyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.vlpT0aCvGs9RSQDajmM0eFzudPk-bBYXRTCgxyXg1Rc)

`{
    "react-native-vision-camera": "3.6.6",
    "react-native-worklets-core": "^0.2.4",
    "react-native-reanimated": "^3.5.4",
}`

francesco-clementi-92, 2023-11-28:
With 3.6.4 is not happeing

Gruak, 2023-11-30:
Same on Lenovo Tab 4 10 Plus

naveenrobo, 2023-12-04:
Same on Samsung Tab A (Android 11). Using frame processor gives the error 

```
Camera.onError(unknown/unknown): [unknown/unknown] Only the original thread that created a view hierarchy can touch its views. [unknown/unknown: [unknown/unknown] Only the original thread that created a view hierarchy can touch its views.]
```

Logcat
```
[1204/222625.460171:ERROR:variations_seed_loader.cc(69)] Failed to open file for reading. Errno: 2
2023-12-04 22:26:27.130 10095-11184 CameraSession           com.rnkycverifyexample               E  Failed to configure CameraSession! Error: Only the original thread that created a view hierarchy can touch its views., Config-Diff: Difference(deviceChanged=true, outputsChanged=true, sidePropsChanged=true)
                                                                                                    android.view.ViewRootImpl$CalledFromWrongThreadException: Only the original thread that created a view hierarchy can touch its views.
                                                                                                    	at android.view.ViewRootImpl.checkThread(ViewRootImpl.java:10750)
                                                                                                    	at android.view.ViewRootImpl.requestLayout(ViewRootImpl.java:2209)
                                                                                                    	at android.view.View.requestLayout(View.java:27055)
                                                                                                    	at android.view.View.requestLayout(View.java:27055)
                                                                                                    	at android.view.View.requestLayout(View.java:27055)
                                                                                                    	at android.view.View.requestLayout(View.java:27055)
                                                                                                    	at android.view.View.requestLayout(View.java:27055)
                                                                                                    	at android.view.View.requestLayout(View.java:27055)
                                                                                                    	at android.view.View.requestLayout(View.java:27055)
                                                                                                    	at android.view.View.requestLayout(View.java:27055)
                                                                                                    	at com.mrousavy.camera.core.PreviewView.setSize(PreviewView.kt:26)
                                                                                                    	at com.mrousavy.camera.core.PreviewView.resizeToInputCamera(PreviewView.kt:54)
                                                                                                    	at com.mrousavy.camera.core.CameraSession.configureCameraDevice(CameraSession.kt:240)
                                                                                                    	at com.mrousavy.camera.core.CameraSession.access$configureCameraDevice(CameraSession.kt:56)
                                                                                                    	at com.mrousavy.camera.core.CameraSession$configureCameraDevice$1.invokeSuspend(Unknown Source:15)
                                                                                                    	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
                                                                                                    	at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:106)
                                                                                                    	at android.os.Handler.handleCallback(Handler.java:938)
                                                                                                    	at android.os.Handler.dispatchMessage(Handler.java:99)
                                                                                                    	at android.os.Looper.loop(Looper.java:246)
                                                                                                    	at android.os.HandlerThread.run(HandlerThread.java:67)
2023-12-04 22:26:27.130 10095-11184 CameraView              com.rnkycverifyexample               E  invokeOnError(...):
2023-12-04 22:26:27.187 10095-11166 ReactNativeJS           com.rnkycverifyexample               E  'Camera.onError(unknown/unknown): [unknown/unknown] Only the original thread that created a view hierarchy can touch its views.', { [unknown/unknown: [unknown/unknown] Only the original thread that created a view hierarchy can touch its views.]
                                                                                                      name: 'unknown/unknown',
                                                                                                      _code: 'unknown/unknown',
                                                                                                      _message: '[unknown/unknown] Only the original thread that created a view hierarchy can touch its views.',
                                                                                                      _cause: 
[... cut ...] pl.kt:33)\n\tat kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:106)\n\tat android.os.Handler.handleCallback(Handler.java:938)\n\tat android.os.Handler.dispatchMessage(Handler.java:99)\n\tat android.os.Looper.loop(Looper.java:246)\n\tat android.os.HandlerThread.run(HandlerThread.java:67)\n',
                                                                                                         message: 'Only the original thread that created a view hierarchy can touch its views.' } }
```

Hy3P, 2023-12-10:
Try adding this line 

```diff
diff --git a/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/PreviewView.kt b/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/PreviewView.kt
index d462abc..b4adcb6 100644
--- a/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/PreviewView.kt
+++ b/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/PreviewView.kt
@@ -9,6 +9,7 @@ import android.view.Gravity
 import android.view.SurfaceHolder
 import android.view.SurfaceView
 import android.widget.FrameLayout
+import com.facebook.react.bridge.UiThreadUtil
 import com.mrousavy.camera.extensions.bigger
 import com.mrousavy.camera.extensions.getMaximumPreviewSize
 import com.mrousavy.camera.extensions.getPreviewTargetSize
@@ -21,11 +22,13 @@ import kotlin.math.roundToInt
 class PreviewView(context: Context, callback: SurfaceHolder.Callback) : SurfaceView(context) {
   var size: Size = getMaximumPreviewSize()
     set(value) {
-      Log.i(TAG, "Resizing PreviewView to ${value.width} x ${value.height}...")
-      holder.setFixedSize(value.width, value.height)
-      requestLayout()
-      invalidate()
-      field = value
+      UiThreadUtil.runOnUiThread {
+        Log.i(TAG, "Resizing PreviewView to ${value.width} x ${value.height}...")
+        holder.setFixedSize(value.width, value.height)
+        requestLayout()
+        invalidate()
+        field = value
+      }
     }
   var resizeMode: ResizeMode = ResizeMode.COVER
     set(value) {

```

francesco-clementi-92, 2023-12-11:
> Try adding this line
>
> ```diff
> diff --git a/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/PreviewView.kt [...]

I confirm that this line the error is not happening

closed 2023-12-12 by mrousavy