margelo/react-native-vision-camera #2186: 🐛 3.6.6: Audio & Video Recording out of sync

yspreen, 2023-11-20:
### What's happening?

Recording a video with audio has weird behavior in 3.6.6 while the same code was working fine in 3.6.4 (with the audio fix mentioned in #2085)

It appears that when starting `record` the video frames in the produced .mov file are actually offset by ~200ms. Which means the video files that are stored were recorded **before** the record button was hit.

The audio frames being stored are in sync with when `record` / `stop` is being hit.

It seems like video frames are cut off at the end. So the extra 200ms at the start are compensated for by the last 200ms of the video being a still frame.

Using 3.6.4 with the workaround is how we're mitigating this issue for now, if anyone else has these issues.

Could not repro in example app because I couldn't get example app to build. Maybe node version mismatch? Gave up trying to get it to run after 1h

### Reproduceable Code

```tsx
const back = useCameraDevice("back")
  const front = useCameraDevice("front")
  const device = (useFront ? front : back) ?? back
  const { hasPermission, requestPermission } = useCameraPermission()
  const { hasPermission: hasMicPermission, requestPermission: requestMicPermission } =
    useMicrophonePermission()
  useEffect(() => {
    hasPermission || requestPermission()
    hasMicPermission || requestMicPermission()
  }, [])
//...

      <Camera
        ref={camera}
        device={device}
        isActive={isFocused}
        style={$rel}
        enableZoomGesture={true}
        video={true}
        audio={true}
        format={format}
        videoStabilizationMode={supportsVideoStabilization ? "cinematic-extended" : "off"}
      >
        {overlay}
      </Camera>
```

### Relevant log output

```shell
VisionCamera.didSetProps(_:): Updating 15 props: [video, flex, onViewReady, cameraId, enableZoomGesture, position, onError, videoStabilizationMode, onInitialized, audio, enableBufferCompression, format, onCodeScanned, enableFrameProcessor, isActive]
VisionCamera.configure(_:): Updating Session Configuration...
VisionCamera.withSessionLock(_:): Beginning CameraSession configuration...
VisionCamera.configure(_:): Beginning AudioSession configuration...
VisionCamera.configureAudioSession(configuration:): Configuring Audio Session...
VisionCamera.configureDevice(configuration:): Configuring Input Device...
VisionCamera.configureDevice(configuration:): Configuring Camera com.apple.avfoundation.avcapturedevice.built-in_video:0...
VisionCamera.configureAudioSession(configuration:): Adding Audio input...
VisionCamera.configureAudioSession(configuration:): Adding Audio Data output...
VisionCamera.configureDevice(configuration:): Successfully configured Input Device!
VisionCamera.configureOutputs(configuration:): Configuring Outputs...
VisionCamera.configureOutputs(configuration:): Adding Video Data output...
VisionCamera.configureOutputs(configuration:): Successfully configured all outputs!
VisionCamera.configure(_:): Committed AudioSession configuration!
VisionCamera.withSessionLock(_:): Committed CameraSession configuration!
VisionCamera.withDeviceLock(_:): Beginning CaptureDevice configuration...
VisionCamera.configureFormat(configuration:device:): Configuring Format (4224x2376 | 1280x720@30.0 (ISO: 55.0..5280.0, Pixel Formats: [VisionCamera.PixelFormat.yuv, VisionCamera.PixelFormat.yuv, VisionCamera.PixelFormat.rgb, VisionCamera.PixelFormat.yuv, VisionCamera.PixelFormat.unknown, VisionCamera.PixelFormat.yuv, VisionCamera.PixelFormat.unknown, VisionCamera.PixelFormat.rgb, VisionCamera.PixelFormat.unknown]))...
VisionCamera.configureFormat(configuration:device:): Successfully configured Format!
VisionCamera.withDeviceLock(_:): Committed CaptureDevice configuration!
VisionCamera.withDeviceLock(_:): Beginning CaptureDevice configuration...
VisionCamera.withDeviceLock(_:): Committed CaptureDevice configuration!
VisionCamera.onSessionInitialized(): Camera initialized!
VisionCamera.startRecording(options:onVideoRecorded:onError:): Starting Video recording...
[... 2155 lines cut ...]
  "position": "back",
  "isMultiCam": false,
  "minZoom": 1
}
```

### Device

iPhone 15 Pro iOS 17.0.3

### VisionCamera Version

3.6.6

### Can you reproduce this issue in the VisionCamera Example app?

No, I cannot reproduce the issue in the Example app

### Additional information

- [X] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

yspreen, 2023-11-20:
video recording of the behavior (see video playing back at the end and times not lining up): https://youtu.be/QfYtFleTjuM

mrousavy, 2023-11-22:
Hey, I just fixed this in [v3.6.9](https://github.com/mrousavy/react-native-vision-camera/releases/tag/v3.6.9)! :)
If you appreciate my dedication and efforts to continuously improve VisionCamera, please consider [💖 sponsoring me on GitHub 💖](https://github.com/sponsors/mrousavy) to keep the open-source project running!

mrousavy, 2023-11-22:
Actually nvm I can still reproduce this issue on iOS 17

mrousavy, 2023-11-23:
Now I finally fixed it! This took way too long, I spent multiple days on researching this, profiling it, browsing Apple Forums, etc. - I even thought it was an Apple LLDB bug!

Here's the fix: https://github.com/mrousavy/react-native-vision-camera/pull/2206

closed 2023-11-23 by mrousavy