margelo/react-native-vision-camera #2187: does the code scanner work for graphene device❓

nicolasburtey, 2023-11-21:
### Question

some users are reporting their QR code scanner doesn't work.

could it be that if a user is using graphene and not using the google api that it may prevent the app to download the `MLKit Vision Barcode Scanning` ?

### VisionCamera Version

3.6.8

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

rodgomesc, 2023-11-21:
bingo!, i think that you got the point, the model will definitely not download if google services is not available, are you saying that those devices don't support google services?

mrousavy, 2023-11-22:
Yup, lazy loading needs the Google Services. Did you update your AndroidManifests to include this?

[image]

mrousavy, 2024-01-15:
Added a new flag to embed the model in the bundle

closed 2024-01-15 by mrousavy