margelo/react-native-vision-camera #2188: Draft: Android crash when permission change at background 🐛

BrianCodeItUp, 2023-11-21:
### What's happening?

I have a screen with code scanner feature. I will ask permission will user enter the screen. After user granted the camera permission. `Camera` component will render. If user go to setting and change camera permission, the app will crash when the app comes back to foreground.

### Reproduceable Code

```tsx
const [isPermissionGranted, setIsPermissionGranted] = useState(false);
useEffect(() => {
  async function handlePermission () {
    const result = await checkPermission();
      if (!result) {
        navigation.goBack();
        return;
      }
   setIsEnablePermission(!!result);
  }
  handlePermission()
 }, []);

const codeScanner = useCodeScanner({
    codeTypes: ['qr'],
    onCodeScanned: handleScanByCamera,
 });
const device = useCameraDevice('back');

return (
  <Camera
    style={StyleSheet.absoluteFill}
    device={device}
    isActive={isFocus && appState === 'active'}
    photo={true}
    codeScanner={codeScanner}
    torch={flashStatus ? 'on' : 'off'}
 />
)
```

### Relevant log output

```shell
2023-11-20 18:24:48.401 24219-25144 CameraManager           app.zqb.android.dev                  E  Camera 0: Error! 3
2023-11-20 18:24:48.404 24219-25144 CameraSession           app.zqb.android.dev                  E  Camera Device android.hardware.camera2.impl.CameraDeviceImpl@3d02098 has been disconnected!
                                                                                                    com.mrousavy.camera.core.CameraDisconnectedError: [session/camera-has-been-disconnected] The given Camera device (id: 0) has been disconnected! Error: CAMERA_IS_DISABLED_BY_ANDROID
                                                                                                    	at com.mrousavy.camera.extensions.CameraManager_openCameraKt$openCamera$2$callback$1.onError(CameraManager+openCamera.kt:49)
                                                                                                    	at android.hardware.camera2.impl.CameraDeviceImpl$CameraDeviceCallbacks.notifyError(CameraDeviceImpl.java:2013)
                                                                                                    	at android.hardware.camera2.impl.CameraDeviceImpl$CameraDeviceCallbacks.lambda$Sm85frAzwGZVMAK-NE_gwckYXVQ(Unknown Source:0)
                                                                                                    	at android.hardware.camera2.impl.-$$Lambda$CameraDeviceImpl$CameraDeviceCallbacks$Sm85frAzwGZVMAK-NE_gwckYXVQ.accept(Unknown Source:8)
                                                                                                    	at com.android.internal.util.function.pooled.PooledLambdaImpl.doInvoke(PooledLambdaImpl.java:271)
                                                                                                    	at com.android.internal.util.function.pooled.PooledLambdaImpl.invoke(PooledLambdaImpl.java:195)
                                                                                                    	at com.android.internal.util.function.pooled.OmniFunction.run(OmniFunction.java:86)
                                                                                                    	at android.os.Handler.handleCallback(Handler.java:883)
                                                                                                    	at android.os.Handler.dispatchMessage(Handler.java:100)
                                                                                                    	at android.os.Looper.loop(Looper.java:224)
                                                                                                    	at android.os.HandlerThread.run(HandlerThread.java:67)
2023-11-20 18:24:48.404 24219-25144 CameraView              app.zqb.android.dev                  E  invokeOnError(...):
2023-11-20 18:24:48.405 24219-25144 System.err              app.zqb.android.dev                  W  com.mrousavy.camera.core.CameraDisconnectedError: [session/camera-has-been-disconnected] The given Camera device (id: 0) has been disconnected! Error: CAMERA_IS_DISABLED_BY_ANDROID
2023-11-20 18:24:48.405 24219-25144 System.err              app.zqb.android.dev                  W  	at com.mrousavy.camera.extensions.CameraManager_openCameraKt$openCamera$2$callback$1.onError(CameraManager+openCamera.kt:49)
2023-11-20 18:24:48.405 24219-25144 System.err              app.zqb.android.dev                  W  	at android.hardware.camera2.impl.CameraDeviceImpl$CameraDeviceCallbacks.notifyError(CameraDeviceImpl.java:2013)
2023-11-20 18:24:48.405 24219-25144 System.err              app.zqb.android.dev                  W  	at android.hardware.camera2.impl.CameraDeviceImpl$CameraDeviceCallbacks.lambda$Sm85frAzwGZVMAK-NE_gwckYXVQ(Unknown Source:0)
2023-11-20 18:24:48.405 24219-25144 System.err              app.zqb.android.dev                  W  	at android.hardware.camera2.impl.-$$Lambda$CameraDeviceImpl$CameraDeviceCallbacks$Sm85frAzwGZVMAK-NE_gwckYXVQ.accept(Unknown Source:8)
2023-11-20 18:24:48.405 24219-25144 System.err              app.zqb.android.dev                  W  	at com.android.internal.util.function.pooled.PooledLambdaImpl.doInvoke(PooledLambdaImpl.java:271)
2023-11-20 18:24:48.405 24219-25144 System.err              app.zqb.android.dev                  W  	at com.android.internal.util.function.pooled.PooledLambdaImpl.invoke(PooledLambdaImpl.java:195)
2023-11-20 18:24:48.405 24219-25144 System.err              app.zqb.android.dev                  W  	at com.android.internal.util.function.pooled.OmniFunction.run(OmniFunction.java:86)
2023-11-20 18:24:48.405 24219-25144 System.err              app.zqb.android.dev                  W  	at android.os.Handler.handleCallback(Handler.java:883)
2023-11-20 18:24:48.405 24219-25144 System.err              app.zqb.android.dev                  W  	at android.os.Handler.dispatchMessage(Handler.java:100)
2023-11-20 18:24:48.405 24219-25144 System.err              app.zqb.android.dev                  W  	at android.os.Looper.loop(Looper.java:224)
2023-11-20 18:24:48.405 24219-25144 System.err              app.zqb.android.dev                  W  	at android.os.HandlerThread.run(HandlerThread.java:67)
2023-11-20 18:24:48.632  1304-1423  InputDispatcher         system_server                        E  channel '17a1751 app.zqb.android.dev/com.wellpay.MainActivity (server)' ~ Channel is unrecoverably broken and will be disposed!
---------------------------- PROCESS ENDED (24219) for package app.zqb.android.dev ----------------------------
[... 117 lines cut ...]
### Camera Device

```json
{}
```

### Device

Redmi

### VisionCamera Version

3.6.6

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

rodgomesc, 2023-11-21:
whoaa, that's interesting

rodgomesc, 2023-11-21:
@mrousavy  i'm wondering if we  should pause the camera session and release resources when the app is on foreground

mrousavy, 2023-11-21:
> CAMERA_IS_DISABLED_BY_ANDROID

interesting, well this should def. be an error (`Camera`'s `onError` prop), but it shouldn't crash the entire app

blueworlds1, 2023-11-28:
Also seeing this error in our logs, but it didn't crash the app

mrousavy, 2024-01-05:
Hey! Thanks for reporting this issue.

I've been working the past days on making sure the Camera lifecycle is safely handled and released when needed to make sure this crash doesn't occur anymore.

I just created a PR for this - can you please this to see if that fixes the issue for you? https://github.com/mrousavy/react-native-vision-camera/pull/2339

If this fixes your issue, please consider [💖 **sponsoring me on GitHub** 💖](https://github.com/sponsors/mrousavy/) to support me / thank me for building VisionCamera and continuously improving it. 

If this does not fix your issue, please clone the repo, check out the branch `fix/blackscreen` (the PR above), run the Example app, reproduce your issue there and share the `adb logcat` logs with me (upload to pastebin or gist) so I can investigate this further.

Thank you! 🙏

closed 2024-01-08 by mrousavy