margelo/react-native-vision-camera #2189: 🐛App crashed after scanning a qrcode

sunnystarh, 2023-11-21:
### What's happening?

I have a crash after scanning the QR code. I haven't found the reason yet. I have released the relevant code and logs. I hope someone can answer it.

### Reproduceable Code

```tsx
const onAuth = async () => {
    const grantedStatus = await Camera.requestCameraPermission();
    if (grantedStatus === 'granted') {
      setHasCameraPermit(true);
    } else {
      setHasCameraPermit(false);
    }
  };
  useEffect(() => {
    onAuth();
  }, []);

  const codeScanner = useCodeScanner({
    codeTypes: ['qr'],
    onCodeScanned: codeData => {
      if (codeData[0].value) {
        verificationCodeData.current = codeData[0].value;
        if (source === 'fromScanIcon') {
          scanCodeFromScanIcon();
        } else if (source === 'fromBindButton') {
          scanCodeFromBindButton();
        }
      }
    },
  });
       
      {device != null && HasCameraPermit (
        <Camera
          photo={false}
          video={false}
          isActive={isFocused}
          device={device}
          codeScanner={codeScanner}
          style={StyleSheet.absoluteFill}
          torch={TorchMode}
        />
      )}
```

### Relevant log output

```shell
--------- beginning of crash
2023-11-21 11:34:14.044 18948-22275 libc                    com.scan                          A  Fatal signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x77c20bc0f7 in tid 22275 (pool-10-thread-), pid 18948 (com.scan)
2023-11-21 11:34:14.078 18948-22278 BufferQueueProducer     com.scan                          E  [ImageReader-1280x720f23m2-18948-18](id:4a040000001e,api:4,p:1554,c:18948) queueBuffer: BufferQueue has been abandoned
2023-11-21 11:34:14.086 18948-22368 BufferQueueProducer     com.scan                          E  [ImageReader-1280x720f23m2-18948-18](id:4a040000001e,api:4,p:1554,c:18948) queueBuffer: BufferQueue has been abandoned
2023-11-21 11:34:14.117 18948-22368 BufferQueueProducer     com.scan                          E  [ImageReader-1280x720f23m2-18948-18](id:4a040000001e,api:4,p:1554,c:18948) queueBuffer: BufferQueue has been abandoned
2023-11-21 11:34:14.185 18948-18961 BufferQueueProducer     com.scan                          E  [ImageReader-1280x720f23m2-18948-18](id:4a040000001e,api:4,p:1554,c:18948) queueBuffer: BufferQueue has been abandoned
2023-11-21 11:34:14.189 18948-22278 BufferQueueProducer     com.scan                          E  [ImageReader-1280x720f23m2-18948-18](id:4a040000001e,api:4,p:1554,c:18948) queueBuffer: BufferQueue has been abandoned
2023-11-21 11:34:14.216 18948-22278 BufferQueueProducer     com.scan                          E  [ImageReader-1280x720f23m2-18948-18](id:4a040000001e,api:4,p:1554,c:18948) queueBuffer: BufferQueue has been abandoned
---------------------------- PROCESS STARTED (24847) for package com.scan ----------------------------
2023-11-21 11:34:14.249 18948-22278 BufferQueueProducer     com.scan                          E  [ImageReader-1280x720f23m2-18948-18](id:4a040000001e,api:4,p:1554,c:18948) queueBuffer: BufferQueue has been abandoned
2023-11-21 11:34:14.284 18948-22278 BufferQueueProducer     com.scan                          E  [ImageReader-1280x720f23m2-18948-18](id:4a040000001e,api:4,p:1554,c:18948) queueBuffer: BufferQueue has been abandoned
2023-11-21 11:34:14.531 24845-24845 DEBUG                   pid-24845                            A  Cmdline: com.scan
2023-11-21 11:34:14.531 24845-24845 DEBUG                   pid-24845                            A  pid: 18948, tid: 22275, name: pool-10-thread-  >>> com.scan <<<
2023-11-21 11:34:14.531 24845-24845 DEBUG                   pid-24845                            A        #00 pc 00000000000ed660  /data/app/~~Pua-MPb3_YSekKggq0wYvw==/com.scan-GEIEq0XRBLMCfIwCAOPiMw==/lib/arm64/libbarhopper_v3.so (BuildId: cf64ac3d5cfd71b04d3bc4112d678e48)
2023-11-21 11:34:14.531 24845-24845 DEBUG                   pid-24845                            A        #01 pc 00000000000d03c8  /data/app/~~Pua-MPb3_YSekKggq0wYvw==/com.scan-GEIEq0XRBLMCfIwCAOPiMw==/lib/arm64/libbarhopper_v3.so (BuildId: cf64ac3d5cfd71b04d3bc4112d678e48)
2023-11-21 11:34:14.531 24845-24845 DEBUG                   pid-24845                            A        #02 pc 00000000000ca240  /data/app/~~Pua-MPb3_YSekKggq0wYvw==/com.scan-GEIEq0XRBLMCfIwCAOPiMw==/lib/arm64/libbarhopper_v3.so (BuildId: cf64ac3d5cfd71b04d3bc4112d678e48)
2023-11-21 11:34:14.531 24845-24845 DEBUG                   pid-24845                            A        #03 pc 00000000000c9f64  /data/app/~~Pua-MPb3_YSekKggq0wYvw==/com.scan-GEIEq0XRBLMCfIwCAOPiMw==/lib/arm64/libbarhopper_v3.so (BuildId: cf64ac3d5cfd71b04d3bc4112d678e48)
2023-11-21 11:34:14.531 24845-24845 DEBUG                   pid-24845                            A        #04 pc 00000000000ca4f0  /data/app/~~Pua-MPb3_YSekKggq0wYvw==/com.scan-GEIEq0XRBLMCfIwCAOPiMw==/lib/arm64/libbarhopper_v3.so (BuildId: cf64ac3d5cfd71b04d3bc4112d678e48)
2023-11-21 11:34:14.531 24845-24845 DEBUG                   pid-24845                            A        #05 pc 00000000000c8aa4  /data/app/~~Pua-MPb3_YSekKggq0wYvw==/com.scan-GEIEq0XRBLMCfIwCAOPiMw==/lib/arm64/libbarhopper_v3.so (BuildId: cf64ac3d5cfd71b04d3bc4112d678e48)
2023-11-21 11:34:14.531 24845-24845 DEBUG                   pid-24845                            A        #06 pc 00000000000c8b44  /data/app/~~Pua-MPb3_YSekKggq0wYvw==/com.scan-GEIEq0XRBLMCfIwCAOPiMw==/lib/arm64/libbarhopper_v3.so (Java_com_google_android_libraries_barhopper_BarhopperV3_recognizeBufferNative+84) (BuildId: cf64ac3d5cfd71b04d3bc4112d678e48)
2023-11-21 11:34:14.531 24845-24845 DEBUG                   pid-24845                            A        #14 pc 00000000001e890c  [anon:dalvik-classes6.dex extracted in memory from /data/app/~~Pua-MPb3_YSekKggq0wYvw==/com.scan-GEIEq0XRBLMCfIwCAOPiMw==/base.apk!classes6.dex]
[... 24 lines cut ...]
### Camera Device

```json
{"hardwareLevel": "full", "hasFlash": true, "hasTorch": true, "id": "0", "isMultiCam": false, "maxZoom": 8, "minZoom": 1, "name": "BACK (0)", "neutralZoom": 1, "physicalDevices": ["wide-angle-camera"], "position": "back", "sensorOrientation": "landscape-right", "supportsFocus": true, "supportsLowLightBoost": true, "supportsRawCapture": true}
```

### Device

honor 70

### VisionCamera Version

v3.6.4

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

yasirlateef, 2023-11-21:
+1

In my case, the app only crashes after the first QR scan. Once I restart the app, it proceeds to scan QR codes without any crash. I just came across this issue yesterday. I'll definitely share any findings under this thread.

apadev666, 2023-11-21:
+ 1 
I have the same issue please tell what we can do

apadev666, 2023-11-21:
@mrousavy please take a look at this

mrousavy, 2023-11-21:
> @mrousavy please take a look at this

@apadev666 why?

yasirlateef, 2023-11-21:
@apadev666 this could happen for a number of reasons. Firstly, it could happen if you don't configure your camera properly. Try removing all the props except for the required ones and try adding them one-by-one to see what breaks.

yasirlateef, 2023-11-21:
There is no apparent bug related to the scanner functionality per-se.

apadev666, 2023-11-21:
I don't have any prop just standard initialization used from docs

apadev666, 2023-11-21:
<Camera
          photo
          device={device}
          isActive={isActive}
          codeScanner={codeScanner}
          style={[styles.absoluteFill]}
          torch={isTorchEnabled ? 'on' : 'off'}
        />
        
        here is code

apadev666, 2023-11-21:
it can crash for the first time but also after for example 10 attempts

rodgomesc, 2023-11-21:
can you guys try those steps and lmk if works

- update/check if google play services is on last version
- Settings > Apps > Google Play Services > Storage > Clear cache 

this seems to be related to mlkit itself

[mlkit/issues/714](https://github.com/googlesamples/mlkit/issues/714)
[mlkit/issues/598](https://github.com/googlesamples/mlkit/issues/598)

yasirlateef, 2023-11-21:
@sunnystarh I can confirm downgrading to `v.3.5.1` resolves this issue for now.

apadev666, 2023-11-21:
@yasirlateef @sunnystarh @rodgomesc for me event downgrade is not working((

apadev666, 2023-11-21:
@rodgomesc now Im using 23.44.14 version of google play service

yasirlateef, 2023-11-21:
> @yasirlateef @sunnystarh @rodgomesc for me event downgrade is not working((

If it were related to the package, downgrading would make a difference for you as well.

apadev666, 2023-11-21:
@yasirlateef it can be after some period, can you please try to scan barcode 10-15 times?

apadev666, 2023-11-21:
@rodgomesc I have tried also clean cache method as you mentioned, but the same crash(

rodgomesc, 2023-11-21:
> @yasirlateef @sunnystarh @rodgomesc for me event downgrade is not working((

are you sure? you answered this on the same minute that yasirlateef did yoy tried that version before?

apadev666, 2023-11-21:
W  W0000 00:00:1700572096.722037   12915 abi_method_util.h:33] not supported: TfLiteInterpreterGetTensor: TF Lite in Google Play services has stable ABI version 1.11.0 which is less than the required version 1.13.0.
2023-11-21 17:08:16.726 12211-12211 CameraSession           com.gingy                            I  Adding 4032 x 3024 Photo Output in Format #256...
2023-11-21 17:08:16.726 12211-12915 libc                    com.gingy                            A  Fatal signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x6e1f79325b in tid 12915 (pool-13-thread-), pid 12211 (com.gingy)
2023-11-21 17:08:16.750 12211-12233 BufferQueueProducer     com.gingy                            E  [ImageReader-1280x720f23m2-12211-30](id:2fb300000026,api:4,p:1331,c:12211) queueBuffer: BufferQueue has been abandoned
2023-11-21 17:08:16.754 12211-12710 BufferQueueProducer     com.gingy                            E  [ImageReader-1280x720f23m2-12211-30](id:2fb300000026,api:4,p:1331,c:12211) dequeueBuffer: BufferQueue has been abandoned
2023-11-21 17:08:16.754 12211-12211 CameraSession           com.gingy                            I  Adding 1280 x 720 CodeScanner Output in Format #35...
2023-11-21 17:08:16.757 12211-12211 CreateCaptureSession    com.gingy                            I  Camera 0: Creating Capture Session #1017... Hardware Level: 3} | Outputs: [android.hardware.camera2.params.OutputConfiguration@355ade1d, android.hardware.camera2.params.OutputConfiguration@5fdb62cc]
2023-11-21 17:08:16.757 12211-12211 CreateCaptureSession    com.gingy                            I  Using new API (>=28)
2023-11-21 17:08:16.776 12211-12233 BufferQueueProducer     com.gingy                            E  [ImageReader-1280x720f23m2-12211-30](id:2fb300000026,api:4,p:1331,c:12211) queueBuffer: BufferQueue has been abandoned
2023-11-21 17:08:17.342 12211-13360 BpBinder                com.gingy                            I  onLastStrongRef automatically unlinking death recipients: <uncached descriptor>
2023-11-21 17:08:17.345 12211-12598 CreateCaptureSession    com.gingy                            I  Camera 0: Capture Session #1017 configured!
2023-11-21 17:08:17.345 12211-12211 CameraSession           com.gingy                            I  Successfully configured Session with 2 outputs for Camera #0!
2023-11-21 17:08:17.345 12211-12598 CreateCaptureSession    com.gingy                            I  Camera 0: Capture Session #1016 closed!
2023-11-21 17:08:17.345 12211-12211 CameraSession           com.gingy                            W  Preview Output is null, aborting...
2023-11-21 17:08:17.345 12211-12211 CameraSession           com.gingy                            I  Successfully updated CameraSession Configuration! isActive: true
2023-11-21 17:08:17.347 12211-12211 CameraView              com.gingy                            I  Updating CameraSession...
2023-11-21 17:08:17.348 12211-12598 CameraSession           com.gingy                            I  Updating CameraSession Configuration...
2023-11-21 17:08:17.348 12211-12598 CameraSession           com.gingy                            I  Successfully updated CameraSession Configuration! isActive: false
2023-11-21 17:08:17.375 12211-12211 Choreographer           com.gingy                            I  Skipped 38 frames!  The application may be doing too much work on its main thread.
2023-11-21 17:08:17.530 12211-12211 PreviewView             com.gingy                            I  Creating PreviewView...
2023-11-21 17:08:17.531 12211-12211 PreviewView             com.gingy                            I  PreviewView onMeasure(0, 0)
2023-11-21 17:08:17.531 12211-12211 PreviewView             com.gingy                            D  coverSize :: 1920x1080 (0.5625), 0x0 (NaN)
2023-11-21 17:08:17.531 12211-12211 PreviewView             com.gingy                            D  Fitted dimensions set: 0x0
2023-11-21 17:08:17.541 12211-12211 PreviewView             com.gingy                            I  Resizing PreviewView to 1920 x 1080...
2023-11-21 17:08:17.542 12211-12211 CameraView              com.gingy                            I  Updating CameraSession...
2023-11-21 17:08:17.542 12211-12598 CameraSession           com.gingy                            I  Updating CameraSession Configuration...
2023-11-21 17:08:17.542 12211-12598 CameraSession           com.gingy                            I  Configuring Camera #0...
2023-11-21 17:08:17.542 12211-12598 CameraManager           com.gingy                            I  Camera 0: Opening...
2023-11-21 17:08:17.545  1331-28017 CameraService           pid-1331                             I  CameraService::connect call (PID -1 "com.gingy", camera ID 0) and Camera API version 2
2023-11-21 17:08:17.608 12211-12211 unknown:Na...chyManager com.gingy                            E  Unable to update properties for view tag 2363
                                                                                                    com.facebook.react.uimanager.IllegalViewOperationException: ViewManager for tag 2363 could not be found.
                                                                                                    
                                                                                                    	at com.facebook.react.uimanager.NativeViewHierarchyManager.resolveViewManager(NativeViewHierarchyManager.java:111)
[... cut ...]
2023-11-21 17:08:17.824  2171-3773  ActivityManager         pid-2171                             I  Process com.gingy (pid 12211) has died: prcp TOP 
2023-11-21 17:08:17.829  1331-2385  Camera2ClientBase       pid-1331                             I  Closed Camera 0. Client was: com.gingy (PID 12211, UID 11285)
2023-11-21 17:08:18.234  2171-2217  ActivityTaskManager     pid-2171                             W  Activity top resumed state loss timeout for ActivityRecord{3491e0 u0 com.gingy/.MainActivity t-1 f}}

here is logcat report @rodgomesc

yasirlateef, 2023-11-21:
@apadev666 You logs indicate no errors thrown from VisionCamera itself. The only exception I can see in your logs is:
```
 E Unable to update properties for view tag 2363
com.facebook.react.uimanager.IllegalViewOperationException: ViewManager for tag 2363 could not be found.
```

It shows that the crash is coming from something related to your JSX/Styles. I would say you're trying to use style properties which are not allowed. It's exactly why downgrading won't do any good for you because this thread isn't relevant to your issue.

apadev666, 2023-11-21:
@yasirlateef  can it be that I'm using Camera inside bottom sheet?
here is render of this component
!!device && (
      <BottomSheetModal
        ref={bottomSheetModalRef}
        snapPoints={[WINDOW_HEIGHT]}
        handleStyle={GlobalStyles.hide}
        onDismiss={() => setIsScannerVisible(false)}>
        <Camera
          photo
          device={device}
          isActive={isActive}
          codeScanner={codeScanner}
          style={[styles.absoluteFill]}
          torch={isTorchEnabled ? 'on' : 'off'}
        />
        <Pressable
          hitSlop={hitSlop}
          style={styles.closeContainer(insets?.top)}
          onPress={() => bottomSheetModalRef.current?.dismiss?.()}>
          <AppIcon width={19} height={19} icon="close" fill={Colors.white} />
        </Pressable>
        {!!isActive && (
          <View style={styles.fillCenter}>
            <AppIcon
              width={350}
              height={250}
              icon="barcodeArea"
              fill={Colors.transparent}
            />
          </View>
        )}
        <TouchableOpacity
          style={styles.lightIconContainer}
          onPress={() => setIsTorchEnabled(prevState => !prevState)}>
          <AppIcon
            width={44}
            height={44}
            fill={Colors.transparent}
            icon={isTorchEnabled ? 'lightOn' : 'lightOff'}
          />
        </TouchableOpacity>
      </BottomSheetModal>

rodgomesc, 2023-11-21:
can you guys please test if this change solves the issue for you?

remove `node_modules` modify vision-camera in `package.json` to be like this and see what happens
```json
"dependencies": {
  "react-native-vision-camera": "github:rodgomesc/react-native-vision-camera#cad3699"
}
```

apadev666, 2023-11-21:
@rodgomesc do I need to remove node modules and after package json update call yarn install?

after this useCameraPermission and other hooks cant be found

rodgomesc, 2023-11-21:
updated my previous post with a step by step

yasirlateef, 2023-11-21:
@rodgomesc Unfortunately, it doesn't solve the issue. After about 4 QR scans, the app crashes. I'll attach my `logcat` output below:

```
2023-11-21 20:05:58.108 23489-23543 OpenGLRenderer          com.kiosk                 E  Unable to match the desired swap behavior.
2023-11-21 20:05:58.128 23489-23543 OpenGLRenderer          com.kiosk                 E  Unable to match the desired swap behavior.
2023-11-21 20:05:58.186 23489-23543 OpenGLRenderer          com.kiosk                 E  Unable to match the desired swap behavior.
2023-11-21 20:06:00.157 23489-23583 chromium                com.kiosk                 E  [1121/200600.156809:ERROR:variations_seed_loader.cc(69)] Failed to open file for reading. Errno: 2
2023-11-21 20:06:03.221 23489-23571 CameraExte...agerGlobal com.kiosk                 E  Timed out while initializing proxy service!
2023-11-21 20:06:08.024 23489-23505 BufferQueueProducer     com.kiosk                 E  [ImageReader-1280x720f23m2-23489-1](id:5bc100000005,api:4,p:1483,c:23489) queueBuffer: BufferQueue has been abandoned
2023-11-21 20:06:08.042 23489-23505 BufferQueueProducer     com.kiosk                 E  [ImageReader-1280x720f23m2-23489-1](id:5bc100000005,api:4,p:1483,c:23489) dequeueBuffer: BufferQueue has been abandoned
2023-11-21 20:06:08.053 23489-23503 BufferQueueProducer     com.kiosk                 E  [ImageReader-1280x720f23m2-23489-1](id:5bc100000005,api:4,p:1483,c:23489) queueBuffer: BufferQueue has been abandoned
2023-11-21 20:06:12.365 23489-23505 BufferQueueProducer     com.kiosk                 E  [ImageReader-1280x720f23m2-23489-4](id:5bc100000009,api:4,p:1483,c:23489) queueBuffer: BufferQueue has been abandoned
2023-11-21 20:06:12.365 23489-23505 BufferQueueProducer     com.kiosk                 E  [ImageReader-1280x720f23m2-23489-4](id:5bc100000009,api:4,p:1483,c:23489) dequeueBuffer: BufferQueue has been abandoned
2023-11-21 20:06:12.551 23489-23725 libc                    com.kiosk                 A  Fatal signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x7a80978193 in tid 23725 (pool-14-thread-), pid 23489 (com.kiosk)
```

rodgomesc, 2023-11-21:
i'll try to find some time today to debug this more closely

apadev666, 2023-11-21:
@rodgomesc Im not able to install( path was not found so all imports are broken

apadev666, 2023-11-21:
@rodgomesc I have tested many ways
I have removed Camera component from bottom sheet, and event after 1000 tries its opening and closing ok
I have tried to add Camera component and event with this with 1000 times but when camera is closed that it can see only black view everything is ok but when camera is opened BUT no any QR code or barcodes are visible just face or just white table its crashing after some attempts, hope this will help you

rodgomesc, 2023-11-21:
that's so weird.  devs in this other pr that seems to be related to this same issue said that solved for them
https://github.com/mrousavy/react-native-vision-camera/issues/2116

i'll wait for more people testing this, can you reproduce this on the example app @apadev666 ?

apadev666, 2023-11-22:
@rodgomesc maybe it can fix, but the whole problem is that I can't install it, so after install all imports are broken and cant import, so I don't know why

animaonline, 2023-11-22:
The logcat is being spammed with this whenever the Camera component is rendered.
`TF Lite in Google Play services has stable ABI version 1.11.0 which is less than the required version 1.13.0.`

animaonline, 2023-11-22:
> > @mrousavy please take a look at this
> 
> @apadev666 why?

@mrousavy 
Wow, that's the best response from a repo owner I've ever seen, hands down, man.

mrousavy, 2023-11-22:
> @mrousavy Wow, that's the best response from a repo owner I've ever seen, hands down, man.

I'm sorry, @animaonline did you pay me for my availability and support? 

I can't reply to every issue and take a look at every bug you guys find in your apps that make money for you. If you want me to fix a bug for your app, then pay me.

I personally maintain this library in my free time and work on stuff that I personally need for my apps, see for example this cool PR: https://github.com/mrousavy/react-native-vision-camera/pull/2173 which I just recently worked on and published for free for everyone to use. If I don't need something in my apps, I would purely work for you, which I don't do for free.

I don't want this to come across as rude, I'm not trying to be rude here - I'm just saying that I can't find the time for every single issue - if you want something prioritized consider sponsoring me on GitHub, then ping me with the issue and I'll try to find the time as soon as I can. That's how this works.

animaonline, 2023-11-22:
@mrousavy wow, just wow man. If you have this kind of attitude, then why do open-source anyway? You're trying not to come off as rude, but you are. I'd rather stay the hell away from a library like this and advice others to follow my example.

rodgomesc, 2023-11-22:
i know that he will continue use this lib in secret 

[image]

apadev666, 2023-11-22:
@animaonline thank you for your response
@rodgomesc @mrousavy  can you tell me do I need to update ML kit or so?

Hector-Chong, 2023-11-24:
It occurred to me as well, but I tried to debug this issue, and it seems that it is a race condition.

The issue is because of the function below. When I change `runBlocking` to `launch(Dispatchers.Main)` to make it run on the main thread, it works.

com.mrousavy.camera.core.CameraSession#destroyPreviewOutputSync
```kotlin
    private fun destroyPreviewOutputSync() {
        Log.i(TAG, "Destroying Preview Output...")
        runBlocking {
            Log.d("DEBUG_CAMERA", "difference: 1")
            configure { config ->
                config.preview = CameraConfiguration.Output.Disabled.create()
            }
            Log.d("DEBUG_CAMERA", "difference: 4")
        }
    }
```

```kotlin
    private fun destroyPreviewOutputSync() {
        Log.i(TAG, "Destroying Preview Output...")
        launch(Dispatchers.Main) {
            Log.d("DEBUG_CAMERA", "difference: 1")
            configure { config ->
                config.preview = CameraConfiguration.Output.Disabled.create()
            }
            Log.d("DEBUG_CAMERA", "difference: 4")
        }
    }
```

alexrififi, 2023-11-24:
@Hector-Chong fail on build:

`Unresolved reference: Dispatchers`

Hector-Chong, 2023-11-24:
> Dispatchers

You should import `Dispatchers` by yourself.

`import kotlinx.coroutines.Dispatchers`

alexrififi, 2023-11-24:
@Hector-Chong
[Your fix](https://github.com/mrousavy/react-native-vision-camera/issues/2189#issuecomment-1825272959) works! Big thanks! ❤️

WitchMad, 2023-11-24:
Still happening here :/ using same version `3.6.8`
`logcat`:
```
11-24 14:01:52.758  2219  2947 I CreateCaptureSession: Camera 0: Capture Session #1014 configured!
11-24 14:01:52.758  1125  3679 I qomx_image_core: OMX_Init:101] Complete 3
11-24 14:01:52.763  2219  2219 I CameraSession: Successfully configured Session with 2 outputs for Camera #0!
11-24 14:01:52.763  2219  2219 I CameraSession: Successfully updated CameraSession Configuration! isActive: false
11-24 14:01:52.763  2219  2219 D DEBUG_CAMERA: difference: 4
11-24 14:01:52.770  2219  2947 I CreateCaptureSession: Camera 0: Capture Session #1013 closed!
11-24 14:01:52.773  1125  3679 I qomx_image_core: OMX_GetHandle:243] get instance pts is 0xe50c9700
11-24 14:01:52.773  1125  3679 D qomx_image_core: OMX_GetHandle:261] handle = 0xe50c9704 Instanceindex = 0,comp_idx 2 g_ptr 0xe7fc9b30
11-24 14:01:52.773  1125  3679 I mm-still: omx_component_set_callbacks: Bad Parameter 0xe50c9704 0x0 0x0 0x0 0x0
11-24 14:01:52.773  1125  3679 I qomx_image_core: OMX_GetHandle:267] Success
11-24 14:01:52.912  1131  1276 I chatty  : uid=1000(system) HwBinder:1131_1 identical 8 lines
11-24 14:01:52.936 31488 31488 D DeviceStatisticsService: chargerType=2 batteryLevel=37 totalBatteryCapacity=4173800
11-24 14:01:52.936 31488 31488 D DeviceInfoHidlClient: isPowerInfoNeverSent()=true
11-24 14:01:52.937 28305 28411 D CmdCenter: - SquareBase: sPrevBatStatus: 2, status: 2
11-24 14:01:52.937 28305 28411 D CmdCenter: - SquareBase: sPrevBatteryPercent: 37 newPercent: 37
11-24 14:01:52.937 28305 28411 D CmdCenter: - SquareBase: sPrevPlugType: 2 plugType: 2
11-24 14:01:53.363  1131  1276 I chatty  : uid=1000(system) HwBinder:1131_1 identical 26 lines
11-24 14:01:53.394 15890 15922 W TimedProcessReaper: Memory state is: 125
11-24 14:01:53.404 12833 10255 W Nearby  : [MBleClient] Ignore outdated onLost runnable from cycle : 1948, current cycle : 3006 [CONTEXT service_id=49 ]
11-24 14:01:53.446  3002  6659 D IpClient/wlan0: addressUpdated: 2804:14d:5494:82bf:246c:7aff:fedd:2896/64 on wlan0 flags 0x00000100 scope 0
11-24 14:01:53.446  3002  6659 D IpClient/wlan0: addressUpdated: 2804:14d:5494:82bf:49b7:3a65:69ce:4c65/64 on wlan0 flags 0x00000001 scope 0
11-24 14:01:53.487 12833 10255 W Nearby  : [MBleClient] Ignore outdated onLost runnable from cycle : 1949, current cycle : 3006 [CONTEXT service_id=49 ]
11-24 14:01:54.563  1131  1276 I chatty  : uid=1000(system) HwBinder:1131_1 identical 64 lines
11-24 14:01:54.590  1137  1163 I android.hardware.thermal@2.0-service.qti: uevent triggered for sensor: msm_therm
11-24 14:01:54.944  2219  3089 F libc    : Fatal signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x7bdaeee7b0 in tid 3089 (pool-22-thread-), pid 2219
```

mrousavy, 2023-11-24:
Hey all - does https://github.com/mrousavy/react-native-vision-camera/pull/2204 fix your issue?

mrousavy, 2023-11-24:
Please change this locally and test it.

alexrififi, 2023-11-24:
> Hey all - does #2204 fix your issue?

@mrousavy Hello! I tried this patch, but it didn't solve the problem. But [this helped](https://github.com/mrousavy/react-native-vision-camera/issues/2189#issuecomment-1825272959)

mrousavy, 2023-11-24:
Ah - gotcha - @Hector-Chong could you please create a PR instead of just posting random patches/diffs here? Appreciate it, thanks!

mrousavy, 2023-11-24:
Honestly I am not 100% sure if this is a true fix though, because once the `destroyPreviewOutputSync` method returns, you can no longer use the `SurfaceView`'s surface to stream into it - so we should not return from that method until we actually stop the `captureSession`.... weird

WitchMad, 2023-11-24:
No one of those solved to me :/
logcat about last test:
```
11-24 14:24:48.825 11847 11847 I CameraSession: Successfully configured Session with 2 outputs for Camera #0!
11-24 14:24:48.825 11847 11847 W CameraSession: Preview Output is null, aborting...
11-24 14:24:48.825 11847 11847 I CameraSession: Successfully updated CameraSession Configuration! isActive: true
11-24 14:24:48.827 11847 11847 I CameraView: Updating CameraSession...
11-24 14:24:48.827 11847 12672 I CameraSession: Updating CameraSession Configuration...
11-24 14:24:48.828  1125 13313 I qomx_image_core: OMX_Init:101] Complete 3
11-24 14:24:48.833 11847 12672 I CameraSession: Successfully updated CameraSession Configuration! isActive: false
11-24 14:24:48.842 11847 12672 I CreateCaptureSession: Camera 0: Capture Session #1011 closed!
11-24 14:24:48.842  1125 13313 I qomx_image_core: OMX_GetHandle:243] get instance pts is 0xe50cc950
11-24 14:24:48.843  1125 13313 D qomx_image_core: OMX_GetHandle:261] handle = 0xe50cc954 Instanceindex = 0,comp_idx 2 g_ptr 0xe7fc9c40
11-24 14:24:48.843  1125 13313 I mm-still: omx_component_set_callbacks: Bad Parameter 0xe50cc954 0x0 0x0 0x0 0x0
11-24 14:24:48.843  1125 13313 I qomx_image_core: OMX_GetHandle:267] Success
11-24 14:24:50.393  1131  1277 I chatty  : uid=1000(system) HwBinder:1131_2 identical 93 lines
11-24 14:24:50.416  3002 13965 D IpClient/wlan0: addressUpdated: 2804:14d:5494:82bf:246c:7aff:fedd:2896/64 on wlan0 flags 0x00000100 scope 0
11-24 14:24:50.418  3002 24395 D IpClient/wlan0: addressUpdated: 2804:14d:5494:82bf:49b7:3a65:69ce:4c65/64 on wlan0 flags 0x00000001 scope 0
11-24 14:24:50.811 11847 12792 F libc    : Fatal signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x7bf0b5dde4 in tid 12792 (pool-27-thread-), pid 11847 (myapp)
11-24 14:24:50.944  1131  1277 I chatty  : uid=1000(system) HwBinder:1131_2 identical 28 lines
11-24 14:24:50.966 13321 13321 E DEBUG   : failed to readlink /proc/12792/fd/199: No such file or directory
11-24 14:24:50.966 13321 13321 E DEBUG   : failed to readlink /proc/12792/fd/201: No such file or directory
11-24 14:24:50.966 13321 13321 E DEBUG   : failed to readlink /proc/12792/fd/203: No such file or directory
11-24 14:24:51.012  1131  1277 I chatty  : uid=1000(system) HwBinder:1131_2 identical 1 line
11-24 14:24:51.178 13321 13321 I crash_dump64: obtaining output fd from tombstoned, type: kDebuggerdTombstone
11-24 14:24:51.186  1004  1004 I tombstoned: received crash request for pid 12792
11-24 14:24:51.190 13321 13321 I crash_dump64: performing dump of process 11847 (target tid = 12792)
11-24 14:24:51.211 13321 13321 F DEBUG   : *** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
11-24 14:24:51.211 13321 13321 F DEBUG   : Build fingerprint: 'motorola/sofiar_retail/sofiar:11/RPES31.Q4U-47-35-12/24aaf7:user/release-keys'
11-24 14:24:51.211 13321 13321 F DEBUG   : Revision: 'pvt'
11-24 14:24:51.211 13321 13321 F DEBUG   : ABI: 'arm64'
11-24 14:24:51.215 13321 13321 F DEBUG   : Timestamp: 2023-11-24 14:24:51-0300
11-24 14:24:51.215 13321 13321 F DEBUG   : pid: 11847, tid: 12792, name: pool-27-thread-  >>> com.myapp <<<
11-24 14:24:51.215 13321 13321 F DEBUG   : uid: 10804
11-24 14:24:51.215 13321 13321 F DEBUG   : signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x7bf0b5dde4
11-24 14:24:51.215 13321 13321 F DEBUG   :     x0  0000007ae4512138  x1  000000000000043c  x2  0000000000000008  x3  0000000000000010
11-24 14:24:51.215 13321 13321 F DEBUG   :     x4  0000007c67724e18  x5  0000007ef82a6c38  x6  0000007dca219cf0  x7  0000007dca219cf0
11-24 14:24:51.215 13321 13321 F DEBUG   :     x8  0000000000000000  x9  000000000000021e  x10 00000000000001b9  x11 b400007bfb35e040
11-24 14:24:51.215 13321 13321 F DEBUG   :     x12 000000000000021d  x13 00000000000c5de4  x14 0000007bf0b5dde4  x15 0000007bf0b5d8e4
11-24 14:24:51.215 13321 13321 F DEBUG   :     x16 0000007bf0a98000  x17 0000007ef82262c0  x18 00000000000e05fc  x19 0000007ae4512258
11-24 14:24:51.216 13321 13321 F DEBUG   :     x20 0000007ae45121f8  x21 0000007ae4512390  x22 000000000000021e  x23 00000000000001b9
11-24 14:24:51.216 13321 13321 F DEBUG   :     x24 b400007ca785f9bc  x25 0000007ae4517000  x26 0000000000001831  x27 0000007ae4512a48
11-24 14:24:51.216 13321 13321 F DEBUG   :     x28 0000007ae4512a88  x29 00000000000006ef
11-24 14:24:51.216 13321 13321 F DEBUG   :     lr  0000007af99289ec  sp  0000007ae4512130  pc  0000007af9928ab4  pst 0000000080000000
11-24 14:24:51.618 13321 13321 F DEBUG   : backtrace:
11-24 14:24:51.618 13321 13321 F DEBUG   :     NOTE: Function names and BuildId information is missing for some frames due
11-24 14:24:51.618 13321 13321 F DEBUG   :     NOTE: to unreadable libraries. For unwinds of apps, only shared libraries
11-24 14:24:51.618 13321 13321 F DEBUG   :     NOTE: found under the lib/ directory are readable.
11-24 14:24:51.618 13321 13321 F DEBUG   :       #00 pc 0000000000128ab4  /data/user_de/0/com.google.android.gms/app_chimera/m/000006cd/dl-VisionBarcode.optional_234414100400.apk
11-24 14:24:51.618 13321 13321 F DEBUG   :       #01 pc 00000000000812d4  /data/user_de/0/com.google.android.gms/app_chimera/m/000006cd/dl-VisionBarcode.optional_234414100400.apk
11-24 14:24:51.618 13321 13321 F DEBUG   :       #02 pc 000000000007af18  /data/user_de/0/com.google.android.gms/app_chimera/m/000006cd/dl-VisionBarcode.optional_234414100400.apk
[... cut ...]
11-24 14:24:51.261  1131  1277 I chatty  : uid=1000(system) HwBinder:1131_2 identical 4 lines
11-24 14:24:51.870  1137  1163 I android.hardware.thermal@2.0-service.qti: uevent triggered for sensor: msm_therm
11-24 14:24:52.420  1004  1004 E tombstoned: Tombstone written to: /data/tombstones/tombstone_27
11-24 14:24:52.424  2403 13326 W ActivityTaskManager:   Force finishing activity com.myapp/.MainActivity
11-24 14:24:52.429  2403 13326 V WindowManager: Changing focus from Window{461b19a u0 com.myapp/com.myapp.MainActivity} to null displayId=0
```

Serck96, 2023-11-24:
[image]

I tried the suggestions, but none of them solve the problem. I added Sentry to the project to get more information and this is the error. @mrousavy

Hector-Chong, 2023-11-25:
> Ah - gotcha - @Hector-Chong could you please create a PR instead of just posting random patches/diffs here? Appreciate it, thanks!

I've added #2213 with more information there.

BrechtBD, 2023-11-27:
> Version: 3.6.8
> 
> It occurred to me as well. I tried to debug this, and it seems to be a race condition.
> 
> The issue is because of the function below. When I change `runBlocking` to `launch(Dispatchers.Main)` to make it run on the main thread, it works.
> 
> com.mrousavy.camera.core.CameraSession#destroyPreviewOutputSync
> 
> ```kotlin
>     private fun destroyPreviewOutputSync() {
>         Log.i(TAG, "Destroying Preview Output...")
>         runBlocking {
>             Log.d("DEBUG_CAMERA", "difference: 1")
>             configure { config ->
>                 config.preview = CameraConfiguration.Output.Disabled.create()
>             }
>             Log.d("DEBUG_CAMERA", "difference: 4")
>         }
>     }
> ```
> 
> ```kotlin
> import kotlinx.coroutines.Dispatchers
> 
>     private fun destroyPreviewOutputSync() {
>         Log.i(TAG, "Destroying Preview Output...")
>         launch(Dispatchers.Main) {
>             Log.d("DEBUG_CAMERA", "difference: 1")
>             configure { config ->
>                 config.preview = CameraConfiguration.Output.Disabled.create()
>             }
>             Log.d("DEBUG_CAMERA", "difference: 4")
>         }
>     }
> ```

This fix seems to work for me

lBroth, 2023-11-27:
> This fix seems to work for me

It works for me too

only1chi, 2023-12-05:
@Hector-Chong 

Have you observed an ANR with your fix? 
I am now observing an ANR (Application Not Responding) when Camera Session is closed (after barcode scanning), and this has to do with `runBlocking` function.

When `CameraView` is destroyed, it calls `cameraSession.close()` which has the following code block:

```
  override fun close() {
    runBlocking {
      mutex.withLock {
        destroy()
        photoOutputSynchronizer.clear()
      }
    }
  }
```

We use sentry to log errors. Here is some of our sentry log:

[image]

only1chi, 2023-12-05:
It is known out there that using `runBlocking` seems to present some issues for Android 12 and 13 devices. In my case, it appears as if the app is suspended because `runBlocking` is keeping the main thread from executing. Unfortunately, whatever it is waiting for never returns, so it hangs.

Here is some [information on Stack Overflow.](https://stackoverflow.com/questions/75738514/whats-the-repercussions-of-using-runblocking-in-an-android-project)

Hector-Chong, 2023-12-06:
> @Hector-Chong
>
> Have you observed an ANR with your fix? I am now observing an ANR (Application Not Responding) when Camera Session is closed (after barcode scanning), and this has to do with [...]

Sadly, I've not experienced either an ANR or another crash within my project.

only1chi, 2023-12-06:
> I need logs for that.

Here are the full sentry logs.

```
{"event_id":"a16a48a75e354732ad8cfca65d6714d2","project":6275770,"release":"com.ststorquefit@4.4.12+295","dist":"295","platform":"java","message":"","datetime":"2023-12-04T15:49:17+00:00","tags":[["device","XP9900"],["device.class","2"],["device.family","XP9900"],["environment","production"],["event.environment","java"],["event.origin","android"],["isSideLoaded","true"],["level","fatal"],["mechanism","AppExitInfo"],["os","Android 12"],["os.name","Android"],["routing.route.name","qrLogin"],["dist","295"],["release","com.ststorquefit@4.4.12+295"],["user","id:2ebcde27-ebc5-45f5-a2df-1ce8984459c5"],["site","glazed-donut"]],"_metrics":{"bytes.ingested.event":104029,"bytes.stored.event":381162},"breadcrumbs":{"values":[{"timestamp":1701704962.473,"type":"session","category":"app.lifecycle","level":"info","data":{"state":"start"}},{"timestamp":1701704962.477,"type":"navigation","category":"app.lifecycle","level":"info","data":{"state":"foreground"}}]},"contexts":{"app":{"app_identifier":"com.ststorquefit","app_name":"Cumulus","app_version":"4.4.12","app_build":"295","in_foreground":true,"type":"app"},"device":{"name":"XP10","family":"XP9900","model":"XP9900","model_id":"10.0.0-17-12.0.0-10.132.10","manufacturer":"Sonimtech","brand":"Sonim","screen_width_pixels":1080,"screen_height_pixels":2016,"screen_density":3.0,"screen_dpi":480,"simulator":false,"memory_size":5747535872,"processor_count":8,"processor_frequency":2208,"archs":["arm64-v8a","armeabi-v7a","armeabi"],"id":"18a585af-5d64-4845-8d63-db1a88992d31","type":"device"},"os":{"name":"Android","version":"12","build":"10.0.0-17-12.0.0-10.132.10","kernel_version":"5.4.210","type":"os"}},"culprit":"jdk.internal.misc.Unsafe in park","environment":"production","exception":{"values":[{"type":"ApplicationNotResponding","value":"ANR","module":"io.sentry.android.core","stacktrace":{"frames":[{"function":"main","module":"com.android.internal.os.ZygoteInit","filename":"ZygoteInit.java","abs_path":"ZygoteInit.java","lineno":1009,"in_app":false,"data":{"orig_in_app":-1,"category":"system"}},{"function":"run","module":"com.android.internal.os.RuntimeInit$MethodAndArgsCaller","filename":"RuntimeInit.java","abs_path":"RuntimeInit.java","lineno":577,"in_app":false,"data":{"orig_in_app":-1,"category":"internals"}},{"function":"invoke","module":"java.lang.reflect.Method","in_app":false,"data":{"orig_in_app":-1,"category":"indirection"}},{"function":"main","module":"android.app.ActivityThread","filename":"ActivityThread.java","abs_path":"ActivityThread.java","lineno":7916,"in_app":false,"data":{"orig_in_app":-1,"category":"threadbase"}},{"function":"loop","module":"android.os.Looper","filename":"Looper.java","abs_path":"Looper.java","lineno":288,"in_app":false,"data":{"orig_in_app":-1,"category":"threadbase"}},{"function":"loopOnce","module":"android.os.Looper","filename":"Looper.java","abs_path":"Looper.java","lineno":201,"in_app":false,"data":{"orig_in_app":-1,"category":"indirection"}},{"function":"dispatchMessage","module":"android.os.Handler","filename":"Handler.java","abs_path":"Handler.java","lineno":99,"in_app":false,"data":{"orig_in_app":-1,"category":"threadbase"}},{"function":"handleCallback","module":"android.os.Handler","filename":"Handler.java","abs_path":"Handler.java","lineno":938,"in_app":false,"data":{"orig_in_app":-1,"category":"threadbase"}},{"function":"run","module":"android.view.Choreographer$FrameDisplayEventReceiver","filename":"Choreographer.java","abs_path":"Choreographer.java","lineno":1105,"in_app":false,"data":{"orig_in_app":-1,"category":"indirection"}},{"function":"doFrame","module":"android.view.Choreographer","filename":"Choreographer.java","abs_path":"Choreographer.java","lineno":854,"in_app":false,"data":{"orig_in_app":-1,"category":"ui"}},{"function":"doCallbacks","module":"android.view.Choreographer","filename":"Choreographer.java","abs_path":"Choreographer.java","lineno":926,"in_app":false,"data":{"orig_in_app":-1,"category":"internals"}},{"function":"run","module":"android.view.Choreographer$CallbackRecord","filename":"Choreographer.java","abs_path":"Choreographer.java","lineno":1118,"in_app":false,"data":{"orig_in_app":-1,"category":"indirection"}},{"function":"doFrame","module":"com.facebook.react.modules.core.ChoreographerCompat$FrameCallback$1","filename":"ChoreographerCompat.java","abs_path":"ChoreographerCompat.java","lineno":85,"in_app":false,"data":{"orig_in_app":-1}},{"function":"doFrame","module":"com.facebook.react.modules.core.ReactChoreographer$ReactChoreographerDispatcher","filename":"ReactChoreographer.java","abs_path":"ReactChoreographer.java","lineno":175,"in_app":false,"data":{"orig_in_app":-1},"lock":{"type":1,"address":"0x0a68de7f","package_name":"java.lang","class_name":"Object"}},{"function":"doFrame","module":"com.facebook.react.uimanager.GuardedFrameCallback","filename":"GuardedFrameCallback.java","abs_path":"GuardedFrameCallback.java","lineno":29,"in_app":false,"data":{"orig_in_app":-1}},{"function":"doFrameGuarded","module":"com.facebook.react.uimanager.UIViewOperationQueue$DispatchUIFrameCallback","filename":"UIViewOperationQueue.java","abs_path":"UIViewOperationQueue.java","lineno":1086,"in_app":false,"data":{"orig_in_app":-1}},{"function":"access$2600","module":"com.facebook.react.uimanager.UIViewOperationQueue","filename":"UIViewOperationQueue.java","abs_path":"UIViewOperationQueue.java","lineno":47,"in_app":false,"data":{"orig_in_app":-1}},{ [... cut ...] ava","abs_path":"LockSupport.java","lineno":252,"in_app":false,"data":{"orig_in_app":-1,"category":"internals"}},{"function":"park","module":"jdk.internal.misc.Unsafe","in_app":false,"data":{"orig_in_app":-1}}],"snapshot":true},"crashed":false,"current":false,"main":false,"state":"TimedWaiting"}]},"timestamp":1701704957.888,"title":"ApplicationNotResponding: ANR","type":"error","user":{"id":"2ebcde27-ebc5-45f5-a2df-1ce8984459c5","username":"marc.glazed.fitter"},"version":"7","location":null}
```

mrousavy, 2023-12-07:
That's a stacktrace, not a log. Do you have native logs in Sentry?

only1chi, 2023-12-07:
> That's a stacktrace, not a log. Do you have native logs in Sentry?

I do understand. I wanted to show the stacktrace, because it was the most obvious suggesting a sequence of events. Since it is an ANR, there isn't a lot of logs. See below, the much I could extract. Pay attention to the suspense or delay freeze. 

-----

[09:32:46] W | ReactNativeJS ▶︎ 'An error occurred! For more details, see the full error text at https://go.apollo.dev/c/err#%7B%22version%22%3A%223.8.8%22%2C%22message%22%3A53%2C%22args%22%3A%5B%5D%7D', '\n    at SyncProvider ([http://localhost:8081/index.bundle//&platform=android&dev=true&minify=false&app=com.ststorquefit&modulesOnly=false&runModule=true:402707:24)\n](http://localhost:8081/index.bundle//&platform=android&dev=true&minify=false&app=com.ststorquefit&modulesOnly=false&runModule=true:402707:24)/n)    at EntitySyncContextProvider ([http://localhost:8081/index.bundle//&platform=android&dev=true&minify=false&app=com.ststorquefit&modulesOnly=false&runModule=true:310537:24)\n](http://localhost:8081/index.bundle//&platform=android&dev=true&minify=false&app=com.ststorquefit&modulesOnly=false&runModule=true:310537:24)/n)    at DeveloperContextProvider ([http://localhost:8081/index.bundle//&platform=android&dev=true&minify=false&app=com.ststorquefit&modulesOnly=false&runModule=true:421583:24)\n](http://localhost:8081/index.bundle//&platform=android&dev=true&minify=false&app=com.ststorquefit&modulesOnly=false&runModule=true:421583:24)/n)    at EnsureSingleNavigator ([http://localhost:8081/index.bundle//&platform=android&dev=true&minify=false&app=com.ststorquefit&modulesOnly=false&runModule=true:396178:24)\n](http://localhost:8081/index.bundle//&platform=android&dev=true&minify=false&app=com.ststorquefit&modulesOnly=false&runModule=true:396178:24)/n)    at BaseNavigationContainer ([http://localhost:8081/index.bundle//&platform=android&dev=true&minify=false&app=com.ststorquefit&modulesOnly=false&runModule=true:395747:28)\n](http://localhost:8081/index.bundle//&platform=android&dev=true&minify=false&app=com.ststorquefit&modulesOnly=false&runModule=true:395747:28)/n)    at ThemeProvider ([http://localhost:8081/index.bundle//&platform=android&dev=true&minify=false&app=com.ststorquefit&modulesOnly=false&runModule=true:400777:21)\n](http://localhost:8081/index.bundle//&platform=android&dev=true&minify=false&app=com.ststorquefit&modulesOnly=false&runModule=true:400777:21)/n)    at NavigationContainerInner ([http://localhost:8081/index.bundle//&platform=android&dev=true&minify=false&app=com.ststorquefit&modulesOnly=false&runModule=true:400656:26)\n](http://localhost:8081/index.bundle//&platform=android&dev=true&minify=false&app=com.ststorquefit&modulesOnly=false&runModule=true:400656:26)/n)    at RootNavigator ([http://localhost:8081/index.bundle//&platform=android&dev=true&minify=false&app=com.ststorquefit&modulesOnly=false&runModule=true:360277:24)\n](http://localhost:8081/index.bundle//&platform=android&dev=true&minify=false&app=com.ststorquefit&modulesOnly=false&runModule=true:360277:24)/n)    at PortalProviderComponent ([http://localhost:8081/index.bundle//&platform=android&dev=true&minify=false&app=com.ststorquefit&modulesOnly=false&runModule=true:420287:33)\n](http://localhost:8081/index.bundle//&platform=android&dev=true&minify=false&app=com.ststorquefit&modulesOnly=false&runModule=true:420287:33)/n)    at BottomSheetModalProviderWrapper ([http://localhost:8081/index.bundle//&platform=android&dev=true&minify=false&app=com.ststorquefit&modulesOnly=false&runModule=true:420460:24)\n](http://localhost:8081/index.bundle//&platform=android&dev=true&minify=false&app=com.ststorquefit&modulesOnly=false&runModule=true:420460:24)/n)    at IntlProvider ([http://localhost:8081/index.bundle//&platform=android&dev=true&minify=false&app=com.ststorquefit&modulesOnly=false&runModule=true:334008:50)\n](http://localhost:8081/index.bundle//&platform=android&dev=true&minify=false&app=com.ststorquefit&modulesOnly=false&runModule=true:334008:50)/n)    at RCTView\n    at View ([http://localhost:8081/index.bundle//&platform=android&dev=true&minify=false&app=com.ststorquefit&modulesOnly=false&runModule=true:44664:43)\n](http://localhost:8081/index.bundle//&platform=android&dev=true&minify=false&app=com.ststorquefit&modulesOnly=false&runModule=true:44664:43)/n)    at UserInactivity ([http://localhost:8081/index.bundle//&platform=android&dev=true&minify=false&app=com.ststorquefit&modulesOnly=false&runModule=true:340622:24)\n](http://localhost:8081/index.bundle//&platform=android&dev=true&minify=false&app=com.ststorquefit&modulesOnly=false&runModule=true:340622:24)/n)    at AppNavigationContainer ([http://localhost:8081/index.bundle//&platform=android&dev=true&minify=false&app=com.ststorquefit&modulesOnly=false&runModule=true:338924:24)\n](http://localhost:8081/index.bundle//&platform=android&dev=true&minify=false&app=com.ststorquefit&modulesOnly=false&runModule=true:338924:24)/n)    at SelectedEntityTypeProvider ([http://localhost:8081/index.bundle//&platform=android&dev=true&minify=false&app=com.ststorquefit&modulesOnly=false&runModule=true:482920:24)\n](http://localhost:8081/index.bundle//&platform=android&dev=true&minify=false&app=com.ststorquefit&modulesOnly=false&runModule=true:482920:24)/n)    at SelectedWorkScopePreferenceProvider ([http://localhost:8081/index.bundle//&platform=android&dev=true&minify=false&app=com.ststorquefit&modulesOnly=false&runModule=true:643950:24)\n](http://localhost:8081/i [... cut ...] &app=com.ststorquefit&modulesOnly=false&runModule=true:665062:22)/n)    at Suspense\n    at Freeze ([http://localhost:8081/index.bundle//&platform=android&dev=true&minify=false&app=com.ststorquefit&modulesOnly=false&runModule=true:665086:23)\n](http://localhost:8081/index.bundle//&platform=android&dev=true&minify=false&app=com.ststorquefit&modulesOnly=false&runModule=true:665086:23)/n)    at DelayedFreeze (http://localhost:8081/index.bundle//&platform=android&dev=true&minify=false&app=com.stst
`

MMYurt, 2023-12-26:
Any updates? Have the latest releases fixed this issue?

closed 2024-01-03 by sunnystarh

parlet, 2024-01-11:
> Any updates? Have the latest releases fixed this issue?

Nope, still reproducible in v3.7.0.

parlet, 2024-01-12:
> @parlet Do you add 'VisionCamera_enableCodeScanner=true' to gradle.properties?

@Serck96, yes, I did.

> I just tried the latest version, 3.7.1, and it worked fine. However, I first uninstalled the preview version of the app, updated the node modules, cleaned the metro cache, and then tried it.

I suppose it doesn't hurt, but that's probably not what it is.

It's possible that you haven't encountered this problem due to its somewhat limited reproducibility. This issue may also depend on the specific devices used and the exact steps involved. Others have reported reproducing this issue in v3.7.0 as well.

By the way, I now see this issue is closed.
A link for further tracking for posterity: #2116.