margelo/react-native-vision-camera #2190: ✨ Allow returning WritableNativeArray from frame processor in v3

xulihang, 2023-11-21:
### What feature or enhancement are you suggesting?

In v2, I can return `WritableNativeArray` on Android from a frame processor like the following:

```java
WritableNativeArray array = new WritableNativeArray();
return array;
```

But this seems not possible and I have to use the following:

```java
List<Object> array = new ArrayList<>();
return array;
```

Since I also need to run a similar native module method , I need to use `WritableNativeArray`.

Allowing returning WritableNativeArray from frame processor in v3 would make the development easier.

### What Platforms whould this feature/enhancement affect?

Android

### Alternatives/Workarounds

I am currently returning WritableNativeArray for a native module method and an ArrayList for a frame processor.

### Additional information

- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar feature requests in this repository](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22✨+enhancement%22) and found none.

rodgomesc, 2023-11-21:
o.O

this is well documented on [the official documentation >> frame-processors-plugins >> overview >> types ](https://react-native-vision-camera.com/docs/guides/frame-processors-plugins-overview#types)

and also in the example app source code shows exactly how to hook up a object to javascript

https://github.com/mrousavy/react-native-vision-camera/blob/main/package/example/android/app/src/main/java/com/mrousavy/camera/example/ExampleFrameProcessorPlugin.java

rodgomesc, 2023-11-21:
i understand @xulihang i don't have authority here to say what being implemented in this lib,  but your request doesn't make sense for me, this was possible in v2 and in v3 we have a faster approach to hook up object to javascript, how this change will benefit hundreds of users that use that lib? i think you should change your approach to solve the problem, or could you provide a use case explaining why exactly `WritableNativeArray` solves your problem and `Map<String, Object>` don't?

xulihang, 2023-11-21:
Allowing the react native bridge's types can make it easy to use previous work like the code here: https://github.com/tony-xlh/vision-camera-dynamsoft-barcode-reader/blob/main/android/src/main/java/com/visioncameradynamsoftbarcodereader/Utils.java#L34

I am just giving a tentative proposal which may benefit some users.

rodgomesc, 2023-11-21:
i see, i'm really trying to help, but how your example can be benefit over this?, it's a easy change, i'm not seeing your point

https://github.com/mrousavy/react-native-vision-camera/blob/deb3c41ee0a55e64cbb83ead912ccdd53d10988f/package/example/android/app/src/main/java/com/mrousavy/camera/example/ExampleFrameProcessorPlugin.java#L29C8-L40C20

rodgomesc, 2023-11-21:
hey @xulihang apologizes for the confusion, i misread `WritableNativeArray` as `WritableNativeMap` since your first message, and you sent an example using `WritableNativeMap` also, really sorry your point makes total sense, we don't seem to have a way of returning arrays from the native  java callback at the moment

xulihang, 2023-11-22:
It turns out that the react native bridge's maps and arrays can be easily converted to the type v3 supports:

```java
map.toHashMap();
```

So there is no need to add this feature.

closed 2023-11-22 by xulihang