margelo/react-native-vision-camera #2193: 🐛

GaylordP, 2023-11-21:
### What's happening?

Hello :)
Thanks you very much for your library and sorry for my bad english.

I know that logs are normally required, but the problem only occurs in production and on certain phones (which I don't own). So I don't have the logs. However, I've run various tests to narrow down the problem.

For over a month, on some phones, adding the photo={true} parameter displays a black screen when using the camera.

[This bug seems very similar to the one I've been following closely (same bug, almost the same phone, but the updates didn't fix it).](https://github.com/mrousavy/react-native-vision-camera/issues/1858)

So far, two users have reported the problem (my application is very small):

- Redmi Note PRO 9
- Redmi Note PRO 11

This code, after requesting permissions, **displays the camera**  @@ : 

    const CameraUpload = () => {
      const { hasPermission, requestPermission } = useCameraPermission();
      const device = useCameraDevice("back");
      const camera = useRef<Camera>(null);

      if (device == null) {
        return <Loader size="large" />;
      } else {
        return (
          <Camera
            ref={camera}
            isActive={true}
            device={device}
            orientation="portrait"
            style={{
              ...StyleSheet.absoluteFillObject,
            }}
          />
        );
      }
    };

### Reproduceable Code

```tsx
This code, after requesting permissions, **does not display the camera** : 

    const CameraUpload = () => {
      const { hasPermission, requestPermission } = useCameraPermission();
      const device = useCameraDevice("back");
      const camera = useRef<Camera>(null);

      if (device == null) {
        return <Loader size="large" />;
      } else {
        return (
          <Camera
            ref={camera}
            isActive={true}
            device={device}
            photo={true}
            orientation="portrait"
            style={{
              ...StyleSheet.absoluteFillObject,
            }}
          />
        );
      }
    };

This code, after requesting permissions, **does not display the camera** : 

    const CameraUpload = () => {
      const { hasPermission, requestPermission } = useCameraPermission();
      const device = useCameraDevice("back");
      const camera = useRef<Camera>(null);

      if (device == null) {
        return <Loader size="large" />;
      } else {
        return (
          <Camera
            ref={camera}
            isActive={true}
            device={device}
            photo={true}
            //orientation="portrait"
            style={{
              ...StyleSheet.absoluteFillObject,
            }}
          />
        );
      }
    };
```

### Relevant log output

```shell
Problem in production, impossible to obtain logs :/
```

### Camera Device

```json
Problem in production, impossible to obtain logs :/
```

### Device

Redmi Note PRO 9, Redmi Note PRO 11

### VisionCamera Version

3.6.8

### Can you reproduce this issue in the VisionCamera Example app?

No, I cannot reproduce the issue in the Example app

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2023-11-21:
Hey - two things:

1. when running in production you can have Sentry or Bugsnag or other tools help you with getting logs. Then just symbolicate them using a tutorial on Sentry to have method names and callstacks.
2. Add an issue title.

tatiana-lonestone, 2023-11-24:
Hello, 
Same problem here on Xiaomi Redmi (MIUU GLOBAL 14.05, Xiaomi Redmi 5g, Xiaomi Redmi 11), but also on REALME C53 and ALCALTEL 5030F.
I'm trying to get some logs

Selcukk70, 2023-11-27:
I have the same problem running on version 3.6.10
The logs:

11-27 13:36:58.098   889 21360 E camera_sof: ================ BEGIN SOF DEBUG DUMP ================
11-27 13:36:58.098   889 21360 E camera_sof: **********clock start**********
11-27 13:36:58.098   889 21360 E camera_sof: opendir clk fails
11-27 13:36:58.094   889   889 W CAM_mct_freeze: type=1400 audit(0.0:477): avc: denied { read } for uid=1047 name="clk" dev="debugfs" ino=14435 scontext=u:r:hal_camera_default:s0 tcontext=u:object_r:debugfs:s0 tclass=dir permissive=0
11-27 13:36:58.098   889 21360 E camera_sof: ================ END SOF DEBUG DUMP ================
11-27 13:36:58.376  3629  3848 I WorkerManager: dispose()
11-27 13:36:58.377  3629  3848 W ThreadPoolDumper: Queue length for executor EventBus is now 11. Perhaps some tasks are too long, or the pool is too small.
11-27 13:36:59.361  2483  2614 I QCNEJ   : |CORE| CNE received action RSSI/Link Changed events: android.net.wifi.RSSI_CHANGED
11-27 13:36:59.361  2483  2614 D QCNEJ   : |CORE| Updating RSSI: -53
11-27 13:36:59.370  7193  7247 I CarrierServices: [334] bur.a: FiST: Wifi connectivity status true
11-27 13:36:59.371  7193  7247 W CarrierServices: [334] bax.a: This RcsEngine is stale. Ignore Connectivity event on this.
11-27 13:36:59.373  7193  7247 W CarrierServices: [334] bax.a: This RcsEngine is stale. Ignore Connectivity event on this.
11-27 13:36:59.374  7193  7247 I CarrierServices: [334] buw.d: Connected state: [1], networkType: [WIFI]
11-27 13:36:59.387   955   968 I mmid    : select timeout: wait for receiving msg
11-27 13:36:59.857   886  1492 D NativeSensors: CapSensor:  mPendingEvent.distance=0.000000
11-27 13:37:00.009  1890  1890 I ActivityManager: Killing 19089:com.google.android.gms.ui/u0a39 (adj 906): empty #17
11-27 13:37:00.009  1890  1890 I ActivityManager: Killing 18641:com.google.android.youtube/u0a138 (adj 906): empty #18
11-27 13:37:00.067 12435 21363 D EventNotificationJob: Running EventNotificationJob, isDetail=true
11-27 13:37:00.070 12435 21363 D ACAccountManager: ACMailAccount with ID 2 not found
11-27 13:37:00.071 12435 21363 D EventNotificationJob: Could not find account for ID: 2
11-27 13:37:00.073 12435 21363 D ACAccountManager: ACMailAccount with ID 4 not found
11-27 13:37:00.073 12435 21363 D EventNotificationJob: Could not find account for ID: 4
11-27 13:37:00.075 12435 21363 D ACAccountManager: ACMailAccount with ID 6 not found
11-27 13:37:00.075 12435 21363 D EventNotificationJob: Could not find account for ID: 6
11-27 13:37:00.077 12435 21363 D ACAccountManager: ACMailAccount with ID 5 not found
11-27 13:37:00.077 12435 21363 D EventNotificationJob: Could not find account for ID: 5
11-27 13:37:00.079 12435 21363 D EventNotificationJob:   finished processing events.
11-27 13:37:00.084  1890  7359 D ConnectivityService: ConnectivityService NetworkRequestInfo binderDied(NetworkRequest [ LISTEN id=103, [ Capabilities: INTERNET&NOT_RESTRICTED&TRUSTED&FOREGROUND] ], android.os.BinderProxy@ea4e176)
11-27 13:37:00.085  1890  2282 E ConnectivityService: RemoteException caught trying to send a callback msg for NetworkRequest [ LISTEN id=103, [ Capabilities: INTERNET&NOT_RESTRICTED&TRUSTED&FOREGROUND] ]
11-27 13:37:00.598   889 21353 E mm-camera: <MCT   ><ERROR> 117: mct_bus_sof_thread_run: FATAL Session 3: SOF Freeze! Sending error message
11-27 13:37:00.598   889 21353 I mm-camera: <MCT   >< INFO> 158: mct_bus_sof_thread_run: Sending HW_ERROR from MCT on session =3
11-27 13:37:00.599   889 21285 E mm-camera: <MCT   ><ERROR> 823: mct_controller_send_cb: FATAL: Sending HW_Error
11-27 13:37:00.599   889 21306 E QCamera : <HAL><ERROR> camEvtHandle: 905: Fatal, camera daemon died
11-27 13:37:00.603   889  1878 E QCamera : <HAL><ERROR> processCaptureRequest: 6131: Unblocked on timeout!!!!
11-27 13:37:00.603   889  1878 E CamDevSession@3.2-impl: processOneCaptureRequest: HAL process_capture_request call failed!
11-27 13:37:00.603  1083 21314 E Camera3-Device: Camera 0: sendRequestsBatch: RequestThread: Unable to submit capture request 3 to HAL device: Function not implemented (-38)
11-27 13:37:00.607 20287 20486 E CameraManager: Camera 0: Error! 4
11-27 13:37:00.609 20287 20486 E CameraSession: Camera Device android.hardware.camera2.impl.CameraDeviceImpl@a2fca39 has been disconnected!
11-27 13:37:00.609 20287 20486 E CameraSession: com.mrousavy.camera.core.CameraDisconnectedError: [session/camera-has-been-disconnected] The given Camera device (id: 0) has been disconnected! Error: UNKNOWN_CAMERA_DEVICE_ERROR
11-27 13:37:00.609 20287 20486 E CameraSession:         at com.mrousavy.camera.extensions.CameraManager_openCameraKt$openCamera$2$callback$1.onError(CameraManager+openCamera.kt:49)
11-27 13:37:00.609 20287 20486 E CameraSession:         at android.hardware.camera2.impl.CameraDeviceImpl$CameraDeviceCallbacks$1.run(CameraDeviceImpl.java:1791)
11-27 13:37:00.609 20287 20486 E CameraSession:         at android.os.Handler.handleCallback(Handler.java:789)
11-27 13:37:00.609 20287 20486 E CameraSession:         at android.os.Handler.dispatchMessage(Handler.java:98)
11-27 13:37:00.609 20287 20486 E CameraSession:         at android.os.Looper.loop(Looper.java:164)
11-27 13:37:00.609 20287 20486 E CameraSession:         at android.os.HandlerThread.run(HandlerThread.java:65)
11-27 13:37:00.609 20287 20486 E CameraView: invokeOnError(...):
[... cut ...] =Cool, POWER_SUPPLY_INPUT_SUSPEND=0}
11-27 13:37:02.043  2778  2939 D HeadsetStateMachine: Disconnected process message: 10, size: 0
11-27 13:37:02.043  2778  2939 D HeadsetStateMachine: Enter processIntentBatteryChanged()
11-27 13:37:02.043  2778  2939 D HeadsetStateMachine: Exit processIntentBatteryChanged()
11-27 13:37:02.043  2778  2939 D HeadsetStateMachine: Exit Disconnected processMessage()
11-27 13:37:02.044  2564  2564 D QtiCarrierConfigHelper: WARNING, no carrier configs on phone Id: 0

daltachange, 2023-11-30:
I tried to solve the problem in the following way：

error log:
[image]

root cause:
[image]

if diff.deviceChanged is true， but the camera id closed. 
Then  execution thie method  ‘configureOutputs(config)’ ，throw exception.

CameraManager:
[image]

fix:
[image]

Maybe this will help you. @GaylordP 

Hi @mrousavy ，please help me to check. Thank you very much.

FazilMuhammed, 2023-12-04:
@mrousavy did you check this issue ? any update from this

baodzzzz, 2023-12-25:
any update from this ?

quocbaooo, 2024-01-02:
> I tried to solve the problem in the following way：
> 
> com.mrousavy.camera.core.CameraSession
> 
> error log: <img alt="image" width="1686" src="https://private-user-images.githubusercontent.com/12230238/286946220-cf87642c-713c-4e13-bdf4-ed21ed54a892.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQxNjc4MzgsIm5iZiI6MTcwNDE2NzUzOCwicGF0aCI6Ii8xMjIzMDIzOC8yODY5NDYyMjAtY2Y4NzY0MmMtNzEzYy00ZTEzLWJkZjQtZWQyMWVkNTRhODkyLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMDIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTAyVDAzNTIxOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTRlODA3NWFiMjc4NmY3ZTQ3NTZjNTllNzFjZDEwY2FiZGRiODE4NmRiZjhmNDljOGJkODQxMjFmZTNhMTExOTgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.VYrHgbfcJLcZWW8X7dP4X8oiGjIhNkc2BrTYUowmMvU">
> 
> root cause: <img alt="image" width="1543" src="https://private-user-images.githubusercontent.com/12230238/286947089-2440b0bf-051b-4c78-ade8-699f4c99debd.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQxNjc4MzgsIm5iZiI6MTcwNDE2NzUzOCwicGF0aCI6Ii8xMjIzMDIzOC8yODY5NDcwODktMjQ0MGIwYmYtMDUxYi00Yzc4LWFkZTgtNjk5ZjRjOTlkZWJkLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMDIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTAyVDAzNTIxOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTUzMjk4MzY1ZDJiNmIwYjlkZWE3ZWYxMGQyNjliYjkxOTJkOTIzZTgwNjY0NTRjN2VmOGRiYWRiYjI3NmRlOTQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.4opsO2QGqxAtSPxFY7ICuH8_88wSuYbQLzIuvPuUlXI">
> 
> if diff.deviceChanged is true， but the camera id closed. Then execution thie method ‘configureOutputs(config)’ ，throw exception.
> 
> CameraManager: <img alt="image" width="1281" src="https://private-user-images.githubusercontent.com/12230238/286948159-3979bc34-3bb8-42c8-8f71-622e8690846b.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQxNjc4MzgsIm5iZiI6MTcwNDE2NzUzOCwicGF0aCI6Ii8xMjIzMDIzOC8yODY5NDgxNTktMzk3OWJjMzQtM2JiOC00MmM4LThmNzEtNjIyZTg2OTA4NDZiLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMDIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTAyVDAzNTIxOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWRjYzA3MWMwYzhhOWJkMjViN2E2OTE5ZGU4NmRmNTQ0ZGVmZTlhYjQ4Mjk2ZTQzMjVhOTQ0N2E5NjczMGFjZTcmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.CJfsvnU93UYUUObT4vu2cp__E-NJFuEdAnCp7kBEK4U">
> 
> fix: <img alt="image" width="1212" src="https://private-user-images.githubusercontent.com/12230238/286949108-bf37d27b-6d62-4b22-a7c9-99c023971f15.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQxNjc4MzgsIm5iZiI6MTcwNDE2NzUzOCwicGF0aCI6Ii8xMjIzMDIzOC8yODY5NDkxMDgtYmYzN2QyN2ItNmQ2Mi00YjIyLWE3YzktOTljMDIzOTcxZjE1LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMDIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTAyVDAzNTIxOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWU0NjEzNTNiNmZjYjk5YzNjY2RlMjM4ODQ0MjViYTI3OWVmNzJiMzc5ODJjMWNkOGE3NTcwNTE0YzZmZDI3NDcmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.5SqCbbWIUKbVYgZO4WG_EfrFDb8sfCFRbA6SqbSEy_I">
> 
> Maybe this will help you. @GaylordP
> 
> Hi @mrousavy ，please help me to check. Thank you very much.

pls tell me what is isCameraClosed?

mrousavy, 2024-01-05:
Hey! Thanks for reporting this issue.

I've been working the past days on making sure the Camera lifecycle is safely handled and released when needed to make sure this crash doesn't occur anymore.

I just created a PR for this - can you please this to see if that fixes the issue for you? https://github.com/mrousavy/react-native-vision-camera/pull/2339

If this fixes your issue, please consider [💖 **sponsoring me on GitHub** 💖](https://github.com/sponsors/mrousavy/) to support me / thank me for building VisionCamera and continuously improving it. 

If this does not fix your issue, please clone the repo, check out the branch `fix/blackscreen` (the PR above), run the Example app, reproduce your issue there and share the `adb logcat` logs with me (upload to pastebin or gist) so I can investigate this further.

Thank you! 🙏