margelo/react-native-vision-camera #2197: 🐛Why camera no reload after standby or block phone

R4tch3t, 2023-11-22:
### What's happening?

The camera is in black screen and no reload after press the block button in the phone, works fine when the app pass to background or navigate in the route 

### Reproduceable Code

```tsx
{device != null && (
        <PinchGestureHandler onGestureEvent={onPinchGesture} enabled={isActive}>
          <Reanimated.View style={StyleSheet.absoluteFill}>
            <TapGestureHandler onEnded={onDoubleTap} numberOfTaps={2}>
              <ReanimatedCamera
                ref={camera}
                style={StyleSheet.absoluteFill}
                device={device}
                format={format}
                fps={fps}
                hdr={enableHdr}
                lowLightBoost={device.supportsLowLightBoost && enableNightMode}
                isActive={isActive}
                onInitialized={onInitialized}
                onError={onError}
                enableZoomGesture={false}
                animatedProps={cameraAnimatedProps}
                enableFpsGraph={true}                                
                photo={true}
                video={true}
                audio={hasMicrophonePermission}
                frameProcessor={frameProcessor}                
                pixelFormat={'yuv'}                
              />
            </TapGestureHandler>
          </Reanimated.View>
        </PinchGestureHandler>
      )}
```

### Relevant log output

```shell
Failed to configure CameraSession! Error: [session/camera-not-ready] The Camera is not ready yet! Wait for the onInitialized() callback!, Config-Diff: Difference(deviceChanged=false, outputsChanged=false, sidePropsChanged=true)
                                                                                                    com.mrousavy.camera.core.CameraNotReadyError: [session/camera-not-ready] The Camera is not ready yet! Wait for the onInitialized() callback!
                                                                                                    	at com.mrousavy.camera.core.CameraSession.configureCaptureRequest(CameraSession.kt:375)
                                                                                                    	at com.mrousavy.camera.core.CameraSession.configure(CameraSession.kt:140)
                                                                                                    	at com.mrousavy.camera.CameraView$update$1.invokeSuspend(CameraView.kt:143)
                                                                                                    	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
                                                                                                    	at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:108)
                                                                                                    	at android.os.Handler.handleCallback(Handler.java:978)
                                                                                                    	at android.os.Handler.dispatchMessage(Handler.java:104)
                                                                                                    	at android.os.Looper.loopOnce(Looper.java:238)
                                                                                                    	at android.os.Looper.loop(Looper.java:357)
                                                                                                    	at android.os.HandlerThread.run(HandlerThread.java:67)
2023-11-21 22:54:44.996 29321-29683 CameraView              com.adonaysoft.stepsV2               E  invokeOnError(...):
```

### Camera Device

```json
{}
```

### Device

motorola edge 20

### VisionCamera Version

3.6.8

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

R4tch3t, 2023-11-22:
Ok I'm sure it's because of camera facial recognition when block the phone. Can I recovery the session or bypass the error, thanks great work?

R4tch3t, 2023-11-22:
Really, if you open react-native-vision-camera and then open the camera in some other app or tool, the session is closed but it is not reopened and the throw is sent...

R4tch3t, 2023-11-22:
Semi-fixed adding 
```
if(!isForeground){
    return <></>
}
```
but still crashing when the device pass to block-screen and the face recognition is on...

shahinyan11, 2023-11-27:
@R4tch3t Hi there. Can you please say what you use for  face detection .  Are you use "vision-camera-face-detector" package ?

mrousavy, 2024-01-05:
Hey! Thanks for reporting this issue.

I've been working the past days on making sure the Camera lifecycle is safely handled and released when needed to make sure this crash doesn't occur anymore.

I just created a PR for this - can you please this to see if that fixes the issue for you? https://github.com/mrousavy/react-native-vision-camera/pull/2339

If this fixes your issue, please consider [💖 **sponsoring me on GitHub** 💖](https://github.com/sponsors/mrousavy/) to support me / thank me for building VisionCamera and continuously improving it. 

If this does not fix your issue, please clone the repo, check out the branch `fix/blackscreen` (the PR above), run the Example app, reproduce your issue there and share the `adb logcat` logs with me (upload to pastebin or gist) so I can investigate this further.

Thank you! 🙏

closed 2024-01-08 by mrousavy