margelo/react-native-vision-camera #2199: App Crashes when frameProcessor prop used🐛

preetamra, 2023-11-22:
### What's happening?

 android app crashes then i try to use frameProcessor.it works fine when I comment out the frameprocessor prop.i have pasted the logs please look into it

### Reproduceable Code

```tsx
function App(): JSX.Element {
  const { hasPermission, requestPermission } = useCameraPermission();
  const device = useCameraDevice('front');
  const frameProcessor = useFrameProcessor(() => {
    'worklet'
 /*    const faces = scanFaces(frame)
    console.log(`Faces in Frame: ${faces}`) */
    console.log("Frame Processor");
  }, [])

  useEffect(() => {
    if(!hasPermission)
    {
      requestPermission().then((res) => {
        console.log(res);
      }).catch((e) => {
        console.log(e);
      })
    }
  })

  return (
    hasPermission?
     <View style={{
      flex:1,
     }}>
      <Camera 
      device={device}
      isActive={true}
      style={{
        flex:1,
      }}
      frameProcessor={frameProcessor}      //app don't crash when i remove the frameProcessor
      />
     </View>:
     <View style={{
      flex:1,
      justifyContent:"center",
      alignItems:"center"
     }}>
     <Text>
     Permission Denied
     </Text>
     </View>
  );
}
```

### Relevant log output

```shell
2023-11-22 15:36:10.095 22446-22577 AndroidRuntime          com.test2                            E  FATAL EXCEPTION: mrousavy/VisionCamera.video
                                                                                                    Process: com.test2, PID: 22446
                                                                                                    com.facebook.jni.CppException: Compiling JS failed: 1:1:invalid empty parentheses '( )' Buffer size 3 starts with: 280a29
                                                                                                    	at com.mrousavy.camera.frameprocessor.FrameProcessor.call(Native Method)
                                                                                                    	at com.mrousavy.camera.core.VideoPipeline._init_$lambda$0(VideoPipeline.kt:115)
                                                                                                    	at com.mrousavy.camera.core.VideoPipeline.$r8$lambda$EMriwS_FKhtjTFtppM84Z5V8tiI(Unknown Source:0)
                                                                                                    	at com.mrousavy.camera.core.VideoPipeline$$ExternalSyntheticLambda0.onImageAvailable(Unknown Source:2)
                                                                                                    	at android.media.ImageReader$1.run(ImageReader.java:837)
                                                                                                    	at android.os.Handler.handleCallback(Handler.java:938)
                                                                                                    	at android.os.Handler.dispatchMessage(Handler.java:99)
                                                                                                    	at android.os.Looper.loopOnce(Looper.java:241)
                                                                                                    	at android.os.Looper.loop(Looper.java:342)
                                                                                                    	at android.os.HandlerThread.run(HandlerThread.java:67)
2023-11-22 15:36:10.095 22446-22577 ExceptionHandle         com.test2                            I  at com.mrousavy.camera.frameprocessor.FrameProcessor.call(Native Method)
2023-11-22 15:36:10.096 22446-22577 ExceptionHandle         com.test2                            I  at com.mrousavy.camera.core.VideoPipeline._init_$lambda$0(VideoPipeline.kt:115)
2023-11-22 15:36:10.096 22446-22577 ExceptionHandle         com.test2                            I  at com.mrousavy.camera.core.VideoPipeline.$r8$lambda$EMriwS_FKhtjTFtppM84Z5V8tiI(Unknown Source:0)
2023-11-22 15:36:10.096 22446-22577 ExceptionHandle         com.test2                            I  at com.mrousavy.camera.core.VideoPipeline$$ExternalSyntheticLambda0.onImageAvailable(Unknown Source:2)
2023-11-22 15:36:10.096 22446-22577 ExceptionHandle         com.test2                            I  at android.media.ImageReader$1.run(ImageReader.java:837)
2023-11-22 15:36:10.096 22446-22577 ExceptionHandle         com.test2                            I  at android.os.Handler.handleCallback(Handler.java:938)
2023-11-22 15:36:10.096 22446-22577 ExceptionHandle         com.test2                            I  at android.os.Handler.dispatchMessage(Handler.java:99)
2023-11-22 15:36:10.096 22446-22577 ExceptionHandle         com.test2                            I  at android.os.Looper.loopOnce(Looper.java:241)
2023-11-22 15:36:10.096 22446-22577 ExceptionHandle         com.test2                            I  at android.os.Looper.loop(Looper.java:342)
2023-11-22 15:36:10.097 22446-22577 ExceptionHandle         com.test2                            I  at android.os.HandlerThread.run(HandlerThread.java:67)
```

### Camera Device

```json
{
  "sensorOrientation": "landscape-left",
  "hardwareLevel": "full",
  "maxZoom": 4,
  "minZoom": 1,
  "supportsLowLightBoost": false,
  "neutralZoom": 1,
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "supportsFocus": true,
  "supportsRawCapture": false,
  "isMultiCam": false,
  "name": "FRONT (1)",
  "hasFlash": true,
  "hasTorch": true,
  "position": "front",
  "id": "1"
}
```

### Device

Infinix X669C

### VisionCamera Version

3.6.8

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

preetamra, 2023-11-23:
are there alternates i can use for frame processing @agiorgiani @whiletruefranz

whiletruefranz, 2023-11-23:
@preetamra It depends, you could capture a frame on an interval, save e.g. a png to the temp directory and then process that file, but that seems really bad performance wise compared to the frame processing worklet. 

Funny thing is: I have a commit with a working version of my file, logging the rgb value at a certain pixel coordinate via frame processor. As soon as I change *anything* inside that file, be it deleting an unused import or changing the backgroundColor of a view, the frame processor errors out and I have to roll back. The error I get (without changing anything on the working function) is the following:

`Regular javascript function '_f' cannot be shared. Try decorating the function with the 'worklet' keyword to allow the javascript function to be used as a worklet.`

@mrousavy I am out of ideas with this, could you chime in? The function used is decorated with the worklet keyword and as I said it works as long as I don't change anything on the component file.

mrousavy, 2023-11-23:
Is the babel plugin properly configured?

mrousavy, 2023-11-23:
maybe check your versions of RN, RN-VisionCamera, RN-Worklets-Core and RN-Reanimated to make sure theyre the same as here in the example app

canibal2, 2023-11-23:
**GUYS, I HAVE GOOD NEWS!!!!!**

This version config is working. Just copy and paste. 
**I will waiting for feedback!!**
I hope it can be help to you.
    
**//package.json**
```
//package.json
{
  "name": "AwesomeFrameFace",
  "version": "0.0.1",
  "private": true,
  "scripts": {
    "android": "react-native run-android",
    "ios": "react-native run-ios",
    "lint": "eslint .",
    "start": "react-native start",
    "test": "jest"
  },
  "dependencies": {
    "react": "18.2.0",
    "react-native": "0.72.3",
    "react-native-reanimated": "^3.5.4",
    "react-native-vision-camera": "3.6.9",
    "react-native-worklets-core": "0.2.4",
    "vision-camera-face-detector-plugin": "github:lenhathieu96/vision-camera-face-detector"
  },
  "devDependencies": {
    "@babel/core": "^7.20.0",
    "@babel/preset-env": "^7.20.0",
    "@babel/runtime": "^7.20.0",
    "@react-native/eslint-config": "^0.72.2",
    "@react-native/metro-config": "^0.72.11",
    "@tsconfig/react-native": "^3.0.0",
    "@types/react": "^18.0.24",
    "@types/react-test-renderer": "^18.0.0",
    "babel-jest": "^29.2.1",
    "eslint": "^8.19.0",
    "jest": "^29.2.1",
    "metro-react-native-babel-preset": "0.76.8",
    "prettier": "^2.4.1",
    "react-test-renderer": "18.2.0",
    "typescript": "4.8.4"
  },
  "engines": {
    "node": ">=16"
  }
}
```
**//babel.config.js**
```
//babel.config.js
module.exports = {
  presets: ['module:metro-react-native-babel-preset'],
  plugins: [
    ['react-native-reanimated/plugin'],
    ['react-native-worklets-core/plugin'],
  ],
};

```
**//app.tsx**
```
//app.tsx
/* eslint-disable react-hooks/exhaustive-deps */
import React, { useCallback, useEffect, useState } from 'react';
import { StyleSheet, Text, SafeAreaView, View, Image } from 'react-native';
import {
  Camera,
  VisionCameraProxy,
  useCameraDevice,
  useCameraPermission,
  useFrameProcessor,
} from 'react-native-vision-camera';
import { Worklets } from 'react-native-worklets-core';

import {
  FaceDetectionStatus,
  FaceDetectorResponse,
  detectFace,
} from 'vision-camera-face-detector-plugin';

const CAMERA_SIZE = 900;

export default function App() {
  const [errMessage, setErrMessage] = useState<string>('');
  const [base64Frame, setBase64Frame] = useState<string>('');
  const [submitting, setSubmitting] = useState<boolean>(false);
  const [status, setStatus] = useState<FaceDetectionStatus>('success');

  const device = useCameraDevice('front');
  const { hasPermission, requestPermission } = useCameraPermission();
  const id = (Date.now() * 1000);
  const submitSever = (frameData: string) => {
    setSubmitting(true);
    setErrMessage('');
    setBase64Frame(frameData);
    return new Promise((res) => {
      setTimeout(() => {
        setSubmitting(false);
        setBase64Frame('');
        res('data ne');
      }, 4000);
    });
  };

  const onGetFaceDetectorResponse = Worklets.createRunInJsFn(
    async (res: FaceDetectorResponse) => {
      setStatus(res.status);
      switch (res.status) {
        case 'error': {
          if (res.error) {
            setErrMessage(res.error.message);
          }
          break;
        }
        case 'success': {
          if (res.frameData) {

            console.log(res.faces)
            submitSever(res.frameData);
          }
          break;
        }
        default:
          break;
      }
    }
  );

  const frameProcessor = useFrameProcessor(
    (frame) => {
      'worklet';
      if (submitting) {
        return;
      }

      const response = detectFace(frame);
      onGetFaceDetectorResponse(response);
    },
    [submitting]
  );

  useEffect(() => {
    if (!hasPermission) {
      requestPermission();
    }
  }, []);

  const renderCamera = useCallback(() => {
    if (device == null || !hasPermission) {
      return <Text>No camera device</Text>;
    }

    return (
      <Camera
        device={device}
        isActive
        style={styles.camera}
        frameProcessor={frameProcessor}
        //ML Kit use YUV format
        pixelFormat="yuv"
      />
    );
  }, [device, hasPermission, frameProcessor]);

  const renderFrame = useCallback(() => {
    return (
      <Image
        style={{ width: 100, height: 100 }}
        source={{ uri: `data:image/png;base64,${base64Frame}` }}
      />
    );
  }, [base64Frame]);

  return (
    <SafeAreaView style={styles.container}>
      <View style={styles.cameraContainer}>{renderCamera()}</View>
      {renderFrame()}
      <Text>{status === 'error' ? `Error: ${errMessage}` : status}</Text>
    </SafeAreaView>
  );
}

const styles = StyleSheet.create({
  container: {
    flex: 1,
    alignItems: 'center',
    justifyContent: 'center',
  },

  cameraContainer: {
    width: CAMERA_SIZE / 2.15,
    height: CAMERA_SIZE - 100,
    borderRadius: CAMERA_SIZE / 2,
    marginVertical: 24,
  },

  camera: {
    flex: 1,
  },
});
```

whiletruefranz, 2023-11-24:
@mrousavy I made sure the versions match the example projects, did not solve a thing unfortunately. 

@canibal2 So you used the Worklets directly instead of the decorator? Will try that.

canibal1, 2023-11-24:
@whiletruefranz actually I tried just in the android side. Right now, I try for IOS. I will give you feed back asap.

rodgomesc, 2023-11-24:
i faced this error only when i forget to configure worklets in babel, you can also try to disable hermes and you probably will get a more helpful description of what going on

canibal1, 2023-11-24:
@whiletruefranz I checked it. As I said it's worked android and ios side. Make sure run

rm -rf node_module
rm -rf package-lock.json
rm -rf yarn.lock

npm install
yarn install 

cd ios
bundle install
bundle exec pod install

Make sure you do as in the config above https://github.com/mrousavy/react-native-vision-camera/issues/2199#issuecomment-1824938125. Just copy and paste.
```
"dependencies": {
     "react": "18.2.0",
     "react-native": "0.72.3",
     "react-native-reanimated": "^3.5.4",
     "react-native-vision-camera": "3.6.9",
     "react-native-worklets-core": "0.2.4",
     "vision-camera-face-detector-plugin": "github:lenhathieu96/vision-camera-face-detector"
   },
```
and 

```
//babel.config.js
module.exports = {
  presets: ['module:metro-react-native-babel-preset'],
  plugins: [
    ['react-native-reanimated/plugin'],
    ['react-native-worklets-core/plugin'],
  ],
};

```
parts is important.

github:lenhathieu96/vision-camera-face-detector package provide the face detection plugin. You can test with this package.

preetamra, 2023-11-25:
I didnt get the same error after updating my packages

whiletruefranz, 2023-11-25:
Thanks to @canibal1 my project works right now. 

    "react-native-worklets-core": "^0.2.4",
    "react-native-vision-camera": "^3.6.8",
    "react-native-reanimated": "^3.5.4",

is the working combination.

closed 2023-11-26 by preetamra

canibal2, 2023-11-26:
@preetamra is it worked for you?

xulihang, 2023-12-14:
Using JSC will output this error: `Frame Processor threw an error: Unexpected token ')'`

I am using the following combination:

```
"react-native": "0.73.0",
"react-native-vision-camera": "^3.6.16",
"react-native-worklets-core": "^0.2.4",
"vision-camera-dynamsoft-document-normalizer": "^1.0.0"
```

bimal1331, 2024-01-16:
> Thanks to @canibal1 my project works right now.
> 
> ```
> "react-native": "0.72.3",
> "react-native-worklets-core": "^0.2.4",
> "react-native-vision-camera": "^3.6.8",
> "react-native-reanimated": "^3.5.4",
> ```
> 
> is the working combination.

Thanks to @whiletruefranz and @canibal1 , I was stuck at this issue for past 2 days, and trying your versions, I was able to make frameprocessors work in my app.

jslok, 2024-02-17:
`yarn start --reset cache` solved it for me