margelo/react-native-vision-camera #2203: 🔧 react/bridging/CallbackWrapper.h' file not found with v3🔧

Filocava99, 2023-11-22:
### How were you trying to build the app?

I was trying to build my Expo app after setting use_Framework to static but now I get the following error: `🔧 react/bridging/CallbackWrapper.h' file not found`. I created a new issue because all the existing ones are specifically for version 2 of RNVC.

### Full build logs

```tsx
Showing Recent Errors Only

Build target VisionCamera of project Pods with configuration Debug

CompileC /Users/filippocavallari/Library/Developer/Xcode/DerivedData/replateapp-djjburazmxijpmazqstsfnxwqwcq/Build/Intermediates.noindex/Pods.build/Debug-iphoneos/VisionCamera.build/Objects-normal/arm64/JSINSObjectConversion.o /Users/filippocavallari/Desktop/Replate/ReplateApp/node_modules/react-native-vision-camera/ios/Frame\ Processor/JSINSObjectConversion.mm normal arm64 objective-c++ com.apple.compilers.llvm.clang.1_0.compiler (in target 'VisionCamera' from project 'Pods')
    cd /Users/filippocavallari/Desktop/Replate/ReplateApp/ios/Pods
    /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/clang -x objective-c++ -ivfsstatcache /Users/filippocavallari/Library/Developer/Xcode/DerivedData/SDKStatCaches.noindex/iphoneos17.0-21A325-1e1d10ddc0bdcd738a98873f48b793a5.sdkstatcache -target arm64-apple-ios12.4 -fmessage-length\=0 -fdiagnostics-show-note-include-stack -fmacro-backtrace-limit\=0 -fno-color-diagnostics -std\=c++17 -stdlib\=libc++ -fobjc-arc -fmodules -gmodules -fmodules-cache-path\=/Users/filippocavallari/Library/Developer/Xcode/DerivedData/ModuleCache.noindex -fmodules-prune-interval\=86400 -fmodules-prune-after\=345600 -fbuild-session-file\=/Users/filippocavallari/Library/Developer/Xcode/DerivedData/ModuleCache.noindex/Session.modulevalidation -fmodules-validate-once-per-build-session -Wnon-modular-include-in-framework-module -Werror\=non-modular-include-in-framework-module -fmodule-name\=VisionCamera -Wno-trigraphs -fpascal-strings -O0 -fno-common -Wno-missing-field-initializers -Wno-missing-prototypes -Werror\=return-type -Wdocumentation -Wunreachable-code -Wno-implicit-atomic-properties -Werror\=deprecated-objc-isa-usage -Wno-objc-interface-ivars -Werror\=objc-root-class -Wno-arc-repeated-use-of-weak -Wimplicit-retain-self -Wno-non-virtual-dtor -Wno-overloaded-virtual -Wno-exit-time-destructors -Wduplicate-method-match -Wno-missing-braces -Wparentheses -Wswitch -Wunused-function -Wno-unused-label -Wno-unused-parameter -Wunused-variable -Wunused-value -Wempty-body -Wuninitialized -Wconditional-uninitialized -Wno-unknown-pragmas -Wno-shadow -Wno-four-char-constants -Wno-conversion -Wconstant-conversion -Wint-conversion -Wbool-conversion -Wenum-conversion -Wno-float-conversion -Wnon-literal-null-conversion -Wobjc-literal-conversion -Wshorten-64-to-32 -Wno-newline-eof -Wno-selector -Wno-strict-selector-match -Wundeclared-selector -Wdeprecated-implementations -Wno-c++11-extensions -Wno-implicit-fallthrough -DPOD_CONFIGURATION_DEBUG\=1 -DDEBUG\=1 -DCOCOAPODS\=1 -DPOD_CONFIGURATION_DEBUG\=1 -DDEBUG\=1 -DSK_METAL\=1 -DSK_GANESH\=1 -DVISION_CAMERA_ENABLE_FRAME_PROCESSORS\=true -D_LIBCPP_ENABLE_CXX17_REMOVED_UNARY_BINARY_FUNCTION -DOBJC_OLD_DISPATCH_PROTOTYPES\=0 -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/iPhoneOS.platform/Developer/SDKs/iPhoneOS17.0.sdk -fstrict-aliasing -Wprotocol -Wdeprecated-declarations -Winvalid-offsetof -g -fvisibility-inlines-hidden -Wno-sign-conversion -Winfinite-recursion -Wmove -Wcomma -Wblock-capture-autoreleasing -Wstrict-prototypes -Wrange-loop-analysis -Wno-semicolon-before-method-body -Wunguarded-availability -index-store-path /Users/filippocavallari/Library/Developer/Xcode/DerivedData/replateapp-djjburazmxijpmazqstsfnxwqwcq/Index.noindex/DataStore -iquote /Users/filippocavallari/Library/Developer/Xcode/DerivedData/replateapp-djjburazmxijpmazqstsfnxwqwcq/Build/Intermediates.noindex/Pods.build/Debug-iphoneos/VisionCamera.build/VisionCamera-generated-files.hmap -I/Users/filippocavallari/Library/Developer/Xcode/DerivedData/replateapp-djjburazmxijpmazqstsfnxwqwcq/Build/Intermediates.noindex/Pods.build/Debug-iphoneos/VisionCamera.build/VisionCamera-own-target-headers.hmap -I/Users/filippocavallari/Library/Developer/Xcode/DerivedData/replateapp-djjburazmxijpmazqstsfnxwqwcq/Build/Intermediates.noindex/Pods.build/Debug-iphoneos/VisionCamera.build/VisionCamera-all-non-framework-target-headers.hmap -ivfsoverlay /Users/filippocavallari/Library/Developer/Xcode/DerivedData/replateapp-djjburazmxijpmazqstsfnxwqwcq/Build/Intermediates.noindex/Pods.build/Debug-iphoneos/VisionCamera.build/all-product-headers.yaml -iquote /Users/filippocavallari/Library/Developer/Xcode/DerivedData/replateapp-djjburazmxijpmazqstsfnxwqwcq/Build/Intermediates.noindex/Pods.build/Debug-iphoneos/VisionCamera.build/VisionCamera-project-headers.hmap -I/Users/filippocavallari/Library/Developer/Xcode/DerivedData/replateapp-djjburazmxijpmazqstsfnxwqwcq/Build/Products/Debug-iphoneos/VisionCamera/include -I/Users/filippocavallari/Library/Developer/Xcode/DerivedData/replateapp-djjburazmxijpmazqstsfnxwqwcq/Build/Products/Debug-iphoneos/ReactCommon-Samples/ReactCommon_Samples.framework/Headers -I/Users/filippocavallari/Library/Developer/Xcode/DerivedData/replateapp-djjburazmxijpmazqstsfnxwqwcq/Build/Products/Debug-iphoneos/ReactCommon/ReactCommon.framework/Headers/react/nativemodule/core -I/Users/filippocavallari/Library/Developer/Xcode/DerivedData/replateapp-djjburazmxijpmazqstsfnxwqwcq/Build/Products/Debug-iphoneos/ReactCommon-Samples/ReactCommon_Samples.framework/Headers/platform/ios -I/Users/filippocavallari/Library/Developer/Xcode/DerivedData/replateapp-djjburazmxijpmazqstsfnxwqwcq/Build/Products/Debug-iphoneos/React-NativeModulesApple/React_NativeModulesApple.framework/Headers -I/Users/filippocavallari/Library/Developer/Xcode/DerivedData/replateapp-djjburazmxijpmazqstsfnxwqwcq/Build/Products/Debug-iphoneos/React-graphics/React_graphics.framework/Headers/react/renderer/graphics/platform/ios -I/Users/filippocavallari/Desktop/Replate/ReplateApp/ios/Pods/Headers/Public -I/Users/filippocavallari/Desktop/Replate/ReplateApp/ios/Pods/Headers/Public/FBLazyVector -I/Users/filippocavallari/Desktop/Replate/ReplateApp/ios/Pods/Headers/Public/RCTRequired -I/Users/filippocavallari/Desktop/Replate/ReplateApp/ios/Pods/Headers/Public/React-callinvoker -I/Users/filippocavallari/Desktop/Replate/ReplateApp/ios/Pods/Headers/Public/React-runtimeexecutor -I/Users/filippocavallari/Desktop/Replate/ReplateApp/ios/Pods/Headers/Public/hermes-engine -I../../node_modules/react-native-vision-camera/cpp -I/Users/filippocavallari/Library/Developer/Xcode/DerivedData/replateapp-djjburazmxijpmazqstsfnxwqwcq/Build/Intermediates.noindex/Pods.build/Debug-iphoneos/VisionCamera.build/DerivedSources-normal/arm64 -I/Users/filippocavallari/Library/Developer/Xcode/DerivedData/replateapp-djjburazmxijpmazqstsfnxwqwcq/Build/Intermediates.noindex/Pods.build/Debug-iphoneos/VisionCamera.build/DerivedSources/arm64 -I/Users/filippocavallari/Library/Developer/Xcode/DerivedData/replateapp-djjburazmxijpmazqstsfnxwqwcq/Build/Intermediates.noindex/Pods.build/Debug-iphoneos/VisionCamera.build/DerivedSources -F/Users/filippocavallari/Library/Developer/Xcode/DerivedData/replateapp-djjburazmxijpmazqstsfnxwqwcq/Build/Products/Debug-iphoneos/VisionCamera -F/Users/filippocavallari/Library/Developer/Xcode/DerivedData/replateapp-djjburazmxijpmazqstsfnxwqwcq/Build/Products/Debug-iphoneos/DoubleConversion -F/Users/filippocavallari/Library/Developer/Xcode/DerivedData/replateapp-djjburazmxijpmazqstsfnxwqwcq/Build/Products/Debug-iphoneos/RCT-Folly -F/Users/filippocavallari/Library/Developer/Xcode/DerivedData/replateapp-djjburazmxijpmazqstsfnxwqwcq/Build/Products/Debug-iphoneos/RCTTypeSafety -F/Users/filippocavallari/Library/Developer/Xcode/DerivedData/replateapp-djjburazmxijpmazqstsfnxwqwcq/Build/Products/Debug-iphoneos/React-Codegen -F/Users/filippocavallari/Library/Developer/Xcode/DerivedData/replateapp-djjburazmxijpmazqstsfnxwqwcq/Build/Products/Debug-iphoneos/React-Core -F/Users/filippocavallari/Library/Developer/Xcode/DerivedData/replateapp-djjburazmxijpmazqstsfnxwqwcq/Build/Products/Debug-iphoneos/React-NativeModulesApple -F/Users/filippocavallari/Library/Developer/Xcode/DerivedData/replateapp-djjburazmxijpmazqstsfnxwqwcq/Build/Products/Debug-iphoneos/React-RCTAnimation -F/Users/filippocavallari/Library/Developer/Xcode/DerivedData/replateapp-djjburazmxijpmazqstsfnxwqwcq/Build/Products/Debug-iphoneos/React-RCTBlob -F/Users/filippocavallari/Libra [... cut ...]
### Additional information

- [X] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

Filocava99, 2023-11-22:
I already tried add the following code to my Podfile:
```
installer.target_installation_results.pod_target_installation_results.each do |pod_name, target_installation_result|
      target_installation_result.native_target.build_configurations.each do |config|
        # For third party modules who have React-bridging dependency to search correct headers
        config.build_settings['HEADER_SEARCH_PATHS'] ||= '$(inherited) '
        config.build_settings['HEADER_SEARCH_PATHS'] << '"$(PODS_ROOT)/Headers/Private/React-bridging/react/bridging" '
        config.build_settings['HEADER_SEARCH_PATHS'] << '"$(PODS_CONFIGURATION_BUILD_DIR)/React-bridging/react_bridging.framework/Headers" '
      end
    end
```
but it didn't work

iSamuIsHere, 2023-11-22:
Same issue on React Native 0.72.3

lucarinaldii, 2023-11-22:
Same issue for me

SmokeLabapp, 2023-11-30:
same issue for me

SmokeLabapp, 2023-11-30:
@Filocava99 were you able to fix the issue?

namti, 2024-01-08:
Same error here

whentao, 2024-01-11:
same issue here

sekhuat, 2024-01-14:
I also faced the same issue and I am using react-native-vision-camera ver 2.16.1. I solved the issue by adding `s.dependency "ReactCommon"` into the `node_modules/react-native-vision-camera/VisionCamera.podspec` at the last line as shown below.

```
Pod::Spec.new do |s|
  ...
  s.dependency "React-callinvoker"
  s.dependency "React"
  s.dependency "React-Core"
  s.dependency "ReactCommon"    <== Add this line
end
```

After made the changes, I execute `pod install` command then recompile the app again. Hope this will help those who using V2 of react-native-vision-camera.

mrousavy, 2024-01-15:
Hey! I think this has already been fixed. If not, feel free to re-open :)

closed 2024-01-15 by mrousavy