margelo/react-native-vision-camera #2207: 🔧 RN 0,73 support

talaikis, 2023-11-23:
### How were you trying to build the app?

```
FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':react-native-vision-camera:compileDebugKotlin'
```

### Full build logs

```tsx
N/A
```

### Project dependencies

```json
"dependencvies": {
  "react-native": "0.73.0-rc.5",
  "react-native-vision-camera": "3.6.8"
}
```

### VisionCamera Version

3.6.8

### Target platforms

Android

### Operating system

Windows

### Can you build the VisionCamera Example app?

No, I cannot build the Example app either

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

jeroenvanhattem, 2023-11-23:
I have the same.
```
"react-native": "0.73.0-rc.5",
"react-native-vision-camera": "^3.6.10"
```

valn1, 2023-12-01:
```
FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':react-native-vision-camera:compileDebugKotlin'.
> A failure occurred while executing org.jetbrains.kotlin.compilerRunner.GradleCompilerRunnerWithWorkers$GradleKotlinCompilerWorkAction
   > Compilation error. See log for more details

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 51s
info Run CLI with --verbose flag for more details.

```
@talaikis is this the error you are seeing?
i am also seeing it on react-native 72.7

talaikis, 2023-12-01:
> @talaikis is this the error you are seeing? i am also seeing it on react-native 72.7

Yeah, seems so.

valn1, 2023-12-01:
@talaikis do you happen to be declaring kotlin version in you build.gradle file?
i was declaring it like kotlinVersion = '1.6.20'
removing it and building again worked for me

valn1, 2023-12-04:
maybe try to update everything, it's what i did

jeroenvanhattem, 2023-12-05:
@mrousavy Is upgrading React Native to support RN0.73 on the roadmap? Or would you recommend going with RN0.72 instead in order to get the package to build?

recallwei, 2023-12-07:
Now, React Native 0.73 is stable, but after upgrading, I got these:
: file:///Users/raipiot/Code/soya-energy-app/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraDevicesManager.kt:74:3 'hasConstants' overrides nothing
e: file:///Users/raipiot/Code/soya-energy-app/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/frameprocessor/VisionCameraProxy.kt:38:58 Only safe (?.) or non-null asserted (!!.) calls are allowed on a nullable receiver of type JavaScriptContextHolder?

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':react-native-vision-camera:compileDebugKotlin'.
> A failure occurred while executing org.jetbrains.kotlin.compilerRunner.GradleCompilerRunnerWithWorkers$GradleKotlinCompilerWorkAction
   > Compilation error. See log for more details

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.
> Get more help at https://help.gradle.org.

ivpusic, 2023-12-07:
@mrousavy urgent fix needed for 0.73

mrousavy, 2023-12-07:
Stop pinging me. If something is urgent, pay me to do it. I don't have unlimited free time.

mrousavy, 2023-12-07:
Fixed in https://github.com/mrousavy/react-native-vision-camera/pull/2260 - if you appreciate my time and dedication, please [consider sponsoring me on GitHub](https://github.com/sponsors/mrousavy) :)