margelo/react-native-vision-camera #2208: 🐛 Video resolution / encoding

j-jonathan, 2023-11-23:
### What's happening?

When the video resolution is too high, recording does not work unless I use the H.265 codec.

Is it possible to know when to use the H.265 codec?

For example:
 - working with h264/h265: {"autoFocusSystem": "contrast-detection", "fieldOfView": 78.63484214515682, "maxExposure": 8, "maxFps": 30, "maxISO": 6400, "maxZoom": 10, "minExposure": -8, "minFps": 1, "minISO": 50, "photoHeight": 2736, "photoWidth": 3648, "pixelFormats": ["yuv", "native"], "supportsDepthCapture": true, "supportsPhotoHdr": false, "supportsVideoHdr": false, "videoHeight": 2056, "videoStabilizationModes": ["off", "standard", "cinematic-extended"], "videoWidth": 3648}
- not working with h264: {"autoFocusSystem": "contrast-detection", "fieldOfView": 78.63484214515682, "maxExposure": 8, "maxFps": 30, "maxISO": 6400, "maxZoom": 10, "minExposure": -8, "minFps": 1, "minISO": 50, "photoHeight": 2736, "photoWidth": 3648, "pixelFormats": ["yuv", "native"], "supportsDepthCapture": true, "supportsPhotoHdr": false, "supportsVideoHdr": false, "videoHeight": 2736, "videoStabilizationModes": ["off", "standard", "cinematic-extended"], "videoWidth": 3648}

### Reproduceable Code

```tsx
const devices = Camera.getAvailableCameraDevices();
const device = devices.find((d) => d.position === 'back');
if (!device) {
    return;
}
const format = getCameraFormat(device, [
    { videoResolution: 'max' },
]);

// ...

return (
    <View style={s.container}>
        <Camera
            ref={this.cameraRef}
            style={{ position: 'absolute', top: 0, left: 0, right: 0, bottom: 0 }}
            device={device}
            format={format}
            isActive={true}
            video={true}
            audio={true}
            resizeMode='contain'
        />
    </View>
);
```

### Relevant log output

```shell
{"cause": {"message": "prepare failed.", "stacktrace": "java.io.IOException: prepare failed.
	at android.media.MediaRecorder._prepare(Native Method)
	at android.media.MediaRecorder.prepare(MediaRecorder.java:1107)
	at com.mrousavy.camera.core.RecordingSession.start(RecordingSession.kt:91)
	at com.mrousavy.camera.core.CameraSession.startRecording(CameraSession.kt:546)
	at com.mrousavy.camera.CameraView_RecordVideoKt.startRecording(CameraView+RecordVideo.kt:56)
	at com.mrousavy.camera.CameraViewModule$startRecording$1.invokeSuspend(CameraViewModule.kt:90)
	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
	at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:106)
	at kotlinx.coroutines.scheduling.CoroutineScheduler.runSafely(CoroutineScheduler.kt:570)
	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.executeTask(CoroutineScheduler.kt:750)
	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.runWorker(CoroutineScheduler.kt:677)
	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.run(CoroutineScheduler.kt:664)
"}, "code": "capture/unknown", "message": "An unknown error occurred while trying to start a video recording! prepare failed.", "userInfo": null}
```

### Camera Device

```json
{
  "sensorOrientation": "landscape-right",
  "hardwareLevel": "limited",
  "maxZoom": 10,
  "minZoom": 1,
  "supportsLowLightBoost": false,
  "neutralZoom": 1,
  "physicalDevices": [
    "wide-angle-camera",
    "ultra-wide-angle-camera",
    "telephoto-camera"
  ],
  "supportsFocus": true,
  "supportsRawCapture": false,
  "isMultiCam": true,
  "name": "BACK (0)",
  "hasFlash": true,
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
```

### Device

HUAWEI P30 (ELE-L29) avec Android 10

### VisionCamera Version

3.6.10

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2023-11-24:
Interesting observation with the codec! Maybe there is a limit for the android recorder?

Have you tried using a really low bitrate?

j-jonathan, 2023-11-27:
Ohhhh, good catch!
It actually doesn't depend on the codec. My phone doesn't support bitrates higher than 80 Mbps.
Since I didn't set a default bitrate, it was taking the one generated by getDefaultBitRate.

```kt
  private fun getDefaultBitRate(): Double {
    var baseBitRate = when (size.width * size.height) {
      in 0..640 * 480 -> 2.0
      in 640 * 480..1280 * 720 -> 5.0
      in 1280 * 720..1920 * 1080 -> 10.0
      in 1920 * 1080..3840 * 2160 -> 30.0
      in 3840 * 2160..7680 * 4320 -> 100.0
      else -> 100.0
    }
    baseBitRate = baseBitRate / 30.0 * (fps ?: 30).toDouble()
    if (this.codec == VideoCodec.H265) baseBitRate *= 0.8
    return baseBitRate
  }
```

The video resolution of the format i used is 3648 x 2736.
So, I had 100 Mbps in h.264, and 80 Mbps in h.265. That's why it worked in h.265 ^^
I tested 80 Mbps in h.264, and it works and 81 Mbps in h.265 doesn't work.

From what I've seen, it may be possible to retrieve it from CamcorderProfile. I need to check that.

Thank you very much for your help :)

mrousavy, 2023-11-27:
Ohh, interesting! I'll try to fix the `getDefaultBitRate()` method by using `CamcorderProfile` instead of computing it on my own then!

mrousavy, 2023-11-27:
Hey @j-jonathan! I spent some time investigating this and debugging the available `CamcorderProfiles`.

I created this PR: https://github.com/mrousavy/react-native-vision-camera/pull/2216 which does two things:

1. It automatically chooses the best/recommended video bit-rate for the given resolution, fps, hdr and codec settings on Android instead of manually calculating it.
2. It up or downscales the computed bit-rate using a multiplier if options like `'low'` (* 0.8) or `'high'` (* 1.2) are passed.

Can you please try this PR and see if that now fixes the problem for you? 

Would be amazing if this now records where it previously failed to record. Thanks for your help! ❤️

mrousavy, 2023-11-27:
And I also added some more options: https://github.com/mrousavy/react-native-vision-camera/pull/2225

j-jonathan, 2023-11-28:
I just tested. Unfortunately, I encountered another error.

```
{"cause": {"message": "Error retrieving camcorder profile params", "stacktrace": "java.lang.RuntimeException: Error retrieving camcorder profile params
	at android.media.CamcorderProfile.native_get_camcorder_profile(Native Method)
	at android.media.CamcorderProfile.get(CamcorderProfile.java:467)
	at com.mrousavy.camera.extensions.RecordingSession_getRecommendedBitRateKt.getRecommendedBitRate(RecordingSession+getRecommendedBitRate.kt:48)
	at com.mrousavy.camera.core.RecordingSession.getBitRate(RecordingSession.kt:132)
	at com.mrousavy.camera.core.RecordingSession.<init>(RecordingSession.kt:37)
	at com.mrousavy.camera.core.CameraSession.startRecording(CameraSession.kt:545)
	at com.mrousavy.camera.CameraView_RecordVideoKt.startRecording(CameraView+RecordVideo.kt:44)
	at com.mrousavy.camera.CameraViewModule$startRecording$1.invokeSuspend(CameraViewModule.kt:91)
	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
	at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:106)
	at kotlinx.coroutines.scheduling.CoroutineScheduler.runSafely(CoroutineScheduler.kt:570)
	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.executeTask(CoroutineScheduler.kt:750)
	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.runWorker(CoroutineScheduler.kt:677)
	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.run(CoroutineScheduler.kt:664)
```

This is due to the 'quality' used (returned by findClosestCamcorderProfileQuality). In my case, it returns CamcorderProfile.QUALITY_4KDCI, and my phone only supports up to CamcorderProfile.QUALITY_2160P.

If I set it to CamcorderProfile.QUALITY_2160P, it works :) and the bitrate is correct! ("Created 3648 x 2736 @ 30 FPS H264 MOV LANDSCAPE_RIGHT 40.0 Mbps RecordingSession (with audio)!")

mrousavy, 2023-11-29:
> In my case, it returns CamcorderProfile.QUALITY_4KDCI, and my phone only supports up to CamcorderProfile.QUALITY_2160P

That is very weird - your Format is 4k (`{ ...other, "videoHeight": 2056, "videoWidth": 3648 }`), then I guess the format is wrong? It shouldn't even be 4k...

mrousavy, 2023-11-29:
Yup okay thanks! I guess then I should edit the `getAvailableCameraDevices()` function to not return sizes that don't have a CamcorderProfile.

So if your device supports a maximum CamcorderProfile of `QUALITY_2160P` - it does not support `QUALITY_4KDCI` or higher. Meaning:

* `QUALITY_2160P` = 3840 * 2160 = **8.294.400 pixels**
* `QUALITY_4KDCI` = 4096 * 2160 = **8.847.370 pixels**

Whereas your camera formats are:

* 3648 * 2056 = **7.500.288 pixels** (that's less than `QUALITY_2160P` so it's fine)
* 3648 * 3723 = **9.980.928 pixels** (that's more than `QUALITY_2160P ` so we can't record videos with that)

So I shouldn't even return the 3648 * 3723 format.

j-jonathan, 2023-11-29:
Actually, the format with { "videoHeight": 2736, "videoWidth": 3648 } works fine. I just checked the resolution of the recorded video, and it's correct (2736x3648), so we can keep this format.

However, for some reason, the call to CamcorderProfile.get(cameraIdInt, quality) with quality=QUALITY_4KDCI is not supported by my phone.

Maybe it's possible to call CamcorderProfile.hasProfile in the findClosestCamcorderProfileQuality to get the closest profile supported by the device 🤔

mrousavy, 2023-11-29:
wtf, that's weird. So it says it's not supported, but actually is kinda supported? Maybe it's not supported in full bit-rate?

PBeloborodov, 2023-12-02:
I am still getting error crashes. On Redmi note 112 device (android 13 )
"react-native-vision-camera":"^3.6.12",

`
  const startRecording = useCallback(() => {
    if (cameraRef.current) {
      try {
        cameraRef.current.stopRecording();
        cameraRef.current.startRecording({
          flash: useFlash ? 'on' : 'off',
          videoCodec: 'h264',
          videoBitRate: 'extra-low',
          onRecordingError: error => {
            console.error('Recording failed!', error);
            // onStoppedRecording();
          },
          onRecordingFinished: video => {
            navigation.navigate('photo-stories', {
              pathSource: video.path,
              typeSource: 'video',
            });
          },
        });
      } catch (e) {
        console.error('failed to start recording!', e, 'camera');
      } finally {
        console.log('endStartRecording');
      }
    }
  }, [cameraRef, navigation, useFlash]);`

`Recording failed! {"cause": {"message": "Error retrieving camcorder profile params", "stacktrace": "java.lang.RuntimeException: Error retrieving camcorder profile params
	at android.media.CamcorderProfile.native_get_camcorder_profile(Native Method)
	at android.media.CamcorderProfile.get(CamcorderProfile.java:528)`

System:
    OS: macOS 13.3
    CPU: (8) arm64 Apple M1
    Memory: 81.53 MB / 8.00 GB
    Shell: 5.9 - /bin/zsh
  Binaries:
    Node: 20.9.0 - /usr/local/bin/node
    Yarn: 1.22.17 - /opt/homebrew/bin/yarn
    npm: 10.1.0 - /opt/homebrew/bin/npm
    Watchman: 2023.11.06.00 - /opt/homebrew/bin/watchman
  Managers:
    CocoaPods: 1.12.1 - /opt/homebrew/bin/pod
  SDKs:
    iOS SDK:
      Platforms: DriverKit 22.4, iOS 16.4, macOS 13.3, tvOS 16.4, watchOS 9.4
    Android SDK: Not Found
  IDEs:
    Android Studio: 2021.1 AI-211.7628.21.2111.8193401
    Xcode: 14.3/14E222b - /usr/bin/xcodebuild
  Languages:
    Java: 11.0.16.1 - /usr/bin/javac
  npmPackages:
    @react-native-community/cli: Not Found
    react: Not Found
    react-native: Not Found
    react-native-macos: Not Found
  npmGlobalPackages:
    *react-native*: Not Found

arslan-Techno, 2023-12-04:
@mrousavy  @j-jonathan  is this issue resolved . I am still getting this issue in my realme 11 pro . In ios it work fine can you please help it is really urgent 

`import { View, Text,Image, Platform, KeyboardAvoidingView,Animated, NativeModules, NativeTouchEvent, Dimensions } from 'react-native'
import React ,{FC, useCallback, useEffect, useRef, useState} from 'react'
import { Modal } from 'react-native'
import { CameraRuntimeError, CameraDevice,PhotoFile, useCameraDevice, useCameraFormat, useFrameProcessor, VideoFile ,Camera,TakePhotoOptions, Point} from 'react-native-vision-camera'
import { StyleSheet } from 'react-native'
import Reanimated, { Extrapolate, interpolate, useAnimatedGestureHandler, useAnimatedProps, useSharedValue } from 'react-native-reanimated'
import { TouchableOpacity } from 'react-native' 
import Colors from '../constant/colors'
import CustomInput from './customInput'
import PhotoEditor from 'react-native-photo-editor'
import CustomText from './customText'
import { commonStyle } from '../constant/commonStyle'
import RNFS from 'react-native-fs';
import Entypo from 'react-native-vector-icons/Entypo'
import Ionicons from 'react-native-vector-icons/Ionicons'
import uuid from 'react-native-uuid';
import Feather from 'react-native-vector-icons/Feather'
import SimpleBtn from './simpleBtn'
import GradientBtn from './gradientBtn'
import { uploadStory } from '../services/api'
import { usePreferredCameraDevice } from '../assets/hooks/usePreferredCameraDevice'
import { CaptureButton } from './CaptureButton'
import Video from 'react-native-video'
import { showEditor } from 'react-native-video-trim';
import { NativeEventEmitter } from 'react-native'
// import VideoTrimmer from './videoTrimer'
// import Animated from 'react-native-reanimated'
// const VideoTrim = NativeModules.VideoTrim
const { RNPhotoEditor } = NativeModules
interface CameraComponentProps {
    navigation?:any,
    setShowCamera:(param:boolean)=>void
    showCamera:boolean
}
const { width, height, scale } = Dimensions.get('window');
const screenAspectRatio = height-150 / width

const generateRandomImagePath = (type?:'photo' | 'video' | '') => {
    const randomString =(uuid.v4()).toString();
    if(type == 'video'){
        return `${RNFS.DocumentDirectoryPath}/${randomString}.mov`;
    }else{
        return `${RNFS.DocumentDirectoryPath}/${randomString}.jpg`;
    }
    
};
const ReanimatedCamera = Reanimated.createAnimatedComponent(Camera)
Reanimated.addWhitelistedNativeProps({
  zoom: true,
})
const CameraComponent:FC<CameraComponentProps> = ({navigation,setShowCamera,showCamera}) => {
    const [capturedImage,setCapturedImage] = useState<any>(null)
    const camera =useRef<any>(null)
    const [VedioEditing,SetVideoEditing] = useState<boolean>(false)
    const [vedioTime,setVedioTime] = useState<number>()
    const [cameraPosition, setCameraPosition] = useState<'front' | 'back'>('back')
    // const device = useCameraDevice(cameraPosition)
    const [isCameraInitialized, setIsCameraInitialized] = useState(false)
    const [mediaType,setMediaType] =useState<'photo' | 'video' |''>('')
    const [filePath, setFilePath] = useState(null);
    const [editedImagePath, setEditedImagePath] = useState(generateRandomImagePath());
    const zoom = useSharedValue(0)
    const [camLocation,setCamLocation] = useState<any>()
    const [flashOn,setFlashon]=useState<true | false>(false)
    const [mode, setMode] = useState('picture');
    const imageSlideAnimation = useRef(new Animated.Value(0)).current;
    const videoSlideAnimation = useRef(new Animated.Value(0)).current;
    const isPressingButton = useSharedValue(false)
[... 934 lines cut ...]
  },
  "devDependencies": {
    "@babel/core": "^7.20.0",
    "@babel/preset-env": "^7.20.0",
    "@babel/runtime": "^7.20.0",
    "@react-native/eslint-config": "^0.72.2",
    "@react-native/metro-config": "^0.72.11",
    "@tsconfig/react-native": "^3.0.0",
    "@types/react": "^18.0.24",
    "@types/react-native-vector-icons": "^6.4.15",
    "@types/react-native-video": "^5.0.16",
    "@types/react-test-renderer": "^18.0.0",
    "babel-jest": "^29.2.1",
    "eslint": "^8.19.0",
    "jest": "^29.2.1",
    "metro-react-native-babel-preset": "0.76.8",
    "prettier": "^2.4.1",
    "react-test-renderer": "18.2.0",
    "typescript": "4.8.4"
  },
  "engines": {
    "node": ">=16"
  }
}
`

ZakirBangash, 2023-12-05:
We are stuck due to this issue. Would a temporary downgrade to a previous version be a viable solution to ensure functionality? I upgraded from v2 to v3 due to iPhone 15 crash

osmankhalid2005, 2023-12-11:
@ZakirBangash 

Use such settings with low resolution and frame rate, it will work:

const device = useCameraDevice('back');

const format = useCameraFormat(device, [
    { videoResolution: { width: 1280, height: 720 } },
  { fps: 30 }
])

MRafayyy, 2023-12-13:
> @ZakirBangash
> 
> Use such settings with low resolution and frame rate, it will work:
> 
> const device = useCameraDevice('back');
> 
> const format = useCameraFormat(device, [ { videoResolution: { width: 1280, height: 720 } }, { fps: 30 } ])

@ZakirBangash  Also set pixelFormat: 'native' and you are good to go.

ZakirBangash, 2023-12-13:
Thank you so much guys  💖
this suggested format works , but only works on android
so I only enable the format for Android then

rodrigodmpa, 2023-12-18:
> @ZakirBangash
> 
> Use such settings with low resolution and frame rate, it will work:
> 
> const device = useCameraDevice('back');
> 
> const format = useCameraFormat(device, [ { videoResolution: { width: 1280, height: 720 } }, { fps: 30 } ])

Any news? Here this solution did not work.

osmankhalid2005, 2023-12-19:
@rodrigodmpa I installed the latest version of the library and it started working.

zx3858359, 2024-01-05:
@rodrigodmpa 
Has this issue been resolved on your end? I also encountered the same problem here. And my version is already the latest 3.6.17.

"react-native-vision-camera": "^3.6.17"
"react": "18.2.0",
"react-native": "0.72.7",
 
 Galaxy S21  SM-G9910
 andriod 13

Elisebassett, 2024-01-05:
^ Same here. We're at the latest version and still getting this error

techied, 2024-01-09:
Also seeing this bug on VisionCamera 3.7.0 released yesterday

JustinHarshbarger, 2024-01-12:
Same problem here. I patched my local copy to only recommend the closest profile supported by my phone and that seemed to resolve my issue. Here is the version I'm using locally:

`
private fun findClosestCamcorderProfileQuality(resolution: Size): Int {
  //First we need to make sure that any profiles we use are supported by this camera:
  val allowedProfiles = (CamcorderProfile.QUALITY_QCIF..CamcorderProfile.QUALITY_8KUHD).filter { profile -> CamcorderProfile.hasProfile(profile)}

  // Iterate through all available CamcorderProfiles and find the one that matches the closest
  val targetResolution = resolution.width * resolution.height
  val closestProfile = allowedProfiles.minBy { profile ->
    val currentResolution = getResolutionForCamcorderProfileQuality(profile)
    return@minBy abs(currentResolution - targetResolution)
  }
  return closestProfile
}
`

whentao, 2024-01-12:
I get this bug on 3.7.1 for OnePlus

mrousavy, 2024-01-15:
Would've been great if you could've created a PR for this.

Anyways, I created a PR to fix this issue: #2389

closed 2024-01-15 by mrousavy

JustinHarshbarger, 2024-01-15:
> Would've been great if you could've created a PR for this.
> 
> Anyways, I created a PR to fix this issue: #2389

I have other priorities that are higher than working on contributing back to this project. I did plan on coming back to do a PR once my app issues were resolved. As a common courtesy I added a comment because I found something that seemed to work for me that wasn't thoroughly tested (and I needed to get my production app fixed that had issues with this and other libraries after IOS/Android updates). I'm also not familiar with the API this library was using, I just know how to run a debugger to set a break point on the error and to try some real-time changes to see if I can resolve the issue.

If you want to flame me for not creating a PR, feel free if that makes you feel better. Just know that responding to people in that way will make them less inclined to suggest a solution going forward.

mrousavy, 2024-01-16:
> I have other priorities that are higher than working on contributing back to this project. I did plan on coming back to do a PR once my app issues were resolved. As a common courtesy I added a [...]

wtf? lol

It's always the same - someone creates an issue, wants it fixed "asap" because it hurts their business, then someone posts a local patch in the comments, and the issue goes stale.
If you use open-source you are using something someone in the community has put hundreds or thousands of hours blood and sweat into, and then offered it to the public/you for free. To keep that project alive and working for the future you can either sponsor/support the devs/maintainers, or contribute back in some way shape or form - this would've been an ideal example.

> I have other priorities that are higher than working on contributing back to this project.

I have other priorities too than to help other people fix issues for their business. 

> If you want to flame me for not creating a PR, feel free if that makes you feel better. Just know that responding to people in that way will make them less inclined to suggest a solution going forward.

lmao, I simply said "Would've been great ....". If you get hurt over such a suggestion that's not my fault.

Doesn't matter anyways, but I think the PR would've taken 2 minutes longer than writing that comment.
I understand that time is valuable, and I understand that you are not required to create a PR, but still - it _would've been great._

Anyways, issue is now resolved since that PR is merged, that's all that matters.