margelo/react-native-vision-camera #2210: 🐛 Android crash - CameraDevice was already closed

mare95, 2023-11-24:
### What's happening?

App crashes on older android devices (eg Nokia 8, Android 8)
Can't reproduce this on my android device but I see lot's of crashes in Firebase Crashlytics

### Reproduceable Code

```tsx
const codeScanner = useCodeScanner({
    codeTypes: 'qr',
    onCodeScanned: barcodes => {
      if (barcodes) {
        onBarCodeRead(barcodes);
      }
    },
  });

  const device = useCameraDevice('back');

<Camera device={device} isActive codeScanner={codeScanner} />
```

### Relevant log output

```shell
Fatal Exception: java.lang.IllegalStateException
CameraDevice was already closed

com.mrousavy.camera.core.CameraSession.destroy (CameraSession.kt:159)
com.mrousavy.camera.core.CameraSession.access$destroy (CameraSession.kt:57)
com.mrousavy.camera.core.CameraSession$close$1.invokeSuspend (CameraSession.kt:101)
kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith (ContinuationImpl.kt:33)

kotlinx.coroutines.BuildersKt.runBlocking$default (Builders.kt:1)
com.mrousavy.camera.core.CameraSession.close (CameraSession.kt:99)
com.mrousavy.camera.CameraView.destroy (CameraView.kt:136)
com.mrousavy.camera.CameraViewManager.onDropViewInstance (CameraViewManager.kt:35)
com.mrousavy.camera.CameraViewManager.onDropViewInstance (CameraViewManager.kt:15)
com.facebook.react.uimanager.NativeViewHierarchyManager.dropView (NativeViewHierarchyManager.java:635)
com.swmansion.reanimated.layoutReanimation.ReanimatedNativeHierarchyManager.dropView (ReanimatedNativeHierarchyManager.java:412)
com.facebook.react.uimanager.NativeViewHierarchyManager.dropView (NativeViewHierarchyManager.java:646)
com.swmansion.reanimated.layoutReanimation.ReanimatedNativeHierarchyManager.dropView (ReanimatedNativeHierarchyManager.java:412)
com.facebook.react.uimanager.NativeViewHierarchyManager.dropView (NativeViewHierarchyManager.java:646)
com.swmansion.reanimated.layoutReanimation.ReanimatedNativeHierarchyManager.dropView (ReanimatedNativeHierarchyManager.java:412)
com.facebook.react.uimanager.NativeViewHierarchyManager.dropView (NativeViewHierarchyManager.java:646)
com.swmansion.reanimated.layoutReanimation.ReanimatedNativeHierarchyManager.dropView (ReanimatedNativeHierarchyManager.java:412)
com.facebook.react.uimanager.NativeViewHierarchyManager$1.onAnimationEnd (NativeViewHierarchyManager.java:491)
com.facebook.react.uimanager.layoutanimation.LayoutAnimationController.deleteView (LayoutAnimationController.java:203)
com.swmansion.reanimated.layoutReanimation.ReaLayoutAnimator.deleteView (ReanimatedNativeHierarchyManager.java:166)
com.facebook.react.uimanager.NativeViewHierarchyManager.manageChildren (NativeViewHierarchyManager.java:481)
com.swmansion.reanimated.layoutReanimation.ReanimatedNativeHierarchyManager.manageChildren (ReanimatedNativeHierarchyManager.java:329)
com.facebook.react.uimanager.UIViewOperationQueue$ManageChildrenOperation.execute (UIViewOperationQueue.java:217)
com.facebook.react.uimanager.UIViewOperationQueue$1.run (UIViewOperationQueue.java:915)
com.facebook.react.uimanager.UIViewOperationQueue.flushPendingBatches (UIViewOperationQueue.java:1026)
com.facebook.react.uimanager.UIViewOperationQueue$DispatchUIFrameCallback.doFrameGuarded (UIViewOperationQueue.java:1086)
com.facebook.react.uimanager.GuardedFrameCallback.doFrame (GuardedFrameCallback.java:29)
com.facebook.react.modules.core.ReactChoreographer$ReactChoreographerDispatcher.doFrame (ReactChoreographer.java:175)
com.facebook.react.modules.core.ChoreographerCompat$FrameCallback$1.doFrame (ChoreographerCompat.java:85)
android.view.Choreographer$CallbackRecord.run (Choreographer.java:1048)

com.android.internal.os.ZygoteInit.main (ZygoteInit.java:876)
```

### Camera Device

```json
I don't log info about the device so i can't provide it.
```

### Device

Nokia 8

### VisionCamera Version

3.6.8

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

spsaucier, 2023-12-02:
I also have cases of this occurring on Android 12 and 13 at least.

spsaucier, 2023-12-02:
Agreed, this PR likely causes the crash: https://github.com/mrousavy/react-native-vision-camera/pull/2174

spsaucier, 2023-12-04:
This is the cause of over 85% of all Android crashes in my production app.

DamienDulac, 2023-12-05:
Hi, I am also facing the same problem on version 3.6.10. 

I don't reproduce it myself, but I see logs uploaded to Firebase.

[image]

tahaipek, 2023-12-06:
https://github.com/mrousavy/react-native-vision-camera/issues/2243#issuecomment-1842920769

Can anyone else try this workaround?

spsaucier, 2023-12-06:
I'll give it a go.

mare95, 2023-12-11:
I'll give it a go as well since i have 500+ crashes in total. If would be good to reduce number of crashes for the beginning. Thanks for sharing your insights @tahaipek  @spsaucier 👍

spsaucier, 2023-12-11:
Now that it's been used for longer, I think it helps but is not a cure-all. Still seeing `Exception java.lang.IllegalStateException: Session has been closed; further changes are illegal.` as nearly half of all Android crashes.

tahaipek, 2023-12-11:
In 2 days of testing, I have not encountered any errors. For now, I have implemented it as a temporary solution. I am open to new suggestions until a permanent solution is developed.

mrousavy, 2024-01-05:
Hey! Thanks for reporting this issue.

I've been working the past days on making sure the Camera lifecycle is safely handled and released when needed to make sure this crash doesn't occur anymore.

I just created a PR for this - can you please this to see if that fixes the issue for you? https://github.com/mrousavy/react-native-vision-camera/pull/2339

If this fixes your issue, please consider [💖 **sponsoring me on GitHub** 💖](https://github.com/sponsors/mrousavy/) to support me / thank me for building VisionCamera and continuously improving it. 

If this does not fix your issue, please clone the repo, check out the branch `fix/blackscreen` (the PR above), run the Example app, reproduce your issue there and share the `adb logcat` logs with me (upload to pastebin or gist) so I can investigate this further.

Thank you! 🙏

closed 2024-01-08 by mrousavy

arslanafzal077, 2024-03-18:
on 3.9.1 it's happening again I'm just using simple camera
 <Camera
            ref={cameraRef}
            style={styles.camera}
            device={device}
            format={format}
            isActive={isFocused}
            photo
            zoom={scale}
            onInitialized={handleInitialized}
            onError={handleError}
          />

mrousavy, 2024-03-19:
Can you try if this is fixed in latest V4 beta?

Sakura-echos, 2024-07-01:
my react-native version is 0.72.3, when I update in latest V4, it cant run the program