margelo/react-native-vision-camera #2214: 🐛 App crash Xiaomi - POCO X3

ReactNativ, 2023-11-25:
### What's happening?

When the camera is opened, the app crashes

### Reproduceable Code

```tsx
<Camera
style={StyleSheet.absoluteFillObject}
ref={cameraRef}
device={device}
isActive={isCameraActive}
onInitialized={() => {}}
photo={true}
orientation={'portrait'}
onError={error => logToSentry(error, true, 'Camera runtime error in capture mode')}
enableHighQualityPhotos={true}
/>
```

### Relevant log output

```shell
{
	  "name": "unknown/unknown",
	  "_code": "unknown/unknown",
	  "_message": "[unknown/unknown] supportsCameraApi:2665: Unknown camera ID 0",
	  "_cause": {
		"cause": {
		  "stacktrace": "android.os.ServiceSpecificException: supportsCameraApi:2665: Unknown camera ID 0 (code 3)\n\tat android.os.Parcel.createExceptionOrNull(Parcel.java:2390)\n\tat android.os.Parcel.createException(Parcel.java:2360)\n\tat android.os.Parcel.readException(Parcel.java:2343)\n\tat android.os.Parcel.readException(Parcel.java:2285)\n\tat android.hardware.ICameraService$Stub$Proxy.supportsCameraApi(ICameraService.java:906)\n\tat android.hardware.camera2.CameraManager.supportsCameraApiLocked(CameraManager.java:1142)\n\tat android.hardware.camera2.CameraManager.supportsCamera2ApiLocked(CameraManager.java:1118)\n\tat android.hardware.camera2.CameraManager.getCameraCharacteristics(CameraManager.java:436)\n\tat com.mrousavy.camera.core.CameraSession.configureOutputs(CameraSession.kt:250)\n\tat com.mrousavy.camera.core.CameraSession.configure(CameraSession.kt:136)\n\tat com.mrousavy.camera.core.CameraSession$createPreviewOutput$1.invokeSuspend(CameraSession.kt:204)\n\tat kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)\n\tat kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:106)\n\tat android.os.Handler.handleCallback(Handler.java:938)\n\tat android.os.Handler.dispatchMessage(Handler.java:99)\n\tat android.os.Looper.loop(Looper.java:236)\n\tat android.os.HandlerThread.run(HandlerThread.java:67)\n",
		  "message": "supportsCameraApi:2665: Unknown camera ID 0"
		},
		"stacktrace": "java.lang.IllegalArgumentException: supportsCameraApi:2665: Unknown camera ID 0\n\tat android.hardware.camera2.CameraManager.throwAsPublicException(CameraManager.java:1089)\n\tat android.hardware.camera2.CameraManager.getCameraCharacteristics(CameraManager.java:495)\n\tat com.mrousavy.camera.core.CameraSession.configureOutputs(CameraSession.kt:250)\n\tat com.mrousavy.camera.core.CameraSession.configure(CameraSession.kt:136)\n\tat com.mrousavy.camera.core.CameraSession$createPreviewOutput$1.invokeSuspend(CameraSession.kt:204)\n\tat kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)\n\tat kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:106)\n\tat android.os.Handler.handleCallback(Handler.java:938)\n\tat android.os.Handler.dispatchMessage(Handler.java:99)\n\tat android.os.Looper.loop(Looper.java:236)\n\tat android.os.HandlerThread.run(HandlerThread.java:67)\nCaused by: android.os.ServiceSpecificException: supportsCameraApi:2665: Unknown camera ID 0 (code 3)\n\tat android.os.Parcel.createExceptionOrNull(Parcel.java:2390)\n\tat android.os.Parcel.createException(Parcel.java:2360)\n\tat android.os.Parcel.readException(Parcel.java:2343)\n\tat android.os.Parcel.readException(Parcel.java:2285)\n\tat android.hardware.ICameraService$Stub$Proxy.supportsCameraApi(ICameraService.java:906)\n\tat android.hardware.camera2.CameraManager.supportsCameraApiLocked(CameraManager.java:1142)\n\tat android.hardware.camera2.CameraManager.supportsCamera2ApiLocked(CameraManager.java:1118)\n\tat android.hardware.camera2.CameraManager.getCameraCharacteristics(CameraManager.java:436)\n\t... 9 more\n",
		"message": "supportsCameraApi:2665: Unknown camera ID 0"
	  }
	}
```

### Camera Device

```json
{}
```

### Device

Xiaomi - POCO X3

### VisionCamera Version

3.6.6

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

bahador14, 2023-11-25:
i have this problem too

Kavosh-m, 2023-11-25:
I have the same issue

cseluzha, 2023-12-12:
Same case for the version 3.6.14, worklets-core: 0.2.4, and react native to 0.72.7
on this devices:
-  Samsung Galaxy A71
-  Xiaomi Poco X3 NFC

The logs are:

Crashed: Thread: SIGSEGV  0x00000073588b4430
#00 pc 0x5d430 libVisionCamera.so (BuildId: 0297815e9897f58169c69de9d43c84c9df99d8d8)
#01 pc 0xe8238 libc++_shared.so (BuildId: fcc246cbb373c5edee634b117c4b4564cc9becf3)
#02 pc 0xe7e6c libc++_shared.so (BuildId: fcc246cbb373c5edee634b117c4b4564cc9becf3)
#03 pc 0xe3f50 libc++_shared.so (BuildId: fcc246cbb373c5edee634b117c4b4564cc9becf3)
#04 pc 0xe3da4 libc++_shared.so (BuildId: fcc246cbb373c5edee634b117c4b4564cc9becf3)
#05 pc 0x482ec libVisionCamera.so (BuildId: 0297815e9897f58169c69de9d43c84c9df99d8d8)
#06 pc 0x487f4 libVisionCamera.so (BuildId: 0297815e9897f58169c69de9d43c84c9df99d8d8)
#07 pc 0x3fd64 libVisionCamera.so (BuildId: 0297815e9897f58169c69de9d43c84c9df99d8d8)
#08 pc 0x3c1b4 libVisionCamera.so (BuildId: 0297815e9897f58169c69de9d43c84c9df99d8d8)
#09 pc 0x3bf88 libVisionCamera.so (BuildId: 0297815e9897f58169c69de9d43c84c9df99d8d8)
#10 pc 0x41094 libVisionCamera.so (BuildId: 0297815e9897f58169c69de9d43c84c9df99d8d8)
#11 pc 0x39f70 libVisionCamera.so (BuildId: 0297815e9897f58169c69de9d43c84c9df99d8d8)

closed 2024-01-15 by mrousavy