margelo/react-native-vision-camera #2223: 🐛 Black screen on camera

Selcukk70, 2023-11-27:
### What's happening?

While the app is trying to open the camera I am getting the following error:

11-27 13:37:00.659 20287 20369 E ReactNativeJS:   _message: '[session/camera-has-been-disconnected] The given Camera device (id: 0) has been disconnected! Error: UNKNOWN_CAMERA_DEVICE_ERROR',

### Reproduceable Code

```tsx
const device = useCameraDevice('back');
const isFocused = useIsFocused();

<Camera
   style={StyleSheet.absoluteFill}
   device={device}
   isActive={isFocused}
     />
```

### Relevant log output

```shell
11-27 13:37:00.084  1890  7359 D ConnectivityService: ConnectivityService NetworkRequestInfo binderDied(NetworkRequest [ LISTEN id=103, [ Capabilities: INTERNET&NOT_RESTRICTED&TRUSTED&FOREGROUND] ], android.os.BinderProxy@ea4e176)
11-27 13:37:00.085  1890  2282 E ConnectivityService: RemoteException caught trying to send a callback msg for NetworkRequest [ LISTEN id=103, [ Capabilities: INTERNET&NOT_RESTRICTED&TRUSTED&FOREGROUND] ]
11-27 13:37:00.598   889 21353 E mm-camera: <MCT   ><ERROR> 117: mct_bus_sof_thread_run: FATAL Session 3: SOF Freeze! Sending error message
11-27 13:37:00.598   889 21353 I mm-camera: <MCT   >< INFO> 158: mct_bus_sof_thread_run: Sending HW_ERROR from MCT on session =3
11-27 13:37:00.599   889 21285 E mm-camera: <MCT   ><ERROR> 823: mct_controller_send_cb: FATAL: Sending HW_Error
11-27 13:37:00.599   889 21306 E QCamera : <HAL><ERROR> camEvtHandle: 905: Fatal, camera daemon died
11-27 13:37:00.603   889  1878 E QCamera : <HAL><ERROR> processCaptureRequest: 6131: Unblocked on timeout!!!!
11-27 13:37:00.603   889  1878 E CamDevSession@3.2-impl: processOneCaptureRequest: HAL process_capture_request call failed!
11-27 13:37:00.603  1083 21314 E Camera3-Device: Camera 0: sendRequestsBatch: RequestThread: Unable to submit capture request 3 to HAL device: Function not implemented (-38)
11-27 13:37:00.607 20287 20486 E CameraManager: Camera 0: Error! 4
11-27 13:37:00.609 20287 20486 E CameraSession: Camera Device android.hardware.camera2.impl.CameraDeviceImpl@a2fca39 has been disconnected!
11-27 13:37:00.609 20287 20486 E CameraSession: com.mrousavy.camera.core.CameraDisconnectedError: [session/camera-has-been-disconnected] The given Camera device (id: 0) has been disconnected! Error: UNKNOWN_CAMERA_DEVICE_ERROR
11-27 13:37:00.609 20287 20486 E CameraSession:         at com.mrousavy.camera.extensions.CameraManager_openCameraKt$openCamera$2$callback$1.onError(CameraManager+openCamera.kt:49)
11-27 13:37:00.609 20287 20486 E CameraSession:         at android.hardware.camera2.impl.CameraDeviceImpl$CameraDeviceCallbacks$1.run(CameraDeviceImpl.java:1791)
11-27 13:37:00.609 20287 20486 E CameraSession:         at android.os.Handler.handleCallback(Handler.java:789)
11-27 13:37:00.609 20287 20486 E CameraSession:         at android.os.Handler.dispatchMessage(Handler.java:98)
11-27 13:37:00.609 20287 20486 E CameraSession:         at android.os.Looper.loop(Looper.java:164)
11-27 13:37:00.609 20287 20486 E CameraSession:         at android.os.HandlerThread.run(HandlerThread.java:65)
11-27 13:37:00.609 20287 20486 E CameraView: invokeOnError(...):
11-27 13:37:00.609 20287 20486 W System.err: com.mrousavy.camera.core.CameraDisconnectedError: [session/camera-has-been-disconnected] The given Camera device (id: 0) has been disconnected! Error: UNKNOWN_CAMERA_DEVICE_ERROR
11-27 13:37:00.609 20287 20486 W System.err:    at com.mrousavy.camera.extensions.CameraManager_openCameraKt$openCamera$2$callback$1.onError(CameraManager+openCamera.kt:49)
11-27 13:37:00.610 20287 20486 W System.err:    at android.hardware.camera2.impl.CameraDeviceImpl$CameraDeviceCallbacks$1.run(CameraDeviceImpl.java:1791)
11-27 13:37:00.610 20287 20486 W System.err:    at android.os.Handler.handleCallback(Handler.java:789)
11-27 13:37:00.610 20287 20486 W System.err:    at android.os.Handler.dispatchMessage(Handler.java:98)
11-27 13:37:00.610 20287 20486 W System.err:    at android.os.Looper.loop(Looper.java:164)
11-27 13:37:00.610 20287 20486 W System.err:    at android.os.HandlerThread.run(HandlerThread.java:65)
11-27 13:37:00.659 20287 20369 E ReactNativeJS: 'Camera.onError(session/camera-has-been-disconnected): [session/camera-has-been-disconnected] The given Camera device (id: 0) has been disconnected! Error: UNKNOWN_CAMERA_DEVICE_ERROR', { [session/camera-has-been-disconnected: [session/camera-has-been-disconnected] The given Camera device (id: 0) has been disconnected! Error: UNKNOWN_CAMERA_DEVICE_ERROR]
11-27 13:37:00.659 20287 20369 E ReactNativeJS:   name: 'session/camera-has-been-disconnected',
11-27 13:37:00.659 20287 20369 E ReactNativeJS:   _code: 'session/camera-has-been-disconnected',
11-27 13:37:00.659 20287 20369 E ReactNativeJS:   _message: '[session/camera-has-been-disconnected] The given Camera device (id: 0) has been disconnected! Error: UNKNOWN_CAMERA_DEVICE_ERROR',
11-27 13:37:00.659 20287 20369 E ReactNativeJS:   _cause: undefined }
11-27 13:37:00.721  1083 21314 W CallStack: update: Failed to unwind callstack.
11-27 13:37:00.722  1083 21314 E ProcessCallStack: getThreadName: Failed to open /proc/self/task/21315/comm
11-27 13:37:00.722  1083 21314 D CameraTraces: Process trace saved. Use dumpsys media.camera to view.
11-27 13:37:00.722  1083 21314 W Camera3-OutputStream: A frame is dropped for stream 2
11-27 13:37:00.722  1083  1083 E Camera3-Device: Camera 0: clearStreamingRequest: Device has encountered a serious error
11-27 13:37:00.722  1083  1083 I Camera3-Device: disconnect: E
11-27 13:37:00.722  1083  1083 E Camera3-Device: Camera 0: disconnect: Shutting down in an error state
11-27 13:37:00.722  1083  1083 I CameraLatencyHistogram: ProcessCaptureRequest latency histogram (3) samples:
11-27 13:37:00.722  1083  1083 I CameraLatencyHistogram:        40     80    120    160    200    240    280    320    360    inf (max ms)
11-27 13:37:00.722  1083  1083 I CameraLatencyHistogram:      33.33   33.33   33.33   0.00   0.00   0.00   0.00   0.00   0.00   0.00 (%)
11-27 13:37:00.723  1083 21314 W Camera3-OutputStream: A frame is dropped for stream 1
11-27 13:37:00.723   889  1878 E CamDevSession@3.2-impl: close: trying to close while there are still 6 inflight buffers!
11-27 13:37:00.723   889  1878 E CamDevSession@3.2-impl: close: trying to close while there are still 3 inflight  RAW boost overrides!
11-27 13:37:00.723   889  1878 I CameraKpiTag: [15745974ms] close_camera_device start
11-27 13:37:00.723   889  1878 I QCameraHalWatchdog: Starting Watchdog Thread (timeout 5000 msec)...
11-27 13:37:00.723   889  1878 I QCamera : <HAL><INFO> close_camera_device: 13573: [KPI Perf]: E camera id 0
11-27 13:37:00.724   913   962 E ANDR-PERF-MPCTL: Invalid profile no. 0, total profiles 0 only
[... 214 lines cut ...]
11-27 13:36:55.301 20287 20369 I ReactNativeJS:         "videoHeight": 2160,
11-27 13:36:55.301 20287 20369 I ReactNativeJS:         "maxFps": 30,
11-27 13:36:55.301 20287 20369 I ReactNativeJS:         "photoHeight": 3024
11-27 13:36:55.301 20287 20369 I ReactNativeJS:       },
```

### Device

Motorola g5

### VisionCamera Version

3.6.10

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

Selcukk70, 2023-11-27:
Removed the following line(446) :
captureRequest.setZoom(config.zoom, cameraCharacteristics)
in node_modules\react-native-vision-camera\android\src\main\java\com\mrousavy\camera\core\CameraSession.kt

Which fixed my camera problem but when scanning a qr code the app crashes(sometimes). 
Yes I can reproduce.

MicaelaWaigel, 2023-11-27:
Same problem here, in devices like Motorola g6 plus (Android 9) (always black screen), Samsung Galaxy s21 FE (Android 13) but in this only when bring the app from background y open the camera again.

`Camera.onError(session/camera-has-been-disconnected): [session/camera-has-been-disconnected] The given Camera device (id: 0) has been disconnected! Error: CAMERA_IS_DISABLED_BY_ANDROID [session/camera-has-been-disconnected: [session/camera-has-been-disconnected] The given Camera device (id: 0) has been disconnected! Error: CAMERA_IS_DISABLED_BY_ANDROID]`

rodgomesc, 2023-11-27:
@MicaelaWaigel i think this is because we are not propertly closing the camera session when it goes to background, need to double check, if someone have more time than me to test this theory please go ahead

MicaelaWaigel, 2023-11-27:
> @MicaelaWaigel i think this is because we are not propertly closing the camera session when it goes to background, need to double check, if someone have more time than me to test this theory please go ahead

We set the isActive according the isFocused and appState, so when the app is in background after a while the error `Camera.onError(session/camera-has-been-disconnected): [session/camera-has-been-disconnected] The given Camera device (id: 0) has been disconnected! Error: CAMERA_IS_DISABLED_BY_ANDROID` occurs. 
When the appState changed to active, then the error is `Camera.onError(unknown/unknown): [unknown/unknown] CameraDevice was already closed [unknown/unknown: [unknown/unknown] CameraDevice was already closed]
`

mrousavy, 2024-01-05:
Hey! Thanks for reporting this issue.

I've been working the past days on making sure the Camera lifecycle is safely handled and released when needed to make sure this crash doesn't occur anymore.

I just created a PR for this - can you please this to see if that fixes the issue for you? https://github.com/mrousavy/react-native-vision-camera/pull/2339

If this fixes your issue, please consider [💖 **sponsoring me on GitHub** 💖](https://github.com/sponsors/mrousavy/) to support me / thank me for building VisionCamera and continuously improving it. 

If this does not fix your issue, please clone the repo, check out the branch `fix/blackscreen` (the PR above), run the Example app, reproduce your issue there and share the `adb logcat` logs with me (upload to pastebin or gist) so I can investigate this further.

Thank you! 🙏

closed 2024-01-08 by mrousavy