margelo/react-native-vision-camera #2227: ❓ Frame Processor threw an error: Property 'xyz' doesn't exist

fazzysyed, 2023-11-28:
### Question

I have created a frame process plugin in Java for Android, but I am encountering an error. Can someone tell me what I am doing wrong, please?

### What I tried

// @ts-nocheck

import {VisionCameraProxy, Frame} from 'react-native-vision-camera';

const plugin = VisionCameraProxy.initFrameProcessorPlugin('xyz');

/**
 * Scans faces.
 *
 *
 */
export function xyz(frame: Frame): string[] {
  'worklet';
  if (plugin == null) {
    console.log('Failed to load Frame Processor Plugin "scanFaces"!');
  } else {
    console.log(plugin);
    return plugin.call(frame);
  }

}

// @ts-nocheck

import React, {useEffect, useState} from 'react';
import {
  StyleSheet,
  Text,
  useWindowDimensions,
  View,
  NativeModule,
  NativeModules,
  Button,
} from 'react-native';

import {
  Camera,
  CameraDevice,
  useFrameProcessor,
} from 'react-native-vision-camera';
import {runOnJS} from 'react-native-reanimated';

import {xyz} from '../../Plugins/TensorFlow';

interface Props {
  device: CameraDevice;
}

const ObjectDetector: React.FC<Props> = ({device}) => {
  const {DetectorModule} = NativeModules;

  const [image, setImage] = useState(null);

  const frameProcessor = useFrameProcessor(frame => {
    'worklet';

    // cos;
    const faces = xyz(frame);
    console.log(`Testing: ${faces}`);
  }, []);

  return (
    <View style={[StyleSheet.absoluteFill]}>
      <Camera
        frameProcessorFps={5}
        frameProcessor={frameProcessor}
        style={StyleSheet.absoluteFill}
        device={device}
        isActive={true}
      />

      {/* {objects?.map(
        ({top, left, width, height, labels}: , index: number) => (
          <View
            key={`${index}`}
            style={[styles.detectionFrame, {top, left, width, height}]}>
            <Text style={styles.detectionFrameLabel}>
              {labels
                .map(label => `${label.label} (${label.confidence})`)
                .join(',')}
            </Text>
          </View>
        ),
      )} */}
    </View>
  );
};

const styles = StyleSheet.create({
  detectionFrame: {
    position: 'absolute',
    borderWidth: 1,
    borderColor: '#00ff00',
    zIndex: 9,
  },
  detectionFrameLabel: {
    backgroundColor: 'rgba(0, 255, 0, 0.25)',
  },
});

export default ObjectDetector;

### VisionCamera Version

3.6.10

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

bglgwyng, 2023-11-28:
Please check if `const plugin = VisionCameraProxy.initFrameProcessorPlugin('xyz');` succeeded.
If not, you might need to add frame processor initialization logic.
And (this page)[https://react-native-vision-camera.com/docs/guides/frame-processors-plugins-android] would be helpful.

fazzysyed, 2023-11-28:
i already did the initialization 

package com.testingapp.xyzframeprocessor;

import android.media.Image;
import android.util.Log;

import androidx.annotation.NonNull;
import androidx.annotation.Nullable;

import com.mrousavy.camera.frameprocessor.Frame;
import com.mrousavy.camera.frameprocessor.FrameProcessorPlugin;

import java.util.Map;

public class XyzFrameProcessorPlugin extends FrameProcessorPlugin {

  public XyzFrameProcessorPlugin(@Nullable Map<String, Object> options) {
    super(options);
  }

package com.testingapp.xyzframeprocessor;

import androidx.annotation.NonNull;

import com.facebook.react.ReactPackage;
import com.facebook.react.bridge.NativeModule;
import com.facebook.react.bridge.ReactApplicationContext;
import com.facebook.react.uimanager.ViewManager;
import com.mrousavy.camera.frameprocessor.FrameProcessorPlugin;
import com.mrousavy.camera.frameprocessor.FrameProcessorPluginRegistry;

import java.util.Collections;
import java.util.List;

public class XyzFrameProcessorPluginPackage implements ReactPackage {
  static {
    FrameProcessorPluginRegistry.addFrameProcessorPlugin(
            "fazzy",
            options -> new XyzFrameProcessorPlugin(options)
    );
  }

  @NonNull
  @Override
  public List<NativeModule> createNativeModules(@NonNull ReactApplicationContext reactContext) {
//    FrameProcessorPluginRegistry.addFrameProcessorPlugin(
//            "xyz",
//            options -> new XyzFrameProcessorPlugin(options)
//    );
//
    return Collections.emptyList();
  }

  @NonNull
  @Override
  public List<ViewManager> createViewManagers(@NonNull ReactApplicationContext reactContext) {
    return Collections.emptyList();
  }
}

  @Nullable
  @Override
  public Object callback(@NonNull Frame frame, @Nullable Map<String, Object> arguments) {

    if (arguments == null) return null;
    Image image = frame.getImage();

    Log.d("ExamplePlugin", image.getWidth() + " x " + image.getHeight() + " Image with format #" + image.getFormat() + ". Logging " + arguments.size() + " parameters:");

//    YoloV5Classifier detector = DetectorFactory.getDetector(, "best-fp16.tflite");
    // code goes here
    return null;
  }
}

mrousavy, 2023-11-28:
Please format your code.

fazzysyed, 2023-11-28:
okay I solved the problem 

const frameProcessorHandler = (frame:Frame) => {
    'worklet';

    if (plugin == null)
      throw new Error('Failed to load Frame Processor Plugin "scanFaces"!');

    return plugin.call(frame);
  };

I have used this function inside the component very the camera logic is and it worked.

but the problem I am facing is I am trying to convert the frame to bitmap to pass it to my yolov5 object detection native code
and now stuck with this error
**Exception in HostFunction: java.lang.ArrayIndexOutOfBoundsException: length=0; index=0**

`public class XyzFrameProcessorPlugin extends FrameProcessorPlugin {

  public XyzFrameProcessorPlugin(@Nullable Map<String, Object> options) {
    super(options);
  }

  @Nullable
  @Override
  public Object callback(@NonNull Frame frame, @Nullable Map<String, Object> arguments) {

    Image image = frame.getImage();
    Log.i("hello" ,"image planes length  "+convertToBitmap(image));

    return "TEstingArray";
  }

  public Bitmap convertToBitmap(Image image) {
    if (image == null) {
      return null;
    }

    Image.Plane[] planes = image.getPlanes();
    ByteBuffer buffer = planes[0].getBuffer();

    int width = image.getWidth();
    int height = image.getHeight();
    int pixelStride = planes[0].getPixelStride();
    int rowStride = planes[0].getRowStride();
    int rowPadding = rowStride - pixelStride * width;

    Bitmap bitmap = Bitmap.createBitmap(width + rowPadding / pixelStride, height, Bitmap.Config.ARGB_8888);
    bitmap.copyPixelsFromBuffer(buffer);

    return bitmap;
  }
}`

mrousavy, 2023-11-28:
Probably `getPlanes()` returns `null` because you are using ImageFormat PRIVATE. Pass `pixelFormat="yuv"` to the Camera

closed 2023-11-28 by mrousavy

fazzysyed, 2023-11-29:
i have passed  pixelFormat="yuv" to the Camera and now the camera isn't working.
```
   <Camera
        frameProcessorFps={5}
        frameProcessor={frameProcessor}
        style={StyleSheet.absoluteFill}
        device={device}
        isActive={true}
        pixelFormat="yuv"
      />
```

 "react-native-vision-camera": "3.6.10",
      

Camera.onError(session/cannot-create-session): [session/cannot-create-session] Failed to create a Camera Session for Camera 0! [session/cannot-create-session: [session/cannot-create-session] Failed to create a Camera Session for Camera 0!]