margelo/react-native-vision-camera #2235: 🐛 [Android] - Recording error (Error retrieving camcorder profile params)

rossicler-hostalky, 2023-11-29:
### What's happening?

When trying to record a video, `onRecordingError` throw an error. The error is:
```
{"cause": {"message": "Error retrieving camcorder profile params", "stacktrace": "java.lang.RuntimeException: Error retrieving camcorder profile params
```

### Reproduceable Code

```tsx
const cameraBack = useCameraDevice("back");
const cameraFront = useCameraDevice("front");
const [device, setDevice] = useState(cameraBack ?? cameraFront);

...

const toggleCamera = () => {
  if (device && device.position === "back" && cameraFront) {
    setDevice(cameraFront);
  } else if (device && device.position === "front" && cameraBack) {
    setDevice(cameraBack);
  }
};

const takeRecording = async () => {
  cameraRef?.current.startRecording({
    flash: flashMode,
    onRecordingFinished: (video) => {
      if (onCaptureVideo) onCaptureVideo(video);
      if (onClose) onClose();
    },
    onRecordingError: (err) => {
      logger("onRecordingError: ", err);
    }
  });
};

...
<Camera
  ref={cameraRef}
  className="flex-1"
  device={device}
  isActive={visible}
  photo
  video={true}
  audio={true}
/>
```
```

### Relevant log output

```shell
onRecordingError:  {"cause": {"message": "Error retrieving camcorder profile params", "stacktrace": "java.lang.RuntimeException: Error retrieving camcorder profile params
	at android.media.CamcorderProfile.native_get_camcorder_profiles(Native Method)
	at android.media.CamcorderProfile.getAll(CamcorderProfile.java:605)
	at com.mrousavy.camera.extensions.RecordingSession_getRecommendedBitRateKt.getRecommendedBitRate(RecordingSession+getRecommendedBitRate.kt:32)
	at com.mrousavy.camera.core.RecordingSession.getBitRate(RecordingSession.kt:132)
	at com.mrousavy.camera.core.RecordingSession.<init>(RecordingSession.kt:37)
	at com.mrousavy.camera.core.CameraSession.startRecording(CameraSession.kt:532)
	at com.mrousavy.camera.CameraView_RecordVideoKt.startRecording(CameraView+RecordVideo.kt:44)
	at com.mrousavy.camera.CameraViewModule$startRecording$1.invokeSuspend(CameraViewModule.kt:91)
	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
	at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:106)
	at kotlinx.coroutines.scheduling.CoroutineScheduler.runSafely(CoroutineScheduler.kt:570)
	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.executeTask(CoroutineScheduler.kt:750)
	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.runWorker(CoroutineScheduler.kt:677)
	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.run(CoroutineScheduler.kt:664)
"}, "code": "capture/unknown", "message": "An unknown error occurred while trying to start a video recording! Error retrieving camcorder profile params", "userInfo": null}
```

### Camera Device

```json
{
  "hardwareLevel": "limited",
  "minExposure": -300,
  "neutralZoom": 1,
  "supportsFocus": true,
  "supportsLowLightBoost": true,
  "hasTorch": true,
  "supportsRawCapture": false,
  "sensorOrientation": "landscape-right",
  "maxZoom": 4,
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "isMultiCam": false,
  "id": "0",
  "position": "back",
  "minZoom": 1,
  "hasFlash": true,
  "maxExposure": 300,
  "name": "BACK (0)"
}
```

### Device

Samsung Galaxy A03 Core

### VisionCamera Version

3.6.12

### Can you reproduce this issue in the VisionCamera Example app?

No, I cannot reproduce the issue in the Example app

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

rossicler-hostalky, 2023-11-29:
The error is happening inside `.../react-native-vision-camera/android/src/main/java/com/mrousavy/camera/extensions/RecordingSession+getRecommendedBitRate.kt`. In this part of the code:
```
fun RecordingSession.getRecommendedBitRate(fps: Int, codec: VideoCodec, hdr: Boolean): Int {
  try {
    val targetResolution = size
    val encoder = codec.toVideoEncoder()
    val bitDepth = if (hdr) 10 else 8
    val quality = findClosestCamcorderProfileQuality(targetResolution)
    Log.i("CamcorderProfile", "Closest matching CamcorderProfile: $quality")

    var recommendedProfile: RecommendedProfile? = null

    if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.S) {
      val profiles = CamcorderProfile.getAll(cameraId, quality)
      if (profiles != null) {
        val best = profiles.videoProfiles.minBy { abs(it.width * it.height - targetResolution.width * targetResolution.height) }

        recommendedProfile = RecommendedProfile(
          best.bitrate,
          best.codec,
          if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.TIRAMISU) best.bitDepth else 8,
          best.frameRate
        )
      }
    }

    if (recommendedProfile == null) {
      val cameraIdInt = cameraId.toIntOrNull()
      val profile = if (cameraIdInt != null) {
        CamcorderProfile.get(cameraIdInt, quality)
      } else {
        CamcorderProfile.get(quality)
      }
      recommendedProfile = RecommendedProfile(
        profile.videoBitRate,
        profile.videoCodec,
        8,
        profile.videoFrameRate
      )
    }

    var bitRate = recommendedProfile.bitRate.toDouble()
    // the target bit-rate is for e.g. 30 FPS, but we use 60 FPS. up-scale it
    bitRate = bitRate / recommendedProfile.fps * fps
    // the target bit-rate might be in 8-bit SDR, but we record in 10-bit HDR. up-scale it
    bitRate = bitRate / recommendedProfile.bitDepth * bitDepth
    if (recommendedProfile.codec == VideoEncoder.H264 && encoder == VideoEncoder.HEVC) {
      // the target bit-rate is for H.264, but we use H.265, which is 20% smaller
      bitRate *= 0.8
    } else if (recommendedProfile.codec == VideoEncoder.HEVC && encoder == VideoEncoder.H264) {
      // the target bit-rate is for H.265, but we use H.264, which is 20% larger
      bitRate *= 1.2
    }
    return bitRate.toInt()
  } catch (e: Throwable) {
    // some Samsung phones throws an error here, returns 8 by default in these cases
    return 8;
  }
}
```

The try/catch was added by me as temporary workaround just to get it working, but it's not a good solution to run in production since when it catches an error it will always return a bitrate of 8, which will probably compromise the picture/video quality.

The problem seems to be anytime it tries to run either `CamcorderProfile.getAll(...)` or `CamcorderProfile.get(...)`.

Brianwebb22, 2023-11-29:
I believe this is being tracked and worked on in #2208 . Been following closely as i've been running into this also.

rossicler-hostalky, 2023-11-29:
> #2208

Oh true, I didn't knew that issue had been opened again with a different error than the original one posted.
I'll close this one and keep track on that one, thanks.

closed 2023-11-29 by rossicler-hostalky

arslan-Techno, 2023-12-04:
> The error is happening inside `.../react-native-vision-camera/android/src/main/java/com/mrousavy/camera/extensions/RecordingSession+getRecommendedBitRate.kt`. In this part of the code:
>
> ```
> fun [...]

worked

osmankhalid2005, 2023-12-11:
Use such settings with low resolution and frame rate, it will work:

const device = useCameraDevice(cameraType);

const format = useCameraFormat(device, [
    { videoResolution: { width: 1280, height: 720 } },
  { fps: 30 }
])

Robogram, 2023-12-12:
> const format = useCameraFormat(device, [
> { videoResolution: { width: 1280, height: 720 } },
> { fps: 30 }
> ])

Crash on stopRecording

MRafayyy, 2023-12-13:
> Use such settings with low resolution and frame rate, it will work:
> 
> const device = useCameraDevice(cameraType);
> 
> const format = useCameraFormat(device, [ { videoResolution: { width: 1280, height: 720 } }, { fps: 30 } ])

Not specifying the format led to this error 'Error retrieving camcorder profile params'. However, after applying the settings below (also specifying pixelFormat), it is working now.

const device = useCameraDevice('back')
    const format = useCameraFormat(device, [
        { videoResolution: { width: 1280, height: 720 },
         pixelFormat: 'native' }, 
        { fps: 30 }
        ])
        
If vid resolution or pixel format won't be compatible then you might get this error: 
 ERROR  {"cause": {"message": "prepare failed.", "stacktrace": "java.io.IOException: prepare failed.
        at android.media.MediaRecorder._prepare(Native Method)
        at android.media.MediaRecorder.prepare(MediaRecorder.java:1320)
        at com.mrousavy.camera.core.RecordingSession.start(RecordingSession.kt:91)
        at com.mrousavy.camera.core.CameraSession.startRecording(CameraSession.kt:533)
        at com.mrousavy.camera.CameraView_RecordVideoKt.startRecording(CameraView+RecordVideo.kt:34)
        at com.mrousavy.camera.CameraViewModule$startRecording$1.invokeSuspend(CameraViewModule.kt:91)
        at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
        at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:106)
        at kotlinx.coroutines.scheduling.CoroutineScheduler.runSafely(CoroutineScheduler.kt:570)
        at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.executeTask(CoroutineScheduler.kt:750)
        at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.runWorker(CoroutineScheduler.kt:677)
        at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.run(CoroutineScheduler.kt:664)
"}, "code": "capture/unknown", "message": "An unknown error occurred while trying to start a video recording! prepare failed.", "userInfo": null}

mrousavy, 2024-01-15:
Fixed in https://github.com/mrousavy/react-native-vision-camera/pull/2389! :)