margelo/react-native-vision-camera #2236: Cade Scanner doesn't scan codes on the left side of screen❓

SharkMang, 2023-11-29:
### Question

I update my react-native app:
react-native 0.69.1 -> 0.72
react-native-vision-camera 2.14.1 -> 3.6.12 

Before updates I use vision-camera-code-scanner for scanning barcodes and scanner scan barcodes on full screen (and the issue was to create custom reaction zone)
But after update the I set the regionOfInterest and it's not work.
Also I found that on the left side of the screen zone where scanner not work (not matter use regionOfInterest or not)

The device.sensorOrientation landscape-right 
The screen orientation portrait 

P.S.  regionOfInterest - not work

[image]
in red not scanned zone 

### What I tried

I tried to set the orientation in props of Camera not help.
The start position of screen in top left side but  start position of camera frame is top right side.

Frame: {"height": 720, "width": 1280} 

Code usage: 
```
const codeScanner = useCodeScanner({
    codeTypes: [
      'code-128',
      'code-39',
      'code-93',
      'codabar',
      'ean-13',
      'ean-8',
      'itf',
      'upc-e',
      'qr',
      'pdf-417',
      'aztec',
      'data-matrix',
    ],
    onCodeScanned: (codes, frame) => {
      console.log('🚀 ~ ScanBarcode ~ frame:', frame);

      console.log('~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~');
      console.log('codes', codes.length);
      console.log('code', codes?.[0]);
      console.log('~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~');
    },
    regionOfInterest: {
      x: layoutPosition.x,
      y: layoutPosition.y,
      width: layoutPosition.width,
      height: layoutPosition.height,
    },
  });
  
   <Camera
      style={StyleSheet.absoluteFill}
      device={device}
      isActive={isFocused}
      audio={false}
      orientation="portrait"
      codeScanner={codeScanner}
    />
```
  
  

### VisionCamera Version

3.6.12

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

SharkMang, 2023-11-30:
@mrousavy

SharkMang, 2023-12-13:
Still facing this issue

mrousavy, 2024-01-15:
Hey! I think this has already been fixed. If not, feel free to re-open :)

closed 2024-01-15 by mrousavy