margelo/react-native-vision-camera #2237: 🐛 Tap to Focus Doesn't work on Android

NuraQ, 2023-11-30:
### What's happening?

Whenever I try to call focus function  I get the following error message in the catch statement "session/camera-not-ready: [session/camera-not-ready] The Camera is not ready yet! Wait for the onInitialized() callback!]" . Although Camera is ready onInitialized is actually triggered

### Reproduceable Code

```tsx
const device = useCameraDevice('back');
....
    const touchToFocus = async (event) => {
        if (!device.supportsFocus || !isCameraFocusReady.current) return;
        let point = {
            x: Math.round(event.pageX - camLocation.x),
            y: Math.round(event.pageY - camLocation.y)
        };
        try {
            isCameraFocusReady.current = false;
            await cameraRef?.current?.focus(point);
            setTimeout(() => {
                isCameraFocusReady.current = true;
            }, 200);
        } catch (error) {
            console.log(error)
        }
    };
.....
    <Camera
      format={format}
      isActive={!paused}
      photo={true}
      orientation={'portrait'}
      focusable={true}
      ref={cameraRef}
      enableHighQualityPhotos={true}
      fps={fps}
      preset='photo'
      enableZoomGesture={true}
      onLayout={handleCameraLayout}
      device={device}
      zoom={device.neutralZoom}
      style={{ flex: 1 }}
       onTouchEnd={(x) => touchToFocus(x.nativeEvent)}
     />
```

### Relevant log output

```shell
session/camera-not-ready: [session/camera-not-ready] The Camera is not ready yet! Wait for the onInitialized() callback!]
```

### Camera Device

```json
{"sensorOrientation":
"landscape-right",
"hardwareLevel": "full",
"maxZoom": 8,
"minZoom":1,
"supportsLowLightBoost":false,
"neutralZoom":1,
"physicalDevices":["wide-angle-camera"],
"supportsFocus":true,
"supportsRawCapture" true,
"isMultiCam":false,
"name":"BACK(0)",
"hasFlash" true,
"has Torch" true,
"position":"back",
"id":"0"}
```

### Device

OnePlus 5T, Android version 10

### VisionCamera Version

3.6.11

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [X] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

nemmtor, 2023-11-30:
I am having the same issue.

karel-suchomel-ed, 2023-12-06:
Same issue, works fine on ios

M4Tdev, 2023-12-15:
Has someone found any solution for this?

M4Tdev, 2023-12-15:
Here is the the answer from @mrousavy in another issue: [#2003](https://github.com/mrousavy/react-native-vision-camera/issues/2003#issuecomment-1843110471)

mrousavy, 2024-01-30:
Yea - duplicate of https://github.com/mrousavy/react-native-vision-camera/issues/2003.

closed 2024-01-30 by mrousavy