margelo/react-native-vision-camera #2240: 🐛 Error: TOO_MANY_OPEN_CAMERAS after reopening camera page

hanna-buhai, 2023-12-01:
### What's happening?

Note: to reproduce the bug on example app, you need to create new initial screen instead of setting Camera as `initialRouteName`. The bug doesn't happen when you navigate to new screens that are being added on top of Camera screen in navigation stack

Steps to reproduce (default cameraPosition is `back`):
1) navigate to Camera Screen
2) switch to `front` cameraPosition
3) navigate back
4) reopen the Camera Screen

The camera screen is black and you get TOO_MANY_OPEN_CAMERAS error. This only happens when the Camera screen is unmounted while the selected cameraPosition doesn't match the default value. So, if the default cameraPosition='front', it happens when you unmount on active back camera.
If you flip the camera multiple times and close the screen while the default camera is active, there is no error.

### Reproduceable Code

```tsx
// here is the device, format and camera configuration

import { useIsFocused } from '@react-navigation/native';
import {
	Camera,
	useCameraDevice,
	getCameraFormat,
	Templates,
} from 'react-native-vision-camera';

export const CameraView = () => {
	const isFocused = useIsFocused();
	const [cameraPosition, setCameraPosition] = useState<'front' | 'back'>(
		'back'
	);

	const device = useCameraDevice(cameraPosition);
	const format = !!device
		? getCameraFormat(device, Templates.Snapchat)
		: undefined;
	const camera = useRef<Camera>(null);

	// ...

	return (
		<Camera
			ref={camera}
			style={StyleSheet.absoluteFill}
			device={device}
			format={format}
			video={true}
			isActive={isFocused}
			audio={true}
			enableZoomGesture={true}
		/>
	);
};
```

### Relevant log output

```shell
12-01 11:47:34.447 19796 19932 I CameraDevices: Camera #0: Unavailable!
12-01 11:47:34.447  1272  1283 D HwBatteryService: cameraUpdateLight enable false
12-01 11:47:34.447   658  1368 I HwCameraServiceEx: disconnect: Disconnected client for camera 0 for PID 19796
12-01 11:47:34.448 19796 19932 I CameraDevices: Camera #0: Available!
12-01 11:47:34.450   658  1368 I         : close camera, send msg to nfc, open nfc
12-01 11:47:34.450  1938  2110 D ScenarioService: camera end : com.app.name, last: com.app.name, pkgs: {com.app.name=27278868}
12-01 11:47:34.450  1938  2329 I PGServer: report state:10019 event type:1 pid:0 uid:0 pkg:com.app.name to pid: 1938
12-01 11:47:34.450   658  1368 I Camera2ClientBase: Closed Camera 0. Client was: com.app.name (PID 19796, UID 10469)
12-01 11:47:34.453 19796 20210 E CameraManager: Camera 0: Error! 2
12-01 11:47:34.457 19796 19796 I ViewRootImpl: jank_removeInvalidNode jank list is null
12-01 11:47:34.457 19796 20210 E CameraSession: Failed to configure CameraSession! Error: [session/camera-cannot-be-opened] The given Camera device (id: 0) could not be opened! Error: TOO_MANY_OPEN_CAMERAS, Config-Diff: Difference(deviceChanged=true, outputsChanged=true, sidePropsChanged=true)
12-01 11:47:34.457 19796 20210 E CameraSession: com.mrousavy.camera.core.CameraCannotBeOpenedError: [session/camera-cannot-be-opened] The given Camera device (id: 0) could not be opened! Error: TOO_MANY_OPEN_CAMERAS
[... 112 lines cut ...]
  "hasTorch": false,
  "position": "front",
  "id": "1"
}
```

### Device

HUAWEI P20 Lite (Android 9), Xiaomi Redmi 9C (Android 10)

### VisionCamera Version

3.6.12

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

paulobunga, 2023-12-01:
Just experienced the same, issue. While trying to Switch between back camera and front camera after navigating to new page

cherish404, 2023-12-07:
I have the same issue in v3.6.13

lewy12121212, 2023-12-15:
same in version 3.6.16 :/

mrousavy, 2024-01-05:
Hey! Thanks for reporting this issue.

I've been working the past days on making sure the Camera lifecycle is safely handled and released when needed to make sure this crash doesn't occur anymore.

I just created a PR for this - can you please this to see if that fixes the issue for you? https://github.com/mrousavy/react-native-vision-camera/pull/2339

If this fixes your issue, please consider [💖 **sponsoring me on GitHub** 💖](https://github.com/sponsors/mrousavy/) to support me / thank me for building VisionCamera and continuously improving it. 

If this does not fix your issue, please clone the repo, check out the branch `fix/blackscreen` (the PR above), run the Example app, reproduce your issue there and share the `adb logcat` logs with me (upload to pastebin or gist) so I can investigate this further.

Thank you! 🙏

closed 2024-01-08 by mrousavy